zigpy/zha-device-handlers #1272: [BUG] OSRAM Plug 01 does not want to be controlled

MichaelBitard, 2022-01-09:
**Describe the bug**
Most of the time it works fine.

Then all of a sudden it decides to not takes my switch on/off commands into account, and there is nothing I can do except retry a few minutes later.

The device is still reachable, I can call a "reconfigure device" from home assistant and it will bind and report (with errors, but I was never able to have a reconfigure on this plug without error (see screenshot)

[image]

I put my home assistant / zha / zigpy / zigpy_zigate in debug mode and here is what I have:
<details><summary>Hidden logs</summary>
<p>

```
2022-01-09 11:14:33 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xBC8C:1:0x0b04]: async_update
2022-01-09 11:14:33 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xBC8C, 260, 2820, 1, 1, 86, b'\x00V\x00\x0b\x05\r\x05\x0f\x05\x08\x05\n\x05\x05\x05', True, False)
2022-01-09 11:14:33 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02bc8c01010b04010400000f0056000b050d050f0508050a050505'
2022-01-09 11:14:33 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x1bA\x02\xbc\x8c\x01\x01\x0b\x04\x01\x04\x00\x00\x0f\x00V\x00\x0b\x05\r\x05\x0f\x05\x08\x05\n\x05\x05\x05'
2022-01-09 11:14:33 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x1bA\x02\x12\xbc\x8c\x02\x11\x02\x11\x02\x1b\x02\x14\x02\x11\x02\x14\x02\x10\x02\x10\x02\x1f\x02\x10V\x02\x10\x02\x1b\x02\x15\x02\x1d\x02\x15\x02\x1f\x02\x15\x02\x18\x02\x15\x02\x1a\x02\x15\x02\x15\x02\x15'
2022-01-09 11:14:33 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000009bc001a0530011a000100
2022-01-09 11:14:33 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'001a0530011a0001' LQI:0
2022-01-09 11:14:33 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 26, 1328, b'\x01\x1a\x00\x01']
2022-01-09 11:14:33 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8012000ab100010102bc8c1a000100
2022-01-09 11:14:33 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8012 b'00010102bc8c1a0001' LQI:0
2022-01-09 11:14:33 ERROR (MainThread) [zigpy_zigate.api] Received unhandled response 0x8012
2022-01-09 11:14:35 DEBUG (MainThread) [homeassistant.components.zha.api] Reconfiguring node with ieee_address: 7c:b0:3e:aa:00:b0:4d:94
2022-01-09 11:14:35 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xF76E](Plug 01): started configuration
2022-01-09 11:14:35 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF76E:ZDO](Plug 01): 'async_configure' stage succeeded
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xF76E, 0, <ZDOCmd.Bind_req: 0x0021>, 0, 0, 87, b'W\x94M\xb0\x00\xaa>\xb0|\x03\x06\x00\x039\x11\x1e\x02\x00\x8d\x15\x00\x01', True, False)
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02f76e00000021000000001657944db000aa3eb07c0306000339111e02008d150001'
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00"v\x02\xf7n\x00\x00\x00!\x00\x00\x00\x00\x16W\x94M\xb0\x00\xaa>\xb0|\x03\x06\x00\x039\x11\x1e\x02\x00\x8d\x15\x00\x01'
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10"v\x02\x12\xf7n\x02\x10\x02\x10\x02\x10!\x02\x10\x02\x10\x02\x10\x02\x10\x16W\x94M\xb0\x02\x10\xaa>\xb0|\x02\x13\x02\x16\x02\x10\x02\x139\x11\x1e\x02\x12\x02\x10\x8d\x15\x02\x10\x02\x11'
2022-01-09 11:14:35 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF76E:3:0x0000]: finished channel configuration
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xF76E, 260, 4096, 1, 3, 88, b'\x01XA\x00', True, False)
2022-01-09 11:14:35 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF76E:3:0x0019]: finished channel configuration
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000009bd001b0530011b020200
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'001b0530011b0202' LQI:0
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 27, 1328, b'\x01\x1b\x02\x02']
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8702000aa2d4000002f76e1b020279
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8702 b'd4000002f76e1b0202' LQI:121
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [212, 0, 0, <Address address_mode=ADDRESS_MODE.NWK address=0xf76e>, 27]
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.zigbee.application] APS Data confirm Fail 27 212
2022-01-09 11:14:35 WARNING (MainThread) [zigpy_zigate.zigbee.application] Unexpected message send failure
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02f76e01031000010400000401584100'
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x10\xb5\x02\xf7n\x01\x03\x10\x00\x01\x04\x00\x00\x04\x01XA\x00'
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x10\xb5\x02\x12\xf7n\x02\x11\x02\x13\x10\x02\x10\x02\x11\x02\x14\x02\x10\x02\x10\x02\x14\x02\x11XA\x02\x10'
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80000009bd001c0530011c030300
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8000 b'001c0530011c0303' LQI:0
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 28, 1328, b'\x01\x1c\x03\x03']
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 8702000aa7d4010302f76e1c030379
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8702 b'd4010302f76e1c0303' LQI:121
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [212, 1, 3, <Address address_mode=ADDRESS_MODE.NWK address=0xf76e>, 28]
2022-01-09 11:14:35 DEBUG (MainThread) [zigpy_zigate.zigbee.application] APS Data confirm Fail 28 212
2022-01-09 11:14:35 WARNING (MainThread) [zigpy_zigate.zigbee.application] Unexpected message send failure
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80110008bd00bc8c010b041a00
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8011 b'00bc8c010b041a' LQI:0
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 0xbc8c, 1, 2820, 26]
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.zigbee.application] ACK Data received 26 0
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.uart] Frame received: 80020035e70001040b04010102bc8c0200001856010b05002900800d05002900800f05002104060805002106000a0500215a0005050021ffff75
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.api] data received 0x8002 b'0001040b04010102bc8c0200001856010b05002900800d05002900800f05002104060805002106000a0500215a0005050021ffff' LQI:117
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.zigbee.application] zigate_callback_handler [0, 260, 2820, 1, 1, <Address address_mode=ADDRESS_MODE.NWK address=0xbc8c>, <Address address_mode=ADDRESS_MODE.NWK address=0x0000>, b'\x18V\x01\x0b\x05\x00)\x00\x80\r\x05\x00)\x00\x80\x0f\x05\x00!\x04\x06\x08\x05\x00!\x06\x00\n\x05\x00!Z\x00\x05\x05\x00!\xff\xff']
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy.zcl] [0xbc8c:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=86 command_id=Command.Read_Attributes_rsp>
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.zigbee.application] request (0xBC8C, 260, 2820, 1, 1, 89, b'\x00Y\x00\x0b\x05\r\x05\x0f\x05\x08\x05\n\x05\x05\x05', True, False)
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.uart] Send: 0x0530 b'02bc8c01010b04010400000f0059000b050d050f0508050a050505'
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.uart] Frame to send: b'\x050\x00\x1bN\x02\xbc\x8c\x01\x01\x0b\x04\x01\x04\x00\x00\x0f\x00Y\x00\x0b\x05\r\x05\x0f\x05\x08\x05\n\x05\x05\x05'
2022-01-09 11:14:36 DEBUG (MainThread) [zigpy_zigate.uart] Frame escaped: b'\x02\x150\x02\x10\x1bN\x02\x12\xbc\x8c\x02\x11\x02\x11\x02\x1b\x02\x14\x02\x11\x02\x14\x02\x10\x02\x10\x02\x1f\x02\x10Y\x02\x10\x02\x1b\x02\x15\x02\x1d\x02\x15\x02\x1f\x02\x15\x02\x18\x02\x15\x02\x1a\x02\x15\x02\x15\x02\x15'
[... cut ...]
This line: `2022-01-09 11:31:35 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF76E:3:0x0006]: command failed: 'off' args: '()' kwargs '{}' exception: ''` tells me the off command failed, but I don't really understand why either.

**Expected behavior**
When I switch on the device, it switches on.
When I switch off the device, it switches off

**Additional context**
I'm using the zigate v1 version 3.21.
This device is quite near my coordinator, so unreachability should not be the issue here.

towo, 2022-01-10:
I've recently had more bits of my ZigBee network suddenly barf up and stop responding, but it wasn't just limited to the OSRAM plugs. @MichaelBitard, do you have other devices in your network that may be affected as well, just to exclude it's merely the OSRAM plug and not something more general?

MichaelBitard, 2022-01-10:
That's a good question, in my case it's really related to this plug only, I can still manage all other zigbee stuff, retrieve my sensors and such.

This is the only one where "Reconfigure device" fails systematically

TheJulianJES, 2022-04-17:
Possibly see: https://github.com/Koenkk/zigbee2mqtt/issues/1792#issuecomment-540741079
I've also had issues with this device in the past and just completely got rid of it. It's possibly a combination of really bad firmware and a bad antenna design.

MichaelBitard, 2022-04-17:
Thanks! I'll look into it.

In the meantime I switched to zigbee2mqtt, then away from zigate and I don't have the issue anymore, so I will close this issue

closed 2022-04-17 by MichaelBitard