zigpy/zha-device-handlers #1275: [Device Support Request] New manufacturerName for DDS238-2 Zigbee

darckly, 2022-01-10:
In one chat I got information that from December 2021 this device now is sending with a new manufacturerName attribute: _TZE200_ewxhg6o9. The new version also contains the fix the switch issue (see the details in the #994 thread) and maybe something else, like EnergyFactor attribute.
I can update the quirck but I have no physically the new version of the DDS238-2 device to test, so if someone has the new version of device and can support me with the testing, I can help with the contribution of this update.

jcor1965, 2022-01-11:
I have one of these devices, how can I help?

[image]

darckly, 2022-01-12:
Great!
Let's try at first to reuse the configuration from the first version of DDS238-2, I only updated the manufactureName with the new one.
Could you do the next for testing:
1. Create "quircks" directory in your /config folder
2. Extract and copy the ts0601_din_power.v2.py file to the /config/quircks directory
[ts0601_din_power.v2.zip](https://github.com/zigpy/zha-device-handlers/files/7852995/ts0601_din_power.v2.zip)
3. Add the following into your configuration.yaml
          zha: custom_quirks_path: '/config/quircks'
4. Restart HA
5. Pair your DDS238-2 device

If all cluster parameters are the same in the new version, you should see the switch, voltage, power and current immediately, total energy should appear after some time.
By this I want to check that at least the minimal information from the relay can be received in the same way.

cristiano695, 2022-01-12:
Hi darckly
I also took this new device and tried several times to do the pairing but I was never able to see the entities.
I tried your solution but it doesn't work.
I attach screenshots of what I did.
[image]

[image]

[image]

Restarted HA and re-paired with the device I still don't see the entities.

[image]

If you want I am available for any test.
Thank's for your job

darckly, 2022-01-13:
Hi,

Looks like the quirk is not loaded, this is how it is visualized on my system :
[image]
In your case, it should start as "Custom quirk:"

but I don't see the mistakes in your configuration...

Could you share the logs from ZHA about the pairing process?

cristiano695, 2022-01-13:
Hi
I did it all over again and the situation is the same as it seems it doesn't load the custom_quirks part even if specified in the configuration.yaml.
The pairing log gives me:

New device 0x23d4 (90:35:ea:ff:fe:ba:9d:02) joined the network
[0x23d4] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x23D4 ieee=90:35:ea:ff:fe:ba:9d:02 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x82\xd4#\x02\x9d\xba\xfe\xff\xea5\x90\x8e'
[0x23d4:zdo] ZDO request ZDOCmd.Device_annce: [0x23D4, 90:35:ea:ff:fe:ba:9d:02, 142]
Tries remaining: 3
[0x23d4] Requesting 'Node Descriptor'
Tries remaining: 2
[0x23d4] Extending timeout for 0x10 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x23D4 ieee=90:35:ea:ff:fe:ba:9d:02 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x10\x00\xd4#\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00'
[0x23d4] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x23d4] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x23D4 ieee=90:35:ea:ff:fe:ba:9d:02 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\x12\x00\xd4#\x01\x01'
[0x23d4] Discovered endpoints: [1]
[0x23d4] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x23d4:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x23D4 ieee=90:35:ea:ff:fe:ba:9d:02 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x14\x00\xd4#\x14\x01\x04\x01Q\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xef\x02\x19\x00\n\x00'
[0x23d4:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x23d4:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=22 command_id=Command.Read_Attributes_rsp>
[0x23d4] Read model 'TS0601' and manufacturer '_TZE200_ewxhg6o9' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x23d4] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_ewxhg6o9' nwk=0x23D4 ieee=90:35:ea:ff:fe:ba:9d:02 is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_ewxhg6o9' nwk=0x23D4 ieee=90:35:ea:ff:fe:ba:9d:02 is_initialized=True>
Checking quirks for _TZE200_ewxhg6o9 TS0601 (90:35:ea:ff:fe:ba:9d:02)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0x23D4:90:35:ea:ff:fe:ba:9d:02 entering async_device_initialized - is_new_join: True
device - 0x23D4:90:35:ea:ff:fe:ba:9d:02 has joined the ZHA zigbee network
[0x23D4](TS0601): started configuration
[0x23D4:ZDO](TS0601): 'async_configure' stage succeeded
Error handling '_save_attribute' event with (90:35:ea:ff:fe:ba:9d:02, 1, 0, 4, '_TZE200_ewxhg6o9') params: FOREIGN KEY constraint failed
[0x23D4:1:0x0000]: finished channel configuration
[0x23D4:1:0x0019]: finished channel configuration
Error handling '_save_attribute' event with (90:35:ea:ff:fe:ba:9d:02, 1, 0, 5, 'TS0601') params: FOREIGN KEY constraint failed
[0x23D4:1:0x0000]: 'async_configure' stage succeeded
[0x23D4:1:0x0019]: 'async_configure' stage succeeded
[0x23D4](TS0601): completed configuration
[0x23D4](TS0601): stored in registry: ZhaDeviceEntry(name='_TZE200_ewxhg6o9 TS0601', ieee='90:35:ea:ff:fe:ba:9d:02', last_seen=1642068283.9499283)
[0x23D4](TS0601): started initialization
[0x23D4:ZDO](TS0601): 'async_initialize' stage succeeded
[0x23D4:1:0x0000]: initializing channel: from_cache: False
[0x23D4:1:0x0000]: finished channel initialization
[0x23D4:1:0x0019]: initializing channel: from_cache: False
[0x23D4:1:0x0019]: finished channel initialization
[... cut ...]
[0x23d4:1:0xef00] ZCL request 0x0001: b'\x00\x11\x12\x02\x00\x04\x00\x00\t\xb1'
[0x23d4:1:0xef00] No handler for cluster command 1
[0x23d4:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=87 command_id=1>
[0x23d4:1:0xef00] Unknown cluster-specific command 1
[0x23d4:1:0xef00] ZCL request 0x0001: b'\x00\x12\x13\x02\x00\x04\x00\x00\x11i'
[0x23d4:1:0xef00] No handler for cluster command 1

darckly, 2022-01-13:
Good thing: based on shared logs, the signature of the new device is the same, and the old quirck should work.

I want to ask to you remove the previous ts0601_din_power.v2.py file from your /config/quircks folder and extract and drop the ts0601_din_power.py instead from the archive below
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7862038/ts0601_din_power.zip)

Restart HA and try to pair the device again.

ziobabba, 2022-01-13:
I did what you asked me and now restarting HA without re-pairing you see the entities (not all the ones that should) and loaded the quirk correctly
[image]

If you want I can do all the tests you want for support.
in the meantime, thank you for making it work :-)
If you want I can re-pair and post you the log.

cristiano695, 2022-01-13:
I did what you asked me and now restarting HA without re-pairing you see the entities (not all the ones that should) and loaded the quirk correctly
[image]

If you want I can do all the tests you want for support.
in the meantime, thank you for making it work :-)
If you want I can re-pair and post you the log.

darckly, 2022-01-13:
I see that all parameters from the first version are also working for your device. This is really good and will simplify the official update.
Next, I want to ask you to perform the following tests:
1. Check that information is always actual and correct:
 - switch off the device in UI, wait for a few minutes. The switch should stay in the same state. Load and Current should show zero after some time. In the first version of device we have an issue: the switch is changing to the ON after a minute, even the device is OFF
 - switch ON the device. After some time (~30 seconds) the current and power should be refreshed
 - switch ON/OFF the relay with the hardware button (press three times for changing the state). The state in UI should reflect the changes with a small delay

2. (more complex task) I'm also wondering are any new parameters the new device sends, besides the old four + switch. For this, please, enable the logging in configurations.yaml and restart HA
_logger:
  default: warning
  logs:
    zhaquirks: debug_

After restart could you find and send me the lines with the text "[zhaquirks.tuya]" from the HA system log. With the old version of the relay these lines have '0x0001', '0x0101', '0x0211', '0x0212', '0x0213' and '0x0214' attributes information. I'm interested in any new '0x02**' that will appear in the log.

Thank you a lot for your help!

jcor1965, 2022-01-14:
I followed what is suggested and I already have the values.

[image]

I cannot test the switch because I have physically disabled this function. My device cannot be turned off accidentally.

darckly, 2022-01-14:
Hi Carlos,
Thank you for your confirmation!

cristiano695, 2022-01-14:
Hi Darckly
I try to answer you in point

- "switch off the device in UI, wait for a few minutes. The switch should stay in the same state. Load and Current should show zero after some time. In the first version of device we have an issue: the switch is changing to the ON after a minute, even the device is OFF"
through UI the switch seems to have no success it seems, also seen from the log, not being able to send it any command.

- switch ON the device. After some time (~30 seconds) the current and power should be refreshed
view point1

- switch ON/OFF the relay with the hardware button (press three times for changing the state). The state in UI should reflect the changes with a small delay
I can not turn off the device as it would also turn off HA because it is already mounted in the home power line

- (more complex task) I'm also wondering are any new parameters the new device sends, besides the old four + switch. For this, please, enable the logging in configurations.yaml and restart HA
From what I can see and what I post to you in the log below, it doesn't seem to be forwarding any new parameters other than the ones you described.

14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.aduro                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.aduro.adurolightncc                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.aurora                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.aurora.aurora_dimmer                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.bitron                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.bitron.thermostat                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.bosch                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.bosch.isw_zdl1_wp11g                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.bosch.motion                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite.cl_3130                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite.cl_3157100                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite.cl_3300S                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite.cl_3305S                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite.cl_3310S                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite.cl_3321S                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite.cl_3460L                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite.ias                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite.motion                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.centralite.motionandtemp                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.const                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.danfoss                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.danfoss.thermostat                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.develco                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.develco.air_quality                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.develco.heat_alarm                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.develco.open_close                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.develco.smoke_alarm                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.echostar                                          
14/01/2022 11:15:31 DEBUG (MainThread) [zhaquirks] Loading quirks module zhaquirks.echostar.bell                                          
[... cut ...]
14/01/2022 11:37:22 DEBUG (MainThread) [zhaquirks.tuya] [0x23d4:1:0xef00] Received value [0, 0 6 234] for attribute 0x0212 (command 0x0001)                                  
14/01/2022 11:37:23 DEBUG (MainThread) [zhaquirks.tuya] [0x23d4:1:0xef00] Received value [0, 0 11 91] for attribute 0x0213 (command 0x0001)                                  
14/01/2022 11:37:24 DEBUG (MainThread) [zhaquirks.tuya] [0x23d4:1:0xef00] Received value [0, 0 9 7] for attribute 0x0214 (command 0x0001)                                  
14/01/2022 11:37:25 DEBUG (MainThread) [zhaquirks.tuya] [0x23d4:1:0xef00] Received value [0, 0 8 107] for attribute 0x0211 (command 0x0001)

cristiano695, 2022-01-14:
> I followed what is suggested and I already have the values.
> 
> [image]
> 
> I cannot test the switch because I have physically disabled this function. My device cannot be turned off accidentally.

Hi Carlos
I might know how you did to physically disable the device shutdown as it would affect me a lot if it did not turn off accidentally.
Thank you

darckly, 2022-01-14:
Hi Christiano,
Thank you a lot for the logs, that's exactly what I need!
In short, no new attributes in the new version of relay, they are the same as in the old.

I'll prepare a Pull Request tomorrow and I think your device with the new manufacturerName will be officially supported in the next HA 2022.02.* version. Till the update, you can use the trick with the custom quirk.

cristiano695, 2022-01-14:
Thanks Alexey for the support and work towards HA

jcor1965, 2022-01-15:
> Hi Carlos I might know how you did to physically disable the device shutdown as it would affect me a lot if it did not turn off accidentally. Thank you

[image]

I unsoldered the relay terminals on the board. Then I isolated them so that they could not cause a short circuit, since they are loose inside the device.
So, if I want to use the device again to turn the power on and off, I just solder the terminals on the board again.

dieneuser, 2022-01-15:
> Hi Darckly I try to answer you in point
>
> * "switch off the device in UI, wait for a few minutes. The switch should stay in the same state. Load and Current should show zero after some time. In the [...]

Hey,

I own the same device as of today and can confirm the results from @cristiano695.

@darckly to the 3rd point:
> * switch ON/OFF the relay with the hardware button (press three times for changing the state). The state in UI should reflect the changes with a small delay

In the GUI nothing happens, the switch remains in the off position.

darckly, 2022-01-15:
Hmm, it's not good... 
@dieneuser Could you try this one? 
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7875976/ts0601_din_power.zip)

dieneuser, 2022-01-16:
> Hmm, it's not good...
> @dieneuser Could you try this one?
> [ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7875982/ts0601_din_power.zip)

@darckly Thank you for the quick fix!
I can confirm, with the last quirks file, the ON/OFF switch works as desired and also shows the correct status in the UI.

darckly, 2022-01-16:
@dieneuser I want to ask you also test the information from sensors - add load and check that information is updated.
Thank you!

dieneuser, 2022-01-16:
@darckly I like to check this during the day, for this I still have to extend the test setup with a consumer.

darckly, 2022-01-17:
@dieneuser could you disconnect and pair the device again? 
Just to be sure that sensors will be detected and created.

dieneuser, 2022-01-17:
@darckly hmm, after removing and repairing, only the switch is visible. A reboot does not change anything.
[image]
I use the last quirk file from you (https://github.com/zigpy/zha-device-handlers/files/7875982/ts0601_din_power.zip)

darckly, 2022-01-17:
Could you try this one?
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7880607/ts0601_din_power.zip)

dieneuser, 2022-01-17:
@darckly
Same behavior - only the switch is visible. The HA log shows a warning:
```
Logger: zhaquirks.tuya
Source: /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py:529
First occurred: 10:25:09 (5 occurrences)
Last logged: 10:25:10

Unsupported command: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=23 command_id=17>
```

darckly, 2022-01-17:
Did you re-pair the device after the update of quirk?

dieneuser, 2022-01-17:
Yes, I did and performed the following steps:

1. copy the file
2. removing the device
3. HA Reboot
4. re-pair device

darckly, 2022-01-17:
@dieneuser Let's try the last change
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7880978/ts0601_din_power.zip)

If it not working, I'll need to find the device locally in Ukraine to debug.
The strange thing, that based on logs, shared from Christiano before, the device sends the information in the same way for switch also, so the original quirk should work with the switch correctly. But appears not...

dieneuser, 2022-01-17:
@darckly
That is very strange. Now there is no sensor anymore.
[image]

I bought the power meter here: https://www.aliexpress.com/item/1005002409588154.html?spm=a2g0s.12269583.0.0.3066540b0xui7e

darckly, 2022-01-17:
@dieneuser 
The device will be only in three-four weeks if ordered from Ali. We will see, maybe someone from the Ukrainian SmartHouse community can lend it for a few days, if we can't solve the issue during this week.
At the same time, the negative result also results, so I returned back to the original code and want to ask you to check this version.
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7882106/ts0601_din_power.zip)

dieneuser, 2022-01-17:
@darckly 
The behavior is a bit strange. I have imported your last file, removed the power meter from ZHA again and restarted HA. After adding the power meter only the on/off switch was visible - then I restarted HA again without repair - now all values are visible and correct. Only the value for the current power is alway 0,0A.
Thank you very much for your effort!

[image]

darckly, 2022-01-17:
@dieneuser 
Good news, the modification which I made was a small trick, which will work also with the old device, so if everything is working now, I can improve the quirk with minimal modifications.
Based on your screenshot, the current is calculated as I=P/U=8/230=0.0347826A, I think the value is rounded. Could you somehow increase the consumption (>23W)?

And please test the Switch behavior, based on my suggested scenarios above.

dieneuser, 2022-01-17:
@darckly 
You are my hero!

[image]

I have increased the load to 155W and now I also see a value from 0.7A. (I=P/U=155/232=0.668103A) 👍
I confirm that all sensors and the switch are now working correctly.

dieneuser, 2022-01-17:
@darckly
Final test successful!

After deleting and rejoining, the sensors and switches are recognized and work as desired.
Great job! Thank you.

SuperBigB, 2022-01-21:
I just recieved six of these. Thanks!

closed 2022-01-23 by dmulcahey

tomkusz, 2022-08-29:
Hello

Would you be so good and add support for the RTX Circuit Breaker 2x16A Energy Meter (ZCB16-2P) _TZE200_fsb6zw01? This one has antoher signature to listed above.

darckly, 2022-08-30:
Hello @tomkusz, did you try to create the custom quirk based on this one: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_din_power.py and extend the section "MODELS_INFO" with your device?

tomkusz, 2022-08-30:
Hello @darckly, I already try this operation, but there is no effect. ZHA discover only lqi and rssi. Could you be so kind and make for me custom quirk with _TZE200_fsb6zw01 device? mayby I'm doing something wrong with this...

darckly, 2022-09-01:
@tomkusz could you open a new issue, because this is closed already?
Also, please enable debug, perform device parring, and add the load for a few minutes (like kettler) - and attach the logs to the new issue.

An4cl3t0, 2023-04-04:
I have a problem with this device since last month, so 2023.02 update.
Now reports only the Power Factor and the switch position.
It never reported the full list of sensor anyway, like here: 
https://gist.github.com/Koenkk/34e60391471f8a6333176594dd80445d

[image]

mirican, 2023-04-21:
On March 11, 2023, the device stopped counting energy. The entity “sensor.licznik_summation_delivered” became unavailable. Up until that moment, everything had been working fine for several months. In the logs, I have the following error message:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/uart.py", line 84, in data_received
    self._api.data_received(frame)
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/api.py", line 373, in data_received
    getattr(self, f"_handle_{command.name}")(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/api.py", line 505, in _handle_aps_data_indication
    self._app.handle_rx(
  File "/usr/local/lib/python3.10/site-packages/zigpy_deconz/zigbee/application.py", line 430, in handle_rx
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 875, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 381, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 370, in handle_message
    self.endpoints[src_ep].handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 373, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 494, in handle_cluster_request
    zvalue = ztype(tuya_data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 96, in __new__
    raise ValueError(
ValueError: 72071 is not an unsigned 16 bit integer

The number 72071 corresponds to the current reading of the meter, which is 720.71 kWh. "I tried to pair the device again, restore a backup from the beginning of March, as well as install on a clean installation.

manyos-robert, 2024-01-13:
Just had the same problem. The following change fixed it:

TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint32_t, True),

PR #1287 (2022-01-23): Update DDS238-2 Din powerMeter with a new version of device
New version of DDS238-2 Zigbee device was released with a new manufacturerName: _TZE200_ewxhg6o9. Device firmware contains a fix of the issue with incorrect Switch information, which the device sends (see details in the tread #994 )
The rest of the information is the same.
PR contains only the new manufacturer name as update, all other information is the same. PR closes #1275 .