zigpy/zha-device-handlers #1284: Fancoil Thermostat TZE200_dzuqwsyg [Device Support Request]

DXBD, 2022-01-14:
This device TZE200_dzuqwsyg has no functions in ZHA. The product manual is listed below. It is known online as BAC-001-ALZB. There are variations on this device which is outlined in page 1 of the product manual which can be found below.

My specific thermostat is for 2 pipe system and has 4 relays (Actuator, Low Fan, Med Fan, High Fan)

This device is supported on Z2M for reference here is the link. 

https://www.zigbee2mqtt.io/devices/BAC-002-ALZB.html

DEVICE SIGNATURE

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_dzuqwsyg",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

PRODUCT MANUAL
[image]
[image]
[image]
[image]

MattWestb, 2022-01-14:
One similar device is being supported for more then one year ago https://github.com/zigpy/zigpy/discussions/653#.

You can trying adding your device ID in the quirk here: https://github.com/zigpy/zha-device-handlers/blob/6227c8b08403f536e13e9c48b647f66600f06145/zhaquirks/tuya/ts0601_electric_heating.py#L140 like this:
```
        MODELS_INFO: [
            ("_TZE200_aoclfnxz", "TS0601"),
            ("_TZE200_dzuqwsyg", "TS0601"),
        ],
```
Configure local quirk https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 download the `ts0601_electric_heating.py` qiurk, putting in the folder for local quirk and modifying it and restarting HA.

Test if its working OK or not.

DXBD, 2022-01-14:
> ("_TZE200_dzuqwsyg", "TS0601"),

The device you referenced is completely different. That thermostat only has 1 relay which controls underfloor heating systems. It seems to be using the smart plug control in the quirk. 

The thermostat I have referenced has 4 control relays. 
Relay 1 = Low Fan Speed 
Relay 2 = Medium Fan Speed 
Relay 3 = High Fan Speed
Relay 4 = Cooling

MattWestb, 2022-01-14:
I was not saying 100% the same only similar.
I think one user have trying getting the cooling working but i was not getting any feedback from hi.

If you is installing the quirk is you  getting some functionality at all ?
If yes remove the power and waiting one minute and putting it back and have debug for     `zhaquirks.: debug` and `zigpy: debug`.
If all is going well is the device sending all DP commands is using and we can start looking how its working and implanting the function of it.

MattWestb, 2022-01-14:
By the way all tuya Zigbee thermostats is using the "plug" or dimmer zigbee module and must being "converted" to normal zigbee device with quirks for working OK.

DXBD, 2022-01-14:
Hi Matt,

Apologies if my message read as being cold and thank you for the super fast response. OK so let me tell you what I did. 

I downloaded the quirk file and edited it as instructed. In my HA config I added the zigbee quirk folder and in that folder I loaded up the .py file then rebooted.

Paired the thermostat with HA and HA didn't recognise it and setup as a router with no entities.

MattWestb, 2022-01-14:
Look in the log ZHA is writing if its not finding ne matching quirk and you is getting "class": "zigpy.device.Device" instead of the name of the quirk.

Did you doing the configure of the local quirk OK in HA ??

DXBD, 2022-01-14:
When I add
      custom_quirks_path: /config/custom_zha_quirks/
 to my config file zha wont load. The folder exists and I have ts0601_electric_heating.py saved in that folder. I wonder what I am doing wrong.

MattWestb, 2022-01-14:
My experience is if the configured folder in HA is **not** existing ZHA is getting panic.
Make sure the folder is right inside your HA configuration folder (its the root for the path there the configuration.yaml is located).

DXBD, 2022-01-14:
Ok so I made a very rookie mistake and didn't properly edit the config.yaml file. Now ZHA loads but it isn't loading the updated ts0601_electric_heating.py file
[image]

ZHA still sees the stat as  "class": "zigpy.device.Device"

DXBD, 2022-01-15:
Ok so after discovering that I made some rookie mistakes. 1: failed to correctly edit config.yaml 2: recent upgrade to windows 11 turned off file extension view. File was named .py.txt grrrr. Fixed these issues and now the device is recognised in HA. However it only shows heating mode and setpoint. This thermostat is for cooling. I am playing around with controls now and will report back with issues.

thanks for your patience Matt.

madspoulsen, 2022-06-08:
I have no idea what im doing really. But I changed the quirk so that I can at least control the power, set-temp and fanspeed for speed 1 and 2 (because i dont know how to set it to value 2 and 3).
It does not work well with the Thermostat card. So I just created a basic entity card. I will most likely just use it to automate different temperatures during the day and night.
If anyone can finish it up, then here are here id's. My thermostat comes in as _TZE200_dzuqwsyg

Power on/off (id:0x0101
0=off
1=on

child_lock (id:0x0128:
0=child lock disabled
1=child lock enabled

mode (id:0x0402:
0=cooling
1=heating
2=ventilate

target_temperature (id:0x0210
value=set-temp

temperature (id:0x0218:
value=current room temp

Fan speed (id:0x041c
0=speed1
1=speed2
2=speed3
3=autospeed

AND HERE IS THE QUIRK FILE THAT WORKS WITH THE BASIC CONTROLS.. 
`

"""Map from manufacturer to standard clusters for thermostatic valves."""
import logging
from typing import Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogOutput,
    Basic,
    BinaryInput,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.hvac import Thermostat

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaPowerConfigurationCluster2AA,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
)

ZONNSMART_MODE_ATTR = (
[... 549 lines cut ...]
                    ZONNSMARTWindowDetection,
                    TuyaPowerConfigurationCluster2AA,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    ZONNSMARTChildLock,
                    ZONNSMARTWindowOpenedTemp,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    ZONNSMARTOnlineMode,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
`

mxilievski, 2022-09-17:
> 

hey, can you try to paste the file contents without the weird formatting. I want to proceed working on this.

madspoulsen, 2022-09-18:
[BAC-002-ALZB_converter.js.txt](https://github.com/zigpy/zha-device-handlers/files/9594973/BAC-002-ALZB_converter.js.txt)

Here it's attached as txt file.
But check this one also.. 
Currently it is working OK. Some times the thermostat doesn't receive my commands, so I have made my rules to repeat 3 times.. Don't know if its just my Zigbee network or what's going on, but all of my 3 are like this.

frodix, 2023-01-15:
Hello,
I have recently installed this thermostat.
I have followed the steps but I get the following error when starting zigbee2mqtt:

/app/data/extension/externally-loaded.js:183
.withSensor(['IN', 'AL', 'OU'], ea.STATE_SET)],
^
TypeError: exposes.climate(...).withSetpoint(...).withSetpoint(...).withLocalTemperature(...).withLocalTemperatureCalibration(...).withSystemMode(...).withFanMode(...).withRunningState(...).withPreset(...).withSensor is not a function
at /app/data/extension/externally-loaded.js:183:6
at Script.runInContext (node:vm:141:12)
at Script.runInNewContext (node:vm:146:17)
at Object.runInNewContext (node:vm:306:38)
at loadModuleFromText (/app/lib/util/utils.ts:150:8)
at loadModuleFromFile (/app/lib/util/utils.ts:157:12)
at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:167:25)
at getExternalConvertersDefinitions.next ()
at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
at new Controller (/app/lib/controller.ts:84:58)

I would appreciate if someone could help me.
Forgive my English, but I use Google translate from Spanish.

MattWestb, 2023-01-15:
Sorry @frodix but Z2M issues you shall reporting in https://github.com/Koenkk/zigbee-herdsman-converters.

closed 2023-07-21 by github-actions[bot]

ProgrammerIamNot, 2023-08-19:
Hi @mxilievski I came across this topic as it seems I have a thermostat with the same signature and functionality. I'm struggling with a custom quirk. I tried to assemble one from the bits you posted before but it's not working probably because of formatting (a grey area for me). I can use a custom quirk for a Tuya heater which turns the thermostat off, on for heating and sets a target temperature. But it would be nice to have control over all features. Would you be able to share the custom quirk as a whole file, please?

mxilievski, 2023-08-19:
Hey @ProgrammerIamNot, I have built "almost" full support for this device. The only thing I don't have a control on is the Fan modes. The reason behind that is that custom fan modes are not supported in ZHA by default. At the point of writing this, only  `[FAN_AUTO, FAN_ON]` are supported https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/climate.py#L202. 

Here you can find my implementation https://gist.github.com/mxilievski/09761353351ecfab89a2e16e46008248 which works for me, however I'm not a Python dev and there are a lot of hacks done in order to get this working, reading the current state from the device https://gist.github.com/mxilievski/09761353351ecfab89a2e16e46008248#file-ts0601_fancoil-py-L79 if the device is powered on by the device directly.

ProgrammerIamNot, 2023-08-19:
Thanks, @mxilievski , for sharing. I got the quirk installed and it works. I was hoping to have control over the fan but that's OK, I'll keep it on low. Should work for me.

ProgrammerIamNot, 2025-08-17:
After having had the thermostat for two years, I got tired of not having a way to control the fan's speed. I decided to pick up the AI's "brain" and, after a few hours of back and forth, I had a way to control the speed of the fan. 

For this, the thermostat must be added to ZHA (likely using a quirk). It appears as a climate entity, and it will not have fan speed control. The entity should allow turning OFF/ON the thermostat, changing the temperature setpoint and heating/cooling mode. 

By adding a separate fan as a template, we'll have a second entity just for the fan speed control.

After creating the Fan Entity, I created four scripts: Transfer OFF, Transfer LOW, Transfer MEDIUM, and Transfer HIGH. Transfer OFF just has the climate.thermostat OFF command. The other scripts have two actions: set target temperature and cooling for the thermostat (*), and the other action is setting the speed of the fan. These four scripts are enough for me to have buttons on the dashboard and run automations.

(*) I use this thermostat as a heat transfer only. I need fan only, and no heating/cooling functionality is required. The climate entity, however, is used for turning the thermostat ON/OFF. I also set the thermostat to cooling and the temperature SP to 7 to ensure the fan turns on. This is done just in case someone messes around with the thermostat itself and changes the mode or the temperature SP. If let's say it's in cooling mode and set temperature is above the current temperature, the fan will not turn on.

```
#FAN TEMPLATE in configuration.yaml, **ieee** needs to be adjusted to match your device
fan:
- platform: template
  fans:
    transfer_fan:
      friendly_name: "Transfer Fan"
      speeds:
        - "off"
        - "low"
        - "medium"
        - "high"
        - "auto"
      speed_list:
        - "low"
        - "medium"
        - "high"
        - "auto"
      turn_on:
        - service: zha.set_zigbee_cluster_attribute
          data:
            ieee: "11:22:33:44:55:66:77:88"
            endpoint_id: 1
            cluster_id: 61184
            attribute: 257
            value: 1
      turn_off:
        - service: zha.set_zigbee_cluster_attribute
          data:
            ieee: "11:22:33:44:55:66:77:88"
            endpoint_id: 1
            cluster_id: 61184
            attribute: 257
            value: 0
      set_speed:
        - service: zha.set_zigbee_cluster_attribute
          data:
            ieee: "11:22:33:44:55:66:77:88"
            endpoint_id: 1
            cluster_id: 61184
            attribute: 1052
            value: >
              {% if speed == 'off' %}
                0
              {% elif speed == 'low' %}
                0
              {% elif speed == 'medium' %}
                1
              {% elif speed == 'high' %}
                2
              {% elif speed == 'auto' %}
                3
              {% endif %}
```

I also created a sensor for monitoring the state of the fan as I found the OFF state of the fan entity doesn't mean the fan is actually stopped. So I look at the thermostat state to know if it's on/off and the fan speed to show the actual speed.
I use a separate file for template sensors (configuration.yaml has a line: template: !include templates.yaml). Change the sensor template accordingly if inserting into configuration.yaml directly.

```
#included into templates.yaml
- sensor:
    - name: "Fan State"
      state: >
        {% set thermostat_state = states('climate.thermostat_thermostat') %}
        {% set fan_mode = state_attr('fan.transfer_fan', 'preset_mode') %}
        {% if thermostat_state == 'off' %}
          OFF
        {% elif fan_mode == 'low' %}
          LOW
        {% elif fan_mode == 'medium' %}
          MEDIUM
        {% elif fan_mode == 'high' %}
          HIGH
        {% else %}
          UNKNOWN
        {% endif %}
      icon: >
        {% set thermostat_state = states('climate.thermostat_thermostat') %}
        {% set fan_mode = state_attr('fan.transfer_fan', 'preset_mode') %}
        {% if thermostat_state == 'off' %}
          mdi:fan-off
        {% elif fan_mode == 'low' %}
          mdi:fan-speed-1
        {% elif fan_mode == 'medium' %}
          mdi:fan-speed-2
        {% elif fan_mode == 'high' %}
          mdi:fan-speed-3
        {% else %}
          mdi:fan
        {% endif %}
```

I used Button Card by [@RomRider](https://github.com/RomRider) for the buttons to trigger the scripts and change the colour of the buttons based on the fan state sensor:
[image]

```
#YAML for OFF button:
type: custom:button-card
entity: script.transfer_off
icon: mdi:fan-off
show_name: false
show_state: false
tap_action:
  action: toggle
styles:
  card:
    - background-color: |
        [[[
          return states['sensor.fan_state'].state === 'OFF' ? 'rgba(255, 250, 0, 0.2)' : 'rgba(0, 0, 0, 0.3)';
        ]]]
```
```
#YAML for Speed LOW button:
type: custom:button-card
entity: script.transfer_low
icon: mdi:fan-speed-1
show_name: false
show_state: false
tap_action:
  action: toggle
styles:
  card:
    - background-color: |
        [[[
          return states['sensor.fan_state'].state === 'LOW' ? 'rgba(53, 255, 0, 0.3)' : 'rgba(0, 0, 0, 0.3)';
        ]]]
```
Speed MEDIUM and HIGH are done similarly to Speed LOW.