zigpy/zha-device-handlers #1286: TZE200_locansqn TS0601 Temp / Humi / Clock

toy3375, 2022-01-15:
Hi everybody, 

My Tuya devices has no entities/controls in home assistant. It´s Zigbee Tuya Temp & Humi Display with integr. Clock.

Thanks in advance for the support.

[image]
[Log.txt](https://github.com/zigpy/zha-device-handlers/files/7874865/Log.txt)

toy3375, 2022-01-15:
Additiional Information:

022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x62e0:1:0x0001] Attribute id: 32 value: 31.1
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x62e0:1:0x0001] Attribute id: 33 value: 200
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0xfb52:1:0x0001] Attribute id: 32 value: 28
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0xf2c7:1:0x0001] Attribute id: 33 value: 94
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x106c:1:0x0006] Attribute id: 0 value: 0
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x106c:1:0x0008] Attribute id: 0 value: 254
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0x0000] Attribute id: 4 value: _TZE200_locansqn
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0x0000] Attribute id: 5 value: TS0601
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0x0001] Attribute id: 49 value: 3
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0x0001] Attribute id: 51 value: 2
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0x0000] Attribute id: 1 value: 67
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0x0000] Attribute id: 65506 value: 49
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0x0000] Attribute id: 65508 value: 0
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0xef00] Attribute id: 513 value: 217
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0xef00] Attribute id: 514 value: 60
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0xef00] Attribute id: 516 value: 100
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0x0001] Attribute id: 33 value: 200
2022-01-15 20:45:22 DEBUG (MainThread) [zigpy.appdb] [0x4f93:1:0x0000] Attribute id: 65503 value: h�u)o�u)ep�u)

toy3375, 2022-01-15:
Attr:  513  Temp /10
Attr: 514 Humidity
Attr: 516 Batterie

toy3375, 2022-01-17:
now i need a little help . The Values don´t appear on HA. 
I dont know how to change the Programm.

I modified this 
[ts0601_temperature.py.txt](https://github.com/zigpy/zha-device-handlers/files/7884632/ts0601_temperature.py.txt)

toy3375, 2022-01-17:
[image]

toy3375, 2022-01-17:
Some Log...

2022-01-17 22:09:41 INFO (MainThread) [ts0601_temperature] toy3375 514, 56
2022-01-17 22:09:41 INFO (MainThread) [ts0601_temperature] toy3375 humid, 56
2022-01-17 22:09:41 ERROR (MainThread) [zigpy_deconz.uart] Unexpected error handling the frame: 'measured_value'
2022-01-17 22:09:42 DEBUG (MainThread) [zhaquirks.tuya] [0x77f3:1:0xef00] Received value [2] for attribute 0x040e (command 0x0002)
2022-01-17 22:09:42 DEBUG (MainThread) [zhaquirks.tuya] [0x77f3:1:0xef00] Received value [0, 0, 0, 100] for attribute 0x0204 (command 0x0002)
2022-01-17 22:09:42 INFO (MainThread) [ts0601_temperature] toy3375 516, 100
2022-01-17 22:09:42 INFO (MainThread) [ts0601_temperature] toy3375 batt, 100

jacekk015, 2022-01-21:
Enable debug logs - if you didn't that already:
```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```
Place that file in zha_custom_quirks
[ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7913051/ts0601_temperature.py.zip)
Restart HA.
Remove and pair device.
Try, and if have errors attach here whole, I mean WHOLE, HA log as a file.

toy3375, 2022-01-21:
[Whole_Log.txt](https://github.com/zigpy/zha-device-handlers/files/7916313/Whole_Log.txt)

jacekk015, 2022-01-22:
@toy3375

unk_model means that device wasn't properly initialized. You need to remove it and add again. Maybe after HA restart.

Your device sends those attributes:
```
2022-01-21 19:19:08 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0] for attribute 0x0409 (command 0x0002)
2022-01-21 19:19:08 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 1, 134] for attribute 0x020a (command 0x0002)
2022-01-21 19:19:08 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x020b (command 0x0002)
2022-01-21 19:19:08 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 0, 60] for attribute 0x020c (command 0x0002)
2022-01-21 19:19:08 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x020d (command 0x0002)
2022-01-21 19:19:08 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 0, 120] for attribute 0x0211 (command 0x0002)
2022-01-21 19:19:08 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 0, 120] for attribute 0x0212 (command 0x0002)
2022-01-21 19:19:08 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 0, 6] for attribute 0x0213 (command 0x0002)
2022-01-21 19:19:09 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 0, 6] for attribute 0x0214 (command 0x0002)
2022-01-21 19:19:19 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 0, 214] for attribute 0x0201 (command 0x0002)
2022-01-21 19:19:19 ERROR (MainThread) [zigpy_deconz.uart] Unexpected error handling the frame: 'measured_value'
2022-01-21 19:19:19 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 0, 51] for attribute 0x0202 (command 0x0002)
2022-01-21 19:19:19 ERROR (MainThread) [zigpy_deconz.uart] Unexpected error handling the frame: 'measured_value'
2022-01-21 19:19:19 DEBUG (MainThread) [zhaquirks.tuya] [0xc1f7:1:0xef00] Received value [0, 0, 0, 100] for attribute 0x0204 (command 0x0002)
```

0x0204 look like battery
0x0201 temperature
0x0202 humidity
It should work for 99.9%

That error for "measured_value" is strange. Which HA version do you use??
First repair it since quirk wasn't properly loaded.

toy3375, 2022-01-23:
Ok.
At first i deleted the Device then i restarted HA.
After few minutes i added Device again.  Now i see the battery stats and on ZigBeee  device the Wifi Signal is on.
But i don´t see Temp and Humi.
here some photos and Log file.

[image]

[image]

[image]

[Whole_Log_20220123_103800.txt](https://github.com/zigpy/zha-device-handlers/files/7920167/Whole_Log_20220123_103800.txt)

toy3375, 2022-01-23:
i found in the Log:

2022-01-23 09:59:17 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.tze200_locansqn_ts0601_23e48c55_power
2022-01-23 09:59:17 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (a4:c1:38:92:55:8c:e4:23, 1, 1029, 0) params: FOREIGN KEY constraint failed
2022-01-23 09:59:17 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (a4:c1:38:92:55:8c:e4:23, 1, 1026, 0) params: FOREIGN KEY constraint failed
2022-01-23 09:59:17 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c4c000c00050002f310ffc3
2022-01-23 09:59:17 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (a4:c1:38:92:55:8c:e4:23, 1, 1, 32) params: FOREIGN KEY constraint failed
2022-01-23 09:59:17 DEBUG (MainThread) [zigpy.appdb] Error handling '_unsupported_attribute_added' event with (a4:c1:38:92:55:8c:e4:23, 1, 1, 33) params: FOREIGN KEY constraint failed
2022-01-23 09:59:24 DEBUG (MainThread) [zigpy.zcl] [0x10f3:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=199, command_id=514, function=0, data=[4, 0, 0, 0, 67])]
2022-01-23 09:59:24 DEBUG (MainThread) [zhaquirks.tuya] [0x10f3:1:0xef00] Received value [0, 0, 0, 67] for attribute 0x0202 (command 0x0002)
2022-01-23 09:59:24 ERROR (MainThread) [zigpy_deconz.uart] Unexpected error handling the frame: 'measured_value'
address_mode=ADDRESS_MODE.NWK address=0x10f3>, 1, 260, 61184, b'\t.\x02\x00\xc6\x01\x02\x00\x04\x00\x00\x00\xdb', 0, 175, 215, 26, 166, 7, 0, -67]
2022-01-23 09:59:24 DEBUG (MainThread) [zigpy.zcl] [0x10f3:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=46 command_id=2>
2022-01-23 09:59:24 DEBUG (MainThread) [zigpy.zcl] [0x10f3:1:0xef00] ZCL request 0x0002: [Command(status=0, tsn=198, command_id=513, function=0, data=[4, 0, 0, 0, 219])]
2022-01-23 09:59:24 DEBUG (MainThread) [zhaquirks.tuya] [0x10f3:1:0xef00] Received value [0, 0, 0, 219] for attribute 0x0201 (command 0x0002)
2022-01-23 09:59:24 ERROR (MainThread) [zigpy_deconz.uart] Unexpected error handling the frame: 'measured_value'

toy3375, 2022-01-23:
After reload ZHA , it founds Temp and Humi without values.

[image]

jacekk015, 2022-01-23:
That code was written for the device, and it worked - it should work for you too.
It was written with new approach, but that should not be a problem here, since you have newest HA version.
Nevertheless here's same code rewritten in the older approach.
Replace and test it.
HA restart it's a minimum - best is to re-pair device.
[ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7921516/ts0601_temperature.py.zip)

toy3375, 2022-01-24:
YEAH.. looks good
1. Remove Tuya Device
2. Replace File
3. Repair Device ( only Battery shown )
4. Reload Intergration ( Temp & Humi avaible )

and now the Bonus-Level. Any Idea for the Clock ( Time ) ? :)

[image]

jacekk015, 2022-01-24:
> and now the Bonus-Level. Any Idea for the Clock ( Time ) ? :)

You think you will ever wanna to get that from the device?
Here's a list of your attributes. You need to check which is what, then everything is possible.
https://github.com/zigpy/zha-device-handlers/issues/1286#issuecomment-1019375279

BTW once paired device should sync wireless with HA time.

jacekk015, 2022-01-24:
Try version below.
Time offset wasn't sent, but I see that your device is making a request for time update.
Device MUST ask for time update - there's no possibility to force it.
Replace, Restart, Re-pair
[ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7927230/ts0601_temperature.py.zip)

toy3375, 2022-01-24:
There´s a new error in the Log:

2022-01-24 18:04:51 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:04:52 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:04:53 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:03 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:03 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:03 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:14 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:14 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Received value [2] for attribute 0x040f (command 0x0002)
**2022-01-24 18:05:14 ERROR (MainThread) [zigpy.device] Failed to parse message (b'098402003901') on cluster 61184, because Data is too short to contain 2 bytes**
**2022-01-24 18:05:25 ERROR (MainThread) [zigpy.device] Failed to parse message (b'098502005b01') on cluster 61184, because Data is too short to contain 2 bytes**
2022-01-24 18:05:35 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:35 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:35 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:35 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:35 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:36 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:36 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:36 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:36 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:36 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:36 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:36 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:37 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:37 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:47 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:47 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:48 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:48 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:48 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:48 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:48 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:48 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:48 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:48 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:49 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:49 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:49 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:05:50 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:06:10 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Received value [0] for attribute 0x0409 (command 0x0002)
2022-01-24 18:06:11 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Received value [0, 0, 1, 134] for attribute 0x020a (command 0x0002)
2022-01-24 18:06:11 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x020b (command 0x0002)
2022-01-24 18:06:11 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Received value [0, 0, 0, 60] for attribute 0x020c (command 0x0002)
2022-01-24 18:06:11 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x020d (command 0x0002)
2022-01-24 18:06:11 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Received value [0, 0, 0, 120] for attribute 0x0211 (command 0x0002)
2022-01-24 18:06:11 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Received value [0, 0, 0, 120] for attribute 0x0212 (command 0x0002)
[... cut ...]
2022-01-24 18:07:46 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:07:46 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:07:57 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:07:57 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)
2022-01-24 18:07:57 DEBUG (MainThread) [zhaquirks.tuya] [0x0c76:1:0xef00] Got set time request (command 0x0024)

jbmfg, 2022-01-25:
First, @jacekk015 thatnks for all the work here, i just got one of these devices and thought i was out of luck.  I have the same results as @toy3375.  I am wondering if could be possible to change the temperature units on the device itself to F.  The limited documentation said it could be done though their hub but im not seeing a way via HA

jacekk015, 2022-01-26:
@jbmfg
> could be possible to change the temperature units on the device itself to F

Probably yes. You just need to tell me which attribute is responsible for the unit change.

jbmfg, 2022-01-26:
These are the attributes i see for the temperature sensor: 

state_class: measurement 
unit_of_measurement: °F 
device_class: temperature

The units are F in the HA UI but on the device itself theyre still C
--

jacekk015, 2022-01-26:
@jbmfg 
I mean Zigbee attribute from the DEBUG logs.

toy3375, 2022-01-26:
Well , I try set some to F or to 1 in HA UI , but I don't know how to set attributes.  
Time cluster don't send Value too.. 
Tomorrow I will receive a TUYA Gateway. I will try to get some data from there..

toy3375, 2022-01-28:
yesterday i received my Tuya Mini Gateway.  
i paired the Device with it .
It sets the time correctly and activates the Temp and Humi Sensor .
In Tuya Smart App you can change the Temp from C to F:

My Workaround for now:  (Maybe for @jbmfg too )

1. Pair device to Tuya Gateway
2. set Device to C or F , wait for current Time.
3. then pair back to ZHA

toy3375, 2022-01-28:
here some Log Data after the workaround

[Log after Workaround.txt](https://github.com/zigpy/zha-device-handlers/files/7956571/Log.after.Workaround.txt)
.

toy3375, 2022-01-28:
a big **Thank you** for your help , @jacekk015 .

jacekk015, 2022-01-28:
Not sure what you did, but there's an attribute that changes value.
```
2022-01-28 07:45:39 DEBUG (MainThread) [zhaquirks.tuya] [0x7dd0:1:0xef00] Received value [1] for attribute 0x040f (command 0x0002)
2022-01-28 08:08:15 DEBUG (MainThread) [zhaquirks.tuya] [0x492a:1:0xef00] Received value [2] for attribute 0x040f (command 0x0002)
```
It's possible that 0x040f is responsible for the C/F setting.
Later I will modify the code and add that attribute, so you will have the possibility to check it.

Time setting is strange. Actually we never need to do anything in code because Tuya code in ZHA managaes Time request itself.

toy3375, 2022-01-28:
Here were are: 

i did some research with the TUYA Gateway. i found most attributes but the Time is still not working in ZHA. The Workaround is semigood . The Device Clock doesnt Sync after repairing to ZHA. Time after few hours begin to "running away"...

Here the results :

Received value [0, 0, 0, 87] for attribute 0x0204 (command 0x0002) // Current Battery power , Value = 87% battery capacity

Received value [0, 0, 0, 250] for attribute 0x0201 (command 0x0002) // Current Temp in C° , Value / 10 = 25,0 C°

Received value [0, 0, 0, 54] for attribute 0x0202 (command 0x0002) // Current Humidity in % , Value = 54%

Received value [0, 0, 0, 254] for attribute 0x020a (command 0x0002) // Max. Temp Alarm , Value / 10 = 25,4 C° , After override Attribute 201 change to: Received value [1] for attribute 0x040e (command 0x0002) , see below

Received value [0, 0, 0, 154] for attribute 0x020b (command 0x0002) // Min. Temp Alarm , Value / 10 = 15,4 C° , After underride Attribute 40E change to: Received value [0] for attribute 0x040e (command 0x0002) , see below

Received value [0, 0, 0, 61] for attribute 0x020c (command 0x0002) // max. Humidity Alarm , Value = 61% max. Humidity 

Received value [0, 0, 0, 21] for attribute 0x020d (command 0x0002) // min. Humidity Alarm , Vakue = 21% min. Humidity 

Received value [0, 0, 0, 6] for attribute 0x0213 (command 0x0002) // Sensitivity temp , Value = 6 / 2 = 3 C° ( Settings Tuya default 3C°  )

Received value [0, 0, 0, 3] for attribute 0x0214 (command 0x0002) // Sensitivity humidity , Value = 3% ( Settings Tuya default 3%  )

Received value [0, 0, 0, 120] for attribute 0x0211 (command 0x0002) // Report Temperature , Value = 120 min ( Settings Tuya default 120 Min. )

Received value [0, 0, 0, 120] for attribute 0x0212 (command 0x0002) // Report Humidity , Value = 120 min ( Settings Tuya default 120 Min. )

Received value [1] for attribute 0x0409 (command 0x0002) //  change C to F  0=C 1=F

Received value [1] for attribute 0x040E (command 0x0002) // alarm Temp 0 = underride , 1 = overide , 2 = all is fine :)

Received value [1] for attribute 0x040F (command 0x0002) // alarm Humidity 0 = underride , 1 = overide , 2 = all is fine :)

toy3375, 2022-01-28:
[image]

kkossev, 2022-01-28:
This device should be very similar (if not the same) as _TZE200_lve3dvpy
The clock synchronization should happen on Tuya request command 0x24, same way as for Tuya thermostats and TRVs

toy3375, 2022-01-28:
yes , i see the time request @0x24 but it doesnt work...

2022-01-28 20:34:21 DEBUG (MainThread) [zhaquirks.tuya] [0x5215:1:0xef00] Received value [0, 0, 0, 45] for attribute 0x0202 (command 0x0002)
2022-01-28 20:34:21 DEBUG (MainThread) [zhaquirks.tuya] [0x5215:1:0xef00] Got set time request (command 0x0024)
2022-01-28 20:34:27 DEBUG (MainThread) [zhaquirks.tuya] [0x5215:1:0xef00] Got set time request (command 0x0024)
2022-01-28 20:34:27 DEBUG (MainThread) [zhaquirks.tuya] [0x5215:1:0xef00] Got set time request (command 0x0024)
2022-01-28 20:34:27 DEBUG (MainThread) [zhaquirks.tuya] [0x5215:1:0xef00] Got set time request (command 0x0024)
2022-01-28 20:34:27 DEBUG (MainThread) [zhaquirks.tuya] [0x5215:1:0xef00] Got set time request (command 0x0024)
2022-01-28 20:34:27 DEBUG (MainThread) [zhaquirks.tuya] [0x5215:1:0xef00] Got set time request (command 0x0024)
2022-01-28 20:34:27 DEBUG (MainThread) [zhaquirks.tuya] [0x5215:1:0xef00] Got set time request (command 0x0024)
2022-01-28 20:34:28 DEBUG (MainThread) [zhaquirks.tuya] [0x5215:1:0xef00] Got set time request (command 0x0024)

toy3375, 2022-01-28:
i set in quirks:
 set_time_offset = 1970 
and i try 
set_time_offset = 2000
but still not working

jacekk015, 2022-01-28:
If you remember I've changed code to old type methods, a bus, instead of direct cluster name call.
Till this time that approach always worked, but for you not.
I see that you use Deconz - it's strange, because I have ZBBridge from Sonoff and 10pcs of Zigbee TRVs and all of them sync the time without problems.

toy3375, 2022-01-28:
ok . do you think i can put the mini Tuya gateway in HA as a device  ? How do use ZBBridge an HA ?

toy3375, 2022-01-28:
ok.. i google it. it's not possible.

toy3375, 2022-01-28:
i give up... :smile: maybe tomorow :crossed_fingers:

toy3375, 2022-01-28:
Logger: zigpy.device
Source: /usr/local/lib/python3.9/site-packages/zigpy/device.py:320
First occurred: 22:19:25 (6 occurrences)
Last logged: 22:19:30

Failed to parse message (b'09d00200bc00') on cluster 61184, because Data is too short to contain 2 bytes
Failed to parse message (b'09d10200bd00') on cluster 61184, because Data is too short to contain 2 bytes
Failed to parse message (b'09d20200be00') on cluster 61184, because Data is too short to contain 2 bytes
Failed to parse message (b'09d30200bf00') on cluster 61184, because Data is too short to contain 2 bytes
Failed to parse message (b'09d40200c000') on cluster 61184, because Data is too short to contain 2 bytes

is this the Time Cluster ?

jacekk015, 2022-01-28:
Need bigger log. 61184 is hex 0xef00 - so it's a Tuya cluster.
Probably not set time command
https://www.rapidtables.com/convert/number/hex-to-decimal.html?x=0xef00

toy3375, 2022-01-28:
[LOG20220128.txt](https://github.com/zigpy/zha-device-handlers/files/7961982/LOG20220128.txt)

Here the LOG...

jacekk015, 2022-01-28:
Replace the quirk file and restart HA.
Then remove and re-pair device.
[ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7962256/ts0601_temperature.py.zip)

mrctorres, 2022-01-29:
> Replace the quirk file and restart HA. Then remove and re-pair device. [ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7962256/ts0601_temperature.py.zip)

There is an error loading:

2022-01-28 21:50:12 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ts0601_temperature
2022-01-28 21:50:12 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry socket://192.168.26.135:8888 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 313, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 403, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/zha_custom_quirks/ts0601_temperature.py", line 49, in <module>
    class CustomTuyaOnOff(LocalDataCluster, OnOff):
NameError: name 'OnOff' is not defined

jbmfg, 2022-01-29:
NoNameError: name 'OnOff' is not defined

^^same error here

toy3375, 2022-01-29:
Yes , i confirm. Error on Startup.

File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 403, in setup
importer.find_module(modname).load_module(modname)
File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
File "<frozen importlib._bootstrap_external>", line 1029, in load_module
File "<frozen importlib._bootstrap_external>", line 854, in load_module
File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
File "<frozen importlib._bootstrap>", line 711, in _load
File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
File "<frozen importlib._bootstrap_external>", line 850, in exec_module
File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
File "/config/quirks/ts0601_temperature.py", line 49, in <module>
class CustomTuyaOnOff(LocalDataCluster, OnOff):
NameError: name 'OnOff' is not defined

toy3375, 2022-01-29:
i add @Imports "OnOff," in Line 21.
Now i see a lot of Functions 😃 ... i will try a little bit..

here the modificated File
[ts0601_temperature.zip](https://github.com/zigpy/zha-device-handlers/files/7964374/ts0601_temperature.zip)

[image]

toy3375, 2022-01-29:
after reload ZHA intergration i see Temp and Humidity on HA.
The communication works only in one way direction. From Device to HA looks good and Values on HA will synced.
But if i change a value and i force a sync ( press Button on device ). It change to old value ( it displays the value from device ).
I hope you understand 😄

toy3375, 2022-01-29:
here Logfile . I seems after changing a value , there´s a error more  in Log File 

[LogFilewith some Errors.txt](https://github.com/zigpy/zha-device-handlers/files/7964434/LogFilewith.some.Errors.txt)

jacekk015, 2022-01-29:
Here's corrected quirk.
[ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7964718/ts0601_temperature.py.zip)

There's no need to press anything on the Device. If the device sends data it overwrites HA data.

Do one thing in a time.
Change one value, if it's edit field hit Enter, after few seconds it should be sent to device.
Write here what value you've changed and post logs.

All errors in the log are connected with Time sync:
```
Got set time request (command 0x0024)
```

mrctorres, 2022-01-29:
> Here's corrected quirk. [ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7964718/ts0601_temperature.py.zip)
>
> There's no need to press anything on the Device. If the [...]

Here I changed the temperature reporting time value to 60:
2022-01-29 18:35:30 DEBUG (MainThread) [zigpy.zcl] [0xab4b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=40 command_id=Command.Default_Response>
2022-01-29 18:35:30 DEBUG (MainThread) [zigpy.zcl] [0xab4b:1:0xef00] ZCL request 0x000b: [0, <Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>]

mrctorres, 2022-01-29:
> Here's corrected quirk. [ts0601_temperature.py.zip](https://github.com/zigpy/zha-device-handlers/files/7964718/ts0601_temperature.py.zip)
>
> There's no need to press anything on the Device. If the [...]

Here I changed the temperature sensitivity to 1:
2022-01-29 18:38:55 DEBUG (MainThread) [zigpy.zcl] [0xab4b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=44 command_id=Command.Default_Response>
2022-01-29 18:38:55 DEBUG (MainThread) [zigpy.zcl] [0xab4b:1:0xef00] ZCL request 0x000b: [0, <Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>]

mrctorres, 2022-01-29:
The battery percentage is also no longer showing in this new version. It's showing up as unknown.

mrctorres, 2022-01-29:
Full logs:

[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/7964842/home-assistant.log)

jacekk015, 2022-01-29:
> Here I changed the temperature sensitivity to 1:
> 2022-01-29 18:38:55 DEBUG (MainThread) [zigpy.zcl] [0xab4b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=44 command_id=Command.Default_Response>
> 2022-01-29 18:38:55 DEBUG (MainThread) [zigpy.zcl] [0xab4b:1:0xef00] ZCL request 0x000b: [0, <Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>]

And the logs start from
```
2022-01-29 18:47:24
```

Only errors I see are for 0x55be eWeLink MS01

mrctorres, 2022-01-29:
> 

This last log was after an HA reset and a command to change a device parameter. I was hoping it might be useful. The eWeLink MS01 is currently turned off. And even after almost an hour, the battery percentage is still unknown.

mrctorres, 2022-01-29:
Now I removed the device, took out its 3 AAA batteries, restarted the HA, added the device again and recharged the ZHA, otherwise the temperature, humidity values ​​do not appear. Now the values ​​of the configurable attributes did not appear. The battery percentage is back. I hope it helps.

[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/7964909/home-assistant.log)

toy3375, 2022-01-30:
After Changing a value (ex. Humidity Alarm ) it will jump back to old value after a while ( at same time LOG Files receive Value from Device with old Value ) 
It seems that Value will not transmit to Device. Can you put some debug String in LOG after Sending or Polling a Value ?
Thank you for your patience :+1:

toy3375, 2022-01-30:
I made value change at 11:34 and 11:35

2022-01-30 11:33:06 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0] for attribute 0x0409 (command 0x0002)
2022-01-30 11:33:06 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 1, 14] for attribute 0x020a (command 0x0002)
2022-01-30 11:33:06 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 100] for attribute 0x020b (command 0x0002)
2022-01-30 11:33:17 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 70] for attribute 0x020c (command 0x0002)
2022-01-30 11:33:17 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 30] for attribute 0x020d (command 0x0002)
2022-01-30 11:33:22 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x0211 (command 0x0002)
2022-01-30 11:33:22 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x0212 (command 0x0002)
2022-01-30 11:33:27 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 6] for attribute 0x0213 (command 0x0002)
2022-01-30 11:33:28 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 6] for attribute 0x0214 (command 0x0002)
2022-01-30 11:33:33 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:33:33 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [1] for attribute 0x040f (command 0x0002)
2022-01-30 11:33:38 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 222] for attribute 0x0201 (command 0x0002)
2022-01-30 11:33:39 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 83] for attribute 0x0202 (command 0x0002)
2022-01-30 11:33:39 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:33:39 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:33:39 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:33:39 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0] for attribute 0x0409 (command 0x0002)
2022-01-30 11:33:39 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:33:40 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:33:40 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:33:43 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0] for attribute 0x0409 (command 0x0002)
2022-01-30 11:33:43 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 1, 14] for attribute 0x020a (command 0x0002)
2022-01-30 11:33:43 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 100] for attribute 0x020b (command 0x0002)
2022-01-30 11:33:48 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 70] for attribute 0x020c (command 0x0002)
2022-01-30 11:33:49 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 30] for attribute 0x020d (command 0x0002)
2022-01-30 11:33:49 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x0211 (command 0x0002)
2022-01-30 11:33:54 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 20] for attribute 0x0212 (command 0x0002)
2022-01-30 11:33:55 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 6] for attribute 0x0213 (command 0x0002)
2022-01-30 11:33:55 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 6] for attribute 0x0214 (command 0x0002)
2022-01-30 11:34:00 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:34:00 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 222] for attribute 0x0201 (command 0x0002)
2022-01-30 11:34:05 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 81] for attribute 0x0202 (command 0x0002)
2022-01-30 11:34:16 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:34:16 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:34:16 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:34:16 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:34:16 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:34:16 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:34:16 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0] for attribute 0x0409 (command 0x0002)
2022-01-30 11:34:17 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 1, 14] for attribute 0x020a (command 0x0002)
2022-01-30 11:34:22 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Received value [0, 0, 0, 100] for attribute 0x020b (command 0x0002)
2022-01-30 11:34:22 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
[... cut ...]
2022-01-30 11:35:54 DEBUG (MainThread) [zhaquirks.tuya] [0x4629:1:0xef00] Got set time request (command 0x0024)
2022-01-30 11:36:28 DEBUG (MainThread) [zhaquirks.tuya] [0x4f42:1:0xef00] Received value [0] for attribute 0x0409 (command 0x0002)
2022-01-30 11:36:28 DEBUG (MainThread) [zhaquirks.tuya] [0x4f42:1:0xef00] Received value [0, 0, 1, 4] for attribute 0x020a (command 0x0002)
2022-01-30 11:36:34 DEBUG (MainThread) [zhaquirks.tuya] [0x4f42:1:0xef00] Received value [0, 0, 0, 190] for attribute 0x020b (command 0x0002)
2022-01-30 11:36:34 DE

toy3375, 2022-01-30:
Maybe useful information.
In TUYA Smartlife APP the value will change after next wake-up. I can force it , when I blow a little bit in to the device.

[image]

mrctorres, 2022-01-30:
From what I could see, the attributes are being updated correctly. Here's the log after setting the temperature reporting time to 60:

2022-01-30 09:40:56 DEBUG (MainThread) [zigpy.zcl] [0x1985:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=236 command_id=Command.Default_Response>
2022-01-30 09:40:56 DEBUG (MainThread) [zigpy.zcl] [0x1985:1:0xef00] ZCL request 0x000b: [0, <Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>]

[image]

mrctorres, 2022-01-30:
Here the log after reading the attribute via the Manage Clusters panel:

2022-01-30 09:45:06 DEBUG (MainThread) [homeassistant.components.zha.api] Get bindable devices: source_ieee: [a4:c1:38:ce:dc:e0:24:35], bindable devices: [[{'ieee': '80:4b:50:ff:fe:0e:a4:6e', 'nwk': 0x0000, 'manufacturer': 'Silicon Labs', 'model': 'EZSP', 'name': 'Silicon Labs EZSP', 'quirk_applied': True, 'quirk_class': 'bellows.zigbee.application.EZSPCoordinator', 'manufacturer_code': 43981, 'power_source': 'Mains', 'lqi': 255, 'rssi': 0, 'last_seen': '2022-01-30T07:34:56', 'available': False, 'device_type': 'Coordinator', 'signature': {'node_descriptor': 'NodeDescriptor(logical_type=<LogicalType.Coordinator: 0>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice|AlternatePanCoordinator: 143>, manufacturer_code=43981, maximum_buffer_size=82, maximum_incoming_transfer_size=128, server_mask=11329, maximum_outgoing_transfer_size=128, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=True, *is_coordinator=True, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)', 'endpoints': {1: {'profile_id': 260, 'device_type': '0xbeef', 'in_clusters': [], 'out_clusters': []}}}, 'entities': [], 'neighbors': [{'device_type': 'EndDevice', 'rx_on_when_idle': 'Off', 'relationship': 'Child', 'extended_pan_id': 'cc:cc:cc:cc:9d:a8:cc:84', 'ieee': 'a4:c1:38:76:9f:6f:a2:a3', 'nwk': '0xF757', 'permit_joining': 'NotAccepting', 'depth': '1', 'lqi': '181'}, {'device_type': 'EndDevice', 'rx_on_when_idle': 'Off', 'relationship': 'Child', 'extended_pan_id': 'cc:cc:cc:cc:9d:a8:cc:84', 'ieee': '00:12:4b:00:22:cf:90:04', 'nwk': '0x82D1', 'permit_joining': 'NotAccepting', 'depth': '1', 'lqi': '112'}, {'device_type': 'EndDevice', 'rx_on_when_idle': 'Off', 'relationship': 'Child', 'extended_pan_id': 'cc:cc:cc:cc:9d:a8:cc:84', 'ieee': '00:12:4b:00:23:96:ae:96', 'nwk': '0x5CA2', 'permit_joining': 'NotAccepting', 'depth': '1', 'lqi': '108'}, {'device_type': 'EndDevice', 'rx_on_when_idle': 'Off', 'relationship': 'Child', 'extended_pan_id': 'cc:cc:cc:cc:9d:a8:cc:84', 'ieee': 'a4:c1:38:ce:dc:e0:24:35', 'nwk': '0x1985', 'permit_joining': 'NotAccepting', 'depth': '1', 'lqi': '139'}, {'device_type': 'EndDevice', 'rx_on_when_idle': 'Off', 'relationship': 'Child', 'extended_pan_id': 'cc:cc:cc:cc:9d:a8:cc:84', 'ieee': '00:12:4b:00:23:ad:5b:b3', 'nwk': '0x58A0', 'permit_joining': 'NotAccepting', 'depth': '1', 'lqi': '233'}], 'endpoint_names': [{'name': 'undefined_0xbeef'}], 'user_given_name': None, 'device_reg_id': 'f883394d03a3d2da871821f400ab4581', 'area_id': 'sala_de_estar'}]]
2022-01-30 09:45:06 DEBUG (MainThread) [homeassistant.components.zha.api] Requested attributes for: cluster_id: 13, cluster_type: 'in', endpoint_id: 12, response: [{'id': 28, 'name': 'description'}, {'id': 65, 'name': 'max_present_value'}, {'id': 69, 'name': 'min_present_value'}, {'id': 81, 'name': 'out_of_service'}, {'id': 85, 'name': 'present_value'}, {'id': 103, 'name': 'reliability'}, {'id': 104, 'name': 'relinquish_default'}, {'id': 106, 'name': 'resolution'}, {'id': 111, 'name': 'status_flags'}, {'id': 117, 'name': 'engineering_units'}, {'id': 256, 'name': 'application_type'}]
2022-01-30 09:45:06 DEBUG (MainThread) [homeassistant.components.zha.api] Requested commands for: cluster_id: 13, cluster_type: 'in', endpoint_id: 12, response: []
2022-01-30 09:45:19 DEBUG (MainThread) [homeassistant.components.zha.api] Requested attributes for: cluster_id: 13, cluster_type: 'in', endpoint_id: 12, response: [{'id': 28, 'name': 'description'}, {'id': 65, 'name': 'max_present_value'}, {'id': 69, 'name': 'min_present_value'}, {'id': 81, 'name': 'out_of_service'}, {'id': 85, 'name': 'present_value'}, {'id': 103, 'name': 'reliability'}, {'id': 104, 'name': 'relinquish_default'}, {'id': 106, 'name': 'resolution'}, {'id': 111, 'name': 'status_flags'}, {'id': 117, 'name': 'engineering_units'}, {'id': 256, 'name': 'application_type'}]
2022-01-30 09:45:19 DEBUG (MainThread) [homeassistant.components.zha.api] Requested commands for: cluster_id: 13, cluster_type: 'in', endpoint_id: 12, response: []
2022-01-30 09:45:32 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [13] cluster_type: [in] endpoint_id: [12] attribute: [85] manufacturer: [None] response: [60.0] failure: [{}],

[image]

mrctorres, 2022-01-30:
Here the log after changing the attribute to 30 via the panel:

2022-01-30 09:48:26 DEBUG (MainThread) [zigpy.zcl] [0x1985:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=240 command_id=Command.Default_Response>
2022-01-30 09:48:26 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x1985](TS0601): set: 30 for attr: 85 to cluster: 13 for ept: 12 - res: ([WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)],)
2022-01-30 09:48:26 DEBUG (MainThread) [homeassistant.components.zha.api] Set attribute for: cluster_id: [13] cluster_type: [in] endpoint_id: [12] attribute: [85] value: [30] manufacturer: [None] response: [([WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)],)]

mrctorres, 2022-01-30:
But update date and time on the device still doesn't work.

jacekk015, 2022-01-30:
@mrctorres I'm getting lost here.
Some say it works, other it doesn't.

Works through HA or only Manage Clusters??
It should work either way.

[226 more replies]

closed 2025-01-23 by TheJulianJES