zigpy/zha-device-handlers #1288: [Device Support Request]

SimpleUserHA, 2022-01-16:
_This post is a copy of another thead in the HA community forums: https://community.home-assistant.io/t/muller-tint-rgb-remote-zha-quirks/379426_

**Additional context**
in the community multiple blueprints were posted already for zigbee2mqtt related to the Müller tint remote.

Examples:
* https://community.home-assistant.io/t/z2m-tint-remote/295520
* https://community.home-assistant.io/t/muller-licht-tint-remote/283190
* https://community.home-assistant.io/t/zigbee2mqtt-tint-remote-fully-functional/356931

Additional links:
* the ZHA quirk for this remote: https://github.com/zigpy/zha-device-handlers/tree/dev/zhaquirks/mli
* the remote at blakadder: https://zigbee.blakadder.com/Muller_Licht_404011.html

**Is your feature request related to a problem? Please describe.**
Since I am using ZHA, I was checking the events **manually**, which were thrown by the remote and documented all of them over here:   https://github.com/SimpleUserHA/MuellerTintRGBRemote

**Describe the solution you'd like**
Does it somehow make sense to wrap some of those parameters and data into the zha quirk, to make button presses easier available and accessible for a wider range of users who don't want to deal with such low level kind of event parameters?

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
(copy from HA device signature,    the one from the log after pairing, I will be able to provide in a few hours, if required)

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4635, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0800",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x1000"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "MLI",
  "model": "ZBT-Remote-ALL-RGBW",
  "class": "zhaquirks.mli.tint.TintRemote"
}`

MattWestb, 2022-01-16:
Its looks like you have "finding" most / all commands the remote is sending.
Adding device automations (DA) in the quirk shall not being so demanding and then user can using it easy and dont knowing every command the remote is sending.
One device with easy done DA is IKEA 5 button remote. Take one look how they is using the commands the remote is sending and do the same with yours and you is getting it working.
https://github.com/zigpy/zha-device-handlers/blob/6227c8b08403f536e13e9c48b647f66600f06145/zhaquirks/ikea/fivebtnremotezha.py#L114

SimpleUserHA, 2022-01-16:
Hi @MattWestb ,

thanks for your hint related to device_automation_triggers. It was a good start to get a first draft for better remote support.

I created a branch in my own fork, which I would use for a pull request later:
https://github.com/SimpleUserHA/zha-device-handlers/blob/SimpleUserHA-tint-dat-1/zhaquirks/mli/tint.py

Until then I have a few issues to solve:
 * I was not coding pything for HA before and never compiled / or developed zha on my own. I was not testing the code on my own and need to find out how to get that cusomized tint.py in my own home assistant first.
 * I added some todos to the code:
    * [ ] [L170](https://github.com/SimpleUserHA/zha-device-handlers/blob/4ef4933a0f937aa8106c64b516310d96049a0bd4/zhaquirks/mli/tint.py#L170): When the color wheel button is pressed, an event is being sent with a random value in the args attribute for the definition of the color. (see https://github.com/SimpleUserHA/MuellerTintRGBRemote#color-wheel)
    Not sure if I have to leave out the color wheel completely or whether the color definition is standardized somewhere.
    * [ ] [L180](https://github.com/SimpleUserHA/zha-device-handlers/blob/4ef4933a0f937aa8106c64b516310d96049a0bd4/zhaquirks/mli/tint.py#L180): When the color temperate buttons are pressed (or long_pressed), an event is being sent multiple times with cycling though different values in the args attribute. (see https://github.com/SimpleUserHA/MuellerTintRGBRemote#color-temperature)
    * [ ] [L221](https://github.com/SimpleUserHA/zha-device-handlers/blob/4ef4933a0f937aa8106c64b516310d96049a0bd4/zhaquirks/mli/tint.py#L221): The brighness up and down button both send the same and equal event on **long_press_release** (see https://github.com/SimpleUserHA/MuellerTintRGBRemote#brightness)
    So I left out the DIM_UP and DIM_DOWN spec for this event.

Maybe you can give me a hint how to handle them correctly?
Any additional feedback to the current code added so far is very welcome.

MattWestb, 2022-01-16:
First configuring local quirl in ZHA https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274. Put you updated quirk in the folder you was making and restarting HA.
If all is going well you can see that the device have loading you new quirk by looking on the device signature and the last line is saying witch quirk is using.

For fixing the formatting so the code is working and passing test then making PRs use back playground https://black.vercel.app/ and putt the code in the right window and you is getting the fixed in the left.

Also then you is making changes it can being that you must deleting the   `__pycache__` in your local quirk folder before restarting HA so the system is not reusing old code and data saved in it and your quirk is misbehaving.

On thing with local quirk is it using all code from the integrated quirks so you can doing changes in your tint.py quirk but if you is also doing changes in the (tint) INIT is your local quirk getting the data from the integrated (tint) INIT and **not** the local ones. If need doing inherages changes it must being done in the HA container for working OK.

Commands / constans used in ZHA you is finding in https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/const.py.

Im not one code master so i cant helping so much then i normally using copy and paste and see if its working or not :-((

Also look in different quirks how they have doing the device automations and you can copy how its working and using it in your device.

MattWestb, 2022-01-16:
If you like have stop command after long pressing dim up and down  with direction (the remote is only sending stop) you can looking on this that is making the stop after up / down (or more correct open close) and shall being easy adapting.
https://github.com/zigpy/zha-device-handlers/blob/6227c8b08403f536e13e9c48b647f66600f06145/zhaquirks/ikea/opencloseremote.py#L47-L76

Make first the easter DA for the "normal" light commands like On/Off and Dimm up/down and getting them working OK before doing the deep digging for the more advanced functions.

SimpleUserHA, 2022-01-17:
Okay, it seems I am not able to get the quirk running using the full quirk from
already.

So I slimmed it down to the scene buttons, which are being made available as device triggers in home assistant, but unfortunately when pressing the buttons HA does not trigger. Don't know whats wrong, also I tried to setup the battery configuration for the 2AA batteries, but these are also not listed in HA correctly.  looks like this is out of my scope at least in a time , which is worth it.

_Remark:  I also deleted the pycache directory in mli, but it was not operating correctly._

so hopefully my branches and the documentation for the remote might help somebody else to implement it.

MattWestb, 2022-01-18:
I dont knowing but  perhaps ZHA is not knowing your new added functions in:
https://github.com/SimpleUserHA/zha-device-handlers/blob/f3a8fd00490fbc3006cde953db583428ab60deed/zhaquirks/mli/tint.py#L47-L51.
Did you getting the "standard" functions working (on/off and dimm up and down) ?

The AAA*2 i have implanting for IKEA https://github.com/zigpy/zha-device-handlers/blob/dcee551996a0a9e388b6afb1e054dada46897ee5/zhaquirks/ikea/__init__.py#L53-L64 (IKEA have one fix for doubling the battery % that comes from the quirk INIT).
Then the attribute is added you need reading them one time (then they is saved in the database) and then ZHA is showing them in GUI under the battery sensor. ZHA is not showing quantity and rated voltage for the moment.

natenjo, 2022-02-12:
I am not sure if I should add this to this issue, or to open a new request:

This is also regarding device support request for a Müller Licht RGB/Color remote control, but   not 100% the same as from the issue creator @SimpleUserHA (https://github.com/SimpleUserHA/MuellerTintRGBRemote).  I am refering to the following one:
**"Tint Fernbedienung weiß"**

- Müller Licht part number: 404049 (while mine actually  states 404049**B** on the back)
- The device pairs as TS1001 (but as not supported)
- On the back, next to the part number there is  a number called "PO" with the value 2100223 (not sure if this is the serial of the device)

Infos, see: 
-  [Amazon](https://www.amazon.de/tint-M%C3%BCller-Licht-Fernbedienung-vorprogrammierte-einstellbare/dp/B09LJ1JVJ5/ref=cm_cr_srp_d_product_top?ie=UTF8&th=1) 
- [Müller Licht](https://www.mueller-licht.de/produktinformationen/artikel/?q=404049)

When pairing with zigbee2mqtt (where this device is also not supported) the following info is logged:
`Device '0x2c1165fffe5e98b6' with Zigbee model 'TS1001' and manufacturer name '_TZ3000_bdbb0fon' is NOT supported,`

I will try and add infos what happens when pairing with ZHA later.

Is there a chance of getting this remote supported?

TheTimeWalker, 2022-03-04:
Hey @SimpleUserHA 
Thank you so much for researching this. For the scene buttons, you have made a very very simple mistake: The args you're expecting are in an array. When you look at the zha_event data, the args are only in a normal object and not in an array. This means you just have to change the scenes to look like this:

```
        # SCENE: worklight
        (SHORT_PRESS, "worklight"): {
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            CLUSTER_ID: 5,
            ENDPOINT_ID: 1,
            ARGS: {
                ATTRIBUTE_ID: 1,
                ATTRIBUTE_NAME: CURRENT_SCENE,
                VALUE: 3
                },
        },
```

With that, the commands should work fine

SimpleUserHA, 2022-03-05:
Hi @TheTimeWalker 
Thanks for your feedback. 👍  I changed it accordingly et voila...  Now it is working...
Going to check for the missing buttons soon as well.

TheJulianJES, 2022-04-17:
Hi, do you currently have a custom working quirk for this remote?
If so, it would be nice if you could PR the changes to this repo or just send the latest quirk file here, so someone else can do it. Then, a custom quirk wouldn't be needed in the future anymore.
Thanks!

SimpleUserHA, 2022-04-19:
Hi @TheJulianJES 

I implemented the fix and the scene buttons were working. 

Due to personal reasons I did not continue to implement the other buttons properly.
Also I was thinking about switching to zigbee2mqtt, since it seems the button and also other devices are implemented propperly there already. 

So the change is available here 
https://github.com/SimpleUserHA/zha-device-handlers/commits/SimpleUserHA-tint-dat-1

Anyone can use it to make a fork and request a pull for the partially working quirk or improve it first.

florianjacob, 2022-10-18:
I'm still interested in device support for the Müller Licht Tint remote control.

tomjschwanke, 2023-01-29:
Model 404049D

Signature:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4635, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0800",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x100f"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "MLI",
  "model": "Remote Control",
  "class": "zigpy.device.Device"
}
```

TheJulianJES, 2023-03-08:
@tomjschwanke Probably won't work, but could you try this as a custom quirk: https://raw.githubusercontent.com/TheJulianJES/zha-device-handlers/tjj/tint_remote/zhaquirks/mli/tint.py

(Added the signature of your remote model at the bottom + changed the `Basic` and `Scenes` cluster like the other remote does. **If** this sends events, they should also be added as a device automation trigger.)

tomjschwanke, 2023-03-08:
@TheJulianJES sadly no, no zha_event that correlates to the device

closed 2023-09-11 by github-actions[bot]

cwm77, 2023-10-13:
This quirk does not work anymore. 

https://github.com/SimpleUserHA/zha-device-handlers/blob/4e85a04163a9f46dca020fa517782efdc8192b1e/zhaquirks/mli/tint.py

No events with zha_event at all.

What has been changed in the meanwhile?

cwm77, 2023-10-13:
> Hi @TheJulianJES
>
> I implemented the fix and the scene buttons were working.
>
> Due to personal reasons I did not continue to implement the other buttons properly. Also I was thinking about switching [...]

I have the exact same MLI remote and have tested the quirk. 

SimpleDescriptor(endpoint=1, profile=260, device_type=2048
device_version=1
input_clusters=[0, 3, 4096]
output_clusters=[4, 3, 6, 8, 768, 4096, 25, 0])

Unfortunately, it doesn't work, and I don't think it ever has.  Unfortunately, I'm not a dev, but with some logic, it's obvious e.g. that a loop was not closed properly (, missing) and within HA no clusters show up with this quirk either. It would be really great if someone with a clue could please take care of this. I am sure that the basic functions such as switching on and off, lighter and darker could be made functional quickly. The hundreds of owners of this remote control who want to use it in ZHA will thank you.