zigpy/zha-device-handlers #1289: [Device Support Request]  Tuya motion sensor RH3040

sergey-suloev, 2022-01-16:
Device info
--------------
RH3040
by TUYATEC-53o41joc
Connected via Zigbee Coordinator
Zigbee info
IEEE: 50:32:5f:ff:fe:49:55:9f
Nwk: 0xb284
Device Type: EndDevice
LQI: Unknown
RSSI: Unknown
Last Seen: 2022-01-16T23:49:28
Power Source: Battery or Unknown

signature 
```javascript
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "TUYATEC-53o41joc",
  "model": "RH3040",
  "class": "zigpy.device.Device"
}

```

sergey-suloev, 2022-01-16:
[image]

sergey-suloev, 2022-01-17:
The device is very well  supported by Z2M, see https://www.zigbee2mqtt.io/devices/RH3040.html

sergey-suloev, 2022-01-19:
Closing the issue as soon as maintainers have not shown interest in this

closed 2022-01-19 by sergey-suloev

kukudemajia, 2022-05-15:
The same issue.The device is very well supported by Z2M.

kukudemajia, 2022-05-15:
Please reopen this issue,thanks.