zigpy/zha-device-handlers #1290: [Device Support Request]  Tuya Gas Leakage sensor  TS0601

sergey-suloev, 2022-01-16:
Device info
TS0601
by _TZE200_ggev5fsl
Connected via Zigbee Coordinator
Zigbee info
IEEE: 04:cd:15:ff:fe:39:76:23
Nwk: 0x215f
Device Type: Router
LQI: Unknown
RSSI: Unknown
Last Seen: 2022-01-17T00:49:27
Power Source: Mains

signature 
```javascript
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0x040d",
        "0x042b",
        "0x042e",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ggev5fsl",
  "model": "TS0601",
  "class": "zhaquirks.tuya.air.ts0601_air_quality.TuyaCO2Sensor"
}
```

sergey-suloev, 2022-01-16:
see related discussion https://github.com/home-assistant/core/issues/64140

sergey-suloev, 2022-01-19:
Closing the issue as soon as maintainers have not shown interest in this

closed 2022-08-03 by github-actions[bot]

pbvdven, 2023-09-12:
I also have a gas leakage sensor _TZE204_mby4kbtq it pairs but no entities are exposed by the device i tried the mentioned quirk but it has an error when loading HA

Logger: zhaquirks
Source: custom_zha_quirks/ts0601_gas.py:22
First occurred: 19:50:35 (1 occurrences)
Last logged: 19:50:35

Unexpected exception importing custom quirk 'ts0601_gas'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 460, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_gas.py", line 22, in <module>
    from . import TuyaManufCluster, TuyaManufClusterAttributes
ImportError: attempted relative import with no known parent package

Device signature
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_mby4kbtq",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

pbvdven, 2023-09-12:
@sergey-suloev @Adminiuga I also have a gas leakage sensor _TZE204_mby4kbtq it pairs but no entities are exposed by the device i tried the mentioned quirk from https://github.com/home-assistant/core/issues/64140 but it has an error when loading HA any idea what i can do? thanks 

Logger: zhaquirks
Source: custom_zha_quirks/ts0601_gas.py:22
First occurred: 19:50:35 (1 occurrences)
Last logged: 19:50:35

Unexpected exception importing custom quirk 'ts0601_gas'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 460, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_gas.py", line 22, in <module>
    from . import TuyaManufCluster, TuyaManufClusterAttributes
ImportError: attempted relative import with no known parent package

Device signature
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_mby4kbtq",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}