zigpy/zha-device-handlers #1291: [Device Support Request] Linkind ZB-MotionSensor-D0003

1jk, 2022-01-16:
**Is your feature request related to a problem? Please describe.**

The sensor is recognized in Home Assistant as a motion sensor and reports motion in the logs as Alarm_1; the sensor also has a bright/dark sensor it reports as Alarm_2 on dark. The problem is that if either Alarm is active Home Assistant marks motion as "on" --- if it is dark the sensor does not time out after motion stops until it is bright again and the motion status updates making it useless for detecting repeated motion in dark or near-dark rooms. Note that it only updates the status of bright/dark when it reports on motion start or motion timeout.

**Describe the solution you'd like**

The motion entity reports motion status (Alarm_1) regardless of the light/dark status (Alarm_2). If possible, an entity that reports the light/dark update status. If possible, An additional entity for Tamper status.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4456, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0500",
        "0x0b05",
        "0xfc81"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "lk",
  "model": "ZB-MotionSensor-D0003",
  "class": "zigpy.device.Device"
}
`

**Additional context**

https://zigbee.blakadder.com/Linkind_ZS110040078.html
https://www.linkind.com/product/pir-motion-sensor-zigbee/

Output from activity log showing various states; bright and dark were achieved by placing sensor in front of computer monitor and changing between white and black background:

_NoTamper, Bright, NoMotion 0-0-0 0 48_
[0x3ab3:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=37 command_id=0>
[0x3ab3:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports: 48>, <bitmap8.0: 0>, 0, 0]
[0x3ab3:1:0x0500]: Updated alarm state: ZoneStatus.0

_NoTamper, Bright, Motion 0-0-1 1 49_
[0x3ab3:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=38 command_id=0>
[0x3ab3:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports|Alarm_1: 49>, <bitmap8.0: 0>, 0, 0]
[0x3ab3:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_1

_NoTamper, Dark, NoMotion 0-2-0 2 50_
[0x3ab3:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=32 command_id=0>
[0x3ab3:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports|Alarm_2: 50>, <bitmap8.0: 0>, 0, 0]
[0x3ab3:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_2

_NoTamper, Dark, Motion 0-2-1 3 51_
[0x3ab3:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=31 command_id=0>
[0x3ab3:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports|Alarm_2|Alarm_1: 51>, <bitmap8.0: 0>, 0, 0]
[0x3ab3:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_2|Alarm_1

_Tamper, Bright, NoMotion 4-0-0 4 52_
[0x3ab3:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=24 command_id=0>
[0x3ab3:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports|Tamper: 52>, <bitmap8.0: 0>, 0, 0]
[0x3ab3:1:0x0500]: Updated alarm state: ZoneStatus.0

_Tamper, Bright, Motion 4-0-1 5 53_
[... 195 lines cut ...]
2022-01-16 12:42:26 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0019] ZCL request 0x0001: [0, 4456, 33019, 1361248258, None]
2022-01-16 12:42:26 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0019] OTA query_next_image handler for 'lk ZB-MotionSensor-D0003': field_control=0, manufacture_id=4456, image_type=33019, current_file_version=1361248258, hardware_version=None
2022-01-16 12:42:26 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0019] No OTA image is available
2022-01-16 12:42:26 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x8009, EmberApsFrame(profileId=260, clusterId=25, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=4), 190, b'\x19\x04\x02\x98')
2022-01-16 12:42:59 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=39), 255, -80, 0x8009, 255, 255, b'\x18\x05\n!\x00 \xa8']
2022-01-16 12:42:59 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=5 command_id=Command.Report_Attributes>
2022-01-16 12:42:59 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0001] ZCL request 0x000a: [[Attribute(attrid=33, value=<TypeValue type=uint8_t, value=168>)]]
2022-01-16 12:42:59 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0001] Attribute report received: battery_percentage_remaining=168
2022-01-16 12:45:06 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1280, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=40), 255, -74, 0x8009, 255, 255, b'\x1ch\x11\x06\n\xfb\xff \x01']
2022-01-16 12:45:06 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=True> manufacturer=4456 tsn=6 command_id=Command.Report_Attributes>
2022-01-16 12:45:06 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0500] ZCL request 0x000a: [[Attribute(attrid=65531, value=<TypeValue type=uint8_t, value=1>)]]
2022-01-16 12:45:06 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0500] Attribute report received: 65531=1
2022-01-16 12:45:06 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1280, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=41), 255, -74, 0x8009, 255, 255, b'\x19\x07\x007\x00\x00\x00\x00\x00']
2022-01-16 12:45:06 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=7 command_id=0>
2022-01-16 12:45:06 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports|Tamper|Alarm_2|Alarm_1: 55>, <bitmap8.0: 0>, 0, 0]
[... cut ...]
2022-01-16 12:46:30 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=9 command_id=0>
2022-01-16 12:46:30 DEBUG (MainThread) [zigpy.zcl] [0x8009:1:0x0500] ZCL request 0x0000: [<ZoneStatus.Restore_reports|Supervision_reports|Alarm_2: 50>, <bitmap8.0: 0>, 0, 0]
2022-01-16 12:46:30 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x8009:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_2
`

jfrux, 2022-01-25:
Great job digging into this.  I'm also interested in getting this sensor working.  Mine seems to get stuck and doesn't want to do anything.

krazos, 2022-02-02:
I bought one of these motion sensors recently, and it has similarly been getting stuck on motion detected and refuses to clear until I remove and re-pair with ZHA.

jfrux, 2022-02-02:
Yeah… wonder if there is some way to solve this or if it’s just their firmware.  I do see there was solutions over on Hubitat forums but not sure what they ended up doing to make it work.

dmackenzie123, 2022-08-02:
Is there any resolution to this issue as yet? I just paired 3 of these motion sensors with ZHA yesterday after a long time running Zigbee2MQTT. Immediately they detected motion they stuck on detected and did not clear without a full delete and repair - only to do it again immediately. They work fine in Zigbee2MQTT. Currently running HA 2022.7.7, which I assume has a reasonably up to date ZHA

TheRealSimonMlr, 2023-01-24:
I would also be genuinely interested in a solution. 
I am **currently trying to learn** how to **write quirks properly**, but this one was maybe a little hard for the beginning 👯 

What I have done so far (for 8 Hours until now):

### → Created a quirk that matches the device and gets applied.
```
"""Linkind ZB-MotionSensor-D0003."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    Ota,
    PollControl,
    PowerConfiguration,
)
from zigpy.zcl.clusters.homeautomation import Diagnostic
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.linkind import LinkindBasicCluster

LINKIND_CLUSTER_ID = 0xFC81

class LinkindD0003(CustomDevice):
    """Linkind ZB-MotionSensor-D0003."""

    signature = {
        MODELS_INFO: [("lk", "ZB-MotionSensor-D0003")],
        # <SimpleDescriptor endpoint=1 profile=260 device_type=1026
        # device_version=1 
        # input_clusters=[0, 1, 3, 32, 1280, 2821, 64641]
        # output_clusters=[25]>
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    IasZone.cluster_id,
                    Diagnostic.cluster_id,
                    LINKIND_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    LinkindBasicCluster,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PollControl.cluster_id,
[... 362 lines cut ...]
          "0x0b05": {
            "endpoint_attribute": "diagnostic",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xfc81": {
            "endpoint_attribute": "manufacturer_specific",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
  ```
  
</details>

javicalle, 2023-01-24:
IMHO the `IasZone` needs a proper way to manage the 'extra' status:
* https://github.com/home-assistant/core/issues/72828

What I would suggest to you here will be:
* import the `MotionWithReset` class
```python
from zhaquirks import MotionWithReset
```
* use it in the replacement part:
```python
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    LinkindBasicCluster,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    MotionWithReset,
                    Diagnostic.cluster_id,
                    LINKIND_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }
```

This class will reset the sensor status at 30 seconds (if no movement detected).
If not working you can implement a custom cluster to do the same.

TheRealSimonMlr, 2023-01-24:
I can't believe it, **seems to work** 💙
[image]

It would be great if someone else verified the current state for a pull request. Even though I might end up creating a custom cluster for this (see below)... But if considered working, I would see this maybe as a basic Version for getting support in ZHA. 

<details>
  <summary>Current quirk</summary>
  
  ```
"""Linkind ZB-MotionSensor-D0003."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    Ota,
    PollControl,
    PowerConfiguration,
)
from zigpy.zcl.clusters.homeautomation import Diagnostic
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.linkind import LinkindBasicCluster
from zhaquirks import MotionWithReset

LINKIND_CLUSTER_ID = 0xFC81

class LinkindD0003(CustomDevice):
    """Linkind ZB-MotionSensor-D0003."""

    signature = {
        MODELS_INFO: [("lk", "ZB-MotionSensor-D0003")],
        # <SimpleDescriptor endpoint=1 profile=260 device_type=1026
        # device_version=1 
        # input_clusters=[0, 1, 3, 32, 1280, 2821, 64641]
        # output_clusters=[25]>
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    IasZone.cluster_id,
                    Diagnostic.cluster_id,
                    LINKIND_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    LinkindBasicCluster,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    MotionWithReset,
                    Diagnostic.cluster_id,
                    LINKIND_CLUSTER_ID,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }
  ```
  
</details>

So, just to get my Head around this: The MotionWithReset is basically taking the IasZone.cluster_id and interpreting the Return values differently (I remember the Translation example from the GitHub-Page)?

Thank you so much @javicalle  🙏
And 2 additional questions (So we even might beat Z2M's attributes):  

1. In Z2M there are also Tamper "Entities". Is there a way to "extract" them too? (I guess this would need a custom cluster, right?)

2. To set the timeout to 2 Minutes (if I remember this correct, this is the default Reporting-Interval for the linkind), something like this would be required?
```
class MotionCluster(MotionWithReset):
    """Motion cluster."""

    reset_s: int = 120
    send_occupancy_event: bool = True
```
(And afterward of course using MotionCluster instead of MotionWithReset in replacement part)

3. There is a Cluster which says, "current_zone_sensitivity_level". Would it be possible to set these values to? If I recall correctly, I saw some Tuya quirks defining an enum where values like 0=low 1=med... were defined. This would require a custom cluster, right?

javicalle, 2023-01-25:
I'm glad that you get it working.

> The MotionWithReset is basically taking the IasZone.cluster_id and interpreting the Return values differently (I remember the Translation example from the GitHub-Page)?

No. What is doing is starting a counter every time that a status report is recieved:
https://github.com/zigpy/zha-device-handlers/blob/d0732016c78c12b4f28930ea6affd385140af44e/zhaquirks/__init__.py#L267-L270

The counter (at `reset_s` seconds) will 'clear' the `IasZone` status (and trigger an event):
https://github.com/zigpy/zha-device-handlers/blob/d0732016c78c12b4f28930ea6affd385140af44e/zhaquirks/__init__.py#L242-L246

About your questions:

> 1. In Z2M there is also a Tamper "Entity". Is there a way to "extract" this too?

IMHO that is a discussion for the ZHA integration and not for the quirk side.

Currently the `ZoneStatus` can report a bunch of events/status:
https://github.com/zigpy/zigpy/blob/297a2b0f364fe7a31de3b4966d70d1a67d432358/zigpy/zcl/clusters/security.py#L41-L54

In ZHA, only the `Alarm_1` and `Alarm_2` are considerated.
I see 2 approachs here:
* that the `IasZone` sensor becomes a multi-state sensor and exposes all possible values
* that new sensors be implemented for the rest of the status (`IasTamper`, `IasBattery`, ...)

The first approach is less friendly to manage and in automatizations.
The second one have the problem that there is no way to know which status will report any device. Then what to do? Create all the entities for all the status? Just a few? Create in disabled mode?
In Z2M has been addressed in their own way and ZHA must do theirs.

> 3\. There is a Cluster which says, "current_zone_sensitivity_level". Would it be possible to set these values to? If I recall correctly, I saw some Tuya quirks defining an enum where values like 0=low 1=med... were defined. This would require a custom cluster, right?

IMHO the goal of any quirk is to fix any functionality of the device that does not work due to a bad implementation of the zigbee standard. Period.
That is a clear example: the device reports the IasZone status but device is not usefull because doesn't restore the 'clear' status. A quirk to the rescue.
Does the device reports/implements any 'sensitivity level'? If it does and isn't working, I would try to fix it. But if the device doesn't implements it, IMHO we should not try to implement it programmatically (or at least add it to the library).

TheRealSimonMlr, 2023-01-25:
Thank you so much @javicalle and the time you put in your answer 💙

That really helped me out understanding how to approach it the ZHA way and gave me the possibility to take a look at some Motion-Quirks which were already created (and understand them better).

Do you have a PayPal-Link where I cloud give you a little something for the time you took explaining this stuff to me? (And for some questions I might have in the future 😁)

javicalle, 2023-01-25:
I'm glad to have been helpful.
Can you create the PR with the working code?

> Do you have a PayPal-Link where I cloud give you a little something for the time you took explaining this stuff to me?

I appreciate the offer but I don't have.
@dmulcahey, are there any tip/coffee/beer jar for this offerings or donations?

PR #2130 (2023-01-26): Added support for Linkind D0003 Motion sensor
Added support for Linkind D0003 Motion sensor, special thanks to @javicalle - Closes #1291
closed 2023-01-26 by dmulcahey