zigpy/zha-device-handlers #1294: [Device Support Request] Zemismart Curtain Motor with Rail (Mains powered) (_TZE200_rmymn92d)

operinko, 2022-01-17:
**Is your feature request related to a problem? Please describe.**
ts0601_cover.py needs a new quirk added to support this device properly.
Currently, it pairs with no entities attached.

**Describe the solution you'd like**
Be able to pair the device and control my curtains through HA.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_rmymn92d",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

operinko, 2022-01-17:
I got it to add the Cover entity with this:
```
class TuyaZemismartCover0601_custom(TuyaWindowCover):
    """Tuya blind controller device."""

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4417,
        #                     maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752,
        #                     maximum_outgoing_transfer_size=66, descriptor_capability_field=0>,
        # "endpoints": {
        # "1": { "profile_id": 260, "device_type": "0x0051", "in_clusters": [ "0x0000", "0x0004","0x0005","0xef00"], "out_clusters": ["0x000a","0x0019"] },
        # "242": { "profile_id": 41440, "device_type": "0x0061", in_clusters": [], "out_clusters": [ "0x0021" ] }
        # },
        # "manufacturer": "_TZE200_rmymn92d",
        # "model": "TS0601",
        # "class": "zigpy.device.Device"
        # }
        MODELS_INFO: [
            ("_TZE200_rmymn92d", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 260,
                DEVICE_TYPE: 0x0051,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

Just.. The buttons all throw:
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 185, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1495, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1530, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 209, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 663, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 896, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 700, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 132, in async_close_cover
    res = await self._cover_channel.down_close()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 59, in wrapper
    result = await command(*args, **kwds)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 1039, in command
    TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_rmymn92d'
```

Kiread-work, 2022-01-18:
@operinko check this https://github.com/zigpy/zha-device-handlers/issues/744#issuecomment-1010750485
Buttons works now without errors, but nothing happens

raphaeujp, 2022-02-17:
Any update on support this device?

TheJulianJES, 2022-02-27:
@operinko When adding a custom quirk, you also want to update the information here: https://github.com/zigpy/zha-device-handlers/blob/e4e62504738ced0c54c983ee5094a3a6af0a741d/zhaquirks/tuya/__init__.py#L94-L111

If I remember correctly the structure for the custom quirk directory should look like this then:
``/config/custom_quirks/tuya``:
-- ``__init.py__`` (modified)
-- ``ts0601_cover.py`` (modified)

Kiread-work, 2022-02-27:
So you will see the device in HA, you can even check the status (open/close). But the buttons will not work
/config/custom_zha_quirks/TS0601_Covertest.py
```
"""Tuya based cover and blinds."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Scenes, Time, GreenPowerProxy
# from zigpy.zcl.clusters.closures import WindowCovering
# from zigpy.quirks import CustomCluster, CustomDevice
# import zigpy.types as t

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)

class TuyaCover0601_TO_GPP(TuyaWindowCover):
    """Tuya blind controller device with time on out and GPP."""

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098,
        #                    maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        #                    maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
        # "endpoints": {
        # "1": { "profile_id": 260, "device_type": "0x0051", "in_clusters": [ "0x0000", "0x0004","0x0005","0xef00"],
        # "out_clusters": ["0x000a","0x0019"] } },
        MODELS_INFO: [
            ("_TZE200_r0jdjrvi", "TS0601"),
            ("_TZE200_rmymn92d", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: 0x0051, # zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                    #WindowCovering.cluster_id
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0061, #97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                    #TuyaCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```
/config/custom_zha_quirks/__init__.py lines 109-110
```
    "_TZE200_r0jdjrvi": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_rmymn92d": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
```

SergeantPup, 2022-04-08:
@Kiread-work.  The new HA version borked my quirk.  Actually the custom quirk borked my ZHA network so I've removed it which means it cannot even track state of _TZE200_rmymn92d.  We have confirmation that the code is working in MQTT and I submitted a PR for the cover inversion in my journey to figure out the difference in the code but I think we're going to need a new custom quirk to use, is that correct?  Something has changed where HA really didn't like that custom quirk being there.

How are things looking for you? still the same?  This is now the last device in my home that HA cannot control :(

Kiread-work, 2022-04-09:
@SergeantPup shame on you =) Just updated HA to core-2022.4.1 on 5.10.108 HAOS and my Xiaomi Humidifier stoped working =))
Maybe I could find some time to restore integration with humidifier and windows cover _TZE200_r0jdjrvi

Kiread-work, 2022-04-13:
@SergeantPup well it works (status, not buttons). Nothing changed.
Button press drops another error now.

SergeantPup, 2022-04-13:
I put it back in and it borked my zha network again so I took it out again.  Tracking the curtain isn't THAT important without control.  This must be something trivial.  Halk tempted to just buy another head unit but I have a 50% shot it might be the same chip :(

TheJulianJES, 2022-04-13:
- Can you send the latest version of the quirk that used to "work"?
- Is there anything in the (debug) logs when the quirk screws up ZHA?

SergeantPup, 2022-04-14:
Hello Julian!

Here is the error I get when the quirk is present but makes the rest of my ZHA unresponsive:

```
Logger: homeassistant.config_entries
Source: custom_zha_quirks/tuya/__init__.py:263
First occurred: 8:04:45 PM (1 occurrences)
Last logged: 8:04:45 PM

Error setting up entry Zigbee Network for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/tuya/__init__.py", line 263, in <module>
    class TuyaManufCluster(CustomCluster):
  File "/usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py", line 91, in __init_subclass__
    raise TypeError(
TypeError: `manufacturer_client_commands` is deprecated. Copy the parent class's `client_commands` dictionary and update it with your manufacturer-specific `client_commands`. Make sure to specify that it is manufacturer-specific through the  appropriate constructor or tuple!
```
Here is the last known working quirk before the April update.
[tuya.zip](https://github.com/zigpy/zha-device-handlers/files/8485321/tuya.zip)

TheJulianJES, 2022-04-14:
You can try this ``tuya`` folder:
[tuya.zip](https://github.com/zigpy/zha-device-handlers/files/8485353/tuya.zip)

SergeantPup, 2022-04-14:
```
Logger: homeassistant.config_entries
Source: custom_zha_quirks/tuya/ts0601_cover.py:101
First occurred: 8:20:25 PM (1 occurrences)
Last logged: 8:20:25 PM

Error setting up entry Zigbee Network for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/tuya/ts0601_cover.py", line 71, in <module>
    class TuyaZemismartSmartCover0601_4(TuyaWindowCover):
  File "/config/custom_zha_quirks/tuya/ts0601_cover.py", line 101, in TuyaZemismartSmartCover0601_4
    OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
NameError: name 'GreenPowerProxy' is not defined

```

TheJulianJES, 2022-04-14:
[tuya.zip](https://github.com/zigpy/zha-device-handlers/files/8485372/tuya.zip)

SergeantPup, 2022-04-14:
ZHA network and device tracking of _TZE200_rmymn92d are concurrently working again with 2022.4.3.  Thank you!

I think I'm getting different errors for the _TZE200_rmymn92d command sends now (which is what the other user was reporting).

Here is a successful command on working device: _TZE200_xaabybja with Nwk: 0xd775 (this device works fine)
```
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 89 (incomingRouteRecordHandler) received: b'75d7abb7d1feff81f68cffb401acc7'
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0xd775, 8c:f6:81:ff:fe:d1:b7:ab, 255, -76, [0xc7ac]]
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0xd775, 8c:f6:81:ff:fe:d1:b7:ab, 255, -76, [0xc7ac])
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef010100010000cfffb775d7ffff0a09530100c7070400010004'
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=207), 255, -73, 0xd775, 255, 255, b'\tS\x01\x00\xc7\x07\x04\x00\x01\x00']
2022-04-13 20:45:41 DEBUG (MainThread) [zigpy.zcl] [0xD775:1:0xef00] Received ZCL frame: b'\tS\x01\x00\xc7\x07\x04\x00\x01\x00'
2022-04-13 20:45:41 DEBUG (MainThread) [zigpy.zcl] [0xD775:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=83, command_id=1, *is_reply=True)
2022-04-13 20:45:41 DEBUG (MainThread) [zigpy.zcl] [0xD775:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:get_data(param=Command(status=0, tsn=199, command_id=1031, function=0, data=[1, 0]))
2022-04-13 20:45:41 DEBUG (MainThread) [zigpy.zcl] [0xD775:1:0xef00] Received command 0x01 (TSN 83): get_data(param=Command(status=0, tsn=199, command_id=1031, function=0, data=[1, 0]))
2022-04-13 20:45:41 DEBUG (MainThread) [tuya] 8c:f6:81:ff:fe:d1:b7:ab Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 199, Command: 0x0407, Function: 0x00, Data: [1, 0]]
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 89 (incomingRouteRecordHandler) received: b'75d7abb7d1feff81f68cffb801acc7'
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0xd775, 8c:f6:81:ff:fe:d1:b7:ab, 255, -72, [0xc7ac]]
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0xd775, 8c:f6:81:ff:fe:d1:b7:ab, 255, -72, [0xc7ac])
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef010100010000d0ffb575d7ffff0d09540100c8030200040000006404'
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=208), 255, -75, 0xd775, 255, 255, b'\tT\x01\x00\xc8\x03\x02\x00\x04\x00\x00\x00d']
2022-04-13 20:45:41 DEBUG (MainThread) [zigpy.zcl] [0xD775:1:0xef00] Received ZCL frame: b'\tT\x01\x00\xc8\x03\x02\x00\x04\x00\x00\x00d'
2022-04-13 20:45:41 DEBUG (MainThread) [zigpy.zcl] [0xD775:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=84, command_id=1, *is_reply=True)
2022-04-13 20:45:41 DEBUG (MainThread) [zigpy.zcl] [0xD775:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:get_data(param=Command(status=0, tsn=200, command_id=515, function=0, data=[4, 0, 0, 0, 100]))
2022-04-13 20:45:41 DEBUG (MainThread) [zigpy.zcl] [0xD775:1:0xef00] Received command 0x01 (TSN 84): get_data(param=Command(status=0, tsn=200, command_id=515, function=0, data=[4, 0, 0, 0, 100]))
2022-04-13 20:45:41 DEBUG (MainThread) [tuya] 8c:f6:81:ff:fe:d1:b7:ab Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 200, Command: 0x0203, Function: 0x00, Data: [4, 0, 0, 0, 100]]
2022-04-13 20:45:41 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xD775:1:0x0102]: Attribute report 'Window Covering'[current_position_lift_percentage] = 100
2022-04-13 20:45:41 DEBUG (MainThread) [tuya] 8c:f6:81:ff:fe:d1:b7:ab Tuya Attribute Cache : [{8: 100}]
2022-04-13 20:45:41 DEBUG (MainThread) [homeassistant.components.zha.cover] setting position: 100
```

Here is the new error from _TZE200_rmymn92d with Nwk: 0xc7ac (this is the device that's accurately tracking state and not responding to button presses). 
```
2022-04-13 20:45:41 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4 [... cut ...]
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0xd775, 8c:f6:81:ff:fe:d1:b7:ab, 255, -72, [0xc7ac]]
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0xd775, 8c:f6:81:ff:fe:d1:b7:ab, 255, -72, [0xc7ac])
2022-04-13 20:45:41 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef010100010000d0ffb575d7ffff0d09540100c8030200040000006404'
```

TheJulianJES, 2022-04-14:
Just to confirm, do your debug logs (from ``[tuya]``) ever log something including the text ``Sending Tuya``?

SergeantPup, 2022-04-14:
I just ran an open/stop/close command on _TZE200_xaabybja 0xb41c.  61 entries for "0xb41c" and 38 entries for "tuya" (none say sending:

```
2022-04-13 21:31:34 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Received command 0x01 (TSN 89): get_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 2]))
2022-04-13 21:31:34 DEBUG (MainThread) [tuya] b4:e3:f9:ff:fe:0c:51:6e Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0401, Function: 0x00, Data: [1, 2]]
2022-04-13 21:31:35 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef010100010000b5ffbf1cb4ffff0a09590100000104000102'
2022-04-13 21:31:35 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=181), 255, -65, 0xb41c, 255, 255, b'\tY\x01\x00\x00\x01\x04\x00\x01\x02']
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Received ZCL frame: b'\tY\x01\x00\x00\x01\x04\x00\x01\x02'
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=89, command_id=1, *is_reply=True)
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:get_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 2]))
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Received command 0x01 (TSN 89): get_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 2]))
2022-04-13 21:31:35 DEBUG (MainThread) [tuya] b4:e3:f9:ff:fe:0c:51:6e Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0401, Function: 0x00, Data: [1, 2]]
2022-04-13 21:31:35 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef010100010000b6ffbf1cb4ffff0a09590100000104000102'
2022-04-13 21:31:35 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=182), 255, -65, 0xb41c, 255, 255, b'\tY\x01\x00\x00\x01\x04\x00\x01\x02']
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Received ZCL frame: b'\tY\x01\x00\x00\x01\x04\x00\x01\x02'
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=89, command_id=1, *is_reply=True)
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:get_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 2]))
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Received command 0x01 (TSN 89): get_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 2]))
2022-04-13 21:31:35 DEBUG (MainThread) [tuya] b4:e3:f9:ff:fe:0c:51:6e Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0401, Function: 0x00, Data: [1, 2]]
2022-04-13 21:31:35 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef010100010000b7ffbf1cb4ffff0a09590100000104000102'
2022-04-13 21:31:35 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=183), 255, -65, 0xb41c, 255, 255, b'\tY\x01\x00\x00\x01\x04\x00\x01\x02']
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Received ZCL frame: b'\tY\x01\x00\x00\x01\x04\x00\x01\x02'
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=89, command_id=1, *is_reply=True)
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:get_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 2]))
2022-04-13 21:31:35 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Received command 0x01 (TSN 89): get_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 2]))
2022-04-13 21:31:35 DEBUG (MainThread) [tuya] b4:e3:f9:ff:fe:0c:51:6e Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0401, Function: 0x00, Data: [1, 2]]
[... cut ...]
2022-04-13 21:31:37 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:get_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 2]))
2022-04-13 21:31:37 DEBUG (MainThread) [zigpy.zcl] [0xB41C:1:0xef00] Received command 0x01 (TSN 89): get_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 2]))
2022-04-13 21:31:37 DEBUG (MainThread) [tuya] b4:e3:f9:ff:fe:0c:51:6e Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0401, Function: 0x00, Data: [1, 2]]
```

SergeantPup, 2022-04-14:
The device that's tracking state but not controlling has "sending tuya" commands:

```
2022-04-13 21:39:18 DEBUG (MainThread) [bellows.ezsp.protocol] Send command nop: ()
2022-04-13 21:39:18 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''
2022-04-13 21:39:18 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=cover, service=open_cover, service_data=entity_id=cover.media_room_curtains_2>
2022-04-13 21:39:18 DEBUG (MainThread) [tuya] a4:c1:38:bd:76:6f:43:4c Sending Tuya Cluster Command.. Manufacturer is _TZE200_rmymn92d Cluster Command is 0x0000, Arguments are ()
2022-04-13 21:39:18 DEBUG (MainThread) [tuya] a4:c1:38:bd:76:6f:43:4c Sending Tuya Command. Paylod values [endpoint_id : 1, Status : 0, TSN: 0, Command: 0x0401, Function: 0, Data: [1, 0]]
2022-04-13 21:39:18 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4417, tsn=58, command_id=0, *is_reply=False)
2022-04-13 21:39:18 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 0]))
2022-04-13 21:39:18 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xC7AC, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=58), 59, b'\x05A\x11:\x00\x00\x00\x01\x04\x00\x01\x00')
2022-04-13 21:39:18 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'001a'
2022-04-13 21:39:18 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'00acc7040100ef0101400100001a3b0000'
2022-04-13 21:39:18 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 51116, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=26), 59, <EmberStatus.SUCCESS: 0>, b'']
2022-04-13 21:39:18 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef010100010000ccfdb6acc7ffff05183a0b008302'
2022-04-13 21:39:18 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=204), 253, -74, 0xc7ac, 255, 255, b'\x18:\x0b\x00\x83']
2022-04-13 21:39:18 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Received ZCL frame: b'\x18:\x0b\x00\x83'
2022-04-13 21:39:18 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=58, command_id=11, *is_reply=True)
2022-04-13 21:39:18 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2022-04-13 21:39:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC7AC:1:0x0102]: executed 'up_open' command with args: '()' kwargs: '{}' result: Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2022-04-13 21:39:20 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=cover, service=stop_cover, service_data=entity_id=cover.media_room_curtains_2>
2022-04-13 21:39:20 DEBUG (MainThread) [tuya] a4:c1:38:bd:76:6f:43:4c Sending Tuya Cluster Command.. Manufacturer is _TZE200_rmymn92d Cluster Command is 0x0002, Arguments are ()
2022-04-13 21:39:20 DEBUG (MainThread) [tuya] a4:c1:38:bd:76:6f:43:4c Sending Tuya Command. Paylod values [endpoint_id : 1, Status : 0, TSN: 0, Command: 0x0401, Function: 0, Data: [1, 1]]
2022-04-13 21:39:20 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4417, tsn=60, command_id=0, *is_reply=False)
2022-04-13 21:39:20 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 1]))
2022-04-13 21:39:20 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xC7AC, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=60), 61, b'\x05A\x11<\x00\x00\x00\x01\x04\x00\x01\x01')
2022-04-13 21:39:20 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'001b'
[... cut ...]
2022-04-13 21:39:20 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=60, command_id=11, *is_reply=True)
2022-04-13 21:39:20 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
```

TheJulianJES, 2022-04-14:
Can you see if there's any difference with this: [tuya.zip](https://github.com/zigpy/zha-device-handlers/files/8485715/tuya.zip)

SergeantPup, 2022-04-14:
[image]
Not sure if it helps but this is how the device adds with this quirk in place

TheJulianJES, 2022-04-14:
Can you completely delete and re-pair the device with the quirk update I've sent a minute ago?

SergeantPup, 2022-04-14:
removed device, swapped quirk, rebooted, added device:

```
2022-04-13 22:16:20 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=cover, service=open_cover, service_data=entity_id=cover.tze200_rmymn92d_ts0601_4c436f76_window_covering>
2022-04-13 22:16:20 DEBUG (MainThread) [tuya] a4:c1:38:bd:76:6f:43:4c Sending Tuya Cluster Command.. Manufacturer is _TZE200_rmymn92d Cluster Command is 0x0000, Arguments are ()
2022-04-13 22:16:20 DEBUG (MainThread) [tuya] a4:c1:38:bd:76:6f:43:4c Sending Tuya Command. Paylod values [endpoint_id : 1, Status : 0, TSN: 0, Command: 0x0401, Function: 0, Data: [1, 0]]
2022-04-13 22:16:20 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4417, tsn=1, command_id=0, *is_reply=False)
2022-04-13 22:16:20 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 0]))
2022-04-13 22:16:20 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xC7AC, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=1), 2, b'\x05A\x11\x01\x00\x00\x00\x01\x04\x00\x01\x00')
2022-04-13 22:16:20 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'0060'
2022-04-13 22:16:20 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'00acc7040100ef01014001000060020000'
2022-04-13 22:16:20 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 51116, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=96), 2, <EmberStatus.SUCCESS: 0>, b'']
2022-04-13 22:16:20 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef010100010000fbfcb9acc7ffff0518010b0083'
2022-04-13 22:16:20 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=251), 252, -71, 0xc7ac, 255, 255, b'\x18\x01\x0b\x00\x83']
2022-04-13 22:16:20 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Received ZCL frame: b'\x18\x01\x0b\x00\x83'
2022-04-13 22:16:20 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=11, *is_reply=True)
2022-04-13 22:16:20 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2022-04-13 22:16:20 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC7AC:1:0x0102]: executed 'up_open' command with args: '()' kwargs: '{}' result: Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2022-04-13 22:16:21 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=cover, service=stop_cover, service_data=entity_id=cover.tze200_rmymn92d_ts0601_4c436f76_window_covering>
2022-04-13 22:16:21 DEBUG (MainThread) [tuya] a4:c1:38:bd:76:6f:43:4c Sending Tuya Cluster Command.. Manufacturer is _TZE200_rmymn92d Cluster Command is 0x0002, Arguments are ()
2022-04-13 22:16:21 DEBUG (MainThread) [tuya] a4:c1:38:bd:76:6f:43:4c Sending Tuya Command. Paylod values [endpoint_id : 1, Status : 0, TSN: 0, Command: 0x0401, Function: 0, Data: [1, 1]]
2022-04-13 22:16:21 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4417, tsn=3, command_id=0, *is_reply=False)
2022-04-13 22:16:21 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 1]))
2022-04-13 22:16:21 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xC7AC, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=3), 4, b'\x05A\x11\x03\x00\x00\x00\x01\x04\x00\x01\x01')
2022-04-13 22:16:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'0061'
2022-04-13 22:16:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'00acc7040100ef01014001000061040000'
[... cut ...]
2022-04-13 22:16:21 DEBUG (MainThread) [zigpy.zcl] [0xC7AC:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2022-04-13 22:16:21 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC7AC:1:0x0102]: executed 'stop' command with args: '()' kwargs: '{}' result: Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
2022-04-13 22:16:22 DEBUG (MainThread) [bellows.ezsp.protocol] Send command nop: ()
2022-04-13 22:16:22 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 5 (nop) received: b''
```

TheJulianJES, 2022-04-14:
I don't know why your working blinds don't log any "Sending Tuya". That would mean that [this code](https://github.com/zigpy/zha-device-handlers/blob/5272b4c3e180f1c0c43c2a412c8428c12549e772/zhaquirks/tuya/__init__.py#L1083-L1166) is never called.

Please check if your logger config (or service call where you set ``tuya: debug``) also contains this:
``
zhaquirks: debug
``

Also, can you send me the device signature from both the working cover that does not output the commands and from the non-working cover that outputs the command?

SergeantPup, 2022-04-14:
That's correct:
```
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
    tuya: debug
```

SergeantPup, 2022-04-14:
Zigbee device signature of controllable unit:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_xaabybja",
  "model": "TS0601",
  "class": "tuya.ts0601_cover.TuyaMoesCover0601"
}
```

Device signature of tracked but non controlled unit:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_rmymn92d",
  "model": "TS0601",
  "class": "tuya.ts0601_cover.TuyaZemismartSmartCover0601_4"
}
```

ruudkoeyvoets, 2022-04-20:
Having simular issue,

2022-04-20 09:30:40 DEBUG (MainThread) [zhaquirks.tuya] a4:c1:38:3a:2a:a7:0e:ee Sending Tuya Cluster Command.. Manufacturer is _TZE200_r0jdjrvi Cluster Command is 0x0001, Arguments are ()
2022-04-20 09:30:40 DEBUG (MainThread) [zhaquirks.tuya] a4:c1:38:3a:2a:a7:0e:ee Sending Tuya Command. Paylod values [endpoint_id : 1, Status : 0, TSN: 0, Command: 0x0401, Function: 0, Data: [1, 2]]
2022-04-20 09:30:40 DEBUG (MainThread) [zigpy.zcl] [0x7B28:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), manufacturer=4417, tsn=218, command_id=0, *is_reply=False)
2022-04-20 09:30:40 DEBUG (MainThread) [zigpy.zcl] [0x7B28:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 2]))
2022-04-20 09:30:40 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x7B28, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=218), 219, b'\x05A\x11\xda\x00\x00\x00\x01\x04\x00\x01\x02')
2022-04-20 09:30:40 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'0058'
2022-04-20 09:30:40 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'00287b040100ef01014001000058db0000'
2022-04-20 09:30:40 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef0101000100003174b9287bffff0518da0b0083'
2022-04-20 09:30:40 DEBUG (MainThread) [zigpy.zcl] [0x7B28:1:0xef00] Received ZCL frame: b'\x18\xda\x0b\x00\x83'
2022-04-20 09:30:40 DEBUG (MainThread) [zigpy.zcl] [0x7B28:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=218, command_id=11, *is_reply=True)
2022-04-20 09:30:40 DEBUG (MainThread) [zigpy.zcl] [0x7B28:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)

Looking at what is send, " b'\x05A\x11\xda\x00\x00\x00\x01\x04\x00\x01\x02')"  This looks to be wrong.

b'\x05A\x11\

xda == 
x00 == Command Word ===== should be 0x04
x0000 == Data length  ===== Should be x0005
x01 == dpID
x04 == Data Type
x0001 == function Length
x02 == Close

Any help would be appriacted

ruudkoeyvoets, 2022-04-20:
Maybe useful_TZE200_r0jdjrvi is from KKMOON. This is Tuya Spec from KKMOON. (but from a different type, so I don't know if applicable
[kkmoon_tuya_cover_spec.pdf](https://github.com/zigpy/zha-device-handlers/files/8520804/kkmoon_tuya_cover_spec.pdf)
)

SergeantPup, 2022-04-21:
Hello,

I've been reviewing the quirk and I have a question:
[image]
On the device tracking and not controlling:
```
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
```

But in the quirk:
```
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
```

I can't find greenpowerproxy or "0x0021" referenced anywhere in the init.py

Does the device just not know what to do with this and that's why it's trying to make a call?

MattWestb, 2022-04-21:
I think its best keeping it in the replacement so user can see that the router have one ZGPP in the future then its being implanted functions for using ZGPP devices.

SergeantPup, 2022-04-21:
yep, it definitely didn't like that.  I'll leave it to the professionals :)

ruudkoeyvoets, 2022-04-21:
The greenpowerproxy can be added in the custom quicks file. See example below

from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Scenes, Time, GreenPowerProxy
# from zigpy.zcl.clusters.closures import WindowCovering
# from zigpy.quirks import CustomCluster, CustomDevice
# import zigpy.types as t

SergeantPup, 2022-04-21:
I noticed some small differences between my zigbee signature and the signature that was provided to create the quirk.  Not sure if this matters or not:
[image]

[image]
https://github.com/zigpy/zha-device-handlers/issues/1294#issuecomment-1014843749

TheJulianJES, 2022-04-21:
Just had a quick look: the original "author" didn't update the comment (in green with the # in front). I think updated some lines when sending an updated quirk here but I'm not sure atm.

The quirk needs to match the signature of the device, as it wouldn't be applied otherwise. Since it's applied, the signature should match.

Kiread-work, 2022-04-24:
Some new information from https://github.com/zigpy/zha-device-handlers/issues/744

```
Hope this helps someone, I just spent about 2 hours figuring it out. You can change the speed of the motor by going to developer tools > services, choosing Zigbee Home Automation: Issue zigbee cluster command and use YAML mode to enter this

service: zha.issue_zigbee_cluster_command
data:
  ieee: IEEE of your device, find it in the integration page
  endpoint_id: 1
  command_type: server
  cluster_id: 258
  command: 0x0006
  args:
      - 0
      - 0
      - 0x69
      - 0
      - [4, 0, 0, 0, {integer between 0 - 100}]
```

ruudkoeyvoets, 2022-04-27:
Is there already someone who understands the UNSUP_MANUF_CLUSTER_COMMAND: 131 some motors give?

SergeantPup, 2022-04-30:
no but I'm suddenly seeing a bunch of this on the logs for this device:

```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py:230
First occurred: 11:24:06 AM (1396 occurrences)
Last logged: 12:45:51 PM

[0xDBAC:1:0xef00] Unknown cluster command 2 b'@\x83\x03\x02\x00\x04\x00\x00\x00\x00'
[0xDBAC:1:0xef00] Unknown cluster command 2 b'@\x83\x01\x04\x00\x01\x01'
[0xDBAC:1:0xef00] Unknown cluster command 2 b'@\x84\x03\x02\x00\x04\x00\x00\x00\x00'
[0xDBAC:1:0xef00] Unknown cluster command 2 b'@\x84\x01\x04\x00\x01\x01'
[0xDBAC:1:0xef00] Unknown cluster command 2 b'@\x85\x03\x02\x00\x04\x00\x00\x00\x00'
```

ruudkoeyvoets, 2022-05-01:
What has changed? I have tried several things, but still no solution

ruudkoeyvoets, 2022-05-04:
Somehow I managed to get the UNSUP_MANUF_CLUSTER_COMMAND resolved.
In the Tuya __init__.py TuyaManufcluster class I changed in server_commands, 0x0000 the is_manufacture_specific from True to False.

For me this works, don’t know if it’s the correct way to solve the issue.

ruudkoeyvoets, 2022-05-04:
Open, Close and Stop does not yet work, goto lift percentage does

SergeantPup, 2022-05-04:
sorry I left you hanging, I'm still wholly invested in this.  When you asked me 'what's changed?' it could literally be a zillion things.  I went on the 2022.05 beta from the start and it's been expansive and time consuming.  Could it be related to that? I have no idea, but I know there were a few different ZWV and ZHA issues reported in that beta that were resolved (including one for curtains pulling initial state).  
I had removed the ZHA logging, I wouldn't think that would've given me additional logs and my Logger was provided by Julien, I just pulled it out because it was making my logs too chatty in beta.

At this point, I don't know if we're on the same quirk or not.  Can you point me to the quirk you're using with these results please?

Kiread-work, 2022-05-04:
@ruudkoeyvoets please provide your custom-quirk
Mine doesn't have the ability to open/close by percentage

ruudkoeyvoets, 2022-05-04:
Open, close and stop also work:

"_TZE200_r0jdjrvi": {0x0000: 0x0002, 0x0001: 0x0000, 0x0002: 0x0001},

ruudkoeyvoets, 2022-05-04:
Custom Quirk is default one:

        MODELS_INFO: [
        ("_TZE200_r0jdjrvi", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
#                    TuyaManufCluster.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

SergeantPup, 2022-05-04:
[image]
doesn't look like it

Kiread-work, 2022-05-04:
@SergeantPup check your model in this section
MODELS_INFO: [
    ("_TZE200_r0jdjrvi", "TS0601"),
    ],

SergeantPup, 2022-05-04:
oh yes
[image]
[image]
:

TheJulianJES, 2022-05-04:
Can you check if the signature changed? Otherwise, enable debug logs in the config file and restart Home Assistant. It'll mention why a quirk doesn't match for any device.

SergeantPup, 2022-05-04:
Something definitely changed from my screenshots a few weeks ago to the current zha sig:
[image]

I had to re-add the logger, rebooting now

TheJulianJES, 2022-05-04:
Mhm, the device_type is changed by the quirk (from SMART_PLUG to WINDOW_COVERING_DEVICE).
Otherwise, the signature looks to be the same?
Not sure why the quirk isn't applying atm.

MattWestb, 2022-05-04:
@SergeantPup deleting the   `__pycache__ `  folder in your custom quirk folder and restating HA and posting `Zigbee device signature` or the downloaded diagnostic.

MattWestb, 2022-05-04:
> @SergeantPup **deleting** the `__pycache__ ` folder in your custom quirk folder and restating HA and posting `Zigbee device signature` and / or the downloaded diagnostic.

DELETING !!!

MattWestb, 2022-05-04:
Also putting longing in HA like this:
```
    zhaquirks: debug
    custom_zha_quirks: debug
```
and you can see if ZHA is loading the local quirk then you restating HA like this:
```
2022-04-29 14:38:54 DEBUG (MainThread) [zhaquirks] Loading custom quirks module ts0601_trv_siterwell
```
for my tuya TRVs.

Kiread-work, 2022-05-04:
Can't see percentage control :( 
But the buttons are working!
[image]

Kiread-work, 2022-05-04:
@SergeantPup check in i_init__.py
>class TuyaManufCluster(CustomCluster)
.....
[image]

SergeantPup, 2022-05-04:
just changed that and buttons work!

TheJulianJES, 2022-05-04:
Do all other curtains still work when all messages are no longer sent as manufacturer specific ones?
Also, is everything working now? (with that additional change)

Kiread-work, 2022-05-04:
How I can remap close/open buttons?
Adding my device to the TUYA_COVER_INVERTED_BY_DEFAULT doesn't help

SergeantPup, 2022-05-04:
You have to invert it on the device itself.  I had to do this also when it started working.

On the remote Press and Hold Open/Close at the same time until the curtains respond back and forth.  Then long press (or press 3 times) the stop button.  Curtain will respond back and forth again, HA should be in synch at that point.  I still have the inverted command line in there.  I removed it and re-added it multiple times and the controls were still backwards but when I reversed the device, everything got aligned.

SergeantPup, 2022-05-04:
I am not getting errors like before, just success commands:
2022-05-04 15:02:20 DEBUG (MainThread) [zigpy.zcl] [0x7332:1:0xef00] Decoded ZCL frame: TuyaManufacturerWindowCover:Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-05-04 15:02:20 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7332:1:0x0102]: executed 'down_close' command with args: '()' kwargs: '{}' result: Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2022-05-04 15:02:20 DEBUG (MainThread) [homeassistant.components.zha.cover] state=closing
2022-05-04 15:02:20 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=cover.media_room_curtains_2, old_state=<state cover.media_room_curtains_2=opening; current_position=99, device_class=curtain, icon=mdi:curtains, friendly_name=Media Room Curtains 2, supported_features=15 @ 2022-05-04T15:01:38.101784-04:00>, new_state=<state cover.media_room_curtains_2=closing; current_position=99, device_class=curtain, icon=mdi:curtains, friendly_name=Media Room Curtains 2, supported_features=15 @ 2022-05-04T15:02:20.884893-04:00>>
2022-05-04 15:02:20 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=cover.media_room_curtains, old_state=<state cover.media_room_curtains=opening; current_position=99, entity_id=['cover.media_room_curtains_1', 'cover.media_room_curtains_2'], assumed_state=True, device_class=curtain, icon=mdi:curtains, friendly_name=Media Room Curtains, supported_features=15 @ 2022-05-04T15:01:38.102413-04:00>, new_state=<state cover.media_room_curtains=closing; current_position=99, entity_id=['cover.media_room_curtains_1', 'cover.media_room_curtains_2'], assumed_state=True, device_class=curtain, icon=mdi:curtains, friendly_name=Media Room Curtains, supported_features=15 @ 2022-05-04T15:02:20.885763-04:00>>
2022-05-04 15:02:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040120000101400100008dffbc2777ffff0508870b0194'
2022-05-04 15:02:21 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=32, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=141), 255, -68, 0x7727, 255, 255, b'\x08\x87\x0b\x01\x94']
2022-05-04 15:02:21 DEBUG (MainThread) [zigpy.zcl] [0x7727:1:0x0020] Received ZCL frame: b'\x08\x87\x0b\x01\x94'
2022-05-04 15:02:21 DEBUG (MainThread) [zigpy.zcl] [0x7727:1:0x0020] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=135, command_id=11, *is_reply=True)
2022-05-04 15:02:21 DEBUG (MainThread) [zigpy.zcl] [0x7727:1:0x0020] Decoded ZCL frame: PollControl:Default_Response(command_id=1, status=<Status.TIMEOUT: 148>)
2022-05-04 15:02:21 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'0027770401200001014001000093880000'
2022-05-04 15:02:21 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 30503, EmberApsFrame(profileId=260, clusterId=32, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=147), 136, <EmberStatus.SUCCESS: 0>, b'']
2022-05-04 15:02:21 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x7727:1:0x0020]: executed 'fast_poll_stop' command with args: '()' kwargs: '{}' result: Default_Response(command_id=1, status=<Status.TIMEOUT: 148>)
2022-05-04 15:02:21 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=cover, service=stop_cover, service_data=entity_id=cover.media_room_curtains_2>
2022-05-04 15:02:21 DEBUG (MainThread) [tuya] a4:c1:38:bd:76:6f:43:4c Sending Tuya Cluster Command.. Manufacturer is _TZE200_rmymn92d Cluster Command is 0x0002, Arguments are ()
2022-05-04 15:02:21 DEBUG (MainThread) [tuya] a4:c1:38:bd:76:6f:43:4c Sending Tuya Command. Paylod values [endpoint_id : 1, Status : 0, TSN: 0, Command: 0x0401, Function: 0, Data: [1, 1]]
2022-05-04 15:02:21 DEBUG (MainThread) [zigpy.zcl] [0x7332:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=139, command_id=0, *is_reply=False)
2022-05-04 15:02:21 DEBUG (MainThread) [zigpy.zcl] [0x7332:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 1]))
2022-05-04 15:02:21 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x7332, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=139), 140, b'\x01\x8b\x00\x00\x00\x01\x04\x00\x01\x01')
2022-05-04 15:02:22 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'0095'
[... cut ...]  0, 0]]
2022-05-04 15:04:11 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x448B:1:0x0102]: Attribute report 'Window Covering'[current_position_lift_percentage] = 0
2022-05-04 15:04:11 DEBUG (MainThread) [tuya] 8c:f6:81:ff:fe:d1:b7:ab Tuya Attribute Cache : [{8: 0}]
2022-05-04 15:04:11 DEBUG (MainThread) [homeassistant.components.zha.cover] setting position: 0

I can control open/stop/close/position on all devices including: _TZE200_rmymn92d and all accurately track state

Kiread-work, 2022-05-04:
@SergeantPup Can it be inverted without external remote? I don't have it =)

[26 more replies]