zigpy/zha-device-handlers #1295: [Device Support Request] Aqara P1 Door and Window Sensor

MiAutomations, 2022-01-18:
**Is your feature request related to a problem? Please describe.**

I have successfully added the device over the ZHA, but the device doesn't change the state from open to close always keep the closed status doesn't change, and doesn't report the battery level    

**Describe the solution you'd like**
I need to have the battery level and the correct status reported from the sensor open or closed

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.magnet.ac01",
  "class": "zigpy.device.Device"
}

**Additional context**
[image]

[image]

Thank you for your support

schwickster, 2022-01-19:
I had the same problem with the E1 counterpart. What ended up solving it for me is updating my Conbee 2 firmware. What Zigbee receiver are you using for your Zigbee network?

That said, it won't suprise me it still needs a quirk.

schwickster, 2022-01-19:
I think on a hardware it will work because it's a Zigbee device. Only thing you can see in the ticket is that it maybe won't work on a software level. I would say give it a try with the Conbee (after using the latest firmware).

MiAutomations, 2022-01-20:
I still facing issues also with conbee with last firmware also with the Sonoff zigbee 3.0 usb dongle plus

MiAutomations, 2022-03-18:
Any news abou this topic ?

Thank you

TheJulianJES, 2022-04-17:
Apparently a new/duplicate issue about this has been opened up: https://github.com/zigpy/zha-device-handlers/issues/1498

VincentMarnier, 2022-04-22:
It does not look related to this issue
This issue is about Aqara P1 Door and Window Sensor (lumi.magnet.ac01)
The one you linked is about Aqara P1 Motion Sensor (lumi.motion.ac02)

PabloGux, 2023-09-24:
Same issue here! Aqara P1 contact sensor (lumi.magnet.ac01) pairs with ZHA but the binary_sensor does not get updated, nor it reads battery information. Please add support in ZHA for this.

hscissors, 2024-01-03:
Same issue. Sonoff Zigbee Adapter-E 3.0 with the lumi.sensor_magnet.aq2. Struggles to pair with ZHA and then immediately fails to report correctly. I have two and oddly one will mesh over time and the work flawlessly. The one that pairs and then fails doesn't seem to have a connection to the adapter in the network overview, but is the exact same distance as the one that starts to work,

TheJulianJES, 2024-01-03:
- https://github.com/zigpy/zha-device-handlers/pull/2822 should have addressed this, the issue wasn't closed yet though.

@hscissors Did you upgrade HA Core to 2024.1.0 (beta) or later? The full release will be made in a few hours. The beta is already available.

hscissors, 2024-01-06:
Updated to `2024.1.1`, deleted the Sonoff Deivce and re-added it and "reset" the network. After struggling to add the devices I got one to connect. It however is showing no connection to the zigbee network.

TheJulianJES, 2024-01-30:
> They however are no connection to the zigbee network.

The map isn't live. It can take hours to update, but you can try to force a refresh through the button in the UI and then come back after 15 minutes or so (and refresh the page).

The device itself is supported now. It becoming unavailable is a network issue, not an issue we can fix with quirks.

Support for the P1 window/door sensor was added in Home Assistant Core 2024.1.0.

closed 2024-01-30 by TheJulianJES