zigpy/zha-device-handlers #1296: [BUG]  Aqara vibration sensor claimed to be supported

sergey-suloev, 2022-01-20:
Aqara vibration sensor claimed to be supported but it is not. See the attached screenshot.
[image]

sergey-suloev, 2022-01-20:
'''javascript
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x000a",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0019",
        "0x0101",
        "0x0500"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x5f02",
      "in_clusters": [
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0012"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.vibration.aq1",
  "class": "zhaquirks.xiaomi.aqara.vibration_aq1.VibrationAQ1"
}
'''

MattWestb, 2022-01-20:
Do you get any `zha_event` and is the ISA Zone changing status ?
The `"class": "zhaquirks.xiaomi.aqara.vibration_aq1.VibrationAQ1"` is saying ZHA have loading one quirk for your device = its supported by one quirk (then its not one Zigbee  certified device = bad working device).

sergey-suloev, 2022-01-20:
@MattWestb  I'm not sure I fully understand what you are saying. My device is 100% official and certified. I bought it from an official Xiaomi store.

sergey-suloev, 2022-01-20:
2022-01-20 11:36:25 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.vibration.aq1 (00:15:8d:00:04:83:5e:06)
2022-01-20 11:36:25 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.vibration_aq1.VibrationAQ1'>
2022-01-20 11:36:25 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 00:15:8d:00:04:83:5e:06: <class 'zhaquirks.xiaomi.aqara.vibration_aq1.VibrationAQ1'>

sergey-suloev, 2022-01-20:
2022-01-20 11:36:26 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.lumi_lumi_vibration_aq1_065e8304_power, old_state=None, new_state=<state sensor.lumi_lumi_vibration_aq1_065e8304_power=44; state_class=measurement, battery_voltage=2.94, unit_of_measurement=%, device_class=battery, friendly_name=LUMI lumi.vibration.aq1 065e8304 power @ 2022-01-20T11:36:26.516517+03:00>>

sergey-suloev, 2022-01-20:
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 battery
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 action
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 strength
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 sensitivity
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 angle_x
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 angle_y
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 angle_z
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 voltage
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 linkquality

MattWestb, 2022-01-20:
Sorry but Aqara (Xiaomi) is not good making Zigbee compliant products but they have doing  some . . . But i cant finding the certificate for your device https://csa-iot.org/csa-iot_products/page/2/?p_keywords=Lumi+United&p_type=14&p_certificate.
But its not interesting then your device have being patches for working OK in ZHA.

Then posting code blocks please put ` ``` ` on one empty line before and after the code so its not losing its formation.

ZHA have finding:
```
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 battery
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 action
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 strength
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 sensitivity
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 angle_x
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 angle_y
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 angle_z
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 voltage
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0x00158d0004835e06 linkquality
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: select 0x00158d0004835e06 sensitivity
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: binary_sensor 0x00158d0004835e06 vibration
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: device_automation 0x00158d0004835e06 action_vibration
2022-01-20 11:36:19 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: device_automation 0x00158d0004835e06 action_tilt
```
Some of the functions have device automations that you finding on the device card and the  `+`  on the side of "Automations".

HA is all about events and if you going to development tools > events > Listen to events and putting in `zha_event` and clicking `start listening` you shall getting all events ZHA is getting from your devices like tilt angle and so on.

sergey-suloev, 2022-01-20:
@MattWestb  It's probably not certified, I have no idea. But you can buy it from any official Xiaomi store.
https://www.zigbee2mqtt.io/devices/DJT11LM.html

MattWestb, 2022-01-20:
Certifying Zigbee devices is costing money and also being member of the alliance so many is not doing it and is putting the Zigbee logo on there products and cont care if the device is working with other system or not.
The latest large is tuya they is one board member but is only certifying products that LIDL and other customer is demanding certificate.

Aqara / Xiaomi was very much problem but then is being better but nearly all ther sensor need quirk for working at all in open systems but the community have learned how they is working and support is normally being fixed but they is doing new versions with changed functions all the time.

Better you is looking if all functions is working OK for your devices then discussing "bad devoices" ;-))

sergey-suloev, 2022-01-20:
@MattWestb  There is no possibility to handle DROP event. And it is not reporting angles. 
Here is what I can see when I click on Automations+ 

[image]

MattWestb, 2022-01-20:
I think both you need listening to `zha_events` to see them and if making automations you shall triggering it on the information you is getting as events.
Its not realistic making device automations for all possible 360° so you need doing them self.
You can also using template sensors and making sensors that doing the listening on events and you is getting the value in HA GUI.

javicalle, 2022-01-20:
It seems that there is not a 'direct' way to get the values that you want.
You will need to handle state changes using zha_events. It's not the friendliest way and it's going to force you to struggle with templates and automations, but it's the way to do it.

You can get a lot of information in the HA forums:
* https://community.home-assistant.io/t/aqara-vibration-sensor-orientation-to-trigger-automation/240737/7
* https://community.home-assistant.io/t/automation-based-on-args/135377
* https://community.home-assistant.io/t/xiaomi-vibration-sensor-a-simple-automation-for-detecting-state-on-and-off/230388

I hope this helps you.

sergey-suloev, 2022-01-20:
@javicalle Thank you for the information. Don't you think though that zha quirk for the device must be improved instead ?

dmulcahey, 2022-01-20:
The angles and the drop event are handled as events. The vibration detection gives you a binary sensor. Drop and the angles are events intentionally. The angles are relative and the drop event is momentary. Having entities for these doesn’t make a ton of sense.

sergey-suloev, 2022-01-20:
@dmulcahey I agree about the drop drop. But for angles Z2M provides sensors which does make sense IMHO

Adminiuga, 2022-01-22:
use `zha_events` for angles. if really need entities, then update a number entity based on zha_event.

sergey-suloev, 2022-01-22:
@Adminiuga  What is the difference between vibration , tilt, drop and angles ? All are events, in general. Why should only vibration be an entity ?

schwickster, 2022-01-25:
I have the same sensor in ZHA and for me it works fine. As in the screenshot I have the same entities. As people are mentioning, for angles, vibrations and axes (coming in the next release) you need to read from ZHA event.

martin3000, 2022-02-19:
I have the problem that the vibration events need 25s to arrive in home assistant.

SamJongenelen, 2022-08-19:
Not stale to me. I would like to also see a more user friendly solution to automate using Aqara devices. 
These (uncertified) devices are used a lot by Zigbee HA users I would assume - and not having a nice way to do automations for it (albeit events or sensors).

terjew, 2022-08-28:
I have browsed through the different threads linked here, but I am still a bit confused as to how I can access the orientation from this device. It seems like many of the threads refer to other integrations such as deconz, and thus are not applicable for zha. Is it possible to create a template sensor that updates some numerical values whenever the orientation changes?

Adminiuga, 2022-08-28:
Create an automation triggered by zha event and update the multiselect helper accordingly

terjew, 2022-08-28:
For anyone with the same use case, I got it working with the following: 
```template:
  - trigger:
      - platform: event
        event_type: "zha_event"
        event_data:
          device_id: 9f43c386f952703e241d4b18d0504b03
          command: 'current_orientation'
    sensor:
      - name: "Garage door orientation X"
        state: "{{ trigger.event.data.args.X }}"
        unique_id: bc6fa84a-1346-4103-b144-b6cf95f94fb2
      - name: "Garage door orientation Y"
        state: "{{ trigger.event.data.args.Y }}"
        unique_id: 3be9e412-c52e-4676-9948-b7807eec62ac
      - name: "Garage door orientation Z"
        state: "{{ trigger.event.data.args.Z }}"
        unique_id: 5b8c75bf-55aa-4317-8ab3-18394d83b62a
```        
However, the problem now is that the current_orientation events are only being fired around once per minute, so it is really not very useful after all. I assume this is a problem on the Aqara side?

SamJongenelen, 2022-08-31:
Could that be a sensitivity setting perhaps? IIRC I tested a Vibration sensor in Zigbee2Mqtt and the sensor updated nearly instant

klepper19, 2023-01-08:
> For anyone with the same use case, I got it working with the following:
> 
> ```
>   - trigger:
>       - platform: event
>         event_type: "zha_event"
>         event_data:
>           device_id: 9f43c386f952703e241d4b18d0504b03
>           command: 'current_orientation'
>     sensor:
>       - name: "Garage door orientation X"
>         state: "{{ trigger.event.data.args.X }}"
>         unique_id: bc6fa84a-1346-4103-b144-b6cf95f94fb2
>       - name: "Garage door orientation Y"
>         state: "{{ trigger.event.data.args.Y }}"
>         unique_id: 3be9e412-c52e-4676-9948-b7807eec62ac
>       - name: "Garage door orientation Z"
>         state: "{{ trigger.event.data.args.Z }}"
>         unique_id: 5b8c75bf-55aa-4317-8ab3-18394d83b62a
> ```
> 
> However, the problem now is that the current_orientation events are only being fired around once per minute, so it is really not very useful after all. I assume this is a problem on the Aqara side?

I appreciate you posted this! My vibration sensors show the orientation in my project and the update rate is nearly instant. Using Conbee 2 with Raspberry pi and ZHA. No further settings changed. Might be a newer firmware on the vibration sensor/ZHA/Conbee?

mox1, 2023-04-06:
So I have this sensor attached to my garage door. I don't care about vibration, I just want Tilt. 
Looking through my home assist debug logs, I see a zha firing "Tilt" events, with an angle. 

Would it be possible simply expose the current Tilt angle (or last reported angle I guess)? This would really help with creating automatons around it. Please not that my sensor is NOT reporting relative values. It is reporting absolute angle where standing on its edge == 9 degrees and lying flat face down == 180. 

`Open garage door:
Apr  6 12:31:43 server1 docker-compose[2470]: homeassistant  | #033[36m2023-04-06 12:31:43.938 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:15:8d:00:07:09:de:ed, unique_id=00:15:8d:00:07:09:de:ed:1:0x0500, device_id=8bf5abeb52b706beefbaf493466b25c1, endpoint_id=1, cluster_id=1280, command=Tilt, args=degrees=104, params=>#033[0m
Close:
Apr  6 12:32:31 server1 docker-compose[2470]: homeassistant  | #033[36m2023-04-06 12:32:31.561 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:15:8d:00:07:09:de:ed, unique_id=00:15:8d:00:07:09:de:ed:1:0x0500, device_id=8bf5abeb52b706beefbaf493466b25c1, endpoint_id=1, cluster_id=1280, command=Tilt, args=degrees=18, params=>#033[0m
Open:
Apr  6 13:00:01 server1 docker-compose[2470]: homeassistant  | #033[36m2023-04-06 13:00:01.366 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:15:8d:00:07:09:de:ed, unique_id=00:15:8d:00:07:09:de:ed:1:0x0500, device_id=8bf5abeb52b706beefbaf493466b25c1, endpoint_id=1, cluster_id=1280, command=Tilt, args=degrees=105, params=>#033[0m
Close:
Apr  6 13:00:34 server1 docker-compose[2470]: homeassistant  | #033[36m2023-04-06 13:00:34.710 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:15:8d:00:07:09:de:ed, unique_id=00:15:8d:00:07:09:de:ed:1:0x0500, device_id=8bf5abeb52b706beefbaf493466b25c1, endpoint_id=1, cluster_id=1280, command=Tilt, args=degrees=18, params=>#033[0m

`

mekaneck, 2023-04-13:
> Would it be possible simply expose the current Tilt angle (or last reported angle I guess)?

It's easier to see & read the events by using the "events" tab in the developer tools and subscribing to `zha_event`. But there's enough info with what you posted. You can use a [template sensor](https://www.home-assistant.io/integrations/template/#turning-an-event-into-a-trigger-based-binary-sensor)
```
template:
  - trigger:
      - platform: event
        event_type: "zha_event"
        event_data:
          device_id: 8bf5abeb52b706beefbaf493466b25c1
          command: 'Tilt'
    sensor:
      - name: "Garage Door Tilt"
        state: "{{ trigger.event.data.args.degrees }}"
        unique_id: garage_door_tilt
        attributes:
          time_fired: "{{ trigger.event.time_fired }}"
```

And if anyone needs the drop sensor, you can use something like this:
```
template:
  - trigger:
      - platform: event
        event_type: "zha_event"
        event_data:
          device_id: 8bf5abeb52b706beefbaf493466b25c1
          command: 'Drop'
    binary_sensor:
      - name: "Aqara Vibration Sensor Drop Event"
        state: "true"
        unique_id: aqara_vibration_sensor_drop_event
        auto_off: 5
        attributes:
          time_fired: "{{ trigger.event.time_fired }}"
```

closed 2023-10-17 by github-actions[bot]