zigpy/zha-device-handlers #1297: [Device Support Request] Ikea TRADFRI bulb 1055lm

parrel, 2022-01-20:
**Is your feature request related to a problem? Please describe.**
The bulb is named "TRADFRIbulbE27WSglobeopal1055lm". According to the manual it supports brightness, warm/cold white and color adjustments. However, the device does not seem to support color in ZHA/HA. 

**Describe the solution you'd like**
I would like to control the color on this bulb. :)

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
Signature from HA:
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xfc57"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRIbulbE27WSglobeopal1055lm",
  "class": "zigpy.device.Device"
}
```
Logs when pairing:
```
[0x1C59:1:0x0500]: Updated alarm state: 0
[0xd68c:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=94 command_id=Command.Report_Attributes>
[0xd68c:1:0x0300] ZCL request 0x000a: [[Attribute(attrid=3, value=<TypeValue type=uint16_t, value=30138>), Attribute(attrid=4, value=<TypeValue type=uint16_t, value=26909>)]]
[0xd68c:1:0x0300] Attribute report received: current_x=30138, current_y=26909
[0x3b1b:11:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=102 command_id=Command.Report_Attributes>
[0x3b1b:11:0x0300] ZCL request 0x000a: [[Attribute(attrid=3, value=<TypeValue type=uint16_t, value=40107>), Attribute(attrid=4, value=<TypeValue type=uint16_t, value=24510>)]]
[0x3b1b:11:0x0300] Attribute report received: current_x=40107, current_y=24510
[0x52A6:1:0x0500]: Updated alarm state: 0
[0x29c7:zdo] ZDO request ZDOCmd.IEEE_addr_rsp: [<Status.SUCCESS: 0>, 7c:b0:3e:aa:0a:07:7e:8e, 0x29C7, None, None, []]
[0x29c7:zdo] No handler for ZDO request:ZDOCmd.IEEE_addr_rsp([<Status.SUCCESS: 0>, 7c:b0:3e:aa:0a:07:7e:8e, 0x29C7, None, None, []])
New device 0xea98 (0c:43:14:ff:fe:9b:e5:af) joined the network
[0xea98] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xEA98 ieee=0c:43:14:ff:fe:9b:e5:af is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x81\x98\xea\xaf\xe5\x9b\xfe\xff\x14C\x0c\x8e'
[0xea98:zdo] ZDO request ZDOCmd.Device_annce: [0xEA98, 0c:43:14:ff:fe:9b:e5:af, 142]
Tries remaining: 3
[0xea98] Requesting 'Node Descriptor'
Tries remaining: 2
[0xea98] Extending timeout for 0x46 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xEA98 ieee=0c:43:14:ff:fe:9b:e5:af is_initialized=False> from ep 0 to ep 0, cluster 32770: b'F\x00\x98\xea\x01@\x8e|\x11RR\x00\x00,R\x00\x00'
[0xea98] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0xea98] Discovering endpoints
Tries remaining: 3
[... 93 lines cut ...]
[0xEA98:1:0x0300]: finished channel initialization
[0xEA98:1:0x0006]: initializing channel: from_cache: False
[0xea98:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=106 command_id=Command.Read_Attributes_rsp>
[0xEA98:1:0x0006]: finished channel initialization
[0xEA98:1:0x0006]: 'async_initialize' stage succeeded
[0xEA98:1:0x0300]: 'async_initialize' stage succeeded
[0xEA98:1:0x0008]: 'async_initialize' stage succeeded
[0xEA98:1:0x1000]: 'async_initialize' stage succeeded
[0xEA98:1:0x0000]: 'async_initialize' stage succeeded
[0xEA98:1:0x0019]: 'async_initialize' stage succeeded
[0xEA98](TRADFRIbulbE27WSglobeopal1055lm): power source: Mains
[0xEA98](TRADFRIbulbE27WSglobeopal1055lm): completed initialization
None: polling current state
[0xea98:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=108 command_id=Command.Read_Attributes_rsp>
[0xea98:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=110 command_id=Command.Read_Attributes_rsp>
[0xEA98:1:0x0008]: received attribute: 0 update with value: 254
[0xea98:1:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=112 command_id=Command.Read_Attributes_rsp>
[0x6327:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=19 command_id=Command.Report_Attributes>
[0x6327:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=80>)]]
[0x6327:1:0x0000] Attribute report received: app_version=80
```

**Additional context**
[Ikea](https://www.ikea.com/nl/nl/p/tradfri-set-met-afstandsbediening-kleur-en-wit-spectrum-60462432/)
[Blakadder](https://zigbee.blakadder.com/Ikea_LED2003G10.html)

MattWestb, 2022-01-20:
IKEA have making 2 color (RGB) lights they is caling CWS (Color White Spectrum) and the first gen was X CWS 600lm and the 3 gen is CWS 800lm then X is different format E27, GU10 and so on).

Your light is not one CWS 800 lm its one WS (White Spectrum) 1055lm 3 gen normally sold in one "family pack" with one styrbar remote.
So it can being you have buying the wrong "family pack" or some have changing the bulb in the package. 
(the CWS 800 is also in "family pack" and looks the same only the color wheel is not on the WS)

PS: I habe baying both packs so habig both types in production ;-)

MattWestb, 2022-01-20:
WS https://www.ikea.com/at/de/p/tradfri-set-mit-fernbedienung-weissspektrum-80513416/
CWS https://www.ikea.com/at/de/p/tradfri-set-mit-fernbedienung-farb-und-weissspektrum-60462432/

parrel, 2022-01-20:
I was also thinking someone might have swapped the bulb (pretty sad if this is the case). This is the box I have.
[image]
Do you have this one in production?

parrel, 2022-01-20:
Yes that's the one I have, I think the bulb got swapped then. I'll close this issue and return the bulb for a new one.

parrel, 2022-01-20:
I did not open it myself, but the person who opened it just told me that the sealing was not there anymore.

closed 2022-01-20 by parrel

parrel, 2022-01-27:
To update on this, got a new bulb from Ikea and it works perfectly 🙂

MattWestb, 2022-01-27:
And IKEA was hard to getting accepting the false light in the package ?
They shall normally not being any problems if not being somthing very strange unrealistic things.

Glaad it was being fixed !!

parrel, 2022-01-27:
Well I  told them the story and she discussed it with her colleagues, after about a minute or 2 I got a refund and went into the store to get a new one 😄

MattWestb, 2022-01-27:
2 minutes for some fue € 👍 
I was buying one integrated micro and the price was not right (20€ to much) then paying for it and was the same only one photo of the reduces price and one colleague and the 20€ back. I can discussing 2 hours for that 20 € and they is losing 220€ (in working time) plus return of the device and bad PR.

parrel, 2022-01-27:
Wow 2 hours that a lot! Glad you got your money back in the end!

jlsjonas, 2022-01-29:
Looks like they had this issue with quite a few bulbs... looks like I had the same issue (IKEA Gent, so not limited to a single store it seems?)