zigpy/zha-device-handlers #1298: [Device Support Request] 3Phase Hiking Tomzn Power Meter DTS238-7 Zigbee (analog of Single Phase DDS238-2 Zigbee)

darckly, 2022-01-21:
This is the three-phases device for Tuya ecosystem DTS238-7 Zigbee Power Meter
[image]

Device signature:

_TZE200_byzdayie TS0601

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_byzdayie",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_din_power.TuyaPowerMeter"
}
```

The device has the same manufacrurerName as the single-phase DDS238-2 PowerMeter device (#994), but different in_cluster content.
With the existing quirk it showers the information only about one phase, and the quirk needs to be extended with the information from the other two phases.

hhh-pixel, 2022-01-21:
> IT also will/should work with the old manufacturerName
> 
> Thank you for the description! I'll try to write the quirk for your device during the weekend. So, if you are not against it, I'll use you as my tester ;) And I have opened a new issue #1298 , let's continue in that tread (because this one is closed already)

Sure) I'm ready to be a tester.

darckly, 2022-01-21:
@hhh-pixel 
Could you enable debug mode and provide the listing after the pairing and waiting for a few minutes?
Enable the logging in configurations.yaml and restart HA
```
logger:
  default: warning
  logs:
    zhaquirks: debug
```
After restart could you find and send me the lines with the text "[zhaquirks.tuya]" from the HA system log?

And (if possible) add loads for all phases, like a lamp, iron, kettle, etc...

hhh-pixel, 2022-01-21:
[ha.log](https://github.com/zigpy/zha-device-handlers/files/7916530/ha.log)

darckly, 2022-01-22:
Thank you, that's what I need.
This is interesting, looks like Hiking starts using the standard Zigbee clusters:
- "0x0006": on-off cluster
- "0x0702": metering cluster to obtain the consumption
- "0x0b04": electric measurement cluster

But also expose the Tuya-specific cluster: "0xef00"
Inside the Tuya cluster the following attributes are reported:
- 0x0101: Switch (Phase A?)
- 0x0210: ? (not present in single phase version) - target temperature, strange?
- 0x0211: Total Energy Phase A
- 0x0212: Current Phase A
- 0x0213: Power Phase A
- 0x0214: Voltage Phase A
- 0x0218: ? (not present in single phase version) - current room temperature, strange?
- 0x0424: ? (not present in single phase version) - sends 1 or 0, don''t know what is this

But let's start from the first assumption, that device can send the power information throught standard Zigbee cluster.
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7918034/ts0601_din_power.zip)
I disabled the Tuya-specific power management part, except switch option, but added the standard clusters in configuration.

Could you load this file as a qustom quirk, delete device, reload HA, re-join device and say/share what you see

hhh-pixel, 2022-01-22:
Hmm. After deleting device and restarting I get this error:
```
Logger: homeassistant.config_entries
Source: quircks/ts0601_din_power.py:5
First occurred: 14:01:35 (1 occurrences)
Last logged: 14:01:35

Error setting up entry Zigbee Stick for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 313, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 403, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/quircks/ts0601_din_power.py", line 5, in <module>
    from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement, Metering
ImportError: cannot import name 'Metering' from 'zigpy.zcl.clusters.homeautomation' (/usr/local/lib/python3.9/site-packages/zigpy/zcl/clusters/homeautomation.py)
```

darckly, 2022-01-22:
@hhh-pixel
Sorry, made typo in the import function
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7918204/ts0601_din_power.zip)

hhh-pixel, 2022-01-22:
Oh, my fault. After rebooting I have more sensors))
[image]

darckly, 2022-01-22:
Hmm, looks like the standard clusters works, but it seems only phase A is displayed
Do you have a load on all phases?

darckly, 2022-01-22:
Hmm, that's explain why they used the same manufacturerName, as for single-phase switch. I'll try to search more about this device...
And you have no a native Tuya gateway to check values in the app, correct?

hhh-pixel, 2022-01-22:
No( I've got only ZigBee USB stick...

darckly, 2022-01-22:
Yeah, no real review I found about this device. It's will be a pity, if it supports only "Phase A" measurement...
BTW., I see your switch is "off" on all screens, and that's strange.

Could you try this quirk?
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7918918/ts0601_din_power.zip)
Delete device, update the quirk, restart HA, pair device...

hhh-pixel, 2022-01-22:
> Yeah, no real review I found about this device. It's will be a pity, if it supports only "Phase A" measurement... BTW., I see your switch is "off" on all screens, and that's strange.

Yes. It's absolutely stupid to make 3phase meter with zigbee or wifi, that is able to measure only 1 phase(

> Could you try this quirk? [ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/7918918/ts0601_din_power.zip) Delete device, update the quirk, restart HA, pair device...
On/off works now! Thank you!)

darckly, 2022-01-23:
@hhh-pixel 

> Yes. It's absolutely stupid to make 3phase meter with zigbee or wifi, that is able to measure only 1 phase(
Wifi version typically has more functions, at least the single-phase version sends much more, compared with Zigbee. So I will not be surprised if finally we found a confirmation that the tree-phase zigbee device sends only information about phase A.

> On/off works now! Thank you!)
Good, this change will be available with PR #1287

darckly, 2022-01-24:
In one ukrainian community I saw confirmation that indeed, this version of device can't control all phases. Maybe change will be in the next version(s)...

hhh-pixel, 2022-01-24:
Sad news(( I asked the seller on ali, but I haven't received any answer yet. Looks like an almost useless device :/

closed 2022-07-30 by github-actions[bot]

endrius84, 2022-08-15:
> 

WOW, where was this device hiding when I was looking for an energy monitoring Meter??!!
I ended up with Frient Energy monitor. It seemed so cool and with futures and is officially supported in ST.
Frient Energy is useless, officially supported by SM but not compatible with ST energy, which doesn't make sense. ST Energy is not an amazing future either and only one tariff pricing option. Do you guys know how could I make better use of the Frient energy monitor?
Is there any smart app I am looking for and can't find?
I have a prepaid meter at home and was hoping with an energy monitor I would be able to set up somehow notifications to top up my account or something but as I mentioned before, Friend energy monitor is overpriced and very basic:(((((((

mmaccaroni, 2022-08-21:
Hi guys a new manufacturer variant is out I opened an issue #1700

evlo, 2022-09-02:
I have one phase version` zhaquirks.tuya.ts0601_din_power.TuyaPowerMeter` where do I upload ts0601 zip file from this thread to make it work?
[image]

Autogarsas, 2022-10-15:
This device isnt compatible yet in z2m? I also asked ali seller, they told its measuring total kWh, cant measure separate phases. Thats ok for me if its supported in z2m, anyone?