zigpy/zha-device-handlers #1301: [BUG] TS0012 not working correctly

Tropaion, 2022-01-22:
I’m using ZHA and added two 2-way switches (TS0012: BSEED Zigbee Smart EU UK RU Einzigen Live Touch Licht Schalter Alexa Smart Lebensdauer Kompatibel Mit Normalen Buchse Hause Dekoration|Switches| - AliExpress) today.
Now I have the problem that sometimes they don’t work correctly.
Sometimes they work very fast then suddnely react very slow, or it switches off/on but in ZHA it says its still on/off or if I turn on/off both sometimes only one gets switched, so basically they often behave weirdly,.

I don't know what the reason could be, in the ZHA Log, there are no errors.

MattWestb, 2022-01-23:
Very likely your switch is not loading the quirk for TS00X devices but not easy saying then  you was not providing any logs and device signature.

You can trying configure local quirks and download this file and change this line and saving it the quirk folder and restarting HA.
https://github.com/zigpy/zha-device-handlers/blob/dcee551996a0a9e388b6afb1e054dada46897ee5/zhaquirks/tuya/ts001x.py#L74
Replace the line with:
```
        MODEL: "TS0012",
```

Tropaion, 2022-01-23:
Today suddenly one of those two devices is NaN, in the Log this appeared:
`Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py:114
First occurred: 00:52:03 (10 occurrences)
Last logged: 00:52:05

[0xd61f:1:0xe001] Unknown cluster-specific command 225`

And this:
`Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:98
First occurred: 04:19:03 (1 occurrences)
Last logged: 04:19:03

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 103, in group_membership_scan
    await ep.group_membership_scan()
  File "/usr/local/lib/python3.9/site-packages/zigpy/endpoint.py", line 163, in group_membership_scan
    groups = {group for group in res[1]}
TypeError: 'Status' object is not iterable`

Thats the device signature:
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_e98krvvk",
  "model": "TS0012",
  "class": "zigpy.device.Device"
}`

MattWestb, 2022-01-23:
` "class": "zigpy.device.Device"` = no quirk for the device is loaded so its not working OK.
Configure local quirks and putting one patched quirk i was writing above. and restarting HA. https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Tropaion, 2022-01-23:
Thanks for your help.

I did it according to the description and changed ```MODELS_INFO: [("_TZ3000_fvh3pjaz", "TS0012")]``` to ```MODELS_INFO: [("_TZ3000_e98krvvk", "TS0012")]``` but the custom quirk didn't load.

I look through my other devices and also saw that the quirk for my TS0011 also didn't load, but surprisingly switching works.
So I also changed the line ```MODELS_INFO: [("_TZ3000_9hpxg80k", "TS0011")]``` to ```MODELS_INFO: [("_TZ3000_hafsqare", "TS0012")]```
Device Signature off TS0011:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_hafsqare",
  "model": "TS0011",
  "class": "zigpy.device.Device"
}
```

But both didn't load.

MattWestb, 2022-01-23:
For the TS0012 you must deleting the old line MODELS_INFO line (its need exact model for triggering) and putting  **in the right line** (74) :
```
        MODEL: "TS0012",
```
or it is not working.

In your last example you have one TS0011and its using the TS0011 class of the quirk and need making the same at this line:
https://github.com/zigpy/zha-device-handlers/blob/dcee551996a0a9e388b6afb1e054dada46897ee5/zhaquirks/tuya/ts001x.py#L26
with:
```
        MODEL: "TS0012",
``` 
and deleting the old MODELS_INFO line.

Tropaion, 2022-01-23:
I did it exactly as you said but sadly the quirks doens't apply to the device, I don't know why.
Does the filename of the quirks matter?

javicalle, 2022-01-23:
The filename doesn't matter.

I think that clusters do not match with your device.

I would suggest another approach:
* download the `ts011f.py` file:
  * https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f.py
* Add inside another class:
```python
class Tuya_2G_Switch_nogp(TuyaSwitch):
    """Tuya 2 gang light switch."""

    signature = {
        MODELS_INFO: [("_TZ3000_e98krvvk", "TS0012")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

It is NOT the definitive solution, but it will serve to see if we make it work.

Delete cache and restart HA.

Tropaion, 2022-01-23:
I don't know if its normal when there was no quirks applied but I don't have a chache file in the quirks folder.

javicalle, 2022-01-23:
If you are working with a `custom_zha_quirks` and it is configured in HA, I think that should be

Tropaion, 2022-01-23:
I tried you suggestion and got another error:
```
Logger: homeassistant.config_entries
Source: custom_zha_quirks/ts011f.py:123
First occurred: 23:26:01 (1 occurrences)
Last logged: 23:26:01

Error setting up entry USB Serial, s/n: n/a for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 313, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 99, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.9/site-packages/zhaquirks/__init__.py", line 403, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 529, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1029, in load_module
  File "<frozen importlib._bootstrap_external>", line 854, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts011f.py", line 119, in <module>
    class Tuya_2G_Switch_nogp(TuyaSwitch):
  File "/config/custom_zha_quirks/ts011f.py", line 123, in Tuya_2G_Switch_nogp
    MODELS_INFO: [("_TZ3000_e98krvvk", "TS0012")],
```

javicalle, 2022-01-23:
> Error setting up entry USB Serial, s/n: n/a for zha

Do you have any other config for the zha integration? O it is just something like:
```yaml
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

MattWestb, 2022-01-23:
@javicalle  If changing to `MODEL` in the TX001 for both TS0011 and TS0012 devices it shall normally loading the quirk but its need adding one new class then hi is having one new cluster "0xe001" that the "normal" is not having.

Tropaion, 2022-01-23:
@javicalle No, I never added something manually to the ZHA config beside this.
@MattWestb I don't know what a cluster is but I often get the error "unknown cluster".

javicalle, 2022-01-23:
> @javicalle If changing to `MODEL` in the TX001 for both TS0011 and TS0012 devices it shall normally loading the quirk but its need adding one new class then hi is having one new cluster "0xe001" that the "normal" is not having.

His device signature has:
```python
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
```

And `ts001x.py` have:
```python
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
```

I think that at least 3 clusters more are needed, don't you think?

javicalle, 2022-01-23:
It could be that @MattWestb is right, but @Tropaion you could try changing the `DEVICE_TYPE` at the `signature` part of both endpoints?

```python
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
```

Tropaion, 2022-01-23:
@javicalle What should I change?
In the commend in "ts001x.py" it says the "device_type=100", the same as in my device signature

javicalle, 2022-01-23:
Perhaps I have created more confusion by trying to intervene.

@MattWestb had suggested you to modify the `ts001x.py` file.

I had made another suggestion to you (independent of Matew's) which was to copy the `ts011f.py` file and add the definition of a new class inside (`Tuya_2G_Switch_nogp`).
In my proposal there was an error in the part of the signature of the device, which is what I have suggested you to modify to test.

Where I said:
```python
    signature = {
        MODELS_INFO: [("_TZ3000_e98krvvk", "TS0012")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```
I should said
```python
    signature = {
        MODELS_INFO: [("_TZ3000_e98krvvk", "TS0012")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

```

But this change must not be tested with the @MattWestb change also implemented.

Tropaion, 2022-01-24:
Okay, I now added the unknown clusters to the TS001x file and it applies the quirks now. Tomorrow I will test how it works.
BTW should I upload the quirk somehow and add it to zigpy? And don't know how this is handled.

javicalle, 2022-01-24:
> BTW should I upload the quirk somehow and add it to zigpy? And don't know how this is handled.

Do you mean to be added in the zha-device-handlers library?
The usual process would be to create a PR. But let's see how it works and then we can discuss it.

Tropaion, 2022-01-24:
So, the farthest one away lost connection overnight and I got these errors, don't know if this is relevant:
```
Logger: homeassistant.components.zha.core.channels.base
Source: components/zha/core/channels/base.py:428
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 01:10:07 (3 occurrences)
Last logged: 01:10:07

[0xB95A:1:0x0702]: async_initialize: all attempts have failed: [DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>')]
[0xB95A:1:0x0b04]: async_initialize: all attempts have failed: [DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>')]
[0xB95A:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>')]
```
The two 2-way switches I currently have installed are still reacting slowly, are the farthest ones away so I'm not sure if this is an aspect. The 1-way switches are one room nearer and work fine.

MattWestb, 2022-01-24:
`  "class": "ts001x.TuyaDoubleNoNeutralSwitch"` = ZHA have loading the quirk for the device and making all changes for it to working.
`  "class": "zigpy.device.Device"` = no matching quirk was found and its using all standard function of the device (if the device is 100% Zigbee standard its the default).

As i was saying the quirk is deleting the time cluster and the device shall start working OK and not  resetting its state after some minute.

Now you only need doing the same for the TS0011 ;-)

Tropaion, 2022-01-24:
@MattWestb I already implemented it yesterday, this is my current quirks file:
[bseed_ts001x.py.txt](https://github.com/zigpy/zha-device-handlers/files/7924990/bseed_ts001x.py.txt)
But I still got these errors:
```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py:95
First occurred: 11:56:44 (21 occurrences)
Last logged: 11:56:44

- Unknown cluster 61184
- Unknown cluster 57345
- Unknown cluster 57344
 ```
BTW does it matter which base class I use? Currently I have ```BSEED_Tuya_Single_No_N(TuyaSwitch)``` but I also saw many which use custom device: ```BSEED_Tuya_Single_No_N(CustomDevice)```. I don't know which to use.

Tropaion, 2022-01-24:
@MattWestb After trial and error I got it working now.
I found that the error is not from these two devices but from the ZigBee Bseed Dimmer (TS0601), which hasn't been implemented yet in ZHA.

But I'm still not sure which base class is better to use ```BSEED_Tuya_Single_No_N(TuyaSwitch)``` or ```BSEED_Tuya_Single_No_N(CustomDevice)```

I don't know if and how to make a PR for this, so I will post the file here: [bseed_ts001x.py.txt](https://github.com/zigpy/zha-device-handlers/files/7925833/bseed_ts001x.py.txt)

MattWestb, 2022-01-24:
### Servus !!
I was changing your last for using MODEL so its working with other tuya devices with the same setting (shall doing the same with the "old" 11 and 12 classes) and running black on it.
The original quirk is using `TuyaSwitch` so i think its better and safer using that  if its working OK with your devices !! 

I think you shall making the class name better than the "old" then its have more functionality and describing it 2.
[bseed_ts001x.py_MW.txt](https://github.com/zigpy/zha-device-handlers/files/7925906/bseed_ts001x.py_MW.txt)

This switches is coming as only controller (as remotes) and also with integrated relays for in wall mounting. Therefore its little tricky not braking both types in the quirk.

You shall getting more attributes on the OnOff cluster but i dont knowing if its working on your model. 
https://github.com/zigpy/zha-device-handlers/blob/db6a9316f08f58631b0efa3a0b049ae7fe90f169/zhaquirks/tuya/__init__.py#L886-L901

https://github.com/zigpy/zha-device-handlers/blob/db6a9316f08f58631b0efa3a0b049ae7fe90f169/zhaquirks/tuya/__init__.py#L827-L857

Great work done !!

If you have getting both working OK then its time making one nice PR for them and perhaps also adding one new class for the TS0013 that is very likely having the same settings but one extra endpoint.

The problem with the TS0601 dimmer is over my head but @javicalle have getting nearly all working on them but hi need little more testing and feedback for them to working with all futures.

One good thing with the TS001X family switches is that you can binding them to light groups and steering the lights also the the system is offline that is not possible with the TS0601 devices.

Tropaion, 2022-01-24:
Ok, I thought ```MODEL``` is a old syntax and can cause collision if there a same definitions.
So, should I change the class name and file name or can I keep it? I don't know if there is a naming convention.
I also added the triple way switch but I don't have a device to test it:  [bseed_ts001x.py.txt](https://github.com/zigpy/zha-device-handlers/files/7926134/bseed_ts001x.py.txt)

> One good thing with the TS001X family switches is that you can binding them to light groups and steering the lights also the the system is offline that is not possible with the TS0601 devices.

Do you mean the ZigBee Groups?
I tried settung one up for all light so I can turn them all off but I can't get it to work, trying it for a few days now -.-
Maybe you have a suggestion for me? There isn't much about it in the net.
[image]

MattWestb, 2022-01-24:
The MODELS is making all device with the right model type and correct cluster config using the quirk.
MODELS_INFO must also the name so can selecting with models shall using the quirk then must adding all IDs or its not working.
Then tuya looks using the same function on all "normal" Zigbee devices its better using MODELS and dont need addinga all variants of them.

Your switch is having on/off cluster as in so ZHA shall doing 1/2 lights on the device card.
If its working OK you can switching them on and off one by one.
If they working normal it shall being possible adding them to light groupes like in your screenshot. But i think the problem is that its one sleeping end device if looking in the device signature: `*is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",` so trying waking it up then sending commands to it and see if its working.
It can being that its not supporting group binding then one sleeping  device.

More "normal" devices like IKEA have support for over 50 groups in there devices and and no problem adding multiple groups. Some less good device cant only being bound to one group or not at all.

Tropaion, 2022-01-24:
@MattWestb So how do I wake it up? It's possible that it's sleeping because it says it's battery powered, probably because its without N wire. But the device has a group cluster, doesn't this imply that it support groups?

MattWestb, 2022-01-24:
Xiaomi / Aqara have doing may switches without neutral and normally they is sleepers.

The device dont have pull control cluster (its Zigbee 3 function that making the end device pulling commands from the coordinator and not losing commands that have timing out) so the routers must keeping your command but they is only doing it for 7 seconds and then throwing it (old not Zigbee 3 way).

Then you is changing state in HA is it working OK or is it very delayed and timing out oft ?

I only having some tuya TRVs that is sleepers that need getting commands but they is normally working OK but can being very lazy.

And still still likely the switches is not supporting groups but must looking in the log if the add group is getting OK or some error (then the device is receiving it).

Then sending commands to the switch is the best pressing one button so its sending somthing and have the radio on and can receiving the command.

Tropaion, 2022-01-24:
The device is often delayed and sometime timing out.
Is it possible to disable sleeping?

MattWestb, 2022-01-24:
Only possible by flashing one new firmware on the device and we dont have the code tuya have baked in there device only the Zigbee stack from Silabs they is using.

I think you must see the without neutral switches and one sleeping remote with relays that is made sending its status / commands and not receiving commands like "normal lights is doing (and i hope tuya switches with neutral is no sleepers).

Look around for information and you is finding some interesting or perhaps bad. I have not doing it then i have finding one good tuya simmer module that can being bonded and have installing it and then IKEA was updating the firmware in there remotes and was losing the group binding and breaking the "remote controlling" but working OK local and also turing all endpoints / gangs on and off.

https://github.com/MattWestb/tuya-Switches-Dimmer

MattWestb, 2022-01-24:
It can helping but very likely not then i dont think they is changing the device type (router, sleeping or not sleeping end device) but you never knowing.

Tropaion, 2022-01-24:
What I don't understand...is there a cluster attribute where I can see which type it is and if it supports groups?

MattWestb, 2022-01-24:
"Normal devices" is having cluster= is working out of the box with ZHA then its Zigbee compatible but with tuya is little "working or not" and also hiding clusters that is working so we dont finding  them so easy.

I fighting with one TS004F that is one dimmer remote but can changing to one "scene controller" but as light controller is sending commands on, off dim up and down OK but its having group cluster as in (its for lights) and out (for sending from remote) and normally you shall binding the  out cluster to one group but with this device its needed adding one group to the in cluster as it was one light.

Its all tuya :-(((

Tropaion, 2022-01-24:
@MattWestb BTW are making the PR or should I?

closed 2022-01-24 by Tropaion

MattWestb, 2022-01-24:
I can making it but i cant testing if its working.
If you is doing one its great then you can testing the finale code for your 2 switches and adding the TS0013 without testing its very OK then all before have looking the same as the TS0012 with one extra endpoint.

Also if you can changing the old to using `MODEL` its being  better then user dont need requesting new devices (as long tuya is not making new versions of them).

If you need help i can hiniting little.

Most important is formatting the code with black and i normally using black playground for doing  that. https://black.vercel.app/

Tropaion, 2022-01-24:
I already did everything you said:
[bseed_ts001x.py.txt](https://github.com/zigpy/zha-device-handlers/files/7929028/bseed_ts001x.py.txt)

MattWestb, 2022-01-24:
Was making one fast look and its looks OK but im not one coding expert.
I think the best is adding the new switches to the exiting quirk by adding the new classes so not having different files for same type of the devices like is being made for other tuya quirks and not making one new file for every new "manufacturer".