zigpy/zha-device-handlers #1302: [Device Support Request] Moes dimmer switches _TZE200_ip2akl4w & _TZE200_fjjbhx9d

Daandeve, 2022-01-23:
This is a Tuya device that has no entities and/or controls in home assistant and its a simple Zigbee Tuya dimmer. It should be very similar to many other devices like this.

1 Gang _TZE200_ip2akl4w
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_ip2akl4w",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

2 Gang _TZE200_fjjbhx9d
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_fjjbhx9d",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

Thanks in advance for the support.

javicalle, 2022-01-23:
It seems that is like others Tuya dimmers but with the green power endpoint.
I believe that can be easily added.

For future reference do you have a link to the product to help visually identify it?

javicalle, 2022-01-23:
The 2 gang will require the PR #1247 to be merged.
The 1 gang can be fixed with PR #1122, but it is a Work In Progress

@Daandeve are you able to test local quirks in your environment?

Daandeve, 2022-01-23:
> It seems that is like others Tuya dimmers but with the green power endpoint. I believe that can be easily added.
> 
> For future reference do you have a link to the product to help visually identify it?

This is the product: https://nl.aliexpress.com/item/1005003619398730.html?gatewayAdapt=glo2nld&spm=a2g0o.9042311.0.0.130d4c4dGYyEsh
I have the 1 and 2 EU gang models

Daandeve, 2022-01-23:
> The 2 gang will require the PR #1247 to be merged. The 1 gang can be fixed with PR #1122, but it is a Work In Progress
> 
> @Daandeve are you able to test local quirks in your environment?

I would like to test it but I do not fully understand how I can test quirks.

javicalle, 2022-01-23:
Depending on the environment there are different ways to do it.
Here are some examples of how to do it:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274
* https://github.com/zigpy/zha-device-handlers#testing-quirks-in-development-in-docker-based-install
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussion-1237451

It is not too difficult, but it does require touching the insides of the installation and it is preferable to do it if you feel comfortable with those kind of modifications.

Daandeve, 2022-01-23:
Thanks, but the device does not show entities and no errors in the logs. I only see this in the logs, I don't think it was there before.
```
[0x80c1:1:0xef00] Unknown cluster-specific command 1
[0x80c1:1:0xef00] Unknown cluster-specific command 36
```

I put the script in `custom_zha_quirks/tuya/mcu/__init__.py`

javicalle, 2022-01-23:
Ok, let's try it.

You need to copy 2 file to the `custom_zha_quirks/tuya/mcu/`:
* https://github.com/zigpy/zha-device-handlers/blob/a4bbbb5f7ffd369755ec97826b897888255e4263/zhaquirks/tuya/mcu/__init__.py
* https://github.com/zigpy/zha-device-handlers/blob/a4bbbb5f7ffd369755ec97826b897888255e4263/zhaquirks/tuya/ts0601_dimmer.py

Then you need to modify the `ts0601_dimmer.py` file:

* add the new `GreenPowerProxy` import:
```python
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, LevelControl, Ota, Scenes, Time
```

* Add 2 new classes at the end of file:
```python
class TuyaSingleSwitchDimmerGP(TuyaDimmerSwitch):
    """Tuya touch switch device."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098,
        # maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        # maximum_outgoing_transfer_size=82, descriptor_capability_field=0>",
        # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1 input_clusters=[0, 4, 5, 61184] output_clusters=[10, 25]>
        MODELS_INFO: [
            ("_TZE200_ip2akl4w", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    TuyaLevelControl,
                    TuyaManufacturerLevelControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

class TuyaDoubleSwitchDimmerGP(TuyaDimmerSwitch):
    """Tuya double channel dimmer device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_fjjbhx9d", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaLevelControlManufCluster,
                    TuyaOnOffMCU,
                    TuyaInWallLevelControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffMCU,
                    TuyaInWallLevelControl,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

```

Also, you can enable debug logs for `custom_zha_quirks`:
```
logger:
  default: warning
  logs:
    .../...
    custom_zha_quirks: debug
```

Delete  the `__pycache__` folder and restart HA.

javicalle, 2022-01-23:
It seems that device also have the `power_on_state` and `backlight_mode`. Not implemented for dimmer yet.
This must be reviewed in order to support it.

javicalle, 2022-01-23:
And probably also support `minimum_level` and `maximum_level`. It is not documented, but the photos from the link suggest it.

Daandeve, 2022-01-23:
I followed all steps but nothing is coming up, no logs and no entities. I tried to remove the zigbee device and add it again, but with no success..

Daandeve, 2022-01-23:
When I reconfigure the device the binding and reporting tables are empty.
[image]

javicalle, 2022-01-23:
Do not have any logs when operate the device? Maybe you need to enable the zigpy logs.
Can you do a power cycle to the device? Usually this way, devices will broadcast their status.

javicalle, 2022-01-23:
And the device signature? It is the quirk loaded?

Daandeve, 2022-01-23:
This is what I get in the logs:
```log
2022-01-24 00:00:21 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=141 command_id=Command.Report_Attributes>
2022-01-24 00:00:21 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=65503, value=<TypeValue type=CharacterString, value=���)e���)i���)i���)i���)i���)i���)i���)i���)i���)i���)i���)i>)]]
2022-01-24 00:00:21 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0x0000] Attribute report received: 65503=���)e���)i���)i���)i���)i���)i���)i���)i���)i���)i���)i���)i

AND

2022-01-24 00:02:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=142 command_id=1>
2022-01-24 00:02:28 WARNING (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] Unknown cluster-specific command 1
2022-01-24 00:02:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL request 0x0001: b'\x00\xd1\x07\x01\x00\x01\x01'
2022-01-24 00:02:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] No handler for cluster command 1
```

Daandeve, 2022-01-23:
> And the device signature? It is the quirk loaded?

How can I see if the quirk is loaded correctly?

javicalle, 2022-01-23:
> How can I see if the quirk is loaded correctly?

From Configuration --> Integrations --> ZHA Devices --> select your device --> Zigbee device signature

It's the same info you put in the first comment.

Daandeve, 2022-01-23:
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_fjjbhx9d",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```
I don't think it is loaded?

javicalle, 2022-01-23:
> I don't think it is loaded

No, it is not.

I believe that I have made a mess copying the files. Give me a second to review it.

Daandeve, 2022-01-23:
I think I did it right
[image]

javicalle, 2022-01-23:
Then you hace to replace at `ts0601_dimmer.py` some imports:
```python
from custom_zha_quirks.tuya.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff as TuyaOnOffMCU,
)
```

I think the rest will be OK.

(I have updated the `TuyaSingleSwitchDimmerGP` signature part)

Daandeve, 2022-01-23:
I updated the files, without success. The quirk is not found in general.
```logs
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_fjjbhx9d TS0601 (a4:c1:38:75:54:c8:df:96)
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1, 242}
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1, 242}
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {11, 13} {1, 242}
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2022-01-24 00:40:53 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
```

javicalle, 2022-01-23:
Just to be sure, are in HA the files placed in `/config/custom_zha_quirks/tuya/...`?

javicalle, 2022-01-23:
No error logs in restart? no `__pycache__` folder in custom_zha_quirks folder?

Daandeve, 2022-01-23:
> Just to be sure, are in HA the files placed in `/config/custom_zha_quirks/tuya/...`?

Yes that is correct

Daandeve, 2022-01-23:
> No error logs in HA restart? no `__pycache__` folder in custom_zha_quirks folder?

No errors and no pycache

javicalle, 2022-01-24:
No idea what could be going on...

Maybe you can try to put all the files at root level `/config/custom_zha_quirks/`. The import from `ts0601_dimmer.py` file would be replaced with:
```python
from . import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff as TuyaOnOffMCU,
)
```
The rest will be fine as is.

javicalle, 2022-01-24:
Are you sure HA is restarting? Sometimes, if there is a validation and other error, the console doesn't tell you and it appears to have restarted, but it really hasn't.
(Sorry to ask, but I don't find a reason way it is not loading the quirk)

Daandeve, 2022-01-24:
Okay now it loads but I get
`ImportError: attempted relative import with no known parent package`

Daandeve, 2022-01-24:
> Are you sure HA is restarting? Sometimes, if there is a validation and other error, the console doesn't tell you and it appears to have restarted, but it really hasn't. (Sorry to ask, but I don't find a reason way it is not loading the quirk)

I thought that was happening as well, but I restarted the full VM. So that is not the problem

javicalle, 2022-01-24:
> Okay now it loads but I get
> `ImportError: attempted relative import with no known parent package`

Well, then put the `__init__.py` file at any place, for example in the original one `/config/custom_zha_quirks/tuya/mcu/` and adapt the import accordly:
```python
from custom_zha_quirks.tuya.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff as TuyaOnOffMCU,
)
```

(not too diferent from previous situation 😅 )

Daandeve, 2022-01-24:
It works!🎇

`__init__.py` location:
`/config/custom_zha_quirks/tuya/mcu/__init__.py`

`ts0601_dimmer.py` location:
`/config/custom_zha_quirks/ts0601_dimmer.py`

And the original import

Daandeve, 2022-01-24:
> Well, then put the `__init__.py` file at any place, for example in the original one `/config/custom_zha_quirks/tuya/mcu/` and adapt the import accordly:
>
> ```python
> from custom_zha_quirks.tuya.mcu [...]

Exactly 😀

javicalle, 2022-01-24:
If you're up for it, I'd like to validate a couple of device features to see if they can be implemented.
If you are predisposed to it, I will contact you in the next few days to validate. But for now, I'm going to rest a bit 😴.

Thanks and regards.

Daandeve, 2022-01-24:
The on/off states report correctly, but they are not controllable. The brightness states report sometimes correctly, but are also not controllable.
When I try to turn it on or off the zha state return a SUCCESS
```log
2022-01-24 01:36:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:2:0x0006] Sending Tuya Cluster Command... Cluster Command is 1, Arguments are ()
2022-01-24 01:36:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=2, cluster_attr='on_off', attr_value=1)
2022-01-24 01:36:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] get_dp_mapping --> found DP: 7
2022-01-24 01:36:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] from_cluster_data: 7, DPToAttributeMapping(ep_attribute='on_off', attribute_name='on_off', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=2)
2022-01-24 01:36:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ztype: Bool.true
2022-01-24 01:36:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] from_value: [1, 1]
2022-01-24 01:36:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] raw: b'\x01'
2022-01-24 01:36:28 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=127, dp=7, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))
2022-01-24 01:36:28 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x80C1:2:0x0006]: executed 'on' command with args: '()' kwargs: '{}' result: Status.SUCCESS
2022-01-24 01:36:28 DEBUG (MainThread) [homeassistant.components.zha.entity] light.bathroom_dimmer_level_on_off_2: turned on: {'on_off': <Status.SUCCESS: 0>}
2022-01-24 01:36:29 DEBUG (MainThread) [zigpy.zcl] [0x8df7:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=59 command_id=Command.Report_Attributes>
2022-01-24 01:36:29 DEBUG (MainThread) [zigpy.zcl] [0x8df7:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
2022-01-24 01:36:29 DEBUG (MainThread) [zigpy.zcl] [0x8df7:1:0x0006] Attribute report received: on_off=0
2022-01-24 01:36:31 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=128 command_id=Command.Default_Response>
```

Daandeve, 2022-01-24:
> If you're up for it, I'd like to validate a couple of device features to see if they can be implemented. If you are predisposed to it, I will contact you in the next few days to validate. But for now, I'm going to rest a bit 😴.
> 
> Thanks and regards.

Yes for sure! Take your time¯\_(ツ)_/¯

MattWestb, 2022-01-24:
If making updated in one normal quirk **and** one INIT its not working putting both in local quirk ford then both "see" the originals in the HA container. The INIT **must** being updated in the container for the local quirk can "see" / using it or putting both in the container.
That's the limit with local quirks.

javicalle, 2022-01-24:
I will need traces from the physical device operation:
* switch on
* switch off
* dimm (with aprox values)

Daandeve, 2022-01-24:
Okay, I don't really know what I'm looking for. But this happened when I turned the switch on, then dimmed the switch and after I turned it off
```log
2022-01-24 16:26:31 DEBUG (MainThread) [zigpy.zcl] [0x8df7:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=239 command_id=Command.Report_Attributes>
2022-01-24 16:26:31 DEBUG (MainThread) [zigpy.zcl] [0x8df7:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
2022-01-24 16:26:31 DEBUG (MainThread) [zigpy.zcl] [0x8df7:1:0x0006] Attribute report received: on_off=0
2022-01-24 16:26:33 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=126 command_id=1>
2022-01-24 16:26:33 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=33, dp=7, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))]
2022-01-24 16:26:34 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=126 command_id=1>
2022-01-24 16:26:34 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=126 command_id=1>
2022-01-24 16:26:34 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x7e seq -- probably duplicate response
2022-01-24 16:26:34 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=126 command_id=1>
2022-01-24 16:26:34 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x7e seq -- probably duplicate response
2022-01-24 16:26:34 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=126 command_id=1>
2022-01-24 16:26:34 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x7e seq -- probably duplicate response
2022-01-24 16:26:35 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=126 command_id=1>
2022-01-24 16:26:35 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x7e seq -- probably duplicate response
2022-01-24 16:26:35 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=126 command_id=1>
2022-01-24 16:26:35 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x7e seq -- probably duplicate response
2022-01-24 16:26:35 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=126 command_id=1>
2022-01-24 16:26:35 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x7e seq -- probably duplicate response
2022-01-24 16:26:37 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=127 command_id=Command.Report_Attributes>
2022-01-24 16:26:37 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=64>), Attribute(attrid=65506, value=<TypeValue type=uint8_t, value=32>), Attribute(attrid=65508, value=<TypeValue type=uint8_t, value=0>)]]
2022-01-24 16:26:37 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0x0000] Attribute report received: app_version=64, 65506=32, 65508=0
2022-01-24 16:26:38 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=127 command_id=Command.Report_Attributes>
2022-01-24 16:26:38 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0x0000] ZCL request 0x000a: [[Attribute(attrid=1, value=<TypeValue type=uint8_t, value=64>), Attribute(attrid=65506, value=<TypeValue type=uint8_t, value=32>), Attribute(attrid=65508, value=<TypeValue type=uint8_t, value=0>)]]
2022-01-24 16:26:38 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0x0000] Attribute report received: app_version=64, 65506=32, 65508=0
2022-01-24 16:26:38 DEBUG (MainThread) [zigpy.util] Duplicate 127 TSN
[... cut ...]
2022-01-24 16:26:43 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x80C1:2:0x0008]: received attribute: 0 update with value: 104
2022-01-24 16:26:43 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=133 command_id=1>
2022-01-24 16:26:43 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=38, dp=7, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x00', *payload=<Bool.false: 0>))]
```

javicalle, 2022-01-24:
That's weird.
You have several send retries and that should not be happening.

Can you turn on the `custom_zha_quirks: debug` log level?

Daandeve, 2022-01-24:
I have the custom_zha_quirks on the debug level but it does not seem to do anything...

MattWestb, 2022-01-24:
Be aware of the tuya is using 2 hardware and 2 software for the Zigbee module.
On the software side its one  no sleeping version and one sleeping version.
TRVs is normally using sleeping firmware and i think switches with no neutral is doing the  same but with neutral is using no sleeping firmware.

Look in the signature if its having `is_receiver_on_when_idle=True` = not sleeping and (from my tuya TRV) `is_receiver_on_when_idle=False` ? sleeping end device.

Daandeve, 2022-01-24:
> Be aware of the tuya is using 2 hardware and 2 software for the Zigbee module. On the software side its one no sleeping version and one sleeping version. TRVs is normally using sleeping firmware and [...]

Okay interesting, but my switches do have neutral wires and they are wired up.

javicalle, 2022-01-24:
Maybe the logger is not like I thought...

Can you please also enable the debug level for `zigpy.zcl`? Maybe this one is too verbose...

Daandeve, 2022-01-24:
> > I have the custom_zha_quirks on the debug level but it does not seem to do anything...
> 
> And you said that HA is updating the state, but you can not control device from HA, correct?

Correct

Daandeve, 2022-01-24:
> Maybe the logger is not like I thought...
> 
> Can you please also enable the debug level for `zigpy.zcl`? Maybe this one is too verbose...

This is what I get, it look the same to me.
```log
2022-01-24 18:39:23 DEBUG (MainThread) [zigpy.device] [0x22a8] Delivery error for seq # 0x0b, on endpoint id 1 cluster 0x0000: message send failure
2022-01-24 18:39:26 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=230 command_id=1>
2022-01-24 18:39:26 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=39, dp=7, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))]
2022-01-24 18:39:26 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=230 command_id=1>
2022-01-24 18:39:26 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=230 command_id=1>
2022-01-24 18:39:26 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0xe6 seq -- probably duplicate response
2022-01-24 18:39:27 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=230 command_id=1>
2022-01-24 18:39:27 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0xe6 seq -- probably duplicate response
2022-01-24 18:39:27 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=230 command_id=1>
2022-01-24 18:39:27 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0xe6 seq -- probably duplicate response
2022-01-24 18:39:27 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=230 command_id=1>
2022-01-24 18:39:27 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0xe6 seq -- probably duplicate response
2022-01-24 18:39:27 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=230 command_id=1>
2022-01-24 18:39:27 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0xe6 seq -- probably duplicate response
2022-01-24 18:39:27 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=230 command_id=1>
2022-01-24 18:39:27 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0xe6 seq -- probably duplicate response
2022-01-24 18:39:30 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=231 command_id=1>
2022-01-24 18:39:30 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=40, dp=8, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x95\x01\x00\x00', *payload=405))]
2022-01-24 18:39:30 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x80C1:2:0x0008]: received attribute: 0 update with value: 103
2022-01-24 18:39:31 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=231 command_id=1>
2022-01-24 18:39:31 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=231 command_id=1>
2022-01-24 18:39:31 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0xe7 seq -- probably duplicate response
2022-01-24 18:39:31 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=231 command_id=1>
2022-01-24 18:39:31 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0xe7 seq -- probably duplicate response
2022-01-24 18:39:31 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=231 command_id=1>
[... cut ...]
2022-01-24 18:39:35 DEBUG (MainThread) [zigpy.zcl] [0x4ee9:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]
2022-01-24 18:39:35 DEBUG (MainThread) [zigpy.zcl] [0x4ee9:1:0x0006] Attribute report received: on_off=1
2022-01-24 18:39:37 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](EZSP): Attempting to checkin with device - missed checkins: 1
2022-01-24 18:39:37 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](EZSP): does not have a mandatory basic cluster
```

javicalle, 2022-01-24:
> This is what I get, it look the same to me.

Yes it is pretty similar. 

You don't get any trace from custom quirk classes and I don't know why.
Also, you are getting retries from device and it shouldn't be happening.

Please, confirm once again that device is loading the quirks, I will look at the code to see what can be happening...

Daandeve, 2022-01-24:
Yes the quirk is loaded.
[image]

MattWestb, 2022-01-24:
Deleting the ` __pycache__ ` folder (in the local quirk directory) between restarts  can being necessarily after doing changes of the code.

Also restarting HA can eliminating some problems like quirk is not loading and so on.

Daandeve, 2022-01-24:
> Deleting the `__pycache__` folder (in the local quirk directory) between restarts can being necessarily after doing changes of the code.
> 
> Also restarting HA can eliminating some problems like quirk is not loading and so on.

Yes I tried that but sadly with no success..
When I turn the switch on in HA it logs this
```log
2022-01-24 22:28:16 DEBUG (MainThread) [zigpy.zcl] [0x80c1:2:0x0006] Sending Tuya Cluster Command... Cluster Command is 1, Arguments are ()
2022-01-24 22:28:16 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=2, cluster_attr='on_off', attr_value=1)
2022-01-24 22:28:16 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] get_dp_mapping --> found DP: 7
2022-01-24 22:28:16 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] from_cluster_data: 7, DPToAttributeMapping(ep_attribute='on_off', attribute_name='on_off', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=2)
2022-01-24 22:28:16 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ztype: Bool.true
2022-01-24 22:28:16 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] from_value: [1, 1]
2022-01-24 22:28:16 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] raw: b'\x01'
2022-01-24 22:28:16 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=9, dp=7, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))
2022-01-24 22:28:16 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x80C1:2:0x0006]: executed 'on' command with args: '()' kwargs: '{}' result: Status.SUCCESS
2022-01-24 22:28:16 DEBUG (MainThread) [homeassistant.components.zha.entity] light.bathroom_dimmer_level_on_off_2: turned on: {'on_off': <Status.SUCCESS: 0>}
2022-01-24 22:28:18 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=10 command_id=Command.Default_Response>
```

When I turn the switch on manually it logs this
```log
2022-01-24 22:28:46 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=156 command_id=1>
2022-01-24 22:28:46 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=50, dp=7, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))]
2022-01-24 22:28:46 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=156 command_id=1>
2022-01-24 22:28:47 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=156 command_id=1>
2022-01-24 22:28:47 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x9c seq -- probably duplicate response
2022-01-24 22:28:47 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=156 command_id=1>
2022-01-24 22:28:47 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x9c seq -- probably duplicate response
2022-01-24 22:28:47 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=156 command_id=1>
2022-01-24 22:28:47 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x9c seq -- probably duplicate response
2022-01-24 22:28:47 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=156 command_id=1>
2022-01-24 22:28:47 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x9c seq -- probably duplicate response
2022-01-24 22:28:47 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=156 command_id=1>
2022-01-24 22:28:47 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x9c seq -- probably duplicate response
2022-01-24 22:28:48 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=156 command_id=1>
2022-01-24 22:28:48 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x9c seq -- probably duplicate response
```

When I turn the switch off manually it logs this
```log
[... cut ...]
2022-01-24 22:28:49 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x9d seq -- probably duplicate response
2022-01-24 22:28:49 DEBUG (MainThread) [zigpy.zcl] [0x80c1:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=157 command_id=1>
2022-01-24 22:28:49 DEBUG (MainThread) [zigpy.device] [0x80c1] Invalid state on future for 0x9d seq -- probably duplicate response
```

javicalle, 2022-01-24:
> I did see this `Logger: zigpy.zcl, Unknown cluster 61184` in my log could this warning be related?

I get that message sometimes during HA restart.
I think it is related to some message that is received before loading the quirks or maybe when initializing the device from the `entity_registry`, but I don't know, just assumptions.

Your logs are saying that the device broadcast their status (or changes), but there is no cluster to handle the message (and that is why it forwards it several times: same TSN).

These are messages from 'switch 2'. On/Off events goes to `dp=7`. Dimm events goes to `dp=8`.

I must have left something half done in the example implementation, but I can't see it.

Does (for that device) 'switch 1' behave the same way?

You have also commented that you have a single-gang button (the `_TZE200_ip2akl4w`), does it behave just as badly?

Daandeve, 2022-01-24:
> I get that message sometimes during HA restart. I think it is related to some message that is received before loading the quirks or maybe when initializing the device from the `entity_registry`, but [...]

The switch 1 & 2 from the 2 gang and the 1 gang switch behave the same.
Here the logs from the 1 gang switch, first I turned it on then dimmed then turned it off.
```txt
2022-01-25 00:15:02 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=128 command_id=1>
2022-01-25 00:15:02 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=7, dp=1, data=TuyaData(dp_type=<TuyaDPType.BOOL: 1>, function=0, raw=b'\x01', *payload=<Bool.true: 1>))]
2022-01-25 00:15:02 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=128 command_id=1>
2022-01-25 00:15:03 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=128 command_id=1>
2022-01-25 00:15:03 DEBUG (MainThread) [zigpy.device] [0xf51b] Invalid state on future for 0x80 seq -- probably duplicate response
2022-01-25 00:15:03 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=128 command_id=1>
2022-01-25 00:15:03 DEBUG (MainThread) [zigpy.device] [0xf51b] Invalid state on future for 0x80 seq -- probably duplicate response
2022-01-25 00:15:03 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=128 command_id=1>
2022-01-25 00:15:03 DEBUG (MainThread) [zigpy.device] [0xf51b] Invalid state on future for 0x80 seq -- probably duplicate response
2022-01-25 00:15:03 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=128 command_id=1>
2022-01-25 00:15:03 DEBUG (MainThread) [zigpy.device] [0xf51b] Invalid state on future for 0x80 seq -- probably duplicate response
2022-01-25 00:15:03 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=128 command_id=1>
2022-01-25 00:15:03 DEBUG (MainThread) [zigpy.device] [0xf51b] Invalid state on future for 0x80 seq -- probably duplicate response
2022-01-25 00:15:04 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=128 command_id=1>
2022-01-25 00:15:04 DEBUG (MainThread) [zigpy.device] [0xf51b] Invalid state on future for 0x80 seq -- probably duplicate response
2022-01-25 00:15:05 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=129 command_id=1>
2022-01-25 00:15:05 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL request 0x0001: [TuyaCommand(status=0, tsn=8, dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'T\x01\x00\x00', *payload=340))]
2022-01-25 00:15:05 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF51B:1:0x0008]: received attribute: 0 update with value: 86
2022-01-25 00:15:06 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=129 command_id=1>
2022-01-25 00:15:06 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=129 command_id=1>
2022-01-25 00:15:06 DEBUG (MainThread) [zigpy.device] [0xf51b] Invalid state on future for 0x81 seq -- probably duplicate response
2022-01-25 00:15:06 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=129 command_id=1>
2022-01-25 00:15:06 DEBUG (MainThread) [zigpy.device] [0xf51b] Invalid state on future for 0x81 seq -- probably duplicate response
2022-01-25 00:15:06 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=129 command_id=1>
[... cut ...]
2022-01-25 00:15:10 DEBUG (MainThread) [zigpy.device] [0xf51b] Invalid state on future for 0x87 seq -- probably duplicate response
2022-01-25 00:15:10 DEBUG (MainThread) [zigpy.zcl] [0xf51b:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=135 command_id=1>
2022-01-25 00:15:10 DEBUG (MainThread) [zigpy.device] [0xf51b] Invalid state on future for 0x87 seq -- probably duplicate response
```

javicalle, 2022-01-24:
I'm sorry to continually ask questions that may seem stupid, but there must be something I've overlooked.

Could you show me the signature of the device with the quirks applied?

It's the same you do in [comment](https://github.com/zigpy/zha-device-handlers/issues/1302#issuecomment-1019588507)

Daandeve, 2022-01-24:
Don't be sorry haha, when I can't find something myself I get that too.
1 gang signature
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_ip2akl4w",
  "model": "TS0601",
  "class": "ts0601_dimmer.TuyaSingleSwitchDimmerGP"
}
```

2 gang signature
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0006",
        "0x0008"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_fjjbhx9d",
  "model": "TS0601",
  "class": "ts0601_dimmer.TuyaDoubleSwitchDimmerGP"
}
```

javicalle, 2022-01-24:
The last silly check for tonight, would you be so kind as to check that the new classes have _exactly_ this definition in the `replacement` part, inside the enpoint 1 definition:
```python
                    TuyaLevelControlManufCluster,
                    TuyaOnOffMCU,
                    TuyaInWallLevelControl,
```

Daandeve, 2022-01-24:
In the `ts0601_dimmer.py`?

javicalle, 2022-01-24:
Yes, in the `ts0601_dimmer.py` file inside the 2 new classes `TuyaSingleSwitchDimmerGP` and `TuyaDoubleSwitchDimmerGP`.

Daandeve, 2022-01-24:
Okay, yes in the single and the double it is correct

javicalle, 2022-01-24:
Aaaaaaargggghhh! 💣

Nothing else comes to mind today.
I'm going to leave it until tomorrow. Thanks for trying.

Daandeve, 2022-01-24:
> Aaaaaaargggghhh! 💣
> 
> Nothing else comes to mind today. I'm going to leave it until tomorrow. Thanks for trying.

😅
Sure, I'll try to look into it as well

Daandeve, 2022-01-25:
I just noticed this in my log `[0x80c1:2:0x0006] Unsupported command_id: 2`

[349 more replies]

PR #1489 (2022-04-19): Implementations for `NO_MANUFACTURER_ID` dimmers
This PR adds device definitions for dimmers that require send commands with the `NO_MANUFACTURER_ID` (or what is the same `manufacturer_specific=False`).

A few comments:
* `tuya/mcu/TuyaLevelControl` now handles `move_to_level_with_on_off` command, sending an `on_off` command before the `current_level` command.
* the `TuyaLevelControlManufCluster` handles DPs up to 3 gangs.
* the new `NoManufacturerCluster` will force the `NO_MANUFACTURER_ID` in commands. This is needed for recent dimmers
* at some point I wanted to migrate the remaining classe [... cut]
closed 2022-04-19 by dmulcahey