zigpy/zha-device-handlers #1303: [BUG]

mdeweerd, 2022-01-23:
**Describe the bug**
"Unstable" UI behavior of LIDL  TS0505A/_TZ3000_dbou1ap4 .

**To Reproduce**
Pictures say more than 1000 words:

There is a strange UI behavior when changing the intensity (the UI switches the light to off but it is still on).  I can set on/off, intensity and temperature despite some strange behaviour;
[image]

The light can be switched on and off normally.
[image]

I tried another light (TI Sample Light) and that's behaving normally.

**Expected behavior**
The intensity change should not result in the device switching off

**Additional context**

I probably upgraded HAOS to 2021.12.3 a few days before reporting this in the Adaptive Lighting component.

I filed an [issue for the Adaptive Lighting component](https://github.com/basnijholt/adaptive-lighting/issues/245).  The behavior is worse when that is enabled, but I think the core issue is not really there..

This might be related to the appearance of https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lidl/rgbcct.py with #1177 .

These are the attributes available with their values (when they can be read) - [TS0505A__TZ3000_dbou1ap4_ff23a460_scan_results.txt](https://github.com/zigpy/zha-device-handlers/files/7921232/TS0505A__TZ3000_dbou1ap4_ff23a460_scan_results.txt)

I am going to try to override the quirk with a dummy quirk and see what happens.

mdeweerd, 2022-01-23:
I added and changed the quirk locally, removing the reference to the LicldRGBCCTColorCluster:
```python
                    Color.cluster_id,  # LidlRGBCCTColorCluster,
```

The replacement is in itself successful, but there is no change in the UI behavior:
[image]

It seems that the quirk is not related to the UI behavior.

MattWestb, 2022-01-23:
The interesting thing is what the device is sending then you is doing changes.
If reporting is set to one second and you is power it off  and it was 100% and have 5 seconds transition time then the device shall  sending 5 reports then the light is dimming down from 100% to zero and off.
If you is having longer reporting  time / threshold you is getting different behaviour in HA

If i remember right shall level and color not being changed then the light is in off state only then its on it shall being possible doing it but with  transition time its the reporting "delayed".

IKEA is (/was i have not testing the new updated firmware so much) is its possible changing level and color then the device is in off state = not Zigbee standard.

MattWestb, 2022-01-23:
What is being sent and received from the device then you is doing the on/off and have changing the color and light level ?
And what is the reporting config in the device for light level and color cluster ?

mdeweerd, 2022-01-23:
Well it looks like CMD 1 (On) is sent to the OnOff cluster and then the device reports that it's off... .

[LIDL.zip](https://github.com/zigpy/zha-device-handlers/files/7921454/LIDL.zip)

I did several Color Temperature settings which are more or less ok (the center values are warm, the extreme right is rather white again).

I did several level settings (in the UI), followed by On commands because the lamp goes off when changing the level.

The light reports "Off" when its on - strange!!!

fullLIDL..log (filtered capture log with regards to LIDL device):
- Frame 90: - Move to level 36 with OnOff (Sequence 201)
- Frame 109: - Success for command (Sequence 201)
- Frame 117:  - Reports Off
- Frame 652: - Command On (Sequence 213)
- Frame 666 - Response to command On: success (Sequence 213)
- Frame 674 - Reports Off

MattWestb, 2022-01-23:
I was scanning the attributes on of my TS0505Aby _TZ3000_riwp3k79 and they is very stranges then looking how they have making the cluster but they is on of the fue tuya devices (nearly all LIDL is) Zigbee certified.

I think it better comparing how some more "normal" light is doing the same for understanding how there is reacting on commands but with devices that is doing reporting  (= no Philips HUE devices).

But i have never seen one device reporting status off after getting and reporting OK on command ON (i have not sniffing all My LIDL devices but perhaps is needed).

Is you is sending one ON the device is reporting off but if you waiting little longer is it reporting ON then the reporting timer is running out ?

mdeweerd, 2022-01-23:
I have now sent sent several On commands in a row using zha-toolkit:

```yaml
service: zha_toolkit.execute
data:
  command: zcl_cmd
  event_done: zha_done
  cluster: 6
  cmd: 1
  ieee: light.tz3000_dbou1ap4_ts0505a_level_light_color_on_off
```

.  The light stays on, but the on status resport switches from on to off to on ...

[LIDL_sendOn_is_Toggle.zip](https://github.com/zigpy/zha-device-handlers/files/7921651/LIDL_sendOn_is_Toggle.zip)

I have a sniffer setup that sniffs 24H/24H all year long.  

That's going to be though to fix in a quirk?  Or should I bring the light back to LIDL indicating that it does not follow the Zigbee specification...

mdeweerd, 2022-01-24:
I did a few more tests.

In order:
1. I power cycled the light - the ZB state is "on" and the light is on.
2. Manual off/on - The light switched off and on.
3. Set level: ZB state is off, light is on:

```json
                            "attribute_id": "0x0000",
                            "attribute_name": "on_off",
                            "value_type": [
                                "Bool",
                                "Discrete"
                            ],
                            "access": "REPORT|READ",
                            "access_acl": 5,
                            "attribute_value": 0
                        },
                        "0x0000": {
                            "attribute_id": "0x0000",
                            "attribute_name": "current_level",
                            "value_type": [
                                "uint8_t",
                                "Analog"
                            ],
                            "access": "REPORT|READ",
                            "access_acl": 5,
                            "attribute_value": 117
                        },
```

4. Switch light "on" in UI: no change in the ZB values (still "off" and level 117).
5. Switch light "off" in UI: ZB values unchanged: still "off" and level 117.

So that's not helpful to know if the light is on or off - even when reading back the attributes.

I notice that cluster 8 has the attribute 0x000F which is not recognized (no name).  That attribute appear in revision 8 and requires that the cluster revision id is 2, but it was left to 1.
Because the command to change the level is 'with on/off" I believe that these bits should not influence the behavior (in theory).  
```jaon
                "0x0008": {
                    "cluster_id": "0x0008",
                    "name": "level",
                    "attributes": {
                        "0x0000": {
                            "attribute_id": "0x0000",
                            "attribute_name": "current_level",
                            "value_type": [
                                "uint8_t",
                                "Analog"
                            ],
                            "access": "REPORT|READ",
                            "access_acl": 5,
                            "attribute_value": 117
                        },
                        "0x000f": {
                            "attribute_id": "0x000f",
                            "attribute_name": "15",
                            "value_type": [
                                "bitmap8",
                                "Discrete"
                            ],
                            "access": "REPORT|WRITE|READ",
                            "access_acl": 7,
                            "attribute_value": 0
                        },
                        "0xfffd": {
                            "attribute_id": "0xfffd",
                            "attribute_name": "65533",
                            "value_type": [
                                "uint16_t",
                                "Analog"
                            ],
                            "access": "REPORT|READ",
                            "access_acl": 5,
                            "attribute_value": 1
                        }
                    },
                    "commands_received": {},
                    "commands_generated": {}
                },
```

The following "Move to level with on/off" turned my light on from the off state (which is expected) - the ZB state was "on" :
````yaml
service: zha_toolkit.execute
data:
  command: zcl_cmd
  event_done: zha_done
  cluster: 8
  cmd: 4 
  args: [ 100, 1 ]
  ieee: light.tz3000_dbou1ap4_ts0505a_level_light_color_on_off
```

Changing the level to 177 was successful as well and the light remained on - and ZB also reported the state as on.

However, command 0 - a simple Move to level also turned the light on from the Off state.  If the Options bit 0 set to 0 would be effective, then the command should not be executed.  However, while the light was switched on, the ZB attribute indicated it was off.
This was the service called to change the level (which changed from 177 to 100 due to that call):
```yaml
service: zha_toolkit.execute
data:
  command: zcl_cmd
  event_done: zha_done
  cluster: 8
  cmd: 0
  args: [ 100, 1 ]
  ieee: light.tz3000_dbou1ap4_ts0505a_level_light_color_on_off
```

Writing the OnOff attribute is not possible: it's read only and the light effectively refuses the write with the next reply:
```raw
WriteAttributesStatusRecord(status=<Status.READ_ONLY: 136>, attrid=0)
```

It almost seems best to ignore the on/off attribute state for this light but that's not the best solution.

MattWestb, 2022-01-24:
The OnOff attribute is RO so the response is OK then its only for reporting and for changing state is the command on the cluster that is being used.

You can trying "reconfiguring device" from the device card but i think its not helping then its looks very bad made firmware in the device.

Still all is tuya devices with lot of surprise :-)))

mdeweerd, 2022-01-24:
I only tried to write to the attribute in an attempt to find something suitable for a quirk 😒 .

IMHO device configuration only sets up binding, reporting and may be some initial states.
Maybe a factory reset would help and then I probably have to pair the device again.

mdeweerd, 2022-01-24:
I sent a Factory Reset which was accepted - the light started to blink to indicate its trying to join a network, which it did.  HA reported it.
Then I tested the Light level setting in the UI.

It didn't fix the issue 😞 .

Command used for the factory reset:
```yaml
service: zha_toolkit.execute
data:
  command: zcl_cmd
  event_done: zha_done
  cluster: 0
  cmd: 0
  ieee: light.tz3000_dbou1ap4_ts0505a_level_light_color_on_off
```

MattWestb, 2022-01-24:
Im very sorry but i from my side factory reset was expected not helping getting the device working as is shall so it 110% one firmware bug :-((

One question how is the data for sending leave with rejoin flag with ýour magic ZHA_Toolkit ?

I need ist for getting the problematic TS004F being in init mode si can changing its working mode so its populating 3 more endpoints in scene mode (most device have one INIT mode then pairing anc can changing its "hardware setting".(= EP nd cluster) that is not possible making after its initiated (ther for ZHA cant reading "new EPs" then it shall normally not being changed).

By the way have you one tuya ZBGW around and can pairing the device with it ?
LIDL 3 AC plugs with  4 usb is working OK without quirk but is getting extra attribute if applying the quirk but tuya have making one new firmware that controlling all plugs for EP 1 but with tuya ZBGW they is doing some magick (perhaps scanning the attribute can helping finding the right one).

mdeweerd, 2022-01-24:
- I didn't put much hope in the factory reset, but in a lot of cases that what support will ask for or do.
- zha_toolkit for leave and rejoin:

```yaml
service: zha_toolkit.execute
data:
  command: rejoin
  event_done: zha_done
  # optional: device to use for the join.
  # command_data: IEEEaddr
  ieee: IEEEaddr or entityname.
```
That will open the network for joining during 1 minute and send a  Mgmt_Leave_req (0x0034) to the device.

- You may want to try "handle_join" first :
```yaml
service: zha_toolkit.execute
data:
  command: handle_join
  # Optional: short network address (if not provided, uses value in ZHA);
  # command_data: 0x4524
  event_done: zha_done
  ieee: IEEEaddr or entityname.
```
- I have a LIDL gateway that I didn't hack yet + I can't spend time on that.  You said that you can sniff, so why not do that and look what is happening?

MattWestb, 2022-01-24:
I trying the leave with rejoin if its working with  the TS004F !!
The system is not needing being open for joining then the device is having all network credential so its technically still one member of the network only its doing on software reset and coming back and is being in "INIT" mode for some seconds (or lesser).

I was one that  was hacking the tuya ZBGW but  Paul was doing the linuy part and i was doing the Zigbee part and getting updated firmware from Gary.
I have one tuya and one LIDL but both is hacked and one running one test network and the other i is using as one network connected sniffer.

Was thinking redoing ne and letting tuya upgrading it but first i must putting one old firmware on the Zigbee module so tuya can using it (the original is using one older EZSP that  have different serial protocol so its not working with old host application).

I reporting back how rejoining is working !!!

And great thanks for your work !!!

MattWestb, 2022-01-24:
Looks like rejoin is sending one leave command but not with rejoin flag.
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x03c8] Extending timeout for 0xf3 request
Device 0x03c8 (60:a4:23:ff:fe:fd:dc:14) left the network
Device 0x03c8 (60:a4:23:ff:fe:fd:dc:14) left the network
```
Then having the network open for joining.
I shall making one sniff so i can see with ZHA was sending.

MattWestb, 2022-01-24:
Ops something it was doing with the device the now if i changing too seen mode i is getting commands from the new endpoints !!!
```
Ignoring message (b'0121fd00') on cluster 6: unknown endpoint or cluster id: 2
Ignoring message (b'0123fd00') on cluster 6: unknown endpoint or cluster id: 3
Ignoring message (b'0122fd00') on cluster 6: unknown endpoint or cluster id: 4
```
Still need doing the sniffing !!!

MattWestb, 2022-01-24:
Sniff of the "rejoining command":
```
ZigBee Encapsulation Protocol, Channel: 16, Length: 59
IEEE 802.15.4 Data, Dst: 0x4b2d, Src: 0x0000
ZigBee Network Layer Data, Dst: 0x5ce9, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
ZigBee Device Profile, Leave Request, Device: SiliconL_ff:fe:fd:dc:14
    Sequence Number: 106
    Extended Address: SiliconL_ff:fe:fd:dc:14 (60:a4:23:ff:fe:fd:dc:14)
    .0.. .... = Remove Children: False
    0... .... = Rejoin: False
```
The rejoin flagg is **not** set and ZHA is opening the network that shall not being needed then the device is already in the network.
```
ZigBee Encapsulation Protocol, Channel: 16, Length: 48
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x0000
ZigBee Network Layer Data, Dst: Broadcast, Src: 0x0000
    Frame Control Field: 0x0208, Frame Type: Data, Discover Route: Suppress, Security Data
    Destination: 0xfffc
    Source: 0x0000
    Radius: 30
    Sequence Number: 140
    [Extended Source: SiliconL_ff:fe:0f:54:89 (84:2e:14:ff:fe:0f:54:89)]
    [Origin: 27]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
    Frame Control Field: Data (0x08)
    Destination Endpoint: 0
    Permit Join Request (Cluster ID: 0x0036)
    Profile: ZigBee Device Profile (0x0000)
    Source Endpoint: 0
    Counter: 179
ZigBee Device Profile, Permit Join Request
    Sequence Number: 104
    Duration: 60
    Significance: 0
```

mdeweerd, 2022-01-24:
We should not be transforming this issue in a zha-toolkit discussion.  The Leave request ~~is~~ was like this: 

   `res = await src.zdo.request(0x0034, src.ieee, 0x01)` 

but apparently the 0x01 is not the parameter.

Replaced with:   
 `res = await src.zdo.leave(remove_children=False, rejoin=True)`

Which results in the following sniffed result, where I notice that tshark is misrepresenting the bits!!!
```
   Command Frame: Leave
        Command Identifier: Leave (0x04)
        ..1. .... = Rejoin: True
        .0.. .... = Request: False
        0... .... = Remove Children: False
```

mdeweerd, 2022-01-24:
I implemented a workaround by ignoring Attribute 0 of cluster 6 - I adapted the original quirk and defined this local one.

The side-effects are gone.

```python
"""Quirk for LIDL RGB+CCT bulb."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class LidlOnOffCluster(CustomCluster, OnOff):
    """Lidl OnOff custom cluster."""
    def _update_attribute(self, attrid, value):
        if attrid != 0:
            super()._update_attribute(attrid, value)
        # else ignore update

class LidlRGBCCTColorCluster(CustomCluster, Color):
    """Lidl RGB+CCT Lighting custom cluster."""

    # Set correct capabilities to ct, xy, hs
    # LIDL bulbs do not correctly report this attribute (comes back as None in Home Assistant)
    _CONSTANT_ATTRIBUTES = {0x400A: 0b11001}

class RGBCCTLightLocal(CustomDevice):
    """Lidl RGB+CCT Lighting device."""

    signature = {
        MODELS_INFO: [("_TZ3000_dbou1ap4", "TS0505A")],
        ENDPOINTS: {
            1: {
                # <SimpleDescriptor endpoint=1 profile=269 device_type=268
                # device_version=1
                # input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096]
                # output_clusters=[10, 25]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # device_version=0
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    LidlOnOffCluster,
                    LevelControl.cluster_id,
                    LidlRGBCCTColorCluster,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

MattWestb, 2022-01-24:
Great !!
Is the device reporting off state then you is sending one off command to it or is it not doing it ?

Comment by our puddly about the leav with reloin flag https://github.com/zigpy/zigpy/issues/831#issuecomment-950373080.

mdeweerd, 2022-01-24:
Hi, I sent several "Off" commands in a row, it seems to report Off when the device is off.  So On = On and Off = Maybe On.

I've used the function mentioned by puddly in the new version, but that is not an explication why the original method did not work.

Here is a history of commands sent and reports received:
```raw
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: On (0x01)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
    Command: On (0x01)
        On/off Control: Off (0x00)
        On/off Control: On (0x01)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
    Command: On (0x01)
        On/off Control: Off (0x00)
        On/off Control: On (0x01)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
        On/off Control: Off (0x00)
    Command: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: On (0x01)
        On/off Control: On (0x01)
        On/off Control: On (0x01)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
        On/off Control: On (0x01)
        On/off Control: On (0x01)
    Command: On (0x01)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: On (0x01)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
        On/off Control: On (0x01)
        On/off Control: On (0x01)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
    Command: On (0x01)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
        On/off Control: On (0x01)
        On/off Control: On (0x01)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
        On/off Control: Off (0x00)
        On/off Control: Off (0x00)
    Command: Move to Level with OnOff (0x04)
        On/off Control: Off (0x00)
    Command: Move to Color Temperature (0x0a)
```

jgclark, 2022-02-14:
I've now found your forum post about this, as I'm finding the same problem.
This is the first time I've needed to use a quirk, and I can't find good instructions on how to add it, as what you've pasted above doesn't look like standard YAML.
So far I've added this to `configuration.yaml`:
```
enable_quirks: true
custom_quirks_path: quirks
```
What do I do next?
Thanks.

mdeweerd, 2022-02-15:
The quirk configuration must be under zha.  I have this in my `configuration.yaml` (you need only the first three lines):

```yaml
zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks
  zigpy_config:
    network:
      channel: 11
      channels: [11]
    ota:
      ikea_provider: false
      ledvance_provider: false
      otau_directory: /config/zb_ota
```

As you see I have chosen an absolute quirks path (/config/zha_quirks), I did not check how this works with relative paths.

You have chosen "quirks", so you need to create the directory "quirks" and add the [code I posted above](https://github.com/zigpy/zha-device-handlers/issues/1303#issuecomment-1020371189) in there.

Then restart HA and check that the quirk is loaded for your device by checking the quirk path in the device details.
I have the following because the file I placed in my quirks path is called rgbcct and the class is called RGBCCTLightLocal.
[image]

I have:

jgclark, 2022-02-15:
Many thanks, @mdeweerd. 
The 'zha:' wasn't obvious (to me). I've also guessed that the quirk file needs to have no extension, and it seems to have rebooted OK. 
And it works 🥳 !

mdeweerd, 2022-02-15:
Actually, the quirks file needs to have the '.py' extension - I just corrected it in my comment above.

I am happy that it works for you, and despite the malfunction, also happy that I do not have the only light behaving like this.

pojlFDlxCOvZ4Kg8y1l4, 2022-02-17:
I've enabled the custom quirk, and it's mentioned in the device details.

-  Quirk: zhaquirks.lidl.cct.CCTLight 

However, it did not change a lot for me. I'll try to rejoin the device, maybe that changes things.

pojlFDlxCOvZ4Kg8y1l4, 2022-02-17:
Reconfiguring works like a charm, but changing brightness still turns it off and on a few times.
Changing the color temperature still is weird, only the middle does warm-ish temps. It still makes the pop up do weird stuff like in the GIFs.
The light being on does not necessarily reflect the on status in HA, but off always does reflect properly in HA.
[image]

pojlFDlxCOvZ4Kg8y1l4, 2022-02-17:
ahh okay, got confused there then.
It is indeed a different zigbee device, the _TZ3000_8uaoilu9.

I figured I'd just throw in the info I could gather from the virtual and physical device, maybe it could be handy to figure out any FW updates:

Device info
TS0502A
by _TZ3000_8uaoilu9
Connected via Texas Instruments CC2531, Z-Stack Home 1.2 (build 20190608)
Zigbee info
IEEE: cc:86:ec:ff:fe:13:57:6b
Nwk: 0x7b5a
Device Type: Router
LQI: 89
RSSI: Unknown
Last Seen: 2022-02-17T18:51:32
Power Source: Mains
Quirk: zhaquirks.lidl.cct.CCTLight

LIDL LIVARNO LED FLOOR LAMP
IAN 368542_2010
Typ 14153905L
man. date 05/21
charge nr bn300965
power adapter model MY-C012-A3 12v1A constant voltage, 9z23

pojlFDlxCOvZ4Kg8y1l4, 2022-02-17:
Hmm, I'm doing something wrong, I can't get it to load the quirk.
It still shows

-  Quirk: zhaquirks.lidl.cct.CCTLight 

What am I missing?
Ctrl+f seems to be finding the correct stuff, correct links and modelnames etc. 

**TS0502A 
_TZ3000_8uaoilu9** 

/config/configuration.yaml
```
zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks
  zigpy_config:
    network:
      channel: 11
      channels: [11]
    ota:
      ikea_provider: false
      ledvance_provider: false
      otau_directory: /config/zb_ota
```

/config/zha_quirks/TS0502A.py

```
"""Quirk for LIDL RGB+CCT bulb."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class LidlOnOffCluster(CustomCluster, OnOff):
    """Lidl OnOff custom cluster."""
    def _update_attribute(self, attrid, value):
        if attrid != 0:
            super()._update_attribute(attrid, value)
        # else ignore update

class LidlRGBCCTColorCluster(CustomCluster, Color):
    """Lidl RGB+CCT Lighting custom cluster."""

    # Set correct capabilities to ct, xy, hs
    # LIDL bulbs do not correctly report this attribute (comes back as None in Home Assistant)
    _CONSTANT_ATTRIBUTES = {0x400A: 0b11001}

class RGBCCTLightLocal(CustomDevice):
    """Lidl RGB+CCT Lighting device."""

    signature = {
        MODELS_INFO: [("_TZ3000_8uaoilu9", "TS0502A")],
        ENDPOINTS: {
            1: {
                # <SimpleDescriptor endpoint=1 profile=269 device_type=268
                # device_version=1
                # input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096]
                # output_clusters=[10, 25]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # device_version=0
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    LidlOnOffCluster,
                    LevelControl.cluster_id,
                    LidlRGBCCTColorCluster,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

Or would it just not work with my specific model, are they too different from each other?
Any ideas?

mdeweerd, 2022-02-17:
I do not think there was an issue with your quirk, maybe you did not restart HA.

Here is an updated version that only ignores changes to "off".

[TS0502A.txt](https://github.com/zigpy/zha-device-handlers/files/8092912/TS0502A.txt)

Is there a __pycache__ directory that has been created in your zha_quirks directory?

pojlFDlxCOvZ4Kg8y1l4, 2022-02-17:
> I do not think there was an issue with your quirk, maybe you did not restart HA.  

Uptime sensor tells me that the last "reboot" was ~ 5hrs ago, which is in line with my last comment from 5 hrs ago.
However, the last boot sensor tells me it was 2 am.  
So maybe something doesn't get rebooted "enough"? 
I'm running HA OS, I'll just reboot the whole VM to be sure.

There is no pycache directory in the zha_quirks directory.

I'll be back in a few minutes and edit this comment for results.

mdeweerd, 2022-02-18:
Other things that you can try (besides checking your configuration):
- Place the original quirk as is in your zha_quirks directory:
  https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/lidl/cct.py
  (To limit doubts about the syntax/etc of the file).
- Add an empty '__init__.py' to your directory (I do not have one, but it might help).

pojlFDlxCOvZ4Kg8y1l4, 2022-02-18:
I've connected through SMB with HA OS, and it shows me a pycache folder. 
It does not show that folder in the file editor, but it the same file editor does show dotfiles. 
I assumed it shows everything including the pycache folder because it shows the dotfiles.
Anyways, there were multiple files in the pycache folder.

- I've deleted the contents and folders in the zha_quirks folder, including pycache.
- added your linked file by using "save as" and pulling it into the folder. 
- changed the name of the saved file to **githubcct.py**.
- I also added an empty init.py with right click > new text file and making sure Windows does not hide the actual extension. 

```
PS Microsoft.PowerShell.Core\FileSystem::\\homeassistant1\config\zha_quirks> dir -r
    Directory: \\homeassistant1\config\zha_quirks
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----         2/18/2022   2:46 PM                __pycache__
-a----         2/18/2022   2:44 PM           3333 githubcct.py
-a----         2/18/2022   2:46 PM              0 init.py
    Directory: \\homeassistant1\config\zha_quirks\__pycache__
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
------         2/18/2022   2:46 PM           1852 githubcct.cpython-39.pyc
------         2/18/2022   2:46 PM            117 init.cpython-39.pyc
PS Microsoft.PowerShell.Core\FileSystem::\\homeassistant1\config\zha_quirks>
```  

Now the light is using **Quirk: githubcct.CCTLight**. 

I'll fiddle a bit with it, it does load the config from the zha_quirks folder so shouldnt be that much of an issue to get it running with your fix.

mdeweerd, 2022-02-18:
Excellent - i should have written '__init__.py' , but apparently that's not needed for you neither.

I wonder why it did not work before.

pojlFDlxCOvZ4Kg8y1l4, 2022-02-18:
Oh, hah, good to know that init.py is not necessary! 

https://github.com/zigpy/zha-device-handlers/issues/1303#issuecomment-1020371189  

I've copied this into a new file, and copy pasted ("_TZ3000_8uaoilu9", "TS0502A") from the cct.py file into yours, replacing your ("_TZ3000_dbou1ap4", "TS0505A") line. 
I deleted the **githubcct.py** file to make sure it does not load that one.

It loads the default quirk again,  **Quirk: zhaquirks.lidl.cct.CCTLight** 
Don't know why it does not like your file. There's no info in the logs regarding it. 
Very strange.
I've compared both files through diff, but I do not see something obvious that would make HA not like it. 

Both files: 
[22020218 TS502A pojlFDlxCOvZ4Kg8y1l4.zip](https://github.com/zigpy/zha-device-handlers/files/8097812/22020218.TS502A.pojlFDlxCOvZ4Kg8y1l4.zip)

I've added following debug settings to HA:

```
logger:
  default: info
  logs:
#    homeassistant.core: debug
    homeassistant.components.zha: debug
#    bellows.zigbee.application: debug
#    bellows.ezsp: debug
#    zigpy: debug
#    zigpy_deconz.zigbee.application: debug
#    zigpy_deconz.api: debug
#    zigpy_xbee.zigbee.application: debug
#    zigpy_xbee.api: debug
#    zigpy_zigate: debug
#    zigpy_znp: debug
    zhaquirks: debug
```

mdeweerd, 2022-02-18:
Diffrences:
- Line endings:
```
$ file githubcct.py stick.py
githubcct.py: Python script, ASCII text executable
stick.py:     Python script, ASCII text executable, with CRLF line terminators
```

```diff
$ diff -w githubcct.py stick.py
1c1
< """Quirk for LIDL CCT bulb."""
---
> """Quirk for LIDL RGB+CCT bulb."""
28,29c28,33
< class LidlCCTColorCluster(CustomCluster, Color):
<     """Lidl CCT Lighting custom cluster."""
---
> class LidlOnOffCluster(CustomCluster, OnOff):
>     """Lidl OnOff custom cluster."""
>     def _update_attribute(self, attrid, value):
>         if attrid != 0:
>             super()._update_attribute(attrid, value)
>         # else ignore update
31c35,39
<     # Remove RGB color wheel for CCT Lighting: only expose color temperature
---
>
> class LidlRGBCCTColorCluster(CustomCluster, Color):
>     """Lidl RGB+CCT Lighting custom cluster."""
>
>     # Set correct capabilities to ct, xy, hs
33c41
<     _CONSTANT_ATTRIBUTES = {0x400A: 16}
---
>     _CONSTANT_ATTRIBUTES = {0x400A: 0b11001}
36,37c44,45
< class CCTLight(CustomDevice):
<     """Lidl CCT Lighting device."""
---
> class RGBCCTLightLocal(CustomDevice):
>     """Lidl RGB+CCT Lighting device."""
40,48c48
<         MODELS_INFO: [
<             ("_TZ3000_49qchf10", "TS0502A"),
<             ("_TZ3000_oborybow", "TS0502A"),
<             ("_TZ3000_9evm3otq", "TS0502A"),
<             ("_TZ3000_rylaozuc", "TS0502A"),
<             ("_TZ3000_el5kt5im", "TS0502A"),
<             ("_TZ3000_oh7jddmx", "TS0502A"),
<             ("_TZ3000_8uaoilu9", "TS0502A"),
<         ],
---
>         MODELS_INFO: [("_TZ3000_8uaoilu9", "TS0502A")],
51c51
<                 # <SimpleDescriptor endpoint=1 profile=260 device_type=268
---
>                 # <SimpleDescriptor endpoint=1 profile=269 device_type=268
56c56
<                 DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,
---
>                 DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
86c86
<                 DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,
---
>                 DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
92c92
<                     OnOff.cluster_id,
---
>                     LidlOnOffCluster,
94c94
<                     LidlCCTColorCluster,
---
>                     LidlRGBCCTColorCluster,
```

I do not think we need an extra comma `MODELS_INFO: [("_TZ3000_8uaoilu9", "TS0502A"),],` as there are quirks without it.

I think it's the signature:

-                 DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
vs
-                 DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,

These have to match the device.

pojlFDlxCOvZ4Kg8y1l4, 2022-02-18:
system logs:
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/8097975/home-assistant.log)

I saw some oddities, it said it loaded all the lights from the default cct.py and saw them seconds or minutes ago.
Maybe something is off there?
I have no clue on how to interpret this.

I think i've sanitized any remotely private data, lines with NMAP etc. 
Those are the gaps.

pojlFDlxCOvZ4Kg8y1l4, 2022-02-18:
> I think it's the signature:
> 
>     * ```
>                   DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
>       ```
> 
> 
> vs
> 
>     * ```
>                   DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,
>       ```
> 
> 
> These have to match the device.

indeed, you're right!

** Quirk: stick.RGBCCTLightLocal ** is now being loaded, stick.py with the different device type copied and pasted. The rest is as-is from your file. 

The light is now controllable without it doing the stuff in the GIFs.
Brightness works well, from what i've seen just now.
Color temperature only gets warm-ish in the middle of the slider, left and right from the middle is cold white again. 

While it's still not completely fixed, it made my light usable. 
Very nice, thank you.

pojlFDlxCOvZ4Kg8y1l4, 2022-02-18:
Okay, how would we do that?

I thought i would be smart and just remove all RGB references and make it all into CCT, and while it works, it did not change a thing lol.

I'll post it anyways, but can't really get the formatting right on mobile:

"""Quirk for LIDL CCT bulb."""
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class LidlOnOffCluster(CustomCluster, OnOff):
    """Lidl OnOff custom cluster."""
    def _update_attribute(self, attrid, value):
        if attrid != 0:
            super()._update_attribute(attrid, value)
        # else ignore update

class LidlCCTColorCluster(CustomCluster, Color):
    """Lidl CCT Lighting custom cluster."""

    # Set correct capabilities to ct, xy, hs
    # LIDL bulbs do not correctly report this attribute (comes back as None in Home Assistant)
    _CONSTANT_ATTRIBUTES = {0x400A: 0b11001}

class CCTLightLocal(CustomDevice):
    """Lidl CCT Lighting device."""

    signature = {
        MODELS_INFO: [("_TZ3000_8uaoilu9", "TS0502A")],
        ENDPOINTS: {
            1: {
                # <SimpleDescriptor endpoint=1 profile=269 device_type=268
                # device_version=1
                # input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096]
                # output_clusters=[10, 25]
                PROFILE_ID: zha.PROFILE_ID,
                #DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # device_version=0
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    LidlOnOffCluster,
                    LevelControl.cluster_id,
                    LidlCCTColorCluster,
                    LightLink.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

mdeweerd, 2022-02-18:
Well, for starters the device type.  The original is                 
                DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,
The replacement is
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,

Let's keep it the same in the replacement:
                DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,

I tried it too, but in my case the original is EXTENDED_COLOR_LIGHT, just COLOR_TEMPERATURE_LIGHT did not help for the color nor for the on/off notification itself.

pojlFDlxCOvZ4Kg8y1l4, 2022-02-18:
Okay, is it also possible to pull settings from zigbee2mqtt?  
I've got the zigbee stick I've ordered yesterday, and it was easier to just spin up zigbee2mqtt instead of getting it up and running with ZHA.  

The light works fine in Z2M.

It seems that the warmest color temperature is still not 3000K, but on the other hand, I've also had other lights that said they could do 3000K but it didn't look like it at all. 6500K, the coldest white setting, seems to be accurate. 
Both the brightness and temperature sliders seem to be working accurately.  

Anyways, it gets recognized as the following light:
[image]

https://www.zigbee2mqtt.io/devices/14153905L.html

mdeweerd, 2022-02-18:
My ligtht color was fine up to the beginning of January - it can be pretty warm.  I assumed something changed in ZHA/zigpy/quirk that made it go bad but at the same time this on-off behavior occurred.

I used and still use the component [Adaptive Lighting](/basnijholt/adaptive-lighting).  I got the correct colors up to about mid-january and then it went broke.  As the colors were also wrong, I thought it had to do with that component, but it does not.

I do not what zigbee2mqtt does.

mdeweerd, 2022-02-20:
It seems that I found another method to mostly fix the fw bug.

I was testing [zha-toolkit](/mdeweerd/zha-toolkit).binds_remove_all on this light to disable the OnOff attribute reporting completely.

After doing so, the binding table was empty, but it was still reporting the OnOff state when using the OnOff command, but not when switching the light on by changing the intensity.

Then I reinitialilsed the device, resulting in a non-empty binding table:
```json
{0: {'src': '60:a4:23:ff:fe:91:fc:9a', 'src_ep': 1, 'cluster_id': '0x0006', 'dst': {'addrmode': 3, 'dst_ieee': '00:12:4b:00:01:6a:41:0c', 'dst_ep': 1}}, 1: {'src': '60:a4:23:ff:fe:91:fc:9a', 'src_ep': 1, 'cluster_id': '0x0008', 'dst': {'addrmode': 3, 'dst_ieee': '00:12:4b:00:01:6a:41:0c', 'dst_ep': 1}}, 2: {'src': '60:a4:23:ff:fe:91:fc:9a', 'src_ep': 1, 'cluster_id': '0x0300', 'dst': {'addrmode': 3, 'dst_ieee': '00:12:4b:00:01:6a:41:0c', 'dst_ep': 1}}
```

It was dysfunctional again with regards to the on/off reporting.

I have no remote on my light, so I do not know if the light reports to the emitter of the command or to the coordinator.

Note: when reinitialising the device, I got the following error:
```
2022-02-20 12:15:04 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0x8800:1:0x1000]: 'async_configure' stage failed: not enough values to unpack (expected 3, got 2)
```

mdeweerd, 2022-02-20:
I suppose that:
```
[homeassistant.components.zha.core.channels.base] [0x8800:1:0x1000]: 'async_configure' stage failed: not enough values to unpack (expected 3, got 2)
```
has to do with:
```
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x40)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .1.. .... = Acknowledgement Request: True
        0... .... = Extended Header: False
    Destination Endpoint: 1
    Cluster: ZLL Commissioning (0x1000)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 3
ZigBee Cluster Library Frame, Command: Default Response, Seq: 149
    Frame Control Field: Profile-wide (0x08)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...0 .... = Disable Default Response: False
    Sequence Number: 149
    Command: Default Response (0x0b)
    Response to Command: 0x41
    Status: Unsupported Cluster (0xc3)
```

MattWestb, 2022-02-20:
I have looking in some tuya ZBGW sniffs and its looks like tuya devices is normally automatic reporting to the coordinator without binding and reporting being stetted up and some time binding / config reporting of SW version on basic cluster or battery on power config cluster and all is reporting OK to the coordinator.

mdeweerd, 2022-02-20:
Maybe an [acceptable solution](https://github.com/zigpy/zha-device-handlers/pull/1360#issuecomment-1041580252) would be to disable requests to bind to the coordinator and also remove bindings to the coordinator on setup.

But that will be for another volunteer.

MattWestb, 2022-02-20:
Also trying disabling extra config is possible in quirks but i have not trying it:
https://github.com/zigpy/zha-device-handlers/blob/f9e25f410da29eb56c9996ca0fcc43bab36c3ba6/zhaquirks/xiaomi/aqara/weather.py#L74-L75

mdeweerd, 2022-02-20:
I checked the LIDL remote that controls another LIDL light (no quirk).
The remote actually only sends On/Off commands and Step commands.

When sending a Step command the light stays off when it's off.
So HA with that remote, it's not possible to set the color light and turn the light on without HA knowing about it.

MattWestb, 2022-02-20:
By Zigbee light and occupancy shall light not doing anything if they is in off state and only reacting on commands then they is in on state (exception on/off/toggle commands and scene commands that can turning the device on) (IKEA device was changing color in off state = not standard) so its look being one OK behaving device.

By the way we like getting sniff from the LIDL remote for doing one quirk for it with all commands in the different modes.

mdeweerd, 2022-02-20:
There is a github issue:
https://github.com/zigpy/zha-device-handlers/issues/613

And a blueprint:
https://community.home-assistant.io/t/zha-lidl-livarnolux-tuya-silvercrest-4-button-remote/372186

And you'ld like sniffed packets?

MattWestb, 2022-02-20:
I like getting sniff from one working device with tuya ZBGW or with HA.
What i understanding its one "normal tuya Zigbee" device = its have 2 different working mode.
If of only pressing ones its can sending dim up and down and off in normal zigbee way.
If pressing the on one second time its start sending color changing commands but no one have posting one sniff so we doing knowing the commands sending and the attributes for changing / getting the mode it having for the moment.

closed 2022-08-26 by github-actions[bot]