zigpy/zha-device-handlers #1306: [BUG] New yooksmart D10110 inverted with quirk

aristeu, 2022-01-24:
**Describe the bug**
I purchased a new yooksmart D10110 cover and paired with home assistant. The controls
seemed inverted and I had to move the bar twice in order to get it to move. I read reports
in the past with the suggestion to unpair and pair again, tried multiple times with no luck.
So I disabled the quirk (apologies for the brute force: moved the file to a different directory
and reloaded) and it works now. For completeness:
Before:
- buttons up and down wouldn't work
- available button would be inverted (e.g.: cover was all the way down and the down button was enabled)
- in order to control the cover I'd move the progress bar all the way to 0 or to 100 then the opposite in order to work
After:
- buttons up and down work
- enabled button matches the direction of the cover: if open, it shows down button enabled

**To Reproduce**
Behavior is consistent across multiple pair/unpair cycles and full home assistant instance restarts

**Additional context**
Something that is possible, since the cover is new, is that they corrected the behavior in their firmware
and the quirk isn't needed anymore.
This device has:  Firmware: 0x10013001

I can provide any debugging necessary. I'm using homeassistant official virtual machine image and keeping
it up to date.

jonnylangefeld, 2022-02-07:
I'm about to buy these and was wondering: how did you disable the quirk? Can you write the steps for a beginner?

Also, are [these](https://www.amazon.com/Yoolax-Motorized-Free-Stop-Cordless-Rechargeable/dp/B07QZDXJBQ?th=1) the right kind? I'm confused about the naming `yooksmart D10110` vs the `Yoolax` on Amazon. But when following the amazon.com link on the [blakadder device database](https://zigbee.blakadder.com/Yookee_D10110.html) it leads to the Yoolax ones on Amazon. Given the `yoo` part is the same I assume that's the same motors?

aristeu, 2022-02-07:
That one should work as well, I picked [this](https://www.amazon.com/gp/product/B07QP5M7YC) one, but it seems
the only change is the style. Pick the P440-US motor that can connect directly through zigbee.

As noted on my initial comment, the way I did is ugly and will be reverted by a future upgrade. These are the
steps:
```
1) Get a shell in the system (I use the kvm image, so virsh console <virtual machine name>):
    homeassistant login: root
    Last login: Tue Jan 25 14:50:00 UTC 2022 on ttyS0
    Last login: Mon Feb  7 14:51:33 on ttyS0
    Welcome to Home Assistant OS.

    Use `ha` to access the Home Assistant CLI.
    # 
2) Move the file to a backup directory:
    # for i in $(find / -name D10110blinds.py); do dir=$(dirname $i); mkdir $dir/.backup; mv $i $dir/.backup/; done
3) Restart homeassistant
```
The correct way would be changing the quirk to not load on this firmware version but since I
just have this one, I can't confirm the ones that need the quirk have a different firmware version.

jonnylangefeld, 2022-02-08:
Awesome thank you so much for the detailed description! Since you recently bought and I’m about to buy I think I’d get the same firmware version where I have to circumvent this quirk.

JeremeLau, 2022-06-17:
I work for yooksmart which is known as the selling brand Yoolax. We have confirmed the issue and will publish new firmware. So you guys need no bother. By the way, the issue is located due to the script of Smartthings which can choose whether or not to invert the direction. we learned the wrong logic obviously.

jonnylangefeld, 2022-06-17:
Cool! How can we update the firmware?
Also, since you work for them: they frequently disconnect for me and reconnecting them always takes a long time of trial and error. Is there any trick to it?

HeroGregg, 2022-06-22:
I was having the same issue, and solved it in a similar fashion.

@JeremeLau, I presume you mean firmware inside the shade. Is that field upgradable? Is there a way to find out what firmware is in the shade? I'm about to purchase six more shades, but I'm concerned I'll end up with shades that behave differently from each other.

I'm not sure if it's relevant, but the shade I installed has the motor/charging port on the left side.

JeremeLau, 2022-06-23:
> Cool! How can we update the firmware? Also, since you work for them: they frequently disconnect for me and reconnecting them always takes a long time of trial and error. Is there any trick to it?

Sorry for the inconvenience. We noticed maybe the paring trigger action is too complex and not easy to make it right every time. We're reconsidering a new paring way, wish can reduce the frequency.
For the new firmware, since we add more interesting funs, it's still under test.

JeremeLau, 2022-06-23:
> I was having the same issue, and solved it in a similar fashion.
>
> @JeremeLau, I presume you mean firmware inside the shade. Is that field upgradable? Is there a way to find out what firmware is in [...]

Yes, indeed. The new firmware I mean is the motor firmware itself. Our motor does have the ability to be upgraded over the air or over the USB port. As I mentioned above, it's still under test. If you don't need them urgently, we suggest you can wait until it's ready. 
Of course, if you would like to test the beta firmware, we'll be appreciated and will guide you to do so.

JeremeLau, 2022-06-29:
> @JeremeLau I’m interested in testing the beta firmware! I have lots of these shades and am really interested in getting it right!

I'm so sorry, the BOSS rejected the application of spread of beta firmware before fully self-test. Pls, be patient. I'll handle this ASAP.

IngoS11, 2022-07-18:
Just FYI: I purchased eight shades. One as a trial about six weeks ago and then seven more. The seven arrived last week, and I mounted them over the weekend. They all have Firmware: 0x12013001 incl the first one that came a few weeks ago, and all suffer from the same issue.

@aristeu : Is the removal of the quirk a one-time thing before registering them, or will I have to remove it with every update to HA? Until 

I assume the firmware update would be the best option for me as my Conbee Stick is in the garage, and the blinds are mounted now. Re-registering would mean I have to take them all off again, would it not?

Until the firmware is available, I have created templates for each shade that correct the problem.

```
platform: template
covers:
  ingo_office_shades_status:
    device_class: blind
    friendly_name: "Ingo Office Shades Status"
    position_template: >
      {{100 - state_attr('cover.ingo_office_shades','current_position')|int(100)}}
    value_template: >
      {{ (100 - state_attr('cover.ingo_office_shades','current_position') | int(100)) > 95 }}
    set_cover_position:
      service: cover.set_cover_position
      data_template:
        entity_id: cover.ingo_office_shades
        postion: >
          {{ 100 - position }}
    open_cover:
      service: cover.open_cover
      data: {}
      target:
        entity_id: cover.ingo_office_shades
    close_cover:
      service: cover.close_cover
      data: {}
      target:
        entity_id: cover.ingo_office_shades
    stop_cover:
      service: cover.stop_cover
      data: {}
      target:
        entity_id: cover.ingo_office_shades
    icon_template: >
      {% if is_state('cover.ingo_office_shades', 'closed') %}
         mdi:blinds-open
      {% else %}
         mdi:blinds
      {% endif %}
```

aristeu, 2022-07-18:
> @aristeu : Is the removal of the quirk a one-time thing before registering them, or will I have to remove it with every update to HA?

Unfortunately an update can overwrite it. Unless we can pinpoint firmware version in order to update the quirk to only apply to
firmwares that need it, I think the only way would be adding an option to disable the quirk manually. At least it'd live through
updates.

jonnylangefeld, 2022-07-28:
I vote for `rm -r zhaquirks/yooksmart`. I just ordered another set of the [Graywind blinds on Amazon](https://a.co/d/iiD27N4) and they came with a new motor and a new firmware. They have the same device identifier `("yooksmart", "D10110")`. But again the motors report the correct percentage and no quirk is needed. I now think this quirk was added in error in the first place and actually more people need to remove the quirk than need the quirk.

My current workaround is to mount an empty `yooksmart` directory via my HA docker compose file like so
```
    volumes:
      - ./deps/zhaquirks/yooksmart:/usr/local/lib/python3.10/site-packages/zhaquirks/yooksmart
```
which seems like an unnecessary workaround.

I have a set of old Yooksmart Zigbee motors that have firmware `0x11013001` and a set of new Yooksmart Zigbee motors that have firmware `0x11206780`.

@JeremeLau are there any instructions how I can upgrade my old firmware motors to the new one?

MasonBergstrom, 2022-08-06:
> My current workaround is to mount an empty `yooksmart` directory via my HA docker compose file like so
> 
> ```
>     volumes:
>       - ./deps/zhaquirks/yooksmart:/usr/local/lib/python3.10/site-packages/zhaquirks/yooksmart
> ```

This workaround worked perfectly for me on my Graywind shades on firmware `0x11206780`. Thanks!

javicalle, 2022-08-11:
I'm curious about this kind of issues.
Can you please try to get the attribute values from the `WindowCovering` cluster:
* 0x0007 'config_status'
* 0x0017 'window_covering_mode'

Thanks in advance.

mverrilli, 2022-10-28:
> > @JeremeLau I’m interested in testing the beta firmware! I have lots of these shades and am really interested in getting it right!
> 
> I'm so sorry, the BOSS rejected the application of spread of beta firmware before fully self-test. Pls, be patient. I'll handle this ASAP.

Hi @JeremeLau, any luck with the new firmware?

rwalker777, 2023-01-05:
Really new to HA and Zigpy, but people above are mis-understanding this issue.  The problem comes from having the blinds in front roll or back roll.  That is what is "reversing" the values for some and not others.  The firmware update @JeremeLau was referring to hopefully fixed this.  Until that happens (if it ever does) an option should be available to reverse the values for blinds that are set to back roll (I have both and front roll appears normal).  Can we re-open this and add a reverse boolean?

ejpenney, 2023-01-05:
I have both front and back roll (six devices), and they are ALL inverted (Home Assistant incorrectly reports state) unless I completely remove the quirk from ZHA.  With the quirk, they all go up when HASS sends cover.close_cover, and vice versa.

Might be helpful if you shared the firmware versions you're running?  Perhaps I just have a lucky combination of firmware/configurations.

For myself, the two back rolling devices are using 0x12013001, and the front roll devices are all running 0x12006780

jonnylangefeld, 2023-01-05:
The matter of the fact is that there is no 'reverse reporting' on any of the firmwares (I shared mine above), which is why this quirk should be removed from this repository. That's also why my workaround described above is necessary until that happens.
The shades allow programming to reverse the direction of the blinds. The reason that's necessary is because you can mount the motors on the right side of the blinds or on the left side of the blinds. Imagine two windows side by side, one has the motor on the right side and one has the motor on the left side. If the motors would spin in the same direction, one shade would go up and one shade would go down. That's why the shades allow re-programming of the direction of the motor. That way you can change the direction of one of the motors and then both sides would go up or go down at the same time.
I scanned the manuals of both firmwares:
* [new `0x11206780` (see page 35 in the PDF (33 in page numbers))](https://drive.google.com/file/d/1f3wGewWysNQAxyMxnKKwUIDnYnmr2kbf/view?usp=sharing)
* [old `0x11013001` (see page 16 in the PDF (15 in page numbers))](https://drive.google.com/file/d/1bOf1sGl63iDOqo-OUnRNOVo1f5qvrKS4/view?usp=sharing)

If the motors are correctly programmed so that the 'up' button on the remote means that the shades go up, then Home Assistant reports the right percentages (0% for closed and 100% for opened).

Hence again my plaidoyer to remove this quirk from ZHA.

jonnylangefeld, 2023-01-05:
> I'm curious about this kind of issues. Can you please try to get the attribute values from the `WindowCovering` cluster:
> 
> * 0x0007 'config_status'
> * 0x0017 'window_covering_mode'
> 
> Thanks in advance.

@javicalle `0x0017` doesn't exist and `0x0007` is labeled as `power_source`.

Here all the clusters:
[image]

and (most because it didn't fit the screenshot) attributes:
[image]

MattWestb, 2023-01-05:
>I'm curious about this kind of issues. Can you please try to get the attribute values from the WindowCovering cluster:

You cant finding it on basic luster (from your screen shots) must selecting window covering 0x0102 and then selecting the attribute 0x0007 and 0x017.

ejpenney, 2023-01-05:
> My current workaround is to mount an empty `yooksmart` directory via my HA docker compose file like so
> 
> ```
>     volumes:
>       - ./deps/zhaquirks/yooksmart:/usr/local/lib/python3.10/site-packages/zhaquirks/yooksmart
> ```

Hi @jonnylangefeld, I have a question about your WA:  I'm running HASS OS, and don't understand where I would make this change.  Please forgive my Docker-related ignorance, but I'm coming to the conclusion that in my particular setup I don't have a Docker compose file, is that correct?

For (other people's) reference, I'm currently I'm doing this:

I have this automation:
```
alias: "Fix: ZHAQuirks/Yooksmart"
description: >-
  Yooksmart ZHAQuirk breaks things with the shades.  Kill it at every opportunity.
trigger:
  - platform: homeassistant
    event: start
  - platform: homeassistant
    event: shutdown
condition: []
action:
  - service: shell_command.rm_yooksmart_quirk
    data: {}
mode: single
```
and in config.yaml:
```
shell_command:
  rm_yooksmart_quirk: if [ -d /usr/local/lib/python3.10/site-packages/zhaquirks/yooksmart ]; then rm -rf /usr/local/lib/python3.10/site-packages/zhaquirks/yooksmart; fi
```

The downside is that this means I need to restart twice every time I do a HASS Core update, once for the update, and again to reload with the quirk removed.

TheJulianJES, 2023-01-05:
You could also just make a custom quirk (those take priority over existing quirks) that just matches the cover and then does nothing. Seems easier than to delete files from libraries

javicalle, 2023-01-05:
> Here all the clusters: [image]
> 

The last one is the `WindowCovering` cluster. 

But it will be easier if you can share the diagnostic information from your devices.

ejpenney, 2023-01-06:
> The last one is the `WindowCovering` cluster.
> 
> But it will be easier if you can share the diagnostic information from your devices.

Let's see if I can help at all...

Firmwares 0x12013001 [(diagnostics here)](https://github.com/zigpy/zha-device-handlers/files/10356387/zha-d0458004cbd34b35a5fd52b770d39592-yooksmart.D10110-7bc5085e99110d700cac680822f0e1a8.json.txt) and 0x12006780 [(diagnostics here)](https://github.com/zigpy/zha-device-handlers/files/10356388/zha-d0458004cbd34b35a5fd52b770d39592-yooksmart.D10110-bb83386a9027576db49ee29122d51c08.json.txt) both provide the same values:
```
config_status (id: 0x0007):
     ConfigStatus.Closed_loop_lift_control|Online|Operational
window_covering_mode (id: 0x0017):
     WindowCoveringMode.<bitmap8.0: 0>
```

Additionally I've attached diagnostics data, this is using a custom quirk that does nothing.

rwalker777, 2023-01-06:
> The matter of the fact is that there is no 'reverse reporting' on any of the firmwares (I shared mine above), which is why this quirk should be removed from this repository. That's also why my [...]

Not sure what you are not understanding, but I'll try one more time.  I have multiple motors all on the same firmware, front roll shows normal, back roll shows reversed.  The current firmware are not reversing the values to the Zigbee controls.  My remotes work fine, but Zigbee controls are backwards (would be the same if you had left mounted front roll vs right roll).  The quirk needs to be changed to by default do nothing, but with a boolean flag to reverse the controls per blind.

Thanks,
Roy

MattWestb, 2023-01-06:
@javicalle Is it possible making one custom attribute (0x0800 ?) on the `WindowCovering` and making it bol and if its 1 the quirk is using the invert and if its 0 its dont ? 
https://github.com/zigpy/zha-device-handlers/blob/22968954cacaff4fc2b00e02ca1bd5622f637393/zhaquirks/yooksmart/D10110blinds.py#L32-L38
If it possible it shall  being possible inverting some of the blinds and other not for getting all working in the same direction if being mounted right or left.

ejpenney, 2023-01-06:
> Not sure what you are not understanding, but I'll try one more time.  I have multiple motors all on the same firmware, front roll shows normal, back roll shows reversed.  The current firmware are [...]

Personally I'm confused that your experience is so different from my own. I have two old firmware devices, both back roll, one is right and the other is left mounted. With the quirk they are both **incorrect** in Home Assistant. Then four new firmware devices, all front rolling, one is left mounted, the other three right mounted. **All** behave incorrectly in Home Assistant with the quirk.

It sounds like my experience is not unique, but yours is under represented in this thread, so I think folks would like to understand why your experience is different . Is it firmware? Device configuration?

I do believe others' assertions that the quirk is sometimes useful, I think the goal is to programmatically determine when it **is** useful...

javicalle, 2023-01-06:
Might be worth reviewing and comparing the value of the attributes:
* 0x0007: config_status
* 0x0017: window_covering_mode

Ref: https://github.com/zigpy/zigpy/blob/a2e5910aab25aceca855e6df0847df4487022818/zigpy/zcl/clusters/closures.py#L596

rwalker777, 2023-01-08:
All 4 of mine are the same:
ConfigStatus.Closed_loop_lift_control|Online|Operational
WindowCoveringMode.<bitmap8.0: 0>

rwalker777, 2023-01-10:
I added comment to the PR, quirk should be removed.  Need to put a note somewhere to users that if their blind is reversed to reset it and set up/down limits and direction again.  I didn't reset zigbee, just the motor settings.

TheJulianJES, 2023-03-04:
@rwalker777
Can you describe exactly what needs to be done to solve the issue with inverted blinds? (without the quirk)

From your PR (with the current "inverting quirk"):
> I was playing with this and reset my blind that was reporting normally... now it is reporting reversed. So all mine are now reversed

So just resetting and re-pairing the device fixed the issue? (Did you delete the device from ZHA before resetting it?)

Also, can you check the attributes again? (now that it's working "properly"):
- `0x0007`: config_status
- `0x0017`: window_covering_mode

rwalker777, 2023-03-04:
Deleted from zha, unpaired the blind. Repaired. That's the problem, the
attributes are the same.

Roy

TheJulianJES, 2023-03-04:
Thanks. You didn't install any firmware updates or anything, right?

rwalker777, 2023-03-04:
No, is there a firmware update available?!

TheJulianJES, 2023-03-04:
I don't know. Was just trying to see if anything changed between the first time you paired the device and now.
So in your opinion, the quirk should be removed and anyone who has an inverted cover should then re-pair the device, right?

rwalker777, 2023-03-04:
Yes.

TheJulianJES, 2023-03-11:
The "inverting quirk" should be removed with HA Core 2023.4.0 (https://github.com/zigpy/zha-device-handlers/pull/1664#issuecomment-1464998229), so you can stop modifying your HA installation or remove any custom quirks when that version is released.

closed 2023-03-11 by TheJulianJES

emayo, 2023-11-24:
> I work for yooksmart which is known as the selling brand Yoolax. We have confirmed the issue and will publish new firmware. So you guys need no bother. By the way, the issue is located due to the [...]

Can you tell me how to make mine work?  I was able to pair my motor through a Sonoff ZBMINI that, in turn, is paired to a Sonoff USB dongle based on the TI CC2652P + CP2102N chip.  There is only about 12 feet distance between the blind motor and the ZBMINI.  After pairing, a LQI of 77 was reported which seems low for only 12 feet.  I get the following error when I try to send commands.

{"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 10348 - 1 - 68 - 258 - 11 after 10000ms)'

How do I update firmware of the motor?   Can this be done over the exposed USB port?

CarlosUrbina, 2024-01-17:
If you are still having problems with an inverted Yooksmart D10110 as of 2024, please know this thread may be a red herring. I had this issue on first setup for a Zebra (the kind, not the brand) Graywind shade. My problem was that I reset the motor, and on setting the up and down limits again, I mistakenly inverted the directions. For my particular shade, the manual states that to invert the direction you press the following sequence (you need to use the provided remote control):

- Press the P button on the controller, the shade jogs up and down once
- Press the up arrow on the controller, the shade jogs up and down once
- Press the down arrow on the controller, the shade jogs up and down once
At this point, the direction on the motor is set, and the states will report properly to Home Assistant

jason-curtis, 2024-01-17:
@CarlosUrbina in your case, were the controls on the handheld remote also inverted, or only the directions in HA?

CarlosUrbina, 2024-01-17:
Curiously, the controller always works as expected, the inversion was only on the HA side

night199uk, 2024-02-08:
folks - I originally added the quirk in zigbee2mqtt for these Yookee / Yoolax blinds after *MUCH* testing with two blinds that acted differently. these blinds are all over the place with the original firmware, and sometimes they are inverted, sometimes not. you can see inverted values from the same blind, reset it, and see regular values. reset it again and see inverted values. it is not that the roll is set incorrectly, at all.  the inversion may come back at some point in the future.

it seems Z2M reverted my hack as some users reported their newer blinds working, but it's not as simple as that and having just upgraded my z2m and being broken again i'm about to post a new quirk for z2m. i even reverted the z2m hack myself a few years back thinking my blinds had fixed themselves only to have to put it back.

I believe @JeremeLau from Yookee / Yoolax / Yooksmart Graywind has event admitted there was a firmware bug. But as yet Yookee have not provided updated firmware on this thread to existing owners. the issues are genuine for some owners and i think having the hack optionally available for owners that need it makes the most sense.

night199uk, 2024-02-08:
@JeremeLau will yookee/yooksmart ever publish the new firmware please? i'd love to buy more of your blinds. they are a great price, look great, and are super functional, but lack of public firmware updates is sucky.

emayo, 2024-02-08:
> @JeremeLau will yookee/yooksmart ever publish the new firmware please? i'd love to buy more of your blinds. they are a great price, look great, and are super functional, but lack of public firmware updates is sucky.

wonder if we can reverse engineer and replace their firmware with our own?  I have a few of their motors I took out of their blinds when I had a range of, frustrating, problems.

PR #1664 (2023-03-11): Remove Yooksmart quirk
There is actually no issue with the percentage reported on the Yooksmart shades.

Fix #1306
commit 06f50ee8 (2023-03-11): Remove Yooksmart quirk (#1664)

There is actually no issue with the percentage reported on the Yooksmart shades.

Fix #1306