zigpy/zha-device-handlers #1311: [BUG] async_update in the logs generates spike to zero

glcos, 2022-01-26:
**Describe the bug**
I'm testing a new device based on PZEM-004T, I made a quirk for it.
It works absolutely fine, apart from an issue I'm really struggling with.
A single entity rms_current has regular spikes to zero, but it does not depend on the sensor itself, because it is sending normal updates to the ZigBee network.

When these spikes happen I notice this entry in the log:

`[homeassistant.components.zha.core.channels.base] [0x0412:1:0x0b04]: async_update`

So it appears that when I see this async_update in the log, the rms_current value is set to zero, then it returns to normal at the next sensor report.

These entries stop if I switch off the sensor.

This is only happening on one single entity (rms_current), all other entities are fine.

**To Reproduce**
Maybe this is linked to clusters ElectricalMeasurement or Metering

**Expected behavior**
Normal entity behaviour, no spikes

**Screenshots**
[image]

Adminiuga, 2022-01-28:
Does the actual async_update() succeeds in reading the attributes?

glcos, 2022-01-28:
> Does the actual async_update() succeeds in reading the attributes?

Not in my case.
That's why I had to disable it.

Adminiuga, 2022-01-29:
What's the reason for attribute read failure?

glcos, 2022-01-29:
I'm trying to figure out, unfortunately I don't see anything in the log that can give me a hint apart from the async_update entry.

closed 2022-08-04 by github-actions[bot]