zigpy/zha-device-handlers #1312: [Device Support Request]Tuya motion sensor zigbee 3.0

vietpv89, 2022-01-27:
**Is your feature request related to a problem? Please describe.**
Hi all, i just bought this zigbee 3.0 sensor:
I achieved to add it using ZHA (I have a SonOff Zigbee bridge as Coordinator)
But there is no entites in the device (no switch to toggle the alarm, no temp and humidity sensors, nothing !), here is a picture :
[image]

**Describe the solution you'd like**

I wan't to all entities support

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

[image]
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
New device 0x9913 (04:cd:15:ff:fe:e0:d1:2b) joined the network
[0x9913] Scheduling initialization
Tries remaining: 3
[0x9913] Requesting 'Node Descriptor'
Tries remaining: 2
[0x9913] Extending timeout for 0x49 request
Device 0x9913 (04:cd:15:ff:fe:e0:d1:2b) joined the network
[0x9913] Scheduling initialization
[0x9913] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9913 ieee=04:cd:15:ff:fe:e0:d1:2b is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x81\x13\x99+\xd1\xe0\xfe\xff\x15\xcd\x04\x8e'
[0x9913:zdo] ZDO request ZDOCmd.Device_annce: [0x9913, 04:cd:15:ff:fe:e0:d1:2b, 142]
Tries remaining: 3
[0x9913] Requesting 'Node Descriptor'
Tries remaining: 2
[0x9913] Extending timeout for 0x4b request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9913 ieee=04:cd:15:ff:fe:e0:d1:2b is_initialized=False> from ep 0 to ep 0, cluster 32770: b'K\x00\x13\x99\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00'
[0x9913] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
[0x9913] Discovering endpoints
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9913 ieee=04:cd:15:ff:fe:e0:d1:2b is_initialized=False> from ep 1 to ep 1, cluster 61184: b'\t\x00\x01\x00\xed\t\x02\x00\x04\x00\x00\x00A'
[0x9913] Received ZCL while uninitialized on endpoint id 1, cluster 0xef00 id, hdr: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=0 command_id=1>, payload: b'\x00\xed\t\x02\x00\x04\x00\x00\x00A'
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9913 ieee=04:cd:15:ff:fe:e0:d1:2b is_initialized=False> from ep 0 to ep 0, cluster 32773: b'M\x00\x13\x99\x01\x01'
[0x9913] Discovered endpoints: [1]
[0x9913] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x9913:1] Discovering endpoint information
Tries remaining: 3
Received frame on uninitialized device <Device model=None manuf=None nwk=0x9913 ieee=04:cd:15:ff:fe:e0:d1:2b is_initialized=False> from ep 0 to ep 0, cluster 32772: b'O\x00\x13\x99\x14\x01\x04\x01Q\x00\x01\x04\x00\x00\x04\x00\x05\x00\x00\xef\x02\x19\x00\n\x00'
[0x9913:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[0, 4, 5, 61184], output_clusters=[25, 10])
Unknown cluster 61184
[0x9913:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=81 command_id=Command.Read_Attributes_rsp>
[0x9913] Read model 'TS0601' and manufacturer '_TZE200_mrf6vtua' from <Endpoint id=1 in=[basic:0x0000, groups:0x0004, scenes:0x0005, None:0xEF00] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x9913] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_mrf6vtua' nwk=0x9913 ieee=04:cd:15:ff:fe:e0:d1:2b is_initialized=True>
Device is initialized <Device model='TS0601' manuf='_TZE200_mrf6vtua' nwk=0x9913 ieee=04:cd:15:ff:fe:e0:d1:2b is_initialized=True>
Checking quirks for _TZE200_mrf6vtua TS0601 (04:cd:15:ff:fe:e0:d1:2b)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
device - 0x9913:04:cd:15:ff:fe:e0:d1:2b entering async_device_initialized - is_new_join: True
device - 0x9913:04:cd:15:ff:fe:e0:d1:2b has joined the ZHA zigbee network
[0x9913](TS0601): started configuration
[0x9913:ZDO](TS0601): 'async_configure' stage succeeded
[0x9913:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=0 command_id=1>
[0x9913:1:0xef00] Unknown cluster-specific command 1
[0x9913:1:0xef00] ZCL request 0x0001: b'\x00\xed\t\x02\x00\x04\x00\x00\x00A'
[0x9913:1:0xef00] No handler for cluster command 1
[... 1918 lines cut ...]
[0x9913:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=2 command_id=1>
[0x9913:1:0xef00] Unknown cluster-specific command 1
[0x9913:1:0xef00] ZCL request 0x0001: b'\x00i\t\x02\x00\x04\x00\x00\x00f'
[0x9913:1:0xef00] No handler for cluster command 1
[0x9913:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=2 command_id=1>
[0x9913:1:0xef00] Unknown cluster-specific command 1
[0x9913:1:0xef00] ZCL request 0x0001: b'\x00i\t\x02\x00\x04\x00\x00\x00f'
[0x9913:1:0xef00] No handler for cluster command 1
[0x9913:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=2 command_id=1>
[0x9913:1:0xef00] Unknown cluster-specific command 1
[0x9913:1:0xef00] ZCL request 0x0001: b'\x00i\t\x02\x00\x04\x00\x00\x00f'
[0x9913:1:0xef00] No handler for cluster command 1
[0x9913:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=2 command_id=1>
[0x9913:1:0xef00] Unknown cluster-specific command 1
[0x9913:1:0xef00] ZCL request 0x0001: b'\x00i\t\x02\x00\x04\x00\x00\x00f'
[0x9913:1:0xef00] No handler for cluster command 1
[0x9913:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=3 command_id=1>
[0x9913:1:0xef00] Unknown cluster-specific command 1
[0x9913:1:0xef00] ZCL request 0x0001: b'\x00j\t\x02\x00\x04\x00\x00\x00a'
[0x9913:1:0xef00] No handler for cluster command 1
[0x9913:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=3 command_id=1>
[0x9913:1:0xef00] Unknown cluster-specific command 1
[0x9913:1:0xef00] ZCL request 0x0001: b'\x00j\t\x02\x00\x04\x00\x00\x00a'
[0x9913:1:0xef00] No handler for cluster command 1
[0x9913:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=

MattWestb, 2022-01-27:
I was patching the tuya motion sensor quirk and adding one new device class for your device.
Configure local quirks https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and put the unziped quirk in the new folder and restarting HA.
If all is going well you is getting one motion sensor on the device card.
[ts0601_motion.zip](https://github.com/zigpy/zha-device-handlers/files/7951292/ts0601_motion.zip)

javicalle, 2022-01-27:
Hi @MattWestb, there is another case of sensor that I bet it is using a diferent DP to send the `MOTION_EVENT`:
* #1236 

I also have made a quirk ot `ts0601_motion.py` but waiting for user to test.

I believe that can be put all together in the same PR (when tested).

Regards.

MattWestb, 2022-01-27:
I dont have the device but only trying helping the user and i was missing your test.
I think is good trying if your quirk is working and if its possible for the user testing if its working OK getting support for the device.

If you is using the new MCU commands is great getting it migrated to it and can cleaning up the dold code.

javicalle, 2022-01-27:
No worries.
My quirk is for another diferent device, but I wanted to comment it to avoid multiple PR on the same class.

> If you is using the new MCU commands 

Not yet, too many open fights right now to start a new one here 😅

MattWestb, 2022-01-27:
Good !
Can you adding this device ID in your quirk so the user can testing it and we is only doing one PR if its working OK ?
And you can doing little code fixes if needed i cant ;-))

javicalle, 2022-01-27:
Sure, but probably will be tomorrow.
Let's see if in the meantime someone can validate the implementations and we upload a clean PR.

vietpv89, 2022-01-28:
> I was patching the tuya motion sensor quirk and adding one new device class for your device. Configure local quirks [#693 (comment)](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274) and put the unziped quirk in the new folder and restarting HA. If all is going well you is getting one motion sensor on the device card. [ts0601_motion.zip](https://github.com/zigpy/zha-device-handlers/files/7951292/ts0601_motion.zip)
i'm already replace file but it only has one entitie and always show clear.
[image]

vietpv89, 2022-01-28:
original device and app have these entities. 
[image]

javicalle, 2022-01-28:
> original device and app have these entities.
One step every time.

Can you post your logs with the new quirk?

javicalle, 2022-01-28:
@vietpv89 Do you have any link to the device store (or whatever) where other users can see and identify the sensor?

vietpv89, 2022-01-28:
> @vietpv89 Do you have any link to the device store (or whatever) where other users can see and identify the sensor?

sorry, that product of China, but i buy it at Vietnam and i don't have any info of that product.
But today i contact to seller and he sent me this file, it working but only have 2 entities
[image]

[ts0601_radar.zip](https://github.com/zigpy/zha-device-handlers/files/7958788/ts0601_radar.zip)

MattWestb, 2022-01-28:
Some have making one quirk that looks very like the tuya Air and putting in normal and with GP endpoint in it :-))

Its helping finding some more DPs for the device but i think its not having all implanted.

@vietpv89 If you is having the device paired and joined and have the quirk loaded OK its logging all in the log file. Then tahe the power of the device for one minute and putting it back. If its making all OK as tuyya device is sending all DPs its using and ZHA is logging it so we can see all the DPs its using.

vietpv89, 2022-01-28:
@MattWestb where can i see the log file ? and if i post the log file, can you help me to modify quirk file to support all entities ?
Thank you !

MattWestb, 2022-01-28:
The HA log file is in your HA config directory and have the name `home-assistant.log`.

vietpv89, 2022-01-28:
> The HA log file is in your HA config directory and have the name `home-assistant.log`.

Thank u, and this is my HA log.
[home-assistant.txt](https://github.com/zigpy/zha-device-handlers/files/7959123/home-assistant.txt)

javicalle, 2022-01-28:
Interesting...
They are using the `TuyaNewManufCluster`. It seems that it would be easy to adapt to my approach.
I will try in the weekend to figure how to.

javicalle, 2022-01-28:
> Thank u, and this is my HA log.
> [home-assistant.txt](https://github.com/zigpy/zha-device-handlers/files/7959123/home-assistant.txt)

There are a lot of DPs here 👍🏻

MattWestb, 2022-01-28:
Great work done brave user @vietpv89  !!!

And @javicalle have getting little new DPs for implanting :-)))

I was baying one LIDL (tuya) motion sensor that is using "normal tuya Zigbee"  and its working OK but is only having motion for IAS Zone and not light level and so on so its being putted on the list for good devices on my system then i like getting all Philips HUE motion sensors that cant staying on one good router and only jumping around in my production system.

### Interesting !!

javicalle, 2022-02-01:
Device has been added to PR #1315.

All already identified DPs has been added 'as is'.
The others DPs has been added as `MmwRadarManufCluster` attributes. Although writing of the attributes has been implemented, it is unknown if the device supports the update.

Some of the attributes (and values) that the device presents are:
- presence_state: presence
- target distance: 1.61m
- illuminance: 250lux
- sensitivity: 9
- minimum_detection_distance: 0.00m
- maximum_detection_distance: 4.05m
- dp_detection_delay: 0.1
- dp_fading_time: 5.0
- ¿illuminance_threshold?: 255lux
- presence_brightness: no control
- no_one_brightness: no control
- current_brightness: off

Maybe the device can be paired with a light and control its brightness based on presence detection (presence_brightness, no_one_brightness, etc.)
(It's just a guess)

javicalle, 2022-02-01:
For future references:

the DPs and values detected are:
```
 dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, raw=b'\x01', *payload=<enum8.undefined_0x01: 1>) +
 dp=2, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, raw=b'\x07\x00\x00\x00', *payload=7) *
 dp=3, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, raw=b'\x00\x00\x00\x00', *payload=0) *
 dp=4, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, raw=b'X\x02\x00\x00', *payload=600) *
 dp=6, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, raw=b'\x01', *payload=<enum8.undefined_0x01: 1>) *
 dp=9, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, raw=b'5\x00\x00\x00', *payload=53) -- 33 < 210
 dp=101, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, raw=b'\x01\x00\x00\x00', *payload=1) *
 dp=102, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, raw=b'\x96\x00\x00\x00', *payload=150) *
 dp=103, data=TuyaData(dp_type=<TuyaDPType.STRING: 3>, raw=b' ', *payload=' ') *
 dp=104, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, raw=b'\r\x00\x00\x00', *payload=13) -- 13 < 17
 dp=105, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>) *
 dp=106, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>) *
 dp=107, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>) *
 dp=108, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, raw=b'\xff\x00\x00\x00', *payload=255) *
```

Legend:
- `+`: a single value
- `*`: pseudo-static value
- `--`: range of detected values

javicalle, 2022-02-01:
@vietpv89 Could the device be similar to one of these?
- https://zigbee.blakadder.com/Tuya_MIR-HE200.html
- https://www.zigbee2mqtt.io/devices/MIR-HE200-TY.html#tuya-mir-he200-ty

vietpv89, 2022-02-04:
> @vietpv89 Could the device be similar to one of these?
> 
> * https://zigbee.blakadder.com/Tuya_MIR-HE200.html
> * https://www.zigbee2mqtt.io/devices/MIR-HE200-TY.html#tuya-mir-he200-ty

I Think my device similar to this device
> * https://www.zigbee2mqtt.io/devices/MIR-HE200-TY.html#tuya-mir-he200-ty
I've just install lastest zha_quirks but it still not working.

javicalle, 2022-02-05:
@vietpv89 Could you test the new quirk version from #1315 and check if the messured values can be accurate?

The values are calculated as they have been extracted from the file that you have attached in:
* https://github.com/zigpy/zha-device-handlers/issues/1312#issuecomment-1024192556

vietpv89, 2022-02-05:
@javicalle yes, i'm already use new version, but  entities same old file .
[image]

javicalle, 2022-02-05:
Are the `occupancy` and `illuminance_lux` sensors working as expected?
What sensors are you missing?

javicalle, 2022-02-05:
There are a lot of DPs that can be sensors values or device settings, but testing is needed to analyze the logs to find out what each one does.
The values shown by the application can give an idea of what functions the device has, but to find out which is which, it is necessary to test.

It is also necessary to do tests to analyze the **values**. For example, does the `sensitivity` accept values of 0-9 ? from 0-15? or will it be 0-100?
Are the `target_distance` values in centimeters? Are they inches? Meters?

All this needs to be analyzed 'in situ' with the device and the logs, it is not something that I can do from here.

vietpv89, 2022-02-05:
@javicalle thanks for support. 
my device have some values and setting: target distance, sensitivity, minimum detection setting,  maximum detection setting, detection delay, fading time, occupancy, illuminance_lux.
Now only occupancy, illuminance_lux working correctly

javicalle, 2022-02-05:
All those attributes should be available as attributes of the `MmwRadarManufCluster` cluster. The problem is that for now no one has been able to relate each of the DPs to the device settings.
However, it should be possible to interact with them from within the appliance cluster management.

I've checked the 'herdsman-converters' implementation, but haven't found any DP mappings that fit this device:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/tuya.js

The `target_distance` value should be reported in the `TuyaAnalogInput` cluster, `present_value` attribute.
HA probably doesn't present it correctly (if it does), but I haven't found any Zigbee cluster that represents such a sensor.

If any value does not scale correctly or is not reported, it is necessary to include the error traces or the expected values (or better, the conversion that each value needs). For me they are all meaningless raw values.

If someone can provide that information about it, I can include it in the PR. 
If not, I am in favor of releasing the current version as is and see if someone else integrates the device and can extract the necessary information.

PR #1315 (2022-02-15): Add some ts0601_motion devices (with Tuya MCU Cluster approach)
This PR will add support for some `ts0601_motion` devices.

The new devices will follow the new 'Tuya MCU Cluster' approach to update device attributes.

Fixes: #1236
Reported motion sensor 'NAS-PD07' manufactured by NEO.
Most of the code came from 'herdsman-converters'
Those DPs that have been possible are reported as a new cluster.
Those that I have not found a match for are reported as attributes of the MCU cluster.

Not sure about possible `tamper` values. Probably there are diferent values >0 with other meanings.

Fixes: #131 [... cut]
closed 2022-02-15 by dmulcahey