zigpy/zha-device-handlers #1313: [Device Support Request] Smart Air House keeper TS0601 _TZE200_dwcarsat

renngurke32, 2022-01-29:
**Is your feature request related to a problem? Please describe.**
Hello all,
I bought a zigbee air quality sensor and I achieved to add it using ZHA. But the sensor got no entities. For zigbee2mqtt there is a solution -> https://github.com/Koenkk/zigbee-herdsman-converters/pull/3745

[image]

**Describe the solution you'd like**
I want to see all entities und use the sensor data.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
[image]

[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=190 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x06\x14\x02\x00\x04\x00\x00\x00\x00'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=190 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x06\x14\x02\x00\x04\x00\x00\x00\x00'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=190 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x06\x14\x02\x00\x04\x00\x00\x00\x00'
[0xbb51:1:0xef00] No handler for cluster command 2
[0x466e:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=181 command_id=Command.Read_Attributes_rsp>
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=191 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x06\x15\x02\x00\x04\x00\x00\x00\x00'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=192 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x06\x16\x02\x00\x04\x00\x00\x01f'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=193 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x04\x02\x02\x00\x04\x00\x00\x00\x04'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=194 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x00\x12\x02\x00\x04\x00\x00\x00\xb8'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=195 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x00\x13\x02\x00\x04\x00\x00\x02\x9f'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=196 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x00\x12\x02\x00\x04\x00\x00\x00\xba'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=196 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x00\x12\x02\x00\x04\x00\x00\x00\xba'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=196 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x00\x12\x02\x00\x04\x00\x00\x00\xba'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=196 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x00\x12\x02\x00\x04\x00\x00\x00\xba'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=196 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x00\x12\x02\x00\x04\x00\x00\x00\xba'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=196 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x00\x12\x02\x00\x04\x00\x00\x00\xba'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=196 command_id=2>
[... 1334 lines cut ...]
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x07\x14\x02\x00\x04\x00\x00\x00\x00'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x07\x14\x02\x00\x04\x00\x00\x00\x00'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x07\x14\x02\x00\x04\x00\x00\x00\x00'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x07\x14\x02\x00\x04\x00\x00\x00\x00'
[0xbb51:1:0xef00] No handler for cluster command 2
[... cut ...]
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x07\x14\x02\x00\x04\x00\x00\x00\x00'
[0xbb51:1:0xef00] No handler for cluster command 2
[0xbb51:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=2>
[0xbb51:1:0xef00] Unknown cluster-specific command 2
[0xbb51:1:0xef00] ZCL request 0x0002: b'\x00\x07\x14\x02\x00\x04\x00\x00\x00\x00'
[0xbb51:1:0xef00] No handler for cluster command 2

closed 2022-02-18 by dmulcahey

e1z0, 2022-02-19:
Does not work either :(

JoryHogeveen, 2022-03-02:
@magnusnordlander 
Just tested this integration (HA just released the compatibility) but sad to say it only registering the VOC sensor, there are 5 more sensors:
- Temperature
- Humidity
- CO2
- CHOH (CH2O)
- PM2.5
- VOC (Works!)

JoryHogeveen, 2022-03-03:
Since this is closed I'll open a new topic.

budcalloway, 2022-03-03:
Hi !!! 
You are all very lucky. I received this device today and can't make it discovered by my CC2531 coordinator... does anyone encountered this problem too ?

budcalloway, 2022-03-06:
Well, I reflashed the CC2531 and have been able to pair it. 
I confirm that the data seems to be mixed up 
[image]

buddy31130, 2022-04-03:
Using the Z2M update (1.25.0-1), the data seem to be good except that I do have a new error message

`Zigbee2MQTT:warn  2022-04-01 20:54:10: zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #20 with data {"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}                 
Zigbee2MQTT:warn  2022-04-01 20:54:14: zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #20 with data {"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}                 
Zigbee2MQTT:warn  2022-04-01 20:54:21: zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #20 with data {"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}`

[image]
This error occurres every 2-3 seconds. I had to remove the device
Maybe the formaldehyde value that is missing ?

cloudbr34k84, 2022-04-21:
so i have a zigbee and wifi version of the 6 in 1.
here is my Zigbee Version 
[image]

Here is My wifi
[image]

I also have the 4 in 1 Wifi Version
[image]

PR #1367 (2022-02-18): Add support for Tuya TS0601_TZE200_dwcarsat
Fixes #1313
commit f9e25f41 (2022-02-18): Add support for Tuya TS0601_TZE200_dwcarsat (#1367)

Fixes #1313