zigpy/zha-device-handlers #1314: [Device Support Request] SilverCrest Smart Home Plug - UK Version"manufacturer": "_TZ3000_00mk2xzy",   "model": "TS011F",   "class": "zigpy.device.Device"

accessiblepixel, 2022-01-29:
**Is your feature request related to a problem? Please describe.**
I am attempting to add some Lidl Silvercrest Smart Plugs to my network. It shows up in ZHA and the switch is there, however it is marked as unavailable.

**Describe the solution you'd like**
I believe this device is already supported as https://zigbee.blakadder.com/Tuya_TS011F.html says it's listed under ZHA, however the device identifier is slightly different, and this is a UK version of that socket.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_00mk2xzy",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}

**Additional context**
I believe if a quirk is added or this device is seen as one of the existing supported ones, it should probably work, however my skill is not great enough to know where to add this model as part of the already supported ones.

Any assistance would be greatly appreciated again!

Kind regards,
Jessica

MattWestb, 2022-01-29:
Hey J !!
Its looks like the normal LIDL plug but with one new device ID butt the quirk shall only matching on endpoints and mode ID. https://github.com/zigpy/zha-device-handlers/blob/c6837b927c7198f23f511905d7f21e1cbf3fe062/zhaquirks/tuya/ts011f_plug.py#L95
The commit was mad 28 december so its not in HA 2021.12 but is coming in next  release in  around 2 weeks.

If you like getting the extra functions before configure local quirks https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and download the quirk and putting it in the locql quirk folder and restart HA and it shall working.

MattWestb, 2022-01-29:
By the way is your tuya TRV working OK or is it cold at home ??

accessiblepixel, 2022-01-29:
@MattWestb Thank you! I'm loading the quirk file now and rebooting. Will report back :)

Yes, my TRV valves are working great, thank you, I removed the custom quirk and used the one that is now in ZHA by default. Still no battery reading (It says Unknown) but I can live with that ;)

I noticed if you set new temperature, the setting updates on the device, but the display in HA jumps back to the old temperature in HA, but if you wait a few minutes, it does show the right (new) heat setting. :)

accessiblepixel, 2022-01-29:
Works perfectly! <3

My eternal gratitude. So you reckon it'll be in Home Assistant 22.01, then? 👯‍♀️

MattWestb, 2022-01-29:
The TRVs is sleeping device and can being little lazy receiving commands and can also being they is timing out but they is fast sending updates to HA.
Sadly we have not getting any easy way getting all extra things easy added for users in ZHA but both the standard and the updated is working very well in my system.
The best with the new is valve jammed detection if one is having problem and cant regulating OK.

You shall now having one tuya OnOff cluster with extra attribute with little more functions lik this:
https://github.com/zigpy/zha-device-handlers/blob/c6837b927c7198f23f511905d7f21e1cbf3fe062/zhaquirks/tuya/__init__.py#L827-L857
Normally tuya have not implanting all function but if its and working you can using all of them (I think the switch mode is not implanted in plugs but you never knowing with tuya).

Dont forgetting deleting the local quirk then upgrading HA for not getting problem in the future and the system is using the old one !!

Its in https://github.com/zigpy/zha-device-handlers/releases/tag/0.0.66 the 5rd from the end  and its in the beta now.

accessiblepixel, 2022-01-29:
Yeah, I just paired all 4 of them - I went with these ones since I saw in the community database support was likely (and I could get them locally, so not anything going around our electrical safety regulations!)

They're dumb switches basically, just on/off, but working fine from what I can see. My plan is to get my smart home not dependent on the cloud and I don't really like localtuya, it's such a pain in the behindus to get working reliably, so I'm moving things over to zigbee when I can, and you and your peers run amazing support here, I have to say. One of the best I've seen from a project.

Do you manage to get any sleep? 😄 

I'm sure I'll be back when I get the next device that doesn't work as expected. I quite like how everyone benefits from reporting things.

accessiblepixel, 2022-01-29:
For the TRVs, I had a slight idea about the "slow to update temperature thing". Maybe when the temperature is adjusted, home assistant could keep displaying the "new" temperature (sort of cache it?) and then update the display to the actual setting once it's been verified?

I have no idea if this is possible, but it might help users (like myself at first!) from thinking "Why's the thing not changing to the new temperature?!" and getting mad with it 😋

MattWestb, 2022-01-29:
Sleep is normally OK only one very close friend was getting one heart attack in February and lymfom cancer diagnose in November so it can being little up and down on my side but its OK.

I must recommending using good routers (I think your plugs is one) around so you can getting the TRVs (and other sleepers) using one router near and not getting problem with interferences and dont need using  so much power for communicating.

I have IKEA plugs as "backbone" for all my sensors and is using "add device via this device"  from the router device card then re/pairing them so they is using the right good router and not need jumping around (if they can, Xiaomi sensors dont).

One user is trying rewriting the tuya "core" for DP devices (your TRVs) and fixing some problem with some devices but its takes time but hi is doing good progress but we cant working around all bad things in all device firmwares and tuya is doing very much of them.

accessiblepixel, 2022-01-30:
I noticed there was an "Add via this device" option that came up when using these new plugs. I kinda knew that mains powered devices could become sorta network "routers", but I always thought that was automatically, but are you saying you have to manually configure it?

I'm only using the cheap (old version) of the Texas Instruments based Zigbee coordinator, since I wanted to have a play around and if I didn't get on with it too well then I hadn't spent too much money, and I know it only supports about 20 devices connected directly but always knew that more devices could be connected if they routed via a mains powered one.

We only have a bungalow so there's never too much distance between the coordinator (or even originally the Tuya gateway) so everything's connected with a pretty good signal, although if I end up using smart plugs nearer the TRVs I'll certainly do what you suggested for an easier network "route" 😃 

I really don't know why Tuya go so outside of the Zigbee specification... Like there's a standard there, but they just kinda use it the way they want it. I can imagine the person rewriting the core Tuya 'quirks' has a big job on their hands though! I just don't like how smart home stuff needs the internet these days, which is kinda useless if it's offline - I know you can still control Tuya stuff if you're on the same network as the devices, but I have them setup on their own isolated network, given all you hear about IoT devices not being the most secure 🤣 

I'm really sorry to hear about your friend, and hope they are okay. My partner's step dad passed away a few years ago, he had leukaemia, which he basically got rid of but then it came back, it was a difficult time for both of us, but I tried to stay strong for my partner and kinda "took over" for a while since he couldn't really concentrate on any big decisions and things like that.

Hope you're having/you've had a great day 👍

MattWestb, 2022-01-30:
Thanks for replay 👍 
Most mains powered devices is being routers (some switches without neutral is normally not). The routers is building and maintaining the meschnetwok autonome. On the (sleeping)end devices is little more tricky like Xiaomi / Aqara is never "jumping" to on new parent then only silent leaving the network.Therefore its good adding them to one good router.

If you going all open zigbee system and dont need the tuya ZBGW and its the ethernet version we have hacking it and using the zigbee module and tunneling serial to it and working with ZHA and Z2M. The Zigbee module is much better the the TI CC-253X and having better radio. But only do it if you if you closing your tuya project or buying one more from LIDL then they is having i in your local store. Some info of the hacking and links to other information https://github.com/zigpy/zigpy/discussions/650.

Have one nice day and take care of you both !

accessiblepixel, 2022-01-30:
Excellent. I'll keep that in mind, thank you for the knowledge :)

Can the Zigbee Wifi Tuya Gateway be modified too, or is it just the Ethernet one?

MattWestb, 2022-01-30:
What i knowing no one have hacking it but it shall having one standard tuya Zigbee module that's not problem as long is one with Silabs chip but we dont knowing how to getting root access to the Linux.

Its needs some with good Linux knowledge and have the hardware for testing and getting ser2net installed.

accessiblepixel, 2022-01-30:
I know Linux pretty well, once I've swapped over to the TI chip based zigbee network, I'll crack the case, and I can at least document what's inside it for you folks, so if I am clueless (quite likely!) then maybe the knowledge of the board and some pictures will help someone more advanced than me :)

It's a TCP Smart branded wireless zigbee gateway, I bought it with the TRVs :)

MattWestb, 2022-01-30:
I was baying one "family pack" with 5 TRV and Eth ZBGW and some weeks after they was changing to the WiFi GW so was having luck !!

If you is getting it opened make one new discussion in Zigpy and posting photos and other information you can getting and perhaps some users with embedded devices knowledge can helping getting inside the chip.
That how i was starting and the Paul (Italian) was starting in the other end (rooting the WiFI SOC) and little help from one Danich unix guru we was getting all in place in some weeks and Gary (Australia) was cooking one updated firmware for the Zigbee module and im Swedish living in Austria so good knowledge sharing and good working product in the end.

closed 2022-08-05 by github-actions[bot]