zigpy/zha-device-handlers #1317: [BUG] TS0001 Tuya switch

adi90x, 2022-01-30:
Hello, 

Thanks for this great projet. 
Just start playing around and it seems Tuya TS0001 are recognized as lamp ( in homeassistant), however those are just switch/relay and not specifically light. 
Is it something on HA side or those it have something to do with the quirk use by this device ? 

Regards,

MattWestb, 2022-01-30:
Lights, plugs relay and so on can being switches on and off from HA and having one off / on state in the GUI. If you like changing the icon to one flash for plugs you can doing it in the device properties.

Controlling devices (battery powered light switches / remotes that is only sending commands) is only having one battery sensor and all events they sending can being triggered in automations but they dont have any state like lights is having so no switch in the GUI for them.

adi90x, 2022-01-30:
Hello,

Not sure to understand your point, my issue is that those switch/relay when paired appeard as light while they are not light ( eventhough they are mainly used to control light)
My question is it normal to have those device reporting as a light while they are not ?
If it is normal then, is there a way to force it only for me ? Maybe a custom quirk ?

Regards,
Adrien

dmulcahey, 2022-01-30:
https://www.home-assistant.io/integrations/zha/#modifying-the-device-type

adi90x, 2022-01-30:
> https://www.home-assistant.io/integrations/zha/#modifying-the-device-type

Hello, yes that is what I use to make them appear as switch, but it need to be done for each device and I was wondering if there is no other global setup available as I have almost 15 of them to enable ?
Moreover, it still don't make sense to have a switch/relay reported as a light...

TheJulianJES, 2022-04-17:
Yes, you would have to add multiple entries to your config like this:
```yaml
zha:
  device_config:
    xx:xx:xx:xx:xx:xx:xx:xx-1:
      type: "switch"
    yy:yy:yy:yy:yy:yy:yy:yy-1:
      type: "switch"
    [...]
```

The Tuya switches report themselves as lights, as they've implemented the type incorrectly.
Since Tuya seems to have done this on multiple models and many users are probably happily using them as lights through ZHA, changing it "globally" at this point by implementing a quirk would completely break it for all of these users. So I think it's best that you just modify them through your config manually.

You could, however, make a [custom quirk](https://www.home-assistant.io/integrations/zha/#custom_quirks_path) for yourself, since the ZHA integration now supports these. Note that It might break at some point if there are any underlying changes in ZHA.

I recommend just adding the 15 switches to your config and be done with it.

closed 2022-10-21 by github-actions[bot]