zigpy/zha-device-handlers #1319: [BUG] TS0601 _TZE200_fzo2pocs Blind Motor Inverted Controls

jameshirka, 2022-01-31:
**Describe the bug**
Zemismart TS0601 _TZE200_fzo2pocs Blind Motorsc connected to Conbee II using ZHA in Home Assistant.
When configured out of the box, the position and up/down arrows in home assistant are inverted.
I have changed cover_inverted cluster in TuyaWindowCoverControl to Bool.true and the position is now working correctly.
However the up and down arrows make the covers open/close inverted and if the blind is fully open or closed, the only way to open it is to change the position.

I've tried setting motor_direction both to true and false but neither change the button controls.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair blind motor
2. set cover_inverted cluster in TuyaWindowCoverControl to Bool.true (the position will now show correctly)
3. try to open or close the blind
4. blind goes the opposite direction to button pressed.

**Expected behavior**
When I press the down/ close button the blind goes down.
When I press the open/ up button the blind goes up.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Device Signature:

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000a",
        "0x0102",
        "0xef00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_fzo2pocs",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_cover.TuyaZemismartSmartCover0601"
}`

When I pushed up (which makes the motor go down) and then stop:

`2022-01-31 13:10:37 DEBUG (MainThread) [zhaquirks.tuya] 5c:02:72:ff:fe:94:31:fe Sending Tuya Cluster Command.. Manufacturer is _TZE200_fzo2pocs Cluster Command is 0x0000, Arguments are ()
2022-01-31 13:10:37 DEBUG (MainThread) [zhaquirks.tuya] 5c:02:72:ff:fe:94:31:fe Sending Tuya Command. Paylod values [endpoint_id : 1, Status : 0, TSN: 0, Command: 0x0401, Function: 0, Data: [1, 0]]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 1 under 2 request id, data: b'050210010000000104000100'
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (27, 2, 0, <DeconzAddressEndpoint address_mode=2 address=0xC72D endpoint=1>, 260, 61184, 1, b'\x05\x02\x10\x01\x00\x00\x00\x01\x04\x00\x01\x00', 2, 0)
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 2]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 2: 00
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x02 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xc72d endpoint=1>, status: 0x00
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [30, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc72d>, 1, 260, 61184, b'\x0c\x02\x10\x01\x0b\x00\x00', 0, 175, 255, 114, 227, 195, 0, -37]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy.zcl] [0xc72d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=True is_reply=True disable_default_response=False> manufacturer=4098 tsn=1 command_id=Command.Default_Response>
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc72d>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'0c0210010b0000'
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc72d>, 1, 260, 61184, b'\t\x08\x01\x00\x00\x01\x04\x00\x01\x00', 0, 175, 255, 114, 227, 195, 0, -37]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy.zcl] [0xc72d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=8 command_id=1>
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy.zcl] [0xc72d:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 0])]
[... 24 lines cut ...]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc72d>, 1, 260, 61184, b'\t\x08\x01\x00\x00\x01\x04\x00\x01\x00', 0, 175, 255, 117, 227, 195, 0, -37]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy.zcl] [0xc72d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=8 command_id=1>
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy.zcl] [0xc72d:1:0xef00] ZCL request 0x0001: [Command(status=0, tsn=0, command_id=1025, function=0, data=[1, 0])]
2022-01-31 13:10:37 DEBUG (MainThread) [zhaquirks.tuya] 5c:02:72:ff:fe:94:31:fe Received Attribute Report. Command is 0x0001, Tuya Paylod values[Status : 0, TSN: 0, Command: 0x0401, Function: 0x00, Data: [1, 0]]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc72d>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'09080100000104000100'
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc72d>, 1, 260, 61184, b'\t\x08\x01\x00\x00\x01\x04\x00\x01\x00', 0, 175, 255, 117, 227, 195, 0, -37]
2022-01-31 13:10:37 DEBUG (MainThread) [zigpy.zcl] [0xc72d:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=8 command_id=1>
[... cut ...]
2022-01-31 13:10:38 DEBUG (MainThread) [zhaquirks.tuya] 5c:02:72:ff:fe:94:31:fe Sending Tuya Cluster Command.. Manufacturer is _TZE200_fzo2pocs Cluster Command is 0x0002, Arguments are ()
2022-01-31 13:10:38 DEBUG (MainThread) [zhaquirks.tuya] 5c:02:72:ff:fe:94:31:fe Sending Tuya Command. Paylod values [endpoint_id : 1, Status : 0, TSN: 0, Command: 0x0401, Function: 0, Data: [1, 1]]
2022-01-31 13:10:38 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 3 under 4 request id, data: b'050210030000000104000101'
`

SweAdelpha, 2022-02-05:
Can confirm I have the same problem with my Zemismart TS0601 _TZE200_fzo2pocs cover blind motors. I've just learned to live with it for six months now. Would be great if we can find a solution though.

TheJulianJES, 2022-04-17:
See https://github.com/zigpy/zha-device-handlers/issues/1399 for a similar issue.

It's possible that ``_TZE200_fzo2pocs`` would just need to be added to ``TUYA_COVER_INVERTED_BY_DEFAULT`` here:
https://github.com/zigpy/zha-device-handlers/blob/16f298e9ce293d241592f86573e909f9c32809d0/zhaquirks/tuya/__init__.py#L129-L137 or perhaps just modify the command sent:
https://github.com/zigpy/zha-device-handlers/blob/16f298e9ce293d241592f86573e909f9c32809d0/zhaquirks/tuya/__init__.py#L106

jameshirka, 2022-04-18:
Thanks @TheJulianJES for the push.
I ended up needing to do both changes to get my blinds to work.
For some reason, I never pushed them back as a pull request.

I'll do a pull request of my fix and hope they don't break the blinds for other users!

blauret, 2022-05-09:
@jameshirka , @dmulcahey,

I would argue the change merged in #1510 is actually a regression and the bug was not there in the first place. Depending on how you actually fit the cover, it will dictate the required command to open or close the cover. Tuya has made some provision for that and allows to invert the orientation via the invert command. I don't think it's linked to variations in the way the Tuya devices are implemented. 

The way to logically invert the device is to send send it an invert command. In order to do that in developperTools->Services, you can send:

invert

```yaml
service: zha.issue_zigbee_cluster_command
data:
  ieee: 'xx:xx:xx:xx:xx:xx:xx:xx'
  endpoint_id: 1
  cluster_id: 258
  cluster_type: in
  command_type: server
  command: 6
  args:
    - 0
    - 0
    - 1029
    - 0
    - [1, 1]
```

stop invert

```yaml
service: zha.issue_zigbee_cluster_command
data:
  ieee: 'xx:xx:xx:xx:xx:xx:xx:xx'
  endpoint_id: 1
  cluster_id: 258
  cluster_type: in
  command_type: server
  command: 6
  args:
    - 0
    - 0
    - 1029
    - 0
    - [1, 0]
```

If we start to modify the code every time someone has their cover inverted it will be a never ending story. 

Now maybe some effort should be done to make it easier to invert the device sense. I don't know where this information should be. Maybe there should some work done in the UI to allow to control that.

Now I wouldn't like to change the implementation again because I went trough the pain to invert my covers to match the latest release. That would be unfortunate to have to change it again.

MattWestb, 2022-05-09:
Its possible implanting one or 2 switches in ZHA GUI like we have made with tuya TRVS but i think its also need changing the key mapping for up and down that is being little more tricky for getting it nice in the GUI then we can describe the switches from the quirk side.
One version is saving the attribute reversed and if its revered the swapping the up and down keys and if its not reversed its using the "original" key mapping but its must being code adding for doing that logic in the quirk handling.

blauret, 2022-05-09:
@MattWestb ,if you can add an invert switch entity in the UI it would be perfect. You don't need to invert the commands in the quirk. I believe the TS0601 will report whether they're inverted or not. Then you can send the appropriate command when switch is toggle. 

I can try to have a look, but I don't have much time for that those days.

jameshirka, 2022-05-09:
@blauret great find.
Weird that I have blinds installed to roll from both sides and yet both were inverted for me.
I tried setting the invert by setting the cover_inverted cluster to no avail and seeing @SweAdelpha having the same issue, I just inverted the controls.

I'll have a look and see what I can do about adding the inverted cluster command and an inverted switch when I have a few hours.

sushant40, 2022-05-10:
@MattWestb @blauret ... I agree - it would be nice to have a invert switch on the HA side to invert blinds.
In the interim i think we should revert changes from PR:... https://github.com/zigpy/zha-device-handlers/pull/1510
I can create the pr in a few hours if necssary.

The reason being - when I add the blind to tuya using their official app it works correctly. The command being sent from ZHA now is inverted and all my blinds and scenes are broken! (probably for other people too). @jameshirka hopefully you don't mind if we do this?

jameshirka, 2022-05-10:
@sushant40 happy to roll it back now that @blauret has a workaround.
Weird that yours work in the app correctly as mine also work in the app correctly but were inverted in HA.
We can then look at an invert switch in the GUI to cover this scenario for others.

sushant40, 2022-05-10:
Cool - i have created https://github.com/zigpy/zha-device-handlers/pull/1555 to revert the change.

almooj, 2022-05-11:
same issue here, everything was fine before

sushant40, 2022-05-16:
The PR to revert the change hasnt been approved? Does anyone know how to nudge it along?

sushant-is-here, 2022-11-12:
Resolved.

sushant-is-here, 2022-11-12:
👍

closed 2023-07-10 by github-actions[bot]

mtwo, 2023-10-04:
FYI for people who end up here after a Google search like I did: inverting the motor via the remote control also fixed this in HA for my Zemismart ZM79-ZB units, meaning that I didn't have to adjust any quirks or send any Zigbee commands. To do this, press the remote set button (hidden behind the battery cover) four times and then press up / open on the remote. You can repeat this process to un-invert the controls as well.

blauret, 2023-10-05:
> FYI for people who end up here after a Google search like I did: inverting the motor via the remote control also fixed this in HA for my Zemismart ZM79-ZB units, meaning that I didn't have to adjust any quirks or send any Zigbee commands. I suspect that this is the same for the TZE200 units. To do this, press the remote set button (hidden behind the battery cover) four times and then press up / open on the remote. You can repeat this process to un-invert the controls as well.

That's also an option, if you have the RF remote control!!!