zigpy/zha-device-handlers #1320: [BUG] Aqara E1 not updating sensor state (EZSP)

agners, 2022-02-01:
**Describe the bug**
After successfully paring the device, the main state (`ias_zone`) doesn't receive updates. Even with debug logs enabled, there seem to be no traffic from the device. Also the batter state seems not to be represented.

Tested with two devices of the same type.

Device Support Request: #1088

**To Reproduce**
Steps to reproduce the behavior:
1. Pair Aqara E1 MCCGQ14LM
2. Observe no battery/no update

**Expected behavior**
Working state updates/battery status.

**Screenshots**
[image]

**Additional context**

- Home Assistant Core 2022.02.b5
  - bellows 0.29.0
  - zha-quirks 0.0.66
- EZSP Radio with EmberZNet version: 6.10.3.0 build 297

Logs during pairing:

<details>

```
2022-02-01 23:10:32 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2022-02-01 23:10:32 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'70f421a9a52af8887e'
2022-02-01 23:10:32 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'00f4a1a9a52a41b200940824e7559349964e24abeece338bc8c66489ff7e3fa7eacddf6f8fffc6dbd7d2698c4623a9ec763ba5ea758241984c2613b1e070381c0e07bbe5ca658a459a4d9e4f9ff7c3d9d46a35a25190482487137e'
2022-02-01 23:10:32 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2022-02-01 23:10:32 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 241 (readCounters) received: b'5400590042014d0001000a0003000300540035000700030000000100010000000100020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'
2022-02-01 23:10:32 DEBUG (MainThread) [bellows.ezsp.protocol] Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2022-02-01 23:10:32 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'01f521a9fe2a16437f7e'
2022-02-01 23:10:32 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'11f5a1a9fe2a15b3c2c4e07e'
2022-02-01 23:10:32 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2022-02-01 23:10:32 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 170 (getValue) received: b'00019b'
2022-02-01 23:10:32 DEBUG (MainThread) [bellows.zigbee.application] Free buffers status EzspStatus.SUCCESS, value: 155
2022-02-01 23:10:32 DEBUG (MainThread) [bellows.zigbee.application] ezsp_counters: [MAC_RX_BROADCAST = 976, MAC_TX_BROADCAST = 957, MAC_RX_UNICAST = 2077, MAC_TX_UNICAST_SUCCESS = 512, MAC_TX_UNICAST_RETRY = 43, MAC_TX_UNICAST_FAILED = 42, APS_DATA_RX_BROADCAST = 26, APS_DATA_TX_BROADCAST = 26, APS_DATA_RX_UNICAST = 323, APS_DATA_TX_UNICAST_SUCCESS = 183, APS_DATA_TX_UNICAST_RETRY = 30, APS_DATA_TX_UNICAST_FAILED = 9, ROUTE_DISCOVERY_INITIATED = 8, NEIGHBOR_ADDED = 12, NEIGHBOR_REMOVED = 11, NEIGHBOR_STALE = 0, JOIN_INDICATION = 13, CHILD_REMOVED = 10, ASH_OVERFLOW_ERROR = 0, ASH_FRAMING_ERROR = 0, ASH_OVERRUN_ERROR = 0, NWK_FRAME_COUNTER_FAILURE = 0, APS_FRAME_COUNTER_FAILURE = 0, UTILITY = 0, APS_LINK_KEY_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 0, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 155]
2022-02-01 23:10:42 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2022-02-01 23:10:42 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'12fa21a9a52ae3787e'
2022-02-01 23:10:42 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'22faa1a9a52a40b203940824e7559349964e24abeece338bc8c66489ff7e3fa7eacddf6f8fffc6dbd7d2698c4623a9ec763ba5ea758241984c2613b1e070381c0e07bbe5ca658a459a4d9e4f9ff7c3d9d46a35a2519048242ade7e'
2022-02-01 23:10:42 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2022-02-01 23:10:42 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 241 (readCounters) received: b'55005a0042014d0001000a0003000300540035000700030000000100010000000100020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'
2022-02-01 23:10:42 DEBUG (MainThread) [bellows.ezsp.protocol] Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2022-02-01 23:10:42 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'23fb21a9fe2a16cdc97e'
2022-02-01 23:10:42 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'33fba1a9fe2a15b3c23b6a7e'
2022-02-01 23:10:42 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2022-02-01 23:10:42 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 170 (getValue) received: b'00019b'
2022-02-01 23:10:42 DEBUG (MainThread) [bellows.zigbee.application] Free buffers status EzspStatus.SUCCESS, value: 155
2022-02-01 23:10:42 DEBUG (MainThread) [bellows.zigbee.application] ezsp_counters: [MAC_RX_BROADCAST = 977, MAC_TX_BROADCAST = 958, MAC_RX_UNICAST = 2077, MAC_TX_UNICAST_SUCCESS = 512, MAC_TX_UNICAST_RETRY = 43, MAC_TX_UNICAST_FAILED = 42, APS_DATA_RX_BROADCAST = 26, APS_DATA_TX_BROADCAST = 26, APS_DATA_RX_UNICAST = 323, APS_DATA_TX_UNICAST_SUCCESS = 183, APS_DATA_TX_UNICAST_RETRY = 30, APS_DATA_TX_UNICAST_FAILED = 9, ROUTE_DISCOVERY_INITIATED = 8, NEIGHBOR_ADDED = 12, NEIGHBOR_REMOVED = 11, NEIGHBOR_STALE = 0, JOIN_INDICATION = 13, CHILD_REMOVED = 10, ASH_OVERFLOW_ERROR = 0, ASH_FRAMING_ERROR = 0, ASH_OVERRUN_ERROR = 0, NWK_FRAME_COUNTER_FAILURE = 0, APS_FRAME_COUNTER_FAILURE = 0, UTILITY = 0, APS_LINK_KEY_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 0, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 155]
2022-02-01 23:10:45 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'43fbb1a9702a817ab2a46325ba117d1d9d4e27ab29e67e'
2022-02-01 23:10:45 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2022-02-01 23:10:45 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 36 (trustCenterJoinHandler) received: b'94c8eb3029001044ef5401000000'
2022-02-01 23:10:45 DEBUG (MainThread) [bellows.zigbee.application] Received trustCenterJoinHandler frame with [0xc894, 54:ef:44:10:00:29:30:eb, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x0000]
2022-02-01 23:10:45 INFO (MainThread) [zigpy.application] New device 0xc894 (54:ef:44:10:00:29:30:eb) joined the network
2022-02-01 23:10:45 DEBUG (MainThread) [zigpy.device] [0xc894] Scheduling initialization
2022-02-01 23:10:45 DEBUG (MainThread) [bellows.ezsp.protocol] Send command findKeyTableEntry: (54:ef:44:10:00:29:30:eb, True)
2022-02-01 23:10:45 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2022-02-01 23:10:45 INFO (MainThread) [zigpy.device] [0xc894] Requesting 'Node Descriptor'
2022-02-01 23:10:45 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2022-02-01 23:10:45 DEBUG (MainThread) [zigpy.device] [0xc894] Extending timeout for 0x9d request
2022-02-01 23:10:45 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 54:ef:44:10:00:29:30:eb/0xc894
2022-02-01 23:10:45 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setExtendedTimeout: (54:ef:44:10:00:29:30:eb, True)
2022-02-01 23:10:45 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'35f821a9212afe8270945a614501935cb87e'
2022-02-01 23:10:45 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'54f8a1a9212aeadbea7e'
2022-02-01 23:10:45 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
[... 570 lines cut ...]
    return convert_to_entity_category(value, raise_report=raise_report)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 214, in convert_to_entity_category
    report(
  File "/usr/src/homeassistant/homeassistant/helpers/frame.py", line 74, in report
    _LOGGER.warning(msg, stack_info=True)
2022-02-01 23:10:52 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zha entity: sensor.lumi_lumi_magnet_acn001_eb302900_basic_lqi
2022-02-01 23:10:52 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event entity_registry_updated[L]: action=create, entity_id=sensor.lumi_lumi_magnet_acn001_eb302900_basic_lqi>
[... cut ...]  255, 255, b'\x18;\n\x00\x00\x10\x00']
2022-02-01 23:10:57 DEBUG (MainThread) [zigpy.zcl] [0x5e77:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=59 command_id=Command.Report_Attributes>
2022-02-01 23:10:57 DEBUG (MainThread) [zigpy.zcl] [0x5e77:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
```
</details>

Adminiuga, 2022-02-02:
The new aqara devices are requiring to see a specific "manufacturer id" of coordinator, when joining the network. During `zigbeed` crashing troubleshooting, I asked you to monkey patch `bellows` and clear a dict in `bellows/zigbee/application.py`. Did you undo that monkey patch?

agners, 2022-02-02:
:see_no_evil: :see_no_evil: I did not! After reverting the changes the E1 started to pair wonderfully in my test instance.

However, I still had problems getting it to work on my production instance. The symptoms were very similar, not working sensor state, no power value. But also the sensor state switched to `Unavailable` pretty quickly. From the logs I found that the sensor would drop off the network shortly after I paired it:

```
2022-02-02 10:06:15 INFO (MainThread) [zigpy.application] Device 0xe405 (54:ef:44:10:00:25:e6:d1) left the network
```

@dmulcahey hinted that it could be that my network got formed before the "hashed TC link key" change. After making a backup using `bellows` and restoring with `--upgrade-to-hashed-tc-link-key` option, that got solved too! Now the sensor works on both instances! :tada: 

Btw, some tips and learnings from pairing the sensor quite often:
- Pressing the reset button ~8s until 3 short blinks resets it and makes it repair
- A short press keeps it awake. That may help getting it Interviewed/Configured correctly.
- When removing the device from the network, a short press also makes sure that it properly gets removed from the network (the device will receive the remove command). After that the device is in pairing mode!
- Make sure to set the device to pairing mode before setting the controller into join mode via "Add device": Reason: Sometimes it's hard to tell if your device is already in pairing mode. When you reset it using long press, the interview/configure process might get interrupted.

Closing this, as both instances turned out to be problem with my local instance.

closed 2022-02-02 by agners