zigpy/zha-device-handlers #1322: [Device Support Request] TS004F Tuya Smart Knob

Tropaion, 2022-02-02:
I recently got this Tuya Smart Knob: [Ali](https://de.aliexpress.com/item/1005003380560022.html)

Pairing works and the Quirks for TS004F is applied, but only the battery cluster works, others don't appear.
Here is the pairing log:
```
Device 0x27a3 (54:0f:57:ff:fe:2e:3c:5f) joined the network
Device 54:0f:57:ff:fe:2e:3c:5f changed id (0x27a3 => 0x27a3)
[0x27a3] Scheduling initialization
Tries remaining: 3
[0x27a3] Requesting 'Node Descriptor'
Tries remaining: 2
[0x27a3] Extending timeout for 0x2e request
Device 0x27a3 (54:0f:57:ff:fe:2e:3c:5f) joined the network
[0x27a3] Scheduling initialization
[0x27a3] Canceling old initialize call
Tries remaining: 3
[0x27a3] Requesting 'Node Descriptor'
Tries remaining: 2
[0x27a3] Extending timeout for 0x2f request
Device 0x27a3 (54:0f:57:ff:fe:2e:3c:5f) joined the network
[0x27a3] Scheduling initialization
[0x27a3] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x27A3 ieee=54:0f:57:ff:fe:2e:3c:5f is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Device_annce: b'\xFF\xA3\x27\x5F\x3C\x2E\xFE\xFF\x57\x0F\x54\x80'
[0x27a3:zdo] ZDO request ZDOCmd.Device_annce: [0x27A3, 54:0f:57:ff:fe:2e:3c:5f, 128]
Tries remaining: 3
[0x27a3] Requesting 'Node Descriptor'
Tries remaining: 2
[0x27a3] Extending timeout for 0x30 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x27A3 ieee=54:0f:57:ff:fe:2e:3c:5f is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Node_Desc_rsp: b'\x30\x00\xA3\x27\x02\x40\x80\x02\x10\x52\x52\x00\x00\x2C\x52\x00\x00'
[0x27a3] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x27a3] Discovering endpoints
Tries remaining: 3
[0x27a3] Extending timeout for 0x31 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x27A3 ieee=54:0f:57:ff:fe:2e:3c:5f is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Active_EP_rsp: b'\x31\x00\xA3\x27\x01\x01'
[0x27a3] Discovered endpoints: [1]
[0x27a3] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x27a3:1] Discovering endpoint information
Tries remaining: 3
[0x27a3] Extending timeout for 0x32 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x27A3 ieee=54:0f:57:ff:fe:2e:3c:5f is_initialized=False> from ep 0 to ep 0, cluster ZDOCmd.Simple_Desc_rsp: b'\x32\x00\xA3\x27\x24\x01\x04\x01\x04\x01\x01\x06\x00\x00\x01\x00\x03\x00\x04\x00\x06\x00\x00\x10\x08\x19\x00\x0A\x00\x03\x00\x04\x00\x05\x00\x06\x00\x08\x00\x00\x10'
[0x27a3:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=260, device_version=1, input_clusters=[0, 1, 3, 4, 6, 4096], output_clusters=[25, 10, 3, 4, 5, 6, 8, 4096])
[0x27a3] Extending timeout for 0x33 request
[0x27a3:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=51 command_id=Command.Read_Attributes_rsp>
[0x27a3] Read model 'TS004F' and manufacturer '_TZ3000_4fjiwweb' from <Endpoint id=1 in=[basic:0x0000, power:0x0001, identify:0x0003, groups:0x0004, on_off:0x0006, lightlink:0x1000] out=[ota:0x0019, time:0x000A, identify:0x0003, groups:0x0004, scenes:0x0005, on_off:0x0006, level:0x0008, lightlink:0x1000] status=<Status.ZDO_INIT: 1>>
[0x27a3] Discovered basic device information for <Device model='TS004F' manuf='_TZ3000_4fjiwweb' nwk=0x27A3 ieee=54:0f:57:ff:fe:2e:3c:5f is_initialized=True>
Device is initialized <Device model='TS004F' manuf='_TZ3000_4fjiwweb' nwk=0x27A3 ieee=54:0f:57:ff:fe:2e:3c:5f is_initialized=True>
Checking quirks for _TZ3000_4fjiwweb TS004F (54:0f:57:ff:fe:2e:3c:5f)
Considering <class 'zhaquirks.tuya.ts004f.TuyaSmartRemote004F'>
Found custom device replacement for 54:0f:57:ff:fe:2e:3c:5f: <class 'zhaquirks.tuya.ts004f.TuyaSmartRemote004F'>
device - 0x27A3:54:0f:57:ff:fe:2e:3c:5f entering async_device_initialized - is_new_join: True
device - 0x27A3:54:0f:57:ff:fe:2e:3c:5f has joined the ZHA zigbee network
[0x27A3](TS004F): started configuration
[0x27A3:ZDO](TS004F): 'async_configure' stage succeeded
[0x27a3] Extending timeout for 0x34 request
[0x27a3] Extending timeout for 0x35 request
[0x27A3:1:0x0000]: finished channel configuration
[0x27a3] Extending timeout for 0x36 request
Error handling '_save_attribute' event with (54:0f:57:ff:fe:2e:3c:5f, 1, 0, 4, '_TZ3000_4fjiwweb') params: FOREIGN KEY constraint failed
Error handling '_save_attribute' event with (54:0f:57:ff:fe:2e:3c:5f, 1, 0, 5, 'TS004F') params: FOREIGN KEY constraint failed
[0x27a3:1:0x1000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=53 command_id=Command.Default_Response>
[0x27a3] Extending timeout for 0x37 request
[0x27A3:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x27a3] Extending timeout for 0x38 request
[0x27A3:1:0x0008]: bound 'level' cluster: Status.SUCCESS
[0x27A3:1:0x0008]: finished channel configuration
[0x27A3:1:0x0019]: finished channel configuration
[0x27a3] Extending timeout for 0x39 request
[0x27A3:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x27A3:1:0x0006]: finished channel configuration
[0x27A3:1:0x0005]: bound 'scenes' cluster: Status.SUCCESS
[... 21 lines cut ...]
[0x27A3:1:0x1000]: finished channel initialization
[0x27A3:1:0x0000]: initializing channel: from_cache: False
[0x27A3:1:0x0000]: finished channel initialization
[0x27A3:1:0x0008]: initializing channel: from_cache: False
[0x27A3:1:0x0008]: finished channel initialization
[0x27A3:1:0x0006]: initializing channel: from_cache: False
[0x27A3:1:0x0006]: finished channel initialization
[0x27A3:1:0x0019]: initializing channel: from_cache: False
[0x27A3:1:0x0019]: finished channel initialization
[0x27A3:1:0x0005]: initializing channel: from_cache: False
[0x27A3:1:0x0005]: finished channel initialization
[0x27a3:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=59 command_id=Command.Read_Attributes_rsp>
[0x27a3] Extending timeout for 0x3c request
[0x27a3:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=60 command_id=Command.Read_Attributes_rsp>
[0x27A3:1:0x0001]: finished channel initialization
[0x27A3:1:0x0001]: 'async_initialize' stage succeeded
[0x27A3:1:0x1000]: 'async_initialize' stage succeeded
[0x27A3:1:0x0000]: 'async_initialize' stage succeeded
[0x27A3:1:0x0008]: 'async_initialize' stage succeeded
[0x27A3:1:0x0006]: 'async_initialize' stage succeeded
[0x27A3:1:0x0019]: 'async_initialize' stage succeeded
[0x27A3:1:0x0005]: 'async_initialize' stage succeeded
[0x27A3](TS004F): power source: Battery or Unknown
[0x27A3](TS004F): completed initialization
```

MattWestb, 2022-02-02:
You is getting a l'impossible devices tuya have making !!!
I have "original" TS004F that i cant getting working with different modes but i hope i can getting up and running with little help.
One user have making on issue but hi was not getting all log needed for getting it working. https://github.com/zigpy/zha-device-handlers/issues/1268

Its have 2 modes and if the quirk is loaded (the updated in dev that is coming it HA 2022.02) you shall being possible switching it between command (dimmer switch) and event (scene switch) also by triple clicking the button.

The command mode shall working OK but i have not getting report if its doing that or not.
Use development tools and events and putting in `zha_event` in listening to and clicking start listening.
If all working OK in dimmer mode you shall getting all commands the original TS004F is sending = On, Off, Dimm up and down and stop dimming.
I need good logs from the color changing so we can implanting it 2 with devices automation in the quirk.
The scene mode is sending no standard command for press, double and long press and i have testing and and getting it working with my switch but not good. But this device have one extra attribute for rotating that i have implantation and testing (by renaming the attribute for my device) and can getting OK data in ZHA.

The best is making one mod working OK before mixing all up and making one large mess of it.

Can you getting logs then having the switch in command mode and trying if the standard commands is working and also logs from color changing (pressing and rotating left and right) ?

For doing changes for the event mode is needed updating files in the HA container (adding the extra attributes) and i have it not finished then have not getting feedback if its working or not but i think its better waiting for the update of HA then its deleting all changes in the container.

I think the last error `Status.MAC_NO_ACK` is the device sleeping and have not replacing on the request.

MattWestb, 2022-02-02:
By the way in ZHA you shall only getting one battery sensor then the "switch part" is not having one state lik lights is having (On/Off) and is sending events that can being used for triggering in automations.

MattWestb, 2022-02-02:
Update: The last errors if from touch link cluster and old ZLL device was having one random group after reset but Zigbee 3 devices is not having it but can being bonded to one group but our devices is tuya so they is doing it by adding groups like lights and i have adding it in the last update of the quirk that is being in HA 2022.02.

I only waiting you is getting one LIDL remote that is having the same mess but noone have getting it working 100% ;-))

Tropaion, 2022-02-02:
In one mode I only get errors:
```
Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zigpy/zcl/__init__.py:114
First occurred: 19:43:11 (43 occurrences)
Last logged: 19:45:39

[0x27a3:1:0x0006] Unknown cluster-specific command 253
[0x27a3:1:0x0006] Unknown cluster-specific command 252
```
In the other mode I get these zha_events but I 

Inrease brightness by one:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:0f:57:ff:fe:2e:3c:5f",
        "unique_id": "54:0f:57:ff:fe:2e:3c:5f:1:0x0008",
        "device_id": "3cc97f6cd8c336b5e52a6070f17fd3e8",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            0,
            25,
            1
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-02T18:38:45.080837+00:00",
    "context": {
        "id": "0fb7ab9b2ef9a42f4f4450ef97e7b785",
        "parent_id": null,
        "user_id": null
    }
}
```
Decrase brightness by one:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:0f:57:ff:fe:2e:3c:5f",
        "unique_id": "54:0f:57:ff:fe:2e:3c:5f:1:0x0008",
        "device_id": "3cc97f6cd8c336b5e52a6070f17fd3e8",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            1,
            13,
            1
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-02T18:40:33.233880+00:00",
    "context": {
        "id": "b2d81f8d78fa063fa3e6df50da9ce91c",
        "parent_id": null,
        "user_id": null
    }
}
```
Push button 1x:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:0f:57:ff:fe:2e:3c:5f",
        "unique_id": "54:0f:57:ff:fe:2e:3c:5f:1:0x0006",
        "device_id": "3cc97f6cd8c336b5e52a6070f17fd3e8",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-02T18:41:13.246422+00:00",
    "context": {
        "id": "a21af6d3c36c78d28ddd66b90f796e20",
        "parent_id": null,
        "user_id": null
    }
}
```

The only events I get are "step" and "toggle, "long press" and "color temperature" don't arrive.
Should I again do some sniffing? :D

MattWestb, 2022-02-02:
The toggle is new the old is sending on and off.
The dimm is also looks OK but is using different parameters so need adding then also in the device automation in the quirk.
Perhaps is the  device not sending the color changing if not have configuring it OK with one tuya light (likely its needs touch link pairing for doing that or some magic from the tuya ZBGW).
Can you looking in the log if you is getting some undefined commands from the device the doing the color changes ?

For "decoding" the "scene" commands you can putting in your local quirk folder (after taking away the .txt) and restarting HA. Its only one hack and is not doing its work nice but you shall getting events then being in scene mode but not the rotating (its using one extra attribute)
[ts004f3.py.txt](https://github.com/zigpy/zha-device-handlers/files/7989432/ts004f3.py.txt)
.

Tropaion, 2022-02-02:
Log when trying to change temperature:
```
2022-02-02 20:23:39 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=768, SrcAddr=0x27A3, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=47, SecurityUse=<Bool.false: 0>, TimeStamp=14099745, TSN=0, Data=b'\x01\x35\x4C\x01\x78\x00\x03\x00\x99\x00\xF4\x01', MacSrcAddr=0x27A3, MsgResultRadius=11)
2022-02-02 20:23:39 DEBUG (MainThread) [zigpy.device] Ignoring message (b'01354c01780003009900f401') on cluster 768: unknown endpoint or cluster id: 'No cluster ID 0x0300 on (54:0f:57:ff:fe:2e:3c:5f, 1)'
2022-02-02 20:23:39 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=768, SrcAddr=0x27A3, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=51, SecurityUse=<Bool.false: 0>, TimeStamp=14122257, TSN=0, Data=b'\x01\x36\x4C\x01\x12\x00\x01\x00\x99\x00\xF4\x01', MacSrcAddr=0x27A3, MsgResultRadius=11)
2022-02-02 20:23:39 DEBUG (MainThread) [zigpy.device] Ignoring message (b'01364c01120001009900f401') on cluster 768: unknown endpoint or cluster id: 'No cluster ID 0x0300 on (54:0f:57:ff:fe:2e:3c:5f, 1)'
```
Log when long press:
```
022-02-02 20:25:45 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=768, SrcAddr=0x27A3, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=25, SecurityUse=<Bool.false: 0>, TimeStamp=6235135, TSN=0, Data=b'\x01\x3F\x04\x01\xC8', MacSrcAddr=0x27A3, MsgResultRadius=11)
2022-02-02 20:25:45 DEBUG (MainThread) [zigpy.device] Ignoring message (b'013f0401c8') on cluster 768: unknown endpoint or cluster id: 'No cluster ID 0x0300 on (54:0f:57:ff:fe:2e:3c:5f, 1)'
2022-02-02 20:25:45 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=768, SrcAddr=0x27A3, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=36, SecurityUse=<Bool.false: 0>, TimeStamp=6258034, TSN=0, Data=b'\x01\x40\x47', MacSrcAddr=0x27A3, MsgResultRadius=11)
2022-02-02 20:25:45 DEBUG (MainThread) [zigpy.device] Ignoring message (b'014047') on cluster 768: unknown endpoint or cluster id: 'No cluster ID 0x0300 on (54:0f:57:ff:fe:2e:3c:5f, 1)'
2022-02-02 20:25:46 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=768, SrcAddr=0x27A3, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=29, SecurityUse=<Bool.false: 0>, TimeStamp=6297664, TSN=0, Data=b'\x01\x41\x01\x01\x0F', MacSrcAddr=0x27A3, MsgResultRadius=11)
2022-02-02 20:25:46 DEBUG (MainThread) [zigpy.device] Ignoring message (b'014101010f') on cluster 768: unknown endpoint or cluster id: 'No cluster ID 0x0300 on (54:0f:57:ff:fe:2e:3c:5f, 1)'
```
Log normal press:
```
2022-02-02 20:26:14 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=768, SrcAddr=0x27A3, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=36, SecurityUse=<Bool.false: 0>, TimeStamp=8071241, TSN=0, Data=b'\x01\x42\x4C\x01\x12\x00\x01\x00\x99\x00\xF4\x01', MacSrcAddr=0x27A3, MsgResultRadius=11)
2022-02-02 20:26:14 DEBUG (MainThread) [zigpy.device] Ignoring message (b'01424c01120001009900f401') on cluster 768: unknown endpoint or cluster id: 'No cluster ID 0x0300 on (54:0f:57:ff:fe:2e:3c:5f, 1)'
```
Log changing brightness:
```
2022-02-02 20:26:52 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=8, SrcAddr=0x27A3, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=40, SecurityUse=<Bool.false: 0>, TimeStamp=10449481, TSN=0, Data=b'\x01\x43\x02\x01\x3D\x02\x00', MacSrcAddr=0x27A3, MsgResultRadius=11)
2022-02-02 20:26:52 DEBUG (MainThread) [zigpy.zcl] [0x27a3:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=67 command_id=2>
2022-02-02 20:26:52 DEBUG (MainThread) [zigpy.zcl] [0x27a3:1:0x0008] ZCL request 0x0002: [1, 61, 2]
2022-02-02 20:26:52 DEBUG (MainThread) [zigpy.zcl] [0x27a3:1:0x0008] No handler for cluster command 2
2022-02-02 20:26:52 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=54:0f:57:ff:fe:2e:3c:5f, unique_id=54:0f:57:ff:fe:2e:3c:5f:1:0x0008, device_id=3cc97f6cd8c336b5e52a6070f17fd3e8, endpoint_id=1, cluster_id=8, command=step, args=[1, 61, 2]>
```

MattWestb, 2022-02-02:
With the updated quirk you is also having group as in cluster = you is seeing it in ZHA grupes and can adding it to groups like other lights (i think its only using the last added group).
If adding the remotes "light" to one group and waking up the remote so it can receiving it it shall sening toggle and dimmering also then HA is offline.

Then have updating the device automation with toggle and the new dimmer parameters it shall being possible using add automations from the device card and also from automation dialogue.

Tropaion, 2022-02-02:
@MattWestb My receiving device is not a zigbee device in my case.
But I still have to get it work first, as seen in the log there are many unknowns :D

MattWestb, 2022-02-02:
You can always triggering one automation on the events the device is sending bt you maust doing it by hand and not with the template. Fixing  the template in the device automation is not one big thing.

I was adding color cluster in the last quirk and hope you is getting real command then.
rename the old quirk so they the system is not using both and restart HA and testing the color commands.
[ts004f3B.py.txt](https://github.com/zigpy/zha-device-handlers/files/7989517/ts004f3B.py.txt)

Tropaion, 2022-02-02:
It is loading but I get many weird errors after boot:
```
  2022-02-02 20:42:07 ERROR (MainThread) [frontend.js.latest.202112291] https://home.mydomain.at/frontend_latest/17dc2469.js:246:519 Uncaught TypeError: Cannot read properties of undefined (reading 'get')
  2022-02-02 20:42:09 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x0006]: Failed to bind 'TS004X_cluster' cluster: 
  2022-02-02 20:42:09 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x0006]: finished channel configuration
  2022-02-02 20:42:09 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=system_log, service=write, service_data=logger=frontend.js.latest.202112291, message=https://home.mydomain.at/frontend_latest/17dc2469.js:246:519 Uncaught TypeError: Cannot read properties of undefined (reading 'get')>
  2022-02-02 20:42:09 ERROR (MainThread) [frontend.js.latest.202112291] https://home.mydomain.at/frontend_latest/17dc2469.js:246:519 Uncaught TypeError: Cannot read properties of undefined (reading 'get')
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x0005]: Failed to bind 'scenes' cluster: 
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x0005]: finished channel configuration
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x0000]: 'async_configure' stage succeeded
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x0001]: 'async_configure' stage succeeded
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x1000]: 'async_configure' stage succeeded
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x0008]: 'async_configure' stage succeeded
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x0006]: 'async_configure' stage succeeded
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x0019]: 'async_configure' stage succeeded
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x27A3:1:0x0005]: 'async_configure' stage succeeded
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x27A3](TS004F): completed configuration
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x27A3](TS004F): stored in registry: ZhaDeviceEntry(name='_TZ3000_4fjiwweb TS004F', ieee='54:0f:57:ff:fe:2e:3c:5f', last_seen=1643830272.64237)
  2022-02-02 20:42:18 DEBUG (MainThread) [zigpy.device] [0x27a3] Extending timeout for 0x0c request
  2022-02-02 20:42:18 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x27A3), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=3, TSN=12, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x01\x0C\x40\x02\x00')
  2022-02-02 20:42:18 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=system_log, service=write, service_data=logger=frontend.js.latest.202112291, message=https://home.mydomain.at/frontend_latest/17dc2469.js:246:519 Uncaught TypeError: Cannot read properties of undefined (reading 'get')>
  2022-02-02 20:42:18 ERROR (MainThread) [frontend.js.latest.202112291] https://home.mydomain.at/frontend_latest/17dc2469.js:246:519 Uncaught TypeError: Cannot read properties of undefined (reading 'get')
  2022-02-02 20:42:18 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
  2022-02-02 20:42:18 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.MAC_NO_ACK: 233>, Endpoint=1, TSN=12)
  2022-02-02 20:42:18 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.MAC_NO_ACK: 233>), retry attempt 1 of 5
  2022-02-02 20:42:20 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x27A3), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=3, TSN=12, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x01\x0C\x40\x02\x00')
  2022-02-02 20:42:20 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
  2022-02-02 20:42:20 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
  2022-02-02 20:42:20 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
  2022-02-02 20:42:20 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.MAC_NO_ACK: 233>, Endpoint=1, TSN=12)
  2022-02-02 20:42:20 DEBUG (MainThread) [zigpy_znp.zigbee.application] Request failed (Unsuccessful request status code: <Status.MAC_NO_ACK: 233>), retry attempt 2 of 5
  2022-02-02 20:42:20 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestSrcRtg.Req(DstAddr=0x27A3, DstEndpoint=1, SrcEndpoint=1, ClusterId=3, TSN=12, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, SourceRoute=[], Data=b'\x01\x0C\x40\x02\x00')
[... cut ...]
2022-02-02 20:47:06 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=768, SrcAddr=0x27A3, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=51, SecurityUse=<Bool.false: 0>, TimeStamp=8075940, TSN=0, Data=b'\x01\x64\x01\x01\x0F', MacSrcAddr=0x27A3, MsgResultRadius=11)
2022-02-02 20:47:06 DEBUG (MainThread) [zigpy.device] Ignoring message (b'016401010f') on cluster 768: unknown endpoint or cluster id: 'No cluster ID 0x0300 on (54:0f:57:ff:fe:2e:3c:5f, 1)'
```

MattWestb, 2022-02-02:
I was doing the inport OK but was pasting the wrong name in replace.
Test this version (same name as last)
[ts004f3B.py.txt](https://github.com/zigpy/zha-device-handlers/files/7989589/ts004f3B.py.txt)
:

Tropaion, 2022-02-02:
Color Temp works now:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:0f:57:ff:fe:2e:3c:5f",
        "unique_id": "54:0f:57:ff:fe:2e:3c:5f:1:0x0300",
        "device_id": "3cc97f6cd8c336b5e52a6070f17fd3e8",
        "endpoint_id": 1,
        "cluster_id": 768,
        "command": "step_color_temp",
        "args": [
            1,
            35,
            1,
            153,
            500
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-02T20:11:52.166788+00:00",
    "context": {
        "id": "e8264ac2c3d516074e72cdf7eb4bfe60",
        "parent_id": null,
        "user_id": null
    }
}
```
But long press is a bit weird, there are three different events happening:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:0f:57:ff:fe:2e:3c:5f",
        "unique_id": "54:0f:57:ff:fe:2e:3c:5f:1:0x0300",
        "device_id": "3cc97f6cd8c336b5e52a6070f17fd3e8",
        "endpoint_id": 1,
        "cluster_id": 768,
        "command": "move_hue",
        "args": [
            1,
            15
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-02T20:12:08.499514+00:00",
    "context": {
        "id": "e2eda71e6d78c27d9696ff36956b17b5",
        "parent_id": null,
        "user_id": null
    }
}
```
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:0f:57:ff:fe:2e:3c:5f",
        "unique_id": "54:0f:57:ff:fe:2e:3c:5f:1:0x0300",
        "device_id": "3cc97f6cd8c336b5e52a6070f17fd3e8",
        "endpoint_id": 1,
        "cluster_id": 768,
        "command": "move_saturation",
        "args": [
            1,
            200
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-02T20:12:22.225590+00:00",
    "context": {
        "id": "1df26be95a4ecb0b4ff03d3db75144e0",
        "parent_id": null,
        "user_id": null
    }
}
````
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:0f:57:ff:fe:2e:3c:5f",
        "unique_id": "54:0f:57:ff:fe:2e:3c:5f:1:0x0300",
        "device_id": "3cc97f6cd8c336b5e52a6070f17fd3e8",
        "endpoint_id": 1,
        "cluster_id": 768,
        "command": "stop_move_step",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-02T20:12:22.285341+00:00",
    "context": {
        "id": "343b3d0d1bac72160453226909cd1086",
        "parent_id": null,
        "user_id": null
    }
}
```

MattWestb, 2022-02-02:
Color command 0x0004 is` 0x0004: ("move_saturation", (t.uint8_t, t.uint8_t), False),  # mode, rate` 
0x0047 = `0x0047: ("stop_move_step", (), False),`
0x0001 = `0x0001: ("move_hue", (t.uint8_t, t.uint8_t), False),  # move mode, rate` 
so looks like normal Zigbee commands and its wath you is getting in in zha_events.
You finding all commands and attributes for color cluster here. https://github.com/zigpy/zigpy/blob/dev/zigpy/zcl/clusters/lighting.py

And the first one is also working OK
```
        0x004C: (
            "step_color_temp",
            (t.bitmap8, t.uint16_t, t.uint16_t, t.uint16_t, t.uint16_t),
            False,
        ),
``` 
I dont knowing the logic tuya is using but its described as stepping the color but device can have many different  color modes and using different commands as you can see.

If you can see the pattern they is using (like step_color_temp, step_color_temp and step_color_temp) its easy getting the automation working. 
If using different sequences every time then its not possible getting the triggering working OK.

I need some sleep but i looking more tomorrow.

Can you looking if the dimmer is sending the same data the turing the button or if its changing if doing little, medium or mutch ?
Then i can doing automations for it (by copy it from IKEA Symphonisk).

Tropaion, 2022-02-02:
@MattWestb Yeah, it works pretty fine, but I don't get why its sending these three command when I do a long press.
It would be nice if the changed triggers are exposed to the gui, now its "on" and "off" but this device only has "toggle".
And there are still three unhandled commands from Cluster 768: 1, 4, 71.
I don't know if its what you meant, but here are different values when turning the knob:
Turning it one click:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:0f:57:ff:fe:2e:3c:5f",
        "unique_id": "54:0f:57:ff:fe:2e:3c:5f:1:0x0008",
        "device_id": "3cc97f6cd8c336b5e52a6070f17fd3e8",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            0,
            13,
            1
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-02T21:04:26.123340+00:00",
    "context": {
        "id": "cf408b8f12d957ddfee4b76484168b8c",
        "parent_id": null,
        "user_id": null
    }
}
```
Turning it ~3 clicks:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:0f:57:ff:fe:2e:3c:5f",
        "unique_id": "54:0f:57:ff:fe:2e:3c:5f:1:0x0008",
        "device_id": "3cc97f6cd8c336b5e52a6070f17fd3e8",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            0,
            37,
            2
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-02T21:05:49.561157+00:00",
    "context": {
        "id": "3899f66a65ff99cf924335862ec06cc5",
        "parent_id": null,
        "user_id": null
    }
}
```
Turning many times:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "54:0f:57:ff:fe:2e:3c:5f",
        "unique_id": "54:0f:57:ff:fe:2e:3c:5f:1:0x0008",
        "device_id": "3cc97f6cd8c336b5e52a6070f17fd3e8",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            0,
            145,
            3
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-02T21:06:24.615029+00:00",
    "context": {
        "id": "63da78a81ea046001d864c0476136a6c",
        "parent_id": null,
        "user_id": null
    }
}
```

Tropaion, 2022-02-02:
@MattWestb I changed the automation triggers and toggling works now:
```
class Tuya4NewButtonTriggers:
    """Tuya 4-button New version remote device triggers."""

    device_automation_triggers = {
        (SHORT_PRESS, TURN_ON): {COMMAND: COMMAND_TOGGLE, CLUSTER_ID: 6, ENDPOINT_ID: 1},
        (SHORT_PRESS, DIM_UP): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 51, 10],
        },
        (LONG_PRESS, DIM_UP): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 51],
        },
        (SHORT_PRESS, DIM_DOWN): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 51, 10],
        },
        (LONG_PRESS, DIM_DOWN): {
            COMMAND: COMMAND_MOVE,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 51],
        },
        (LONG_RELEASE, DIM_DOWN): {
            COMMAND: COMMAND_STOP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
        },
    }
```
But I can't get the brightness controll running, but I found out the meaning of the three parameters:
```ARGS: [x1, x2, x3]```
x1 = Direction, 0 = clockwise, 1 = anticlockwise
x2 = Dimmvalue:
- 13 = 1 step
- 24 = 2 steps
- 37 = 3 steps
- 49 = 4 steps
- 61 = 5 steps
- 73 = 6 steps
- 85 = 7 steps
x3 = rotation speed, 1 = slow, 2 = middle, 3 = fast

MattWestb, 2022-02-03:
Servus !
### First great logging and testing !!!
I think we need doing more device classes then we is getting many not working device automations if implanting all in the same class.
For testing its OK making all crazy things but in the end its better getting it structured and good working for the end user.

For the dimmering it shall being "translated" to rotate like have being done with the IKEA rotator:
https://github.com/zigpy/zha-device-handlers/blob/23a166902e50d5c32019e0d4d32a968715041bb6/zhaquirks/ikea/symfonisk.py#L96-L113
The question is shall only triggering on one step or manny and also the speed ?
I thinking making one automation for each combination is little overkill.
I is more or less copy and paste then coding sometime little more advance but i think you have more knowledge / experience of working with coding.
Perhaps Right / Left. Slow / medium and fast and trying getting the steps as parameter in the automation if its possible. What is the maximum of steps the device is sending before it stop or start from beginning ?

The device is sending color changes for more modes because its not knowing witch mode the lights is controlling is supporting.
"move_to_hue", "move_saturation" and "step_color_temp".
I wondering if "stop_move_step" is one stop command the is being sent then releasing the button because hue and saturation is relative to what the light is having setted for the moment (thinking = not knowing) and the Color Temperature (CT) is sending "step_color_temp" and can needoing on stop command.
Its not so important now but good looking at it later.
I think we shall only using one of the 3 color command sent and i think the CT is best if we can getting the stop command as one release after long press.

I chall testing making more device classed and see if i can getting your device being triggered with `MODELS_INFO:` before its loading the "standard" MODEL: "TS004F" so we can moving the device automations to the device class and not getting one mess with working and not working ones for all devices.
The added color cluster is not making any harm with my device then its not  sending any command on that cluster but i like its more clean and not one mess of all.

Tropaion, 2022-02-03:
@MattWestb It's difficult to find out by rotating so fast, but what I have seen the ```Max. Value = 253``` which would make sense, assuming that it is using a 8-bit register.
I think the speed parameter is not that important for use, but if you can implement it, why not? :D
BTW are you making a new class or should I try? You have more knowledge about this device.

MattWestb, 2022-02-03:
I have making 2 new classes and moving the device automation into the device class so its only working for the device and not all.
The "normal is last and is being loaded if no of the first is not matching (i have testing 5 times and its looks wrong OK) and first is your new class with the ID (and without my test clusters) of your device and in the middle is my test class.

I have adding the color cluster as import and also in your class and its being loaded OK in HA and is not crashing.

You need putting in your new device automations and doing the inport for toggle and other things you is needing for it.

Line 98 is for light mode commands and the outcommented line 99 is for deicing tuya scene commands that is sending no standard commands for its "scenes" (we is coming to it later)
[ts004f3B.py.txt](https://github.com/zigpy/zha-device-handlers/files/7995520/ts004f3B.py.txt)

Now we can doing changes and not braking the other device types :-)).

Tropaion, 2022-02-03:
@MattWestb Sorry, but I can't help with that, I never worked with the automation trigger and don't know how they work.
But isn't it possible the calculate with mathemical equation?
BTW is there a reason why you removed the toggle I posted?
And why does this device have two disabled entities?
[image]

MattWestb, 2022-02-03:
No i was making one "clean" quirk from dev and copy the last classes i have making in it and adding my changes but i dont knowing all changes you have making in the device automation so need adding them and the imports. Sorry for deleting but i was thinking better have one clean start and not my old test with much crap in.

The 2 disables ebterys is not from My its new functions like the first one (send identifying) its LQI and RSSI for the last received message. I haven't have time updating my 4 HA so i dont  have  them so i dont knowing if and how they is working.

Tropaion, 2022-02-03:
@MattWestb Here are my changes for trigger, short press and log press work with it: [ts004f3B.py.txt](https://github.com/zigpy/zha-device-handlers/files/7995940/ts004f3B.py.txt)
But I don't like that there are three different events when doing a long press.
Is it possible to just make one event with the command "LONG_PRESS"?

MattWestb, 2022-02-03:
Long press is the color changing i think ?
The device is sending 3 commands and we can changing that.
But we shall only using one for doing device automations and ignoring the other that  we dont need.
I have putting your automations in my quirk but  i need looking for the functions then they need being imported from some place or the tests is not working.

I have updating my production system to HA 2022.2 and need do little testing and then some food for dinner and also updating my 2 test setups for being in phase with you.

MattWestb, 2022-02-03:
I dont knowing if we have doing right with long press and press rotated.
Long press is "set color" and press rotate is "set the color temperature".
Can you looking if the long press is sending all Color command 0x0001, 0x0004, 0x0047 and 0x007C one time and not later. 
Also what is the device what is its sending by press and rotating left and right ?

Tropaion, 2022-02-03:
> In the device automations you must using names the system is knowing or its not working making automations with them.
> You is finding them here https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/const.py

It is working with string too, I already tested my implementation, but we can use button, looks better.

> "stop_move_step" `COMMAND_STEP_COLOR_TEMP = "step_color_temp"` if using the CT for triggering the command.
> Release from long press can being being `stop_move_step` but i think its not defined in the constans :-((

None of the three commands the long press sends are implemented, but it works with strings, the way I implemented it, too.
When we make a PR we can just add the new commands to const.py

MattWestb, 2022-02-03:
If dont finding any suitable then adding it in the PR.
I think the strings is making the PR failing the tests and can t being merged.
I have updated my test systems and with your device automations i must adding     `COMMAND_TOGGLE.` in import or ZHA is not starting.

Its looks like all device is getting the LQI and RSSI but they is disabled by default on all device i have looking on.

Tropaion, 2022-02-03:
@MattWestb Just add the three new constants to const.py and then import them and make for both files a PR.
But why are you already making a PR? The dimming function is still not working.

MattWestb, 2022-02-03:
I was looking how IKEA quirk is made and was defining it in the quirk the its was made so but later i can moving it to constant then is used of more quirks.
Also adding left and right and button.
```
from zhaquirks.const import (
    ARGS,
    BUTTON,
    CLUSTER_ID,
    COMMAND,
    COMMAND_MOVE,
    COMMAND_OFF,
    COMMAND_ON,
    COMMAND_STEP,
    COMMAND_STOP,
    COMMAND_TOGGLE,
    DEVICE_TYPE,
    DIM_DOWN,
    DIM_UP,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LEFT,
    LONG_PRESS,
    LONG_RELEASE,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    RIGHT,
    SHORT_PRESS,
    TURN_OFF,
    TURN_ON,
)
from zhaquirks.tuya import TuyaZBOnOffAttributeCluster, TuyaSmartRemoteOnOffCluster

ROTATED = "device_rotated"
```
Is the changed imports.

I was putting the new named in your automations and ZHA is starting OK.
```
    device_automation_triggers = {
        (SHORT_PRESS, BUTTON): {COMMAND: COMMAND_TOGGLE, CLUSTER_ID: 6, ENDPOINT_ID: 1},
        (LONG_PRESS, BUTTON): {COMMAND: "stop_move_step", CLUSTER_ID: 768, ENDPOINT_ID: 1},
        (ROTATED, RIGHT): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 13, 1],
        },
        (ROTATED, LEFT): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 13, 1],
        },
```
Its only triggering on one step left and right but is interesting is its working in real.

I only thinking not making  somethings that is working but is not possible getting thru the test then making on PR
I was adding "new" variables like ROTATED as variable and its easier making one first PR with all functions we like have and then moving the the constans to constans in one following up PR.

Back to DA is it looking OK that i have doing and is it working in real ?

Tropaion, 2022-02-03:
@MattWestb If you send me your current version I can test it.

> Back to DA is it looking OK that i have doing and is it working in real ?

What do you mean?

MattWestb, 2022-02-03:
Different way thinking how to making PRs is not important but better getting the Device Automations working OK.
I trying getting long press working with one of the 2 commands its sending.

Tropaion, 2022-02-03:
@MattWestb Long press is sending three commands and automation I did works, I even tested it with a lamp.

MattWestb, 2022-02-03:
Good.

Is the rotating right and left one step slow working ?

Tropaion, 2022-02-03:
@MattWestb right and left one step is not working correctly.
Because rotated right is also triggered when doing left rotation and left is trigger when doing right rotation.
So no matter what direction I rotate, both triggers will be activated.
And I get this error:
```
Logger: homeassistant.config
Source: config.py:454
First occurred: 19:47:20 (2 occurrences)
Last logged: 19:47:20

Invalid config for [automation]: Invalid trigger configuration. Got OrderedDict([('default_config', {}), ('tts', [OrderedDict([('platform', 'google_translate')])]), ('group', {}), ('automation', [OrderedDict([('alias', 'Set theme at startup'), ('trigger', OrderedDict([('platform', 'homeassistant'), ('event', 'start')])), ('action', OrderedDict([('service', 'frontend.set_theme'), ('data', OrderedDict([('name', 'noctis')]))])), ('id', 'f62a8e53c0074e80a89b4d6e26a4591e')]), OrderedDict([('id', '1643843094508'), ('alias', 'Fabian_Deckenlicht_On/Off'), ('descripti.... (See /config/configuration.yaml, line 9).
```

MattWestb, 2022-02-03:
How is your HA` /config/configuration.yaml, line 9` looking ? Is it related to ZHA ?
I must looking if i have making the args OK in the automation.

MattWestb, 2022-02-03:
I have making one version with one step with slow and fast but need 
ROTATED_SLOW = "device_rotated_slow"
ROTATED_FAST = "device_rotated-fast"
defined for working OK.
```
        (ROTATED_SLOW, RIGHT): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 13, 1],
        },
        (ROTATED_SLOW, LEFT): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 13, 1],
        },
        (ROTATED, RIGHT): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 13, 2],
        },
        (ROTATED, LEFT): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 13, 2],
        },
        (ROTATED_FAST, RIGHT): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [0, 13, 3],
        },
        (ROTATED_FAST, LEFT): {
            COMMAND: COMMAND_STEP,
            CLUSTER_ID: 8,
            ENDPOINT_ID: 1,
            ARGS: [1, 13, 3],
        },
```
If its not working can your posting the `zha_event` its sending for the command so can getting the ARGs OK.

Tropaion, 2022-02-03:
@MattWestb Your ARGs are ok, the error is something different.
I already posted all commands yesterday.

MattWestb, 2022-02-03:
Yes and very good documents / analyzed !!
I was dubble checking the args and looking on your posted events and its looks very OK.

Deleting the ` __pycache__ ` in your local quirk folder and restarting HA so it cant reusing old cashed code and data.

Tropaion, 2022-02-03:
I don't have a ```__pychache___```. My ZHA never generates one :D.
But I will try again if I find something.

MattWestb, 2022-02-03:
That is very strange and is only happening if ZHA is not loading the .py files is not generated if it was deleted.
My path is looking like this `C:\msys32\Portainer\homeassistant\custom_zha_quirks\__pycache__` and having HA in `C:\msys32\Portainer\`.
If you deleting the local quirk (not the folder) is ZHA loading you quirk or the standard one from HA container ?
You can see the quirk name loaded for the device on the device card on the last line on the left side.
Your local quirkname  = TuyaSmartRemote004F2A
ZHA quirk name = TuyaSmartRemote004F

Tropaion, 2022-02-03:
@MattWestb It's definitely loading the loca quirk since I have multiple ones in use as you know.
And all changes and tests always apply, so this isn't a problem.

MattWestb, 2022-02-03:
Try fixing or renaming `automations.yaml` then the name of the automation tiger name has being  changed and is breaking your saved automation.

Tropaion, 2022-02-03:
I deleted all automations, reboot and did a few test.
- Button works without problems
I created with rotate left a dimm down automation, works too.
Then I added with rotate right a dimm up automation, works BUT after creating it, dimm down doesn't work anymore.
When I deactivate rotate right, rotate left works again.
So I think there should be a problem with rotate right.
This is my yaml entry:
```
- id: '1643918693820'
  alias: Fabian_Deckenlicht_On/Off
  description: ''
  trigger:
  - device_id: 3cc97f6cd8c336b5e52a6070f17fd3e8
    domain: zha
    platform: device
    type: remote_button_short_press
    subtype: button
  condition: []
  action:
  - type: toggle
    device_id: ded8c256a221c111531ecce08e8ddeb9
    entity_id: light.fabian_deckenlicht_on_off
    domain: light
  mode: single
- id: '1643918722609'
  alias: Fabian_Deckenlicht_DimmDown
  description: ''
  trigger:
  - device_id: 3cc97f6cd8c336b5e52a6070f17fd3e8
    domain: zha
    platform: device
    type: device_rotated
    subtype: left
  condition: []
  action:
  - device_id: ded8c256a221c111531ecce08e8ddeb9
    domain: light
    entity_id: light.fabian_deckenlicht_on_off
    type: brightness_decrease
  mode: single
- id: '1643918785173'
  alias: Fabian_Deckenlicht_DimmUp
  description: ''
  trigger:
  - device_id: 3cc97f6cd8c336b5e52a6070f17fd3e8
    domain: zha
    platform: device
    type: device_rotated
    subtype: right
  condition: []
  action:
  - device_id: ded8c256a221c111531ecce08e8ddeb9
    domain: light
    entity_id: light.fabian_deckenlicht_on_off
    type: brightness_increase
  mode: single
```

MattWestb, 2022-02-03:
I dont knowing how the automations code is working.
My feeling is that somthing is not filtering the events OK and the automations is triggering al all that is coming from the device or some filtering is working.

Did you putting the missing defined variables in the constans.py in the HA container and importing then in the quirk ?

Tropaion, 2022-02-03:
@MattWestb No, I just added the constant in the quirk since it doesn't matter for python where you define it.

MattWestb, 2022-02-03:
I dont knowing how python is working i only doing copy and paste and little modification that mostly is working ;-(((
I understand that Zigpy is understanding the defining but how do HA getting that information from the local quirk ? I think constans.py or the quirk.INIT is having some code for HA to understanding it then its living in one other "universe" and must getting all information for working OK.

Can you trying putting the constants in the constants,py in the HA container and see if its working the same ?

Tropaion, 2022-02-03:
@MattWestb I'm sure it isn't because of constants but I think I know why it isn't working.
The IKEA device uses the command ```COMMAND_MOVE``` which is compatible with the the subtype left and right.
But this devices uses the command ```COMMAND_STEP``` which isn't compatible with the the subtype left and right.

MattWestb, 2022-02-03:
The IKEA 5 button quirk is using both commands and they is imported from `zhaquirks/const.py`. 
The rotated is defined in `zhaquirks/ikea/__init__.py` and is imported from it but i cant see any strange things done there.
Left and right is from `const.py.`

I have finding traces of the defining rotated, right and left in ZHA https://github.com/home-assistant/core/blob/65d14909ee3678b942ed87308a55cad576119ea9/homeassistant/components/zha/translations/en.json but not whrer the code is implanted.

HA is not knowing the ARGS: [0, 13, 2], so the (quirk ) device automations is doing the triggering of  them and sending the rotate and right / left to HA and doing the automations of it.
So the device automation trigger is not working OK in the quirk.
If doing one "manual automation" in HA it must triggering on the ARGS it getting from `zha_event` then its dont knowing the rotated and left / right. 

Sorry but its only thinkings and not knowing :-(((

Tropaion, 2022-02-03:
@MattWestb I don't know much either since I never programmed in python just C/C++ and the documentation of zha is not that good. But I will try a few thing tomorrow afternoon after my totally uninteresting exam 😢

MattWestb, 2022-02-03:
Is it so bad in JKU ? 
I hope you have one short  day and not so boring.
I must also getting some sleep and doing some work tomorrow.

Tropaion, 2022-02-03:
I'm in a different university now, extremely specialised in IT and yeah, the exams currently are hard, but tomorrow is the last and most hated one, signal theorie and mathematics 😢

Tropaion, 2022-02-06:
@MattWestb Do you know what this missing cluster command is?
```
[0xd9d4:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=17 command_id=2>
[0xd9d4:1:0x0008] ZCL request 0x0002: [0, 13, 1]
[0xd9d4:1:0x0008] No handler for cluster command 2
```
And why I got these two exclamation marks?
[image]

Tropaion, 2022-02-06:
And I still don't understand why we have this error:
```
Logger: homeassistant.components.zha.core.channels.base
Source: components/zha/core/channels/base.py:428
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 00:23:31 (2 occurrences)
Last logged: 00:23:44

[0xD9D4:1:0x1000]: 'async_configure' stage failed: not enough values to unpack (expected 3, got 2)
```

Tropaion, 2022-02-07:
@MattWestb I also found something interesting in the logs.
When I rotate the knob right, the automation DimmUp is triggered:
```
2022-02-07 01:19:47 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=8, SrcAddr=0xD9D4, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=21, SecurityUse=<Bool.false: 0>, TimeStamp=8705888, TSN=0, Data=b'\x01\x45\x02\x00\x0D\x01\x00', MacSrcAddr=0xD9D4, MsgResultRadius=11)
2022-02-07 01:19:47 DEBUG (MainThread) [zigpy.zcl] [0xd9d4:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=69 command_id=2>
2022-02-07 01:19:47 DEBUG (MainThread) [zigpy.zcl] [0xd9d4:1:0x0008] ZCL request 0x0002: [0, 13, 1]
2022-02-07 01:19:47 DEBUG (MainThread) [zigpy.zcl] [0xd9d4:1:0x0008] No handler for cluster command 2
2022-02-07 01:19:47 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=54:0f:57:ff:fe:2e:3c:5f, unique_id=54:0f:57:ff:fe:2e:3c:5f:1:0x0008, device_id=3cc97f6cd8c336b5e52a6070f17fd3e8, endpoint_id=1, cluster_id=8, command=step, args=[0, 13, 1]>
2022-02-07 01:19:47 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event automation_triggered[L]: name=Fabian_SmartKnob_Deckenlicht_DimmUp, entity_id=automation.fabian_smartknob_deckenlicht_dimmup, source=event 'zha_event'>
2022-02-07 01:19:47 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=automation.fabian_smartknob_deckenlicht_dimmup, old_state=<state automation.fabian_smartknob_deckenlicht_dimmup=on; last_triggered=2022-02-07T01:18:43.108709+01:00, mode=single, current=0, id=1644190324288, friendly_name=Fabian_SmartKnob_Deckenlicht_DimmUp @ 2022-02-07T01:07:58.149267+01:00>, new_state=<state automation.fabian_smartknob_deckenlicht_dimmup=on; last_triggered=2022-02-07T01:19:47.506486+01:00, mode=single, current=1, id=1644190324288, friendly_name=Fabian_SmartKnob_Deckenlicht_DimmUp @ 2022-02-07T01:07:58.149267+01:00>>
2022-02-07 01:19:47 INFO (MainThread) [homeassistant.components.automation.fabian_smartknob_deckenlicht_dimmup] Fabian_SmartKnob_Deckenlicht_DimmUp: Running automation actions
2022-02-07 01:19:47 INFO (MainThread) [homeassistant.components.automation.fabian_smartknob_deckenlicht_dimmup] Fabian_SmartKnob_Deckenlicht_DimmUp: Executing step device automation
2022-02-07 01:19:47 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_on, service_data=entity_id=light.fabian_deckenlicht_on_off, brightness_step_pct=10>
2022-02-07 01:19:47 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=automation.fabian_smartknob_deckenlicht_dimmup, old_state=<state automation.fabian_smartknob_deckenlicht_dimmup=on; last_triggered=2022-02-07T01:19:47.506486+01:00, mode=single, current=1, id=1644190324288, friendly_name=Fabian_SmartKnob_Deckenlicht_DimmUp @ 2022-02-07T01:07:58.149267+01:00>, new_state=<state automation.fabian_smartknob_deckenlicht_dimmup=on; last_triggered=2022-02-07T01:19:47.506486+01:00, mode=single, current=0, id=1644190324288, friendly_name=Fabian_SmartKnob_Deckenlicht_DimmUp @ 2022-02-07T01:07:58.149267+01:00>>
2022-02-07 01:19:48 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.fabian_deckenlicht_on_off, old_state=<state light.fabian_deckenlicht_on_off=on; min_mireds=153, max_mireds=370, supported_color_modes=['brightness', 'color_temp'], color_mode=color_temp, brightness=26, color_temp=212, hs_color=(26.792, 23.244), rgb_color=(255, 222, 195), xy_color=(0.384, 0.354), friendly_name=On/Off, supported_features=32 @ 2022-02-07T01:07:51.562526+01:00>, new_state=<state light.fabian_deckenlicht_on_off=on; min_mireds=153, max_mireds=370, supported_color_modes=['brightness', 'color_temp'], color_mode=color_temp, brightness=51, color_temp=212, hs_color=(26.792, 23.244), rgb_color=(255, 222, 195), xy_color=(0.384, 0.354), friendly_name=On/Off, supported_features=32 @ 2022-02-07T01:07:51.562526+01:00>>
2022-02-07 01:19:48 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.fabian_deckenlicht_current_consumption, old_state=<state sensor.fabian_deckenlicht_current_consumption=1.8; state_class=measurement, unit_of_measurement=W, device_class=power, friendly_name=Aktueller Verbrauch @ 2022-02-07T01:18:43.651344+01:00>, new_state=<state sensor.fabian_deckenlicht_current_consumption=2.8; state_class=measurement, unit_of_measurement=W, device_class=power, friendly_name=Aktueller Verbrauch @ 2022-02-07T01:19:48.059870+01:00>>
```
But when I rotate left, first the automation DimmUp is called and then DimmDown:
```
22-02-07 01:18:43 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=8, SrcAddr=0xD9D4, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=14, SecurityUse=<Bool.false: 0>, TimeStamp=4680024, TSN=0, Data=b'\x01\x44\x02\x01\x0D\x01\x00', MacSrcAddr=0xD9D4, MsgResultRadius=11)
2022-02-07 01:18:43 DEBUG (MainThread) [zigpy.zcl] [0xd9d4:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=68 command_id=2>
[... cut ...]
2022-02-07 01:18:43 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.fabian_deckenlicht_current_consumption, old_state=<state sensor.fabian_deckenlicht_current_consumption=2.8; state_class=measurement, unit_of_measurement=W, device_class=power, friendly_name=Aktueller Verbrauch @ 2022-02-07T01:18:27.935860+01:00>, new_state=<state sensor.fabian_deckenlicht_current_consumption=1.8; state_class=measurement, unit_of_measurement=W, device_class=power, friendly_name=Aktueller Verbrauch @ 2022-02-07T01:18:43.651344+01:00>>
```

MattWestb, 2022-02-07:
The 2 first warning signs i thinks the new LIQ and RISS sensors.
The cluster 0x1000 is LightLink cluster and we have not  "touching" it = not our fault.
The dimmer device trigger looks promise and the data i was putting it looks right.
If you using the device automation from the device card is it working OK or is it always being  triggered ?

I looking more in the afternoon.

MattWestb, 2022-02-07:
I was using one IKEA Symphonisk that also is one rotater and is sending nearly the same commands and was doing one automation from the device card with  rotating right and fire one zha_event right:1 and one more with left.
```
2022-02-07 14:08:59 DEBUG (MainThread) [zigpy.zcl] [0x7dbc:1:0x0008] ZCL request 0x0001: [0, 195]
2022-02-07 14:08:59 DEBUG (MainThread) [zigpy.zcl] [0x7dbc:1:0x0008] No handler for cluster command 1
2022-02-07 14:08:59 INFO (MainThread) [homeassistant.components.automation.sf_right] SF Right: Running automation actions
2022-02-07 14:08:59 INFO (MainThread) [homeassistant.components.automation.sf_right] SF Right: Executing step zha_event

2022-02-07 14:08:57 DEBUG (MainThread) [zigpy.zcl] [0x7dbc:1:0x0008] ZCL request 0x0001: [1, 195]
2022-02-07 14:08:57 DEBUG (MainThread) [zigpy.zcl] [0x7dbc:1:0x0008] No handler for cluster command 1
2022-02-07 14:08:57 INFO (MainThread) [homeassistant.components.automation.sf_left] SF Left: Running automation actions
2022-02-07 14:08:57 INFO (MainThread) [homeassistant.components.automation.sf_left] SF Left: Executing step zha_event
```
And i is getting right:1 and left:1 nicely as events and also in the log.

The IKEA DA:
https://github.com/zigpy/zha-device-handlers/blob/23a166902e50d5c32019e0d4d32a968715041bb6/zhaquirks/ikea/symfonisk.py#L102-L113

I have not looking but shall but my feeling is that the Device Automation and what the device is sending is OK but perhaps problem with the automation you have done.

My automation made from the device card, add automation for this device rotating Right / left is looking like this in the automations.yaml:
```
- id: '1644238923083'
  alias: SF Right
  description: ''
  trigger:
  - device_id: 1018f5c2e28bde1044a8ee2449db3bcd
    domain: zha
    platform: device
    type: device_rotated
    subtype: right
  condition: []
  action:
  - event: zha_event
    event_data:
      Right: 1
  mode: single
- id: '1644239212807'
  alias: SF Left
  description: ''
  trigger:
  - device_id: 1018f5c2e28bde1044a8ee2449db3bcd
    domain: zha
    platform: device
    type: device_rotated
    subtype: left
  condition: []
  action:
  - event: zha_event
    event_data:
      Left: 1
  mode: single
```
Im looking little more in you logs and reporting if i can see some interesting in it.

MattWestb, 2022-02-08:
I was looking little in your log with double commands sent and its looks its only one triggering of the code. But i dont understand way its making 2 actions from it.

Can you taking away your action and putting in one single `zha_event` instead so its cant being something strange with the action part that is linked to other things ?

Tropaion, 2022-02-08:
@MattWestb What do you mean with one single ```zha_event```. You mean I should remove the other automations?

MattWestb, 2022-02-08:
Put in like this and the automation is only doing one zha_event.
```
  action:
  - event: zha_event
    event_data:
      Left: 1
  mode: single
```
Its feels very spooky that the automation is triggered one time and is doing 2 actions.
Only for eliminating its some problem with the action part.

MattWestb, 2022-02-08:
I agree but in the log its 2 actions that is only one trigger.
Therefor eliminating somthing strange happens in the action part (somthing linked that  we dont see / knowing).

Its easy testing and eliminating other things that can doing bad things.

**If** the zha_events is working OK = the quirk and the device automation is OK and somthing is wrong with the with the `entity_id: light.fabian_deckenlicht_on_off` `brightness_increase` but we dont knowing that before testing it.

grinono, 2022-02-09:
i just tried to connect this device, it's found by ZHE,  but the button is not working. Is that a known issue?

[image]

Tropaion, 2022-02-09:
@grinono If you have the same device as me, you have to use the quirk discussed here.

[29 more replies]

closed 2022-05-27 by Tropaion