zigpy/zha-device-handlers #1325: [Device Support Request] IKEA 4 btn remote

JoryHogeveen, 2022-02-03:
**Describe the solution you'd like**
The 5 btn remote is already supported. The newer 4 button remote is almost the same: https://community.home-assistant.io/t/zha-ikea-four-button-remote-for-lights/347657

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc57"
      ],
      "out_clusters": [
        "0x0003",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "Remote Control N2",
  "class": "zhaquirks.ikea.fourbtnremote.IkeaTradfriRemote"
}
```

**Additional context**

MattWestb, 2022-02-03:
Its some problem with it then is spamming the network with most coordinators (perhaps not with TI ones) and also the long press left and right is making more commands then we like but we cant changing how the firmware in the device is working

The long press left and right is used by IKEA for resetting all light in the group to one default setting if they have getting out of sync and the same is for the 5 button then long pressing the toggle button.

JoryHogeveen, 2022-02-03:
Hi @MattWestb 
I've got the remote working great through automation using a custom blueprint, so for now I'm fine.
The only thing is that I'd like to bind this remote directly to the bulb so it will still work if the hub is down.

Binding is something that isn't very user friendly in ZHA yet. I can't tell if it actually binds or does anything at all. That's why I thought it might be an issue to report here :) I didn't realize there is already an integration for this remote so I guess it's more a general ZHA issue.

MattWestb, 2022-02-03:
Before all IKEA controllers was only supporting group binding but the last update of the "classical" controllers was loosing that function but must being bonded to devices instead (more devices is possible).
But the Styrbar is only supporting group binding so need doing it OK from the device card and manage cluster and group binding.
If i remember right is changing binding it can being necessarily doing one reconfigure of the device from the device card or its not reporting OK to the coordinator.

Dont forgetting waking up the device then sending commands to it or its not getting the new configuration !!
Also if doing binding of the remote its can being problem if your automations is sending "other commands" to the device and making problems (I think On/Off and light level shall not being used at the same time with binding and automations).

JoryHogeveen, 2022-02-07:
Hi @MattWestb 

I tried binding the Styrbar to the light group (only one bulb).
The binding itself seems to work, however I can only use on/off, all other remote functionality isn't working.
Any ideas? I did disable the automation before binding.

Also, is there any location where a user can view the current/active bindings that are made through ZHA?

Thanks!

MattWestb, 2022-02-07:
I can being that you need doing one reconfiguration of the device **if** its stooping reporting to ZHA after doing / changing binding of IKEA controllers.

The dimmer function shall also working with all "normal" lights in the bounded group.

Back to the "scene" problem. IKEA is sending commands from the scene buttons its bounded group **but** is tagging them group 0xff09 and scene and with manufacture commands that only IKEA lights is understanding.

If making one group 0xff09 and adding IKEA lights in it (they can being member of more groups) and they is getting the commands, but after the lights have being retested they dont have the scenes and for the moment we cant adding them then some group and scene commands is not working in ZHA.
One work around that was working (before the firmware update for 2 mouths ago) was doing one firmware downgrade (working with local OTA files plus patching ZHA) and then the lights is / was adding the 3 or more scenes in there scene table and was working OK util resetting the light.

If getting the scene add command working in ZHA all shall being possible and Z2M and deCONZ its working if doing all right adding more scenes in IKEA light that working with the remotes.

JoryHogeveen, 2022-02-07:
Hi @MattWestb 

Thank you for the information! Totally explains why the scenes didn't work.
I just tried and indeed, dimming is also working.

However, there is a few seconds latency when using the remote through binding and sometimes it doesn't even respond at all.
Using an automation is a lot faster and also a lot more stable.

Any idea how/why this is possible? As far as I know the remote should communicate directly to the lamp instead of going through HA right?

JoryHogeveen, 2022-02-07:
Hi @MattWestb 
That explains a bit more.
Btw. I just got a 5 button remote with 3 bulbs as well. This remote instantly worked with binding and responds very quick. Works great! I think the mentioned issue is only related to the 4 button remote.

Totally unrelated to this topic (I'll open a new one if needed):
One of the 3 bulbs on my 5 button remote doesn't respond to scene switches. The other two are. I'm not sure how to fix this. Any ideas are very appreciated!

MattWestb, 2022-02-07:
The 5 button is not spamming the network so its normally quicker then the network is not limiting the traffic.
If you is updating the firmware if the remote is loosing the possibility doing group binding and you must do device binding that is limiting the function on some device is controlling.

I thing 2 of your lights is being updated of ZHA and restored the default scenes that working with the remote and the 3 is not or the 3 have being retested and loosing the "default scenes".

JoryHogeveen, 2022-02-07:
Hi @MattWestb 
You are correct! The firmware on the 2 working bulbs is `0x20029623` and on the 3rg not working bulb `0x20022623` so I guess this one needs an update.

And just to be sure, are you telling me **not** to update the 5 button remote?

MattWestb, 2022-02-07:
I have not testing the "scene" commands with the new firmware then i was having most of my IKEA controllers on one bad test firmware that was not possible updating without flashing one old version with J-Tag and then i was not having  time sniffing how they is working.
If updating some bug is fixed (list network on battery drained) but you is loosing group binding and must do device binding and i dont knowing if the scenes is working with that.

All IKEA controllers is blocked for firmware downgrading so cant redo it easy.
The lights was not but i have not testing with the last update but its easy and is described in the Zogpy Wiki https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates#downgrading-your-device-firmware.

JoryHogeveen, 2022-02-07:
Thank you so much for your help. I'll check these resources and try it out!