zigpy/zha-device-handlers #1329: [Device Support Request] Revisiting Develco Frient ZHEMI101 Electricity Meter Reader

shawnbarton, 2022-02-04:
Two previous issues have tried to address this device in the past but both were seemingly closed before I could gleam an answer to my issue. The issues are #949 and #1310.

Device information is already available in these issues so I won't re-post here. If this is just a gap in my newbie knowledge then this thread can at least serve to document the solution.

I cannot find a way of setting important configuration options on this device via ZHA, namely `interface_mode` (I would like to run this device in `IEC62056-21` mode instead of pulse/kwh) and `pulse_configuration` (sets the number of pulses per kwh, in case the other mode doesn't work).

Zigbee2MQTT seems to have these implemented and a message must be published to set the configuration: https://www.zigbee2mqtt.io/devices/ZHEMI101.html

In HomeAssistant, I am unable to locate these options in the Manage Clusters view of this device in the Clusters, Cluster Attributes, and Cluster Commands dropdowns.

Is this simply not possible with ZHA or am I overlooking something?

Thanks in advance.

rossdargan, 2022-03-09:
It's not there - It looks like a quirk might add it but it doesn't look easy! If someone could give some pointers I could have a try at it.

This manual might help: http://docplayer.net/135845279-Technical-manual-for-zhemi101-zigbee-external-meter-interface-ha.html - the top of page 9

d-schultz, 2022-05-05:
Hi, I am also interested in changing the puls rate of my device. Conceptually, it should be easy: In ZHA GUI I go to the the metering cluster and search for attribute at 0x03000. I enter the manufacturer code 4117, ask for the attribute value and get  1000. This should work correctly, since 1000 imp/kWh is the default value of the device. Then I enter 10000 and klick the button to set the attribute. The GUI feedback looks like success. However, when asking for the attribute value, I get 1000 again.

Do I do something wrong? Is it possible at all to change the pulse rate using ZHA? If so, could someone show how it works, please?

richheitech, 2022-05-15:
Hi, I'm also having the same issues with ZHA and this device. My meter needs 800 imp/kWh. The option to change the pulse config isn't available and the divisor is fixed at 1000.

RefineryX, 2022-08-02:
I am also looking to change the pulse from the default 1000 to 3200.

Did anyone come up with a solution on how to change this?

fuskaps, 2022-08-07:
Hi
I got it fixed, go to Home Assistant, Developer Tools, Servises
[image]
change to "friendly name" on your meter and change pulse to desired, press "call service"

RefineryX, 2022-08-07:
> Hi
> I got it fixed, go to Home Assistant, Developer Tools, Servises
> [image]
> change to "friendly name" on your meter and change pulse to desired, press "call service"
> [image]

This looks like it is zigbee2mqtt and not zha?

SiM22, 2022-08-11:
This does work with zha. The way I fixed it is I upgraded the firmware on it. If I remember correctly I plugged my conbee into my windows machine. Paired the reader to it and ran some updates. Then plugged it back in to home assistant and paired the reader with HA. Sounds flakey but for whatever reason it worked

RefineryX, 2022-08-11:
> This does work with zha. The way I fixed it is I upgraded the firmware on it. If I remember correctly I plugged my conbee into my windows machine. Paired the reader to it and ran some updates. Then [...]

Can you confirm what entities get exposed? Also, are you able to successfully change the pulse configuration?

SiM22, 2022-08-18:
[image]
[image]

RefineryX, 2022-08-18:
This is super helpful, thanks @SiM22! I followed the above and bumped into a few issues.

My meter runs at the pulse of 3200. I used this [website](https://www.rapidtables.com/convert/number/decimal-to-hex.html) to convert ``3200`` into the 'Hex signed 2's complement' which gave ``0C80``. From your instructions above, this should now read as ``0x0C80``.

In clusters, I get the default ``enum8.undefined_0x00``. I set ``0x0C80`` and click ``SET ZIGBEE ATTRIBUTE`` and I receive this error on my logs ```Failed to convert attribute 0x0300 from 0x0C80 (<class 'str'>) to type <enum 'enum8'>: 3200 is not an unsigned 8 bit integer``` and another

```
Error executing service: <ServiceCall zha.set_zigbee_cluster_attribute (c:01GAS4ZPJ1SZ4GJ1C15P42NK4D): ieee=00:15:bc:00:1b:02:1f:79, endpoint_id=2, cluster_id=1794, cluster_type=in, attribute=768, value=0x0C80>
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1731, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1750, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 746, in admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 1118, in set_zigbee_cluster_attributes
    response = await zha_device.write_zigbee_attribute(
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 639, in write_zigbee_attribute
    response = await cluster.write_attributes(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 577, in write_attributes
    return await self.write_attributes_raw(attrs, manufacturer)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 583, in write_attributes_raw
    result = await self._write_attributes(attrs, manufacturer=manufacturer)
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 325, in request
    return await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

SiM22, 2022-08-18:
So the documentation states nothing about that attribute having to be 8bit, only other ones. This just might be a case of people doing what they want with the zigbee standard. In this case do the following and lets do a trial and error.

[image]

Get that attribute and do a get, does it return 1000? If yes underneath set it to 3200 and click Set. In theory that should not throw an error. I wonder if they are using the divisor as the number used for pulses. God knows....try that and get back to me

RefineryX, 2022-08-18:
Thanks. Checking the ``divisor`` and it is ``1000``.

[image]

I tried to update this too ``3200`` and it did not throw an error. However, although it appears that it accept it... it does not save. I also tried ``3200`` and using the ``Manufacturer Code Override`` of ``4117`` - unfortunately it does not save and clicking the ``Get Zigbee Attribute`` returns back to ``1000``.

I did create a ticket last week with some logs here, if that helps - https://github.com/zigpy/zha-device-handlers/issues/1681
 
[image]

SiM22, 2022-08-18:
when you do the get, leave the 1000 as is. put the 3200 where you put 4117

RefineryX, 2022-08-18:
Like this?

[image]

If so, I get no error on the logs but it does not accept the number. Clicking ``GET ZIGBEE ATTRIBUTE`` shows ``1000`` remains.

Checking also back to  ``unit_of_measure`` shows ``enum8.undefined_0x00``.

SiM22, 2022-08-18:
Let me get back to you. I done this a long time ago and need to job me memory

RefineryX, 2022-08-18:
Greatly appreciate it, thanks so much! Looking forward to get this working.

SiM22, 2022-08-18:
I am trying to understand where the manufacturer code came from in this issue. When i look at the device signature for the reader. It shows 0

[image]

Maybe try change the 1000 to 3200 again, and then in manufacturer code add a 0

RefineryX, 2022-08-18:
Just tried that nothing saves, 1000 sticks.

Are you able to successfully change the pulse rate on your side?

The manufacturer code of 4117 was recommended by someone else. Section 3.2 of the technical manual reveals the Develco ZigBee Manufacturer code 0x1015. This converted is 4117 (also tried putting 0x1015).

SiM22, 2022-08-19:
So I done something and it changed something. I will list what I tried so far as it changed something but even though i changed it back to what it was (1000) the readings are all messed up ha!

Logs showing the changes:

[image]

The readings are messed up now because when i made the change from 1000 to 1001, it cleared all the energy reading history in the UI and it started fresh but it starts with the last known reading which is something stupidly high. In essence it believes in (in the UI) that i used thousands upon thousands of kWh in one day.

[image]

Anyway, i reverted the value. I will wait 24 hours for the new day to start and the Home Assistant UI should start recording energy as normal.

Story over, now to what i have been trying.

I changed the divisor from 1000 to 1001 using the manufacturer code 4117. It seemed to work, but if I look in the technical manual it shows 1015 as the manufacturer code...image below.

[image]

[image]

It also states it is for a specific software version of 2.4.0 or higher, and then somewhere else it states 6.4.1. 

I can not say for sure what version i am running as it shows as an unsupported attribute.

```
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0000": {
                "attribute_name": "zcl_version",
                "value": 3
              },
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "Develco"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "ZHEMI101"
              },
              "0x0007": {
                "attribute_name": "power_source",
                "value": 3
              }
            },
            "unsupported_attributes": {
              "0x4000": {
                "attribute_name": "sw_build_id"
              },
              "0x0001": {
                "attribute_name": "app_version"
              },
              "0x0002": {
                "attribute_name": "stack_version"
              },
              "0x0003": {
                "attribute_name": "hw_version"
              },
              "0x0006": {
                "attribute_name": "date_code"
              },
```

I
 
Also the more i google the more I think what fixed it for me, and what lets me actually change values was using deconz to update it. They seem to have some things in deconz that allows/updates/tweaks something - https://forum.phoscon.de/t/lixee-zlinky-tic-ddf-file/1164/142
So I would say, try plugging your zigbee coordinator into another machine and use deconz. Pair and do all the updates you can.

RefineryX, 2022-08-19:
Thanks for that.

So, I went down a rabbit hole with it this morning.

First, I updated the firmware on my Conbee II stick to deCONZ_ConBeeII_0x26780700.bin.GCF released May 15th ([here](https://deconz.dresden-elektronik.de/deconz-firmware/?C=M;O=D)). After the successful upgrade, I was still unable to update the pulse config to ``3200`` or even ``1001`` used in your example.

Second, I installed Deconz, paired my Frient device and connected this to Home Assistant. There is no way to update the firmware of the Frient directly from Deconz nor was I able to update the pulse config.

So back to square one, this time with upgraded Conbee II firmware. It appears that the Frient device works fine in both ``deconz`` and ``zha`` **only if** your meter pulses at the default rate of ``1000 / kwh``. 

I was also looking at a similar issue from https://github.com/Koenkk/zigbee2mqtt/issues/8638. Although the same issue, it appears this was fixed with changes the codebase. Not sure if it helps this case but wanted to share what I had found.

pfefferle, 2022-12-09:
Is it maybe possible to change the divisor using a custom/local quirk?

fnogatz, 2023-02-16:
Apparently what works is to connect to the electricity meter only once via zigbee2mqtt, set the configuration, and then use the device as normal in Home Assistant with zha. This way I got it working for my pulse rate of 10'000. I'm using a SONOFF Zigbee 3.0 USB Dongle with my Home Assistant system with zha on a Raspberry Pi. So, here's what I've done, using my normal linux workstation for the configuration via zigbee2mqtt:

1. Pair the electricity meter as normal with your HA system with zha. You should be able to see the (wrong) values given the standard pulse rate of 1'000.
1. Unplug the SONOFF Zigbee stick from the HA device and plug it into your workstation.
1. Install zigbee2mqtt on the workstation (*not* the device running HA). For me, this was a straight-forward following of the [official guide](https://www.zigbee2mqtt.io/guide/getting-started/#installation), and only a matter of minutes.
1. Pair the electricity meter with zigbee2mqtt.
1. In zigbee2mqtt, you can easily adjust the pulse rate in the device's settings (via web-interface! :open_mouth: ). It will directly show you the correctly calculated values in the zigbee2mqtt web interface.
1. Stop zigbee2mqtt, and plug in the Zigbee stick in your normal HA device again. Optionally restart HA.
1. HA now reports the values with the correct pulse rate.

galmok, 2023-10-08:
I may have a solution, but it requires that "ZHA Toolkit" is installed (via HACS). With this toolkit, you get access to "ZHA Toolkit: Write Attribute" which is similar to the built-in Write Attribute, but with more flexibility. The built-in Write Attribute lacks the "Attribute Type" field and assumes the wrong type and the EMI ignores the message as a result. I had to setup my EMI to read the infrared signal (get the actual meter value (current consumption) and instantaneous demand) and had to push a manufacturer specific value to one of the registers.

This picture shows how I set it up:

[image]

closed 2024-04-12 by github-actions[bot]