zigpy/zha-device-handlers #1335: [Device Support Request] Smart valve controller - Tuya - TS0601 - TZE200_d0ypnbvn

JMan7777, 2022-02-08:
Hi,

I got a smart valve controller (Tuya - TS0601 - TZE200_d0ypnbvn) and it seems to be not supported in HA at the moment.

[image]

It shows no entities after pairing:

[image]

Also, after pairing it keeps writing the following log entries:

```
[0x5859:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=218 command_id=2>
[0x5859:1:0xef00] Unknown cluster-specific command 2
[0x5859:1:0xef00] ZCL request 0x0002: b'\x00\x00\x1a\x05\x00\x01\x01'
[0x5859:1:0xef00] No handler for cluster command 2
```

It would be great to get this device supported. 

Thx a million.

**My HA setup (On Raspberry Pi 4b - 8GB with Nortek HUSBZB-1 USB stick):**
Home Assistant 2022.2.3
Home Assistant OS 7.3
Supervisor Version 2022.01.1

**Device signature:**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_d0ypnbvn",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

**Add device log:**
```
New device 0x5859 (a4:c1:38:47:9c:df:dc:05) joined the network
[0x5859] Scheduling initialization
Tries remaining: 3
[0x5859] Requesting 'Node Descriptor'
Tries remaining: 2
[0x5859] Extending timeout for 0xe5 request
[... 280 lines cut ...]
[0x5859:1:0xef00] ZCL request 0x0002: b'\x00\x00\x1a\x05\x00\x01\x01'
[0x5859:1:0xef00] No handler for cluster command 2
[0x5859:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=218 command_id=2>
[0x5859:1:0xef00] Unknown cluster-specific command 2
[0x5859:1:0xef00] ZCL request 0x0002: b'\x00\x00\x1a\x05\x00\x01\x01'
[0x5859:1:0xef00] No handler for cluster command 2
[0x5859:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=219 command_id=Command.Read_Attributes>
[0x5859:1:0x000a] ZCL request 0x0000: [[7]]
[0x5859:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=220 command_id=2>
[0x5859:1:0xef00] Unknown cluster-specific command 2
[0x5859:1:0xef00] ZCL request 0x0002: b'\x00\x00\x1a\x05\x00\x01\x01'
[0x5859:1:0xef00] No handler for cluster command 2
[0x5859:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=220 command_id=2>
[0x5859:1:0xef00] Unknown cluster-specific command 2
[0x5859:1:0xef00] ZCL request 0x0002: b'\x00\x00\x1a\x05\x00\x01\x01'
[0x5859:1:0xef00] No handler for cluster command 2
[0x5859:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=220 command_id=2>
[0x5859:1:0xef00] Unknown cluster-specific command 2
[0x5859:1:0xef00] ZCL request 0x0002: b'\x00\x00\x1a\x05\x00\x01\x01'
[0x5859:1:0xef00] No handler for cluster command 2
[0x5859:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=220 command_id=2>
[0x5859:1:0xef00] Unknown cluster-specific command 2
[0x5859:1:0xef00] ZCL request 0x0002: b'\x00\x00\x1a\x05\x00\x01\x01'
[0x5859:1:0xef00] No handler for cluster command 2
```

JMan7777, 2022-04-30:
Any chance to get this supported in the near future?

Thx a million.

sdavinci69, 2022-05-19:
I just bought the same, hoping I could control it with Home Assistant, but same problem as above, I dont find the switch to turn it on or off🥲in the device entities

kasik96, 2022-08-25:
Same issue here.

JMan7777, 2022-12-04:
Any chance to get this solved in the near future? 😉

remoteassist, 2023-01-25:
I was able to get it working combining code from #1571 and #https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_switch.py , Thanks @javicalle for the initial code.
    I am not good with coding, but know it's working for my device. So might not be able to help you if the quirks does not work for your device. See code in "ts0601_water_valve.py"

How to setup local Quirks https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226

[image]

<details>
  <summary>ts0601_water_valve.py</summary>
  
  ```
"""Tuya Valve."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, OnOff, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    MoesSwitchManufCluster,
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaOnOffNM,
)

class TuyaValveManufCluster(TuyaMCUCluster):
    """On/Off Tuya cluster with extra device attributes."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF01: ("time_left", t.uint32_t, True),
            0xEF02: ("state", t.enum8, True),
            0xEF03: ("last_valve_open_duration", t.uint32_t, True),
            0xEF04: ("dp_6", t.uint32_t, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOffNM.ep_attribute,
            "on_off",
            dp_type=TuyaDPType.BOOL,
        ),
        6: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_6",
            TuyaDPType.VALUE,
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "time_left",
            TuyaDPType.VALUE,
        ),
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "state",
            TuyaDPType.VALUE,
        ),
        15: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "last_valve_open_duration",
            TuyaDPType.VALUE,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        6: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
        15: "_dp_2_attr_update",
    }

class TuyaValve(CustomDevice):
    """Tuya valve device."""

    signature = {
        MODELS_INFO: [("_TZE200_d0ypnbvn", "TS0601")],
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1,
        #       "in_clusters": ["0x0000",,"0x0004","0x0005","0xef00"], output_clusters=[10, 25])
        
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaValveManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # "profile_id": 41440,
                # "device_type": "0x0061",
                # "in_clusters": [],
                # "out_clusters": ["0x0021"]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

  ```
</details>

kkossev, 2023-03-26:
Is anyone using this TS0601 TZE200_d0ypnbvn valve in ZHA?
With another Home Automation hub this device sends the valve open/closed status every 3 seconds 24/7 ... : ( 
Anyone experiencing the same issue?

MattWestb, 2023-03-26:
Is the spammer reporting with or without your "tuya magic spell" applied ?

kkossev, 2023-03-26:
With the magic spell.. But will double-check with the user again to make sure the spell is cased during the initial pairing of the device, while it is still in pairing mode.
The possibility to change drivers/quirks on-the-fly is good, but sometimes it does not have the expected effect ..

MattWestb, 2023-03-26:
We is trying redoing the spell and sending it then bindings is done also the re configuring the device then its not in INIT mode and my TS004F is working then getting it some hours later after joining the network (and rested and taking the battery out) so its little strange.
More here https://github.com/zigpy/zha-device-handlers/pull/2287#issuecomment-1484089207

kkossev, 2023-03-26:
It is not just this Tuya device that spams the Zigbee coordinator, but also the mmWave radars (especially the big black square 24Ghz radar with the annoying LED), the CO2 sensors, the clamp energy meter... the list can be continued.

I just can't understand how it is possible that otherwise good products are spoiled that badly with bad spammy firmware.

javicalle, 2023-03-26:
Had someone tried to add the device to the `TuyaSingleSwitch_GP` quirk?
Can be tested with a local quirk like this one:
```python
"""Tuya DP based switches."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    MoesSwitchManufCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)

class TuyaSingleSwitch_GP(TuyaSwitch):
    """Tuya single channel switch with GreenPowerProxy cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_d0ypnbvn", "TS0601"),  # reported in #1335
            ("_TZE200_gbagoilo", "TS0601"),  # reported in #1634
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=51 device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

JMan7777, 2023-07-25:
I tried the quirk and it works perfectly. The valve shows up initially as a light but this is just cosmetic. Thx a lot for the quirk.

javicalle, 2023-07-25:
> I tried the quirk and it works perfectly. The valve shows up initially as a light but this is just cosmetic. Thx a lot for the quirk.

Have you changed the representation? Which one have you choose?

Could you try with this `DEVICE_TYPE` in the replacement part?:
```python
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

JMan7777, 2023-07-25:
Hi, thx for the reply.
Unfortunately I can't test this for the next 3 months, but I will try later on. 
Thanks again. 👍

mintjberry, 2024-01-06:
> Had someone tried to add the device to the `TuyaSingleSwitch_GP` quirk? Can be tested with a local quirk like this one:
>
> ```python
> """Tuya DP based switches."""
> from zigpy.profiles import zha
> from [...]

Works well, thanks. I'm a newby to these quirks, and want to get at least one more of these valves. Is there anything special that you need to do in order to control two separate units? Or will the one quirk cover them all and allow for individual control?

PR #3351 (2024-09-24): Add Moes Tuya valve controller `_TZE204_d0ypnbvn`, `_TZE200_d0ypnbvn`
## Proposed change

This implement support for valve `_TZE204_d0ypnbvn`.

## Additional information

This device is also known as `_TZE200_d0ypnbvn`, I don't have it handy, but it should work for it.

I haven't added test as the code should already be covered as I only added a device to an existing definition.

Fixes #3291, fixes #3220, fixes #1335.

## Checklist
- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that  [... cut]
closed 2024-09-24 by TheJulianJES