zigpy/zha-device-handlers #1337: [BUG] Aqara plug (lumi.plug.maeu01) generates errors post-2022.2

tetele, 2022-02-08:
**Describe the bug**
I use this plug with HA 2022.2.3, where it's been updated to use the quirk for lumi.plug.mmeu01 after this commit.

There are errors popping up in the log after this update.

```
Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: 4:34:56 PM (16 occurrences)
Last logged: 4:55:26 PM

Exception in async_state_changed when dispatching 'LUMI lumi.plug.maeu01_54:ef:44:10:00:0e:52:9d_available_entity': () Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/zha/entity.py", line 107, in async_state_changed self.async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 530, in async_write_ha_state self._async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 563, in _async_write_ha_state state = self._stringify_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 536, in _stringify_state if (state := self.state) is None: File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 371, in state value = self.native_value File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 175, in native_value return self.formatter(raw_state) File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 472, in formatter return self._channel.summa_formatter(value) File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/smartenergy.py", line 196, in _formatter_function return self._summa_format.format(value).lstrip() AttributeError: 'NoneType' object has no attribute 'format'
```

**To Reproduce**
Steps to reproduce the behavior: unknown

**Additional context**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.maeu01",
  "class": "zhaquirks.xiaomi.aqara.plug_mmeu01.Plug"
}
```

hermanops, 2022-02-14:
I also have problems with this device on version 2022.2.6

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.maeu01",
  "class": "zhaquirks.xiaomi.aqara.plug_mmeu01.Plug"
}`

mazerte, 2022-02-21:
Same here! I did a post on Home Assistant community to track my research around it 
https://community.home-assistant.io/t/aqara-smart-plug-reporting-energy-unavailable/395392

andrekrit, 2022-02-27:
@EagleStClair, I'm in the same boat. I've [posted](https://community.home-assistant.io/t/aqara-plug-sp-euc01-lumi-plug-maeu01-not-updating/393110) in the HA forums, but the only other people replying are the ones also experiencing the same or similar issues. I've got not idea how to get the people with the proper knowledge to actually take a look at it. 

I'm struggling with the same questions about the quirks and also can't seem to find the exact files to use to for custom quirks. I wish the documentation was a bit better.

leonardpitzu, 2022-02-27:
There are many more having this issue but keep only a watching eye on github. Me personally I jumped from deconz to zha and after being happy for 2 days all went down the drain because power and energy measurements are gone…

hermanops, 2022-02-27:
Not sure if it's a solution for everyone, but I switched to using Zigbee2mqtt instead of ZHA and now my device works perfect!

dmulcahey, 2022-02-27:
The error is because the `_summa_format` attribute for the smart energy metering cluster is `None`. This cluster is not replaced by the quirk from what I can tell so I am unsure how this was working previously. I will try to look at this closer later tonight.

tetele, 2022-02-28:
@dmulcahey it was NOT working previously. Prior to this upgrade, the maeu01 plug only had on/off functionality (at least in my tests), but did not produce any errors in the log. It had NO quirk applied, as the device signature did not match anything.

It's true that Aqara devices take a lot of attempts to be paired properly (i.e. by exposing all endpoints/clusters) and it might be that I didn't go through enough iterations to actually have all endpoints exposed, but before HA 2022.2, I've never had power monitoring, energy consumption etc., not even as non-functioning entities.

mazerte, 2022-02-28:
@dmulcahey and @tetele, on my side I have 2 smart plugs, 1 add to home assistant 1 months ago and working perfectly and 1 added 2 weeks ago reporting only on/off.

I dive deep on the issue (as far I can go) and you the error seems to be linked the `_summa_format` but both report the same value for Metering cluster (0x0702) attribute summa_formatting (0x0303).

So basically everything seems to work when a call the cluster directly but Home Assistant couldn't get those metrics. Maybe it's something a bit fragile at the onboarding the HA must be more resilant like putting a default value for `_summa_format`.

dmulcahey, 2022-02-28:
Can you read the attribute from the manage cluster screen and post a screenshot? Then if a value is returned for it, restart HA and tell me what happens with the non working device

mazerte, 2022-02-28:
Desk Power is the working one and Workshop Power have unavailable power metrics

[image]
[image]

tetele, 2022-02-28:
I have the exact same situation as @mazerte. I've also restarted HA and looked in the logs. I found these errors related to ZHA:

(my plugs have the following NWKs - 0xfcff for the one that's reporting energy consumption and 0x08bf for the one that doesn't)

```
Logger: homeassistant.components.zha.core.channels.base
Source: components/zha/core/channels/base.py:428
Integration: Zigbee Home Automation ([documentation](https://www.home-assistant.io/integrations/zha), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+zha%22))
First occurred: 18:29:44 (5 occurrences)
Last logged: 18:29:46

[0x930F:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('[0x930f:1:0x0006]: Message send failure'), DeliveryError('[0x930f:1:0x0006]: Message send failure'), DeliveryError('[0x930f:1:0x0006]: Message send failure'), DeliveryError('[0x930f:1:0x0006]: Message send failure')]
[0x930F:1:0x0002]: async_initialize: all attempts have failed: [DeliveryError('[0x930f:1:0x0002]: Message send failure'), DeliveryError('[0x930f:1:0x0002]: Message send failure'), DeliveryError('[0x930f:1:0x0002]: Message send failure'), DeliveryError('[0x930f:1:0x0002]: Message send failure')]
[0x69C8:1:0x0002]: async_initialize: all attempts have failed: [DeliveryError('[0x69c8:1:0x0002]: Message send failure'), DeliveryError('[0x69c8:1:0x0002]: Message send failure'), DeliveryError('[0x69c8:1:0x0002]: Message send failure'), DeliveryError('[0x69c8:1:0x0002]: Message send failure')]
[0x69C8:2:0x0006]: async_initialize: all attempts have failed: [DeliveryError('[0x69c8:2:0x0006]: Message send failure'), DeliveryError('[0x69c8:2:0x0006]: Message send failure'), DeliveryError('[0x69c8:2:0x0006]: Message send failure'), DeliveryError('[0x69c8:2:0x0006]: Message send failure')]
[0x69C8:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('[0x69c8:1:0x0006]: Message send failure'), DeliveryError('[0x69c8:1:0x0006]: Message send failure'), DeliveryError('[0x69c8:1:0x0006]: Message send failure'), DeliveryError('[0x69c8:1:0x0006]: Message send failure')]
```

```
Logger: homeassistant.components.sensor
Source: components/zha/core/channels/smartenergy.py:196
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 18:29:46 (1 occurrences)
Last logged: 18:29:46

Error adding entities for domain sensor with platform zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 614, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 782, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 539, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 572, in _async_write_ha_state
    state = self._stringify_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 545, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 371, in state
    value = self.native_value
  File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 174, in native_value
    return self.formatter(raw_state)
  File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 471, in formatter
    return self._channel.summa_formatter(value)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/smartenergy.py", line 196, in _formatter_function
    return self._summa_format.format(value).lstrip()
AttributeError: 'NoneType' object has no attribute 'format'

```

```
Logger: homeassistant
Source: components/zha/core/channels/smartenergy.py:196
First occurred: 18:29:46 (1 occurrences)
Last logged: 18:29:46

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 614, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 782, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 539, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 572, in _async_write_ha_state
    state = self._stringify_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 545, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 371, in state
    value = self.native_value
  File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 174, in native_value
    return self.formatter(raw_state)
  File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 471, in formatter
    return self._channel.summa_formatter(value)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/smartenergy.py", line 196, in _formatter_function
    return self._summa_format.format(value).lstrip()
AttributeError: 'NoneType' object has no attribute 'format'
```

dmulcahey, 2022-02-28:
After reading that value can you restart again please?

tetele, 2022-03-01:
Yes, but what info do you need after the second restart?

dmulcahey, 2022-03-01:
Once the attribute is read and cached that error should be gone after restarting and I’m wondering if the entities will be back

tetele, 2022-03-01:
So, I read the `summa_format` attribute on the device that's not reporting energy consumption, then restarted HA.

I can't find any ZHA-related log entries after restart (except for something regarding a device that ran out of battery, but that's expected).

The plug still doesn't report the energy.

[image]

Note that `summation_delivered`, `apparent_power`, `rms_current` and `rms_voltage` are `unavailable`, `smartenergy_metering` is `unknown`.

dmulcahey, 2022-03-01:
The unknown entity was previously unavailable right? Can you turn in the washing machine so it draws power or can you plug something else into the plug that uses power for a bit? We can address the other entities next but I want to know if that unknown flips to a value after the attribute is reported

tetele, 2022-03-01:
The unknown entity has been unknown for a while, IIRC.

I've plugged in an electric heater and nothing happens after a couple of minutes. At 1kW, it should start moving the needle pretty quickly, I guess.

[image]

tetele, 2022-03-01:
Here's something interesting about the other plug (the one that, apparently, DOES report energy consumption): I've plugged in the same heater for more than 1/10 of an hour, and the energy consumption (measured in kWh with single decimal precision) has not increased with 0.1kWh, as expected.

[image]

More over, here's the log

```
Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: 13:32:19 (54 occurrences)
Last logged: 14:25:19

Exception in async_state_changed when dispatching 'LUMI lumi.plug.maeu01_54:ef:44:10:00:0e:52:9d_available_entity': () Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/zha/entity.py", line 107, in async_state_changed self.async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 539, in async_write_ha_state self._async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 572, in _async_write_ha_state state = self._stringify_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 545, in _stringify_state if (state := self.state) is None: File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 371, in state value = self.native_value File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 174, in native_value return self.formatter(raw_state) File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 471, in formatter return self._channel.summa_formatter(value) File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/smartenergy.py", line 196, in _formatter_function return self._summa_format.format(value).lstrip() AttributeError: 'NoneType' object has no attribute 'format'
```

which is, basically, the same error as in the originally reported issue.

EagleStClair, 2022-03-02:
> 

That is what I'm seeing as well. Does it update if you get the zigbee attribute manually?
Mine does if I read current_summ_delivered (id: 0x0000):
[image]

Then the corresponding sensor is updated as well:
[image]

But HA never updates the sensors without me manually reading the attribute.

That's probably why ZHA consider my plug unavailable every 2 hours - no values are read/reported so 7200 sec is the limit for considering a mains powered device unavailable.
[image]

tetele, 2022-03-02:
For me it just reports 0 even if I query the attribute (in the Manage Clusters screen as well as in the HA interface).

Lampyridae1984, 2022-03-06:
I have similar problems. All sensors on the SP-EUC01 (lumi.plug.maeu01) show "unknown" as the value and turn completely unavailable after a reboot. The On/Off toggle function works just fine, but recently ZHA drops the entire device and I have to re-pair it. 😞

vinhui, 2022-03-06:
I have the same here. All the states show as unknown. Getting the attribute manually gives me a value.
[image]

Lampyridae1984, 2022-03-06:
> I have the same here. All the states show as unknown. Getting the attribute manually gives me a value.

This is precisely what I'm seeing as well and a short time after getting the value manually (somewhere around 1h), ZHA just drops/disconnects from the device completely. But then again, if I don't get the value manually, ZHA still disconnects eventually for no good reason. 🤷‍♂️

vinhui, 2022-03-07:
At the office we also have one of these plugs (the reason that I bought mine) that does work.
[image]

vinhui, 2022-03-07:
My home setup:
## System Health

version | core-2022.3.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.92-v7
arch | armv7l
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.22.0
Stage | running
Available Repositories | 1005
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | March 17, 2022, 1:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | eu-west-2-3.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 13.9 GB
disk_used | 8.7 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (10.0.2), Backup Hassio to Google Drive (1.7.10), File editor (5.3.3), AdGuard Home (4.4.5), Nextcloud Backup (0.16.8)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 1
views | 2
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

At the office:
(But we have had it working on different versions as well at the office.)
## System Health

version | core-2022.3.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.92-v8
arch | aarch64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.23.0
Stage | running
Available Repositories | 1004
Downloaded Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 13.8 GB
disk_used | 4.4 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.3.0), File editor (5.3.3)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 2
views | 3
mode | storage

</details>

Lampyridae1984, 2022-03-07:
Hmmm... what Zigbee Gateways are being used (both at home and at your workplace)? I'm using a Conbee II.

vinhui, 2022-03-07:
Both are conbee 2 sticks
Home:
[image]
Office:
[image]

But it seems like it sees it differently. This is also what I see in the device filter in Office HA:
[image]

Lampyridae1984, 2022-03-07:
Yeah, I'm not sure what to make of this. It seems completely random how the device is being recognized and what sensors are functioning. I'm unfortunately not skilled enough to make any further assessment. But it's good to have all the above posted data. I'm going to add my own tonight after work. Thank you @vinhui for all the info.

vinhui, 2022-03-07:
That seems to have worked somewhat, but I'm not getting the 'power usage at this moment' entity. In contrast, the device seems to report a temperature but I'n not sure if that is correct. 
[image]

slo1958, 2022-03-07:
If this help, I've observed similar issues with three plugs  "lumi.plug.maeu01". 

I'm running Home Assistant using docker on a Raspberry 3B+
Installation using only a SD card
Zigbee interface: dresden elektronik USB stick ConBee II
Mapped using  /dev/ttyACM0:/dev/ttyACM0
Config folder mapped outside the container, the same  configuration folder is used for all the tests described below.

1/ using HA 2022.3.1: when pairing a socket, I get one control, six sensors and two entries for diagnostic.  All sensors are 'unknown', only the control to switch the socket on/off and the diagnostic 'identify' works (the led on the socket is flashing when "press" is pressed) Same behaviour for all three sockets. Stopped the container.  

2/ using  HA 2022.2.9, remove the sockets, added them again. Same issue as above

3/ using  HA 2021.12.10, remove the sockets, added them again: working. I only have  one control and two sensors (current power and total power)

4/ stopped 2021.12.10 and restarted 2022.3.1. The socket still seems to properly update the two sensors. I get power consumption when the socket is on and the value seems correct. I tried by switching on and off the socket. Got two messages in the log (see below)

I removed and paired again one socket. Same issues as described under (1) 

Hope this helps. 

Log file attached

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:96 
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 19:31:32 (2 occurrences) 
Last logged: 19:31:32

[1871751104] Received invalid command: zha/device

2022-03-07 18:31:32 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [1871751104] Received invalid command: zha/device
2022-03-07 18:31:32 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [1871751104] Received invalid command: zha/device

[logfile.txt](https://github.com/zigpy/zha-device-handlers/files/8200097/logfile.txt)

vinhui, 2022-03-07:
That doesn't seem to change anything.

Lampyridae1984, 2022-03-07:
> For the maeu01 models: I am fairly certain I know what is wrong. is anyone willing to try to modify the quirk locally? if so, remove this line:
> 
> https://github.com/zigpy/zha-device-handlers/blob/b0bbbfb6b1f267b7d87d833b1ba5a5b0b127a9a1/zhaquirks/xiaomi/aqara/plug_mmeu01.py#L93
> 
> , then save the file and restart HA. After HA is back up, reconfigure the device via the reconfigure button on the device card.

My HA setup is a HA Blue. How do I access this file? With a Terminal addon?

tetele, 2022-03-08:
@Lampyridae1984 https://developers.home-assistant.io/docs/operating-system/debugging/

@dmulcahey it didn't do anything for me, apparently.

I have a feeling that it depends a lot on how the plug was initially paired (i.e. what data it exchanged with the coordinator). I say that because neither of my plugs had the signature that the quirk demanded initially, but after multiple pairing attempts, it sort of worked.

I would like to debug this in more detail, but I don't know exactly how to set things up. @dmulcahey are there any docs/tutorial for setting up and debugging zigpy? I have a bit of experience in software development and debugging, but not with Python. However, I'm willing to give it a go, as long as I can learn a bit about how to set things up.

Stefano0042, 2022-03-09:
@dmulcahey Hi David,  I like to pass you a feedback, ..strictly followed your suggestion (removed line 93), but without any relevant result, no sensors found.
[image]

Stefano0042, 2022-03-11:
@dmulcahey by updating the aqara(xiaomi) firmware to [09.2020 ](https://github.com/Koenkk/zigbee-OTA/blob/master/images/Xiaomi/OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946.ota) I got a small improvement:
quirks is detected, but after the first HAcore 2022.3.3 reboot few sensors, some of them not used by the plug (see rms_voltage and rms_current...), are now on unknow status.
[image]

tetele, 2022-03-11:
@Stefano0042 how did you update the firmware? Do you need an Aqara hub to do so?

Stefano0042, 2022-03-12:
@tetele Hi Tudor if you do not have a Deconz/Conbee II  instance running somewhere, you can go for the official [zigpy  OTA update procedure](https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates#enabling-ota-updates).

tetele, 2022-03-14:
> you can go for the official [zigpy OTA update procedure](https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates#enabling-ota-updates).

Thanks, @Stefano0042! My bad, the question was more related to where you got the firmware from, rather than how to upload it to the plug :D I take Deconz takes care of that automatically, and I don't have it.

Stefano0042, 2022-03-14:
Hi, normally you can have a chance in Koenkk repository [(just click here) 09.2020](https://github.com/Koenkk/zigbee-OTA/blob/master/images/Xiaomi/OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946.ota)

cloudbr34k84, 2022-03-17:
I have 3 of these and this is my signature

manufacturer": "LUMI",
  "model": "lumi.plug",
  "class": "zhaquirks.xiaomi.aqara.plug.Plug"

[image]

Lampyridae1984, 2022-03-17:
@cloudbr34k84 your screenshot is not loading

cloudbr34k84, 2022-03-17:
> @cloudbr34k84 your screenshot is not loading

Sorted, thanks mate

Lampyridae1984, 2022-03-18:
@cloudbr34k84, let me see if I understand you correctly. Are you saying that you have three "lumi.plug.maeu01" integrated on the latest build of HA? AND The quirk used is "zhaquirks.xiaomi.aqara.plug.Plug"? Did you change the quirk or is this how it got configured? 

I bought mine quite recently and this is my configuration through ZHA with a Conbee II:
[image]

cloudbr34k84, 2022-03-18:
> @cloudbr34k84, let me see if I understand you correctly. Are you saying that you have three "lumi.plug.maeu01" integrated on the latest build of HA; and on which HA version (if you can remember) did you initially start using them? Did you change the quirk or is this how it got configured?
> 
> I bought mine quite recently and this is my configuration with a Conbee II: [image]

Hey sorry mate, i have updated my original post and answered these questions.

Lampyridae1984, 2022-03-18:
> Hey sorry mate, i have updated my original post and answered these questions.

Cheers mate. Interesting. I'm just guessing now, so either the AU plugs hardware is not 100% equal to the EU version or they might use significantly different firmware.

Does anyone know if you can manually change the quirk?

Lampyridae1984, 2022-03-19:
The following issue seems to be linked/same as this one:
[https://github.com/home-assistant/core/issues/65292](url)

martbock, 2022-03-22:
I had the same problem with a freshly installed Home Assistant (running as a container on a RPi 4 with ZHA and a RaspBee II) and the SP-EUC01 Aqara smart plug.

> For the maeu01 models: I am fairly certain I know what is wrong. is anyone willing to try to modify the quirk locally? if so, remove this line:
> 
> https://github.com/zigpy/zha-device-handlers/blob/b0bbbfb6b1f267b7d87d833b1ba5a5b0b127a9a1/zhaquirks/xiaomi/aqara/plug_mmeu01.py#L93
> , then save the file and restart HA. After HA is back up, reconfigure the device via the reconfigure button on the device card.
> 
> z2m binds the clusters on these: https://github.com/Koenkk/zigbee-herdsman-converters/blob/66bfcdc9405abebe70906b279e96f12de73f26a7/devices/xiaomi.js#L1048

 After applying this patch, most identified sensors are still showing "Not available" – however `smartenergy_metering_summation_delivered` now actually shows a value whereas before it was dead too.

What I find strange is the alleged device temperature value of 0,2°C which doesn't make any sense.

[image]

Stefano0042, 2022-03-22:
@dmulcahey ..summarizing an unwanted breaking change in 2023.m.r has pushed aqara plug into the "red sea", adding lumi.plug.maeu01 in the quirks models list has not been enough to rescue the plug.

husjon, 2022-03-22:
Hi, I purchased 3 of these EU plugs today hoping to monitor a few of my appliances.

After going through all the comments I was able to get the plugs to report summed metering, but that's it.
Following these steps.
1. Comment out: https://github.com/zigpy/zha-device-handlers/blob/b0bbbfb6b1f267b7d87d833b1ba5a5b0b127a9a1/zhaquirks/xiaomi/aqara/plug_mmeu01.py#L93
2. Remove existing device from the ZHA integration
3. Add device

Before commenting out `SKIP_CONFIGURATION`:
[image]

After commenting out `SKIP_CONFIGURATION`:
[image]

Stefano0042, 2022-03-24:
@husjon Hi, but here we are still missing a fundamental sensor: active power, and I'm quite sure a multiplier factor  x100 on device temperature.

husjon, 2022-03-24:
@Stefano0042, yep. I was mainly adding my comment as a summary of what was mentioned previously.
I'm still hoping for a solution as I mainly bought these plugs for appliance monitoring in addition to the switching.

I haven't been able to do much troubleshooting lately, but if someone could point me in the right direction I'll try to assist as needed.

jinovak, 2022-03-25:
I have the same problem with newly added plugs. First one i bought and connect in January works well. Others were bought & connected in february and these are not working. All sensors are unavailable.   

First plug connected in January (dont remember version of HA) is still working well
[image]

Lately installed are unavailable - only on/off is available
[image]

RossEm0174, 2022-03-25:
Hello, today I got my Aqara socket, and its currently also not working...

[image]

dmulcahey, 2022-04-09:
Can someone with the MAEU01 plug please test the quirk in https://github.com/zigpy/zha-device-handlers/pull/1479

DO NOT USE THE CUSTOM DEPS ADDON

1 - Remove the existing device via the remove button on the device card in HA
2 - configure the quirk as a local quirk
3 - restart HA
4 - pair the plug to HA again

vinhui, 2022-04-09:
I did that, but it didn't seem to make a difference for me.

The device after I re-paired it.
[image]

The changes I made:
[image]

torge70, 2022-04-09:
> Can someone with the MAEU01 plug please test the quirk in #1479
> 
> DO NOT USE THE CUSTOM DEPS ADDON
> 
> 1 - Remove the existing device via the remove button on the device card in HA 2 - configure the quirk as a local quirk 3 - restart HA 4 - pair the plug to HA again

Device is added. On_Off works.
Power meter does not work.

husjon, 2022-04-09:
On a clean 2022.4 docker instance I did the following steps:
1. Patch local quirk
2. Remove device
3. Restart HA
4. Add device

It is now missing `sensor.electrical_measurement`. (this was there with no quirk assigned)
[image]

Also the device temperature is still off by a factor of 10. (same as no quirk)
[image]

husjon, 2022-04-09:
To add to my previous comment.
Here is the same plug with no quirk assigned. (this includes the same steps as above, but with the quick assignment removed)

[image]

torge70, 2022-04-10:
Looks like this cluster value gives the current power consumtion in W. I don'd understand how to modify the quirk to pass this value correctly to HA.

ElectricalMeasurementCluster (Endpoint id: 1, Id: 0x0b04, Type: in)
active_power (id: 0x050b)
Value: 936

bendikrb, 2022-04-13:
Same problem here. I have two (seemingly) identical plugs - one of which doesn't get quirk applied..
The [quirk patch](https://github.com/zigpy/zha-device-handlers/issues/1337#issuecomment-1094012936) from @dmulcahey does not appear to make any difference here..
Following log output (with patch applied) which I suppose could reveal the/an issue (?);
```
2022-04-13 17:41:34 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='lumi.plug.maeu01' manuf='LUMI' nwk=0xEBCA ieee=54:ef:44:10:00:2c:7b:6d is_initialized=True>
2022-04-13 17:41:34 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.plug.maeu01 (54:ef:44:10:00:2c:7b:6d)
2022-04-13 17:41:34 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.plug_mmeu01.Plug'>
2022-04-13 17:41:34 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 242} {1, 242, 21, 31}
```
Signature:
```json
{
  "node_descriptor":"NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints":{
    "1":{
      "profile_id":260,
      "device_type":"0x0051",
      "in_clusters":[
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xfcc0"
      ],
      "out_clusters":[
        "0x000a",
        "0x0019"
      ]
    },
    "21":{
      "profile_id":260,
      "device_type":"0x0051",
      "in_clusters":[
        "0x000c"
      ],
      "out_clusters":[
        
      ]
    },
    "31":{
      "profile_id":260,
      "device_type":"0x0051",
      "in_clusters":[
        "0x000c"
      ],
      "out_clusters":[
        
      ]
    },
    "242":{
      "profile_id":41440,
      "device_type":"0x0061",
      "in_clusters":[
        
      ],
      "out_clusters":[
        "0x0021"
      ]
    }
  }
}

```

[86 more replies]

PR #1479 (2022-04-22): Fix Xiaomi MAEU01 plug
Fixes #1337
Fixes: https://github.com/home-assistant/core/issues/67644
Fixes: https://github.com/home-assistant/core/issues/65292
closed 2022-04-22 by dmulcahey