zigpy/zha-device-handlers #1341: Add Quirks for Sunricher SG9092 Thermostat

Maxcrouz73, 2022-02-10:
Hello,

I try to integrated this thermostat, [SR-ZG9092A instruction.pdf](https://github.com/zigpy/zha-device-handlers/files/8042115/SR-ZG9092A.instruction.pdf).

Everythings work great exept to set the away mode (i able to read it), i seem to be normal because it's not a standard attribut:

[image]

[image]

So on the endpoint 1, on the cluster thermostat 0x0201, on the attribut 0x2002 i cannot set any order because it's not a standard attribut, i have no clue how to make a quirks...

Here is my Json:
```
{
    "name": "Sunricher.ZG9092",
    "configuration": {
        "heating": [],
        "cooling": [],
        "stoping": [],
        "window": [],
        "failure": [],
        "failureActuator": [],
        "orderChange": [],
        "existingMode": []
    },
    "commands": [
        {
            "logicalId": "1::513::0",
            "generic_type": "THERMOSTAT_TEMPERATURE",
            "name": "Température",
            "type": "info",
            "subType": "numeric",
            "isHistorized": "1",
            "unite": "°C",
            "configuration": {
                "repeatEventManagement": "never",
                "calculValueOffset": "#value# \/ 100",
                "historizeRound": "1"
            },
            "isVisible": "1"
        },
        {
            "logicalId": "1::513::18",
            "generic_type": "THERMOSTAT_SETPOINT",
            "name": "Consigne ",
            "type": "info",
            "subType": "numeric",
            "isHistorized": "0",
            "unite": "°C",
            "configuration": {
                "calculValueOffset": "#value# \/ 100"
            },
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "isVisible": "0"
        },
        {
            "logicalId": "attributes::1::in::513::18::#slider#*100",
            "generic_type": "THERMOSTAT_SET_SETPOINT",
            "name": "Regalge consigne",
            "type": "action",
            "subType": "slider",
            "isHistorized": "0",
            "unite": "°C",
            "configuration": {
                "minValue": "6",
                "maxValue": "28"
            },
            "template": {
                "dashboard": "core::button",
                "mobile": "core::button"
            },
            "value": "Consigne ",
            "isVisible": "1"
        },
        {
            "logicalId": "1::513::20",
            "name": "Consigne Hors Gel",
            "type": "info",
            "subType": "numeric",
            "isHistorized": "1",
            "configuration": {
                "calculValueOffset": "#value# \/ 100",
                "repeatEventManagement": "never"
            },
            "isVisible": "1"
        },
        {
            "logicalId": "attributes::1::in::513::20::#slider#",
            "name": "Réglage Consigne Hors Gel",
            "type": "action",
            "subType": "slider",
            "isHistorized": "0",
            "configuration": {
                "minValue": "0",
                "maxValue": "15",
                "calculValueOffset": "#value#*100",
                "actionConfirm": "0"
            },
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "value": "Consigne Hors Gel",
            "isVisible": "1"
        },
        {
            "logicalId": "1::513::28",
            "generic_type": "THERMOSTAT_STATE_NAME",
            "name": "Mode ",
            "type": "info",
            "subType": "string",
            "isHistorized": "0",
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "isVisible": "1"
        },
        {
            "logicalId": "attributes::1::in::513::28::0",
            "name": "Arrêt",
            "type": "action",
            "subType": "other",
            "isHistorized": "0",
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "value": "Mode ",
            "isVisible": "1"
        },
        {
            "logicalId": "attributes::1::in::513::28::1",
            "name": "Auto",
            "type": "action",
            "subType": "other",
            "isHistorized": "0",
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "value": "Mode ",
            "isVisible": "1"
        },
        {
            "logicalId": "attributes::1::in::513::28::4",
            "name": "Manu",
            "type": "action",
            "subType": "other",
            "isHistorized": "0",
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "value": "Mode ",
            "isVisible": "1"
        },
        {
            "logicalId": "attributes::1::in::513::28::8",
            "name": "Dry",
            "type": "action",
            "subType": "other",
            "isHistorized": "0",
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "value": "Mode ",
            "isVisible": "1"
        },
        {
            "logicalId": "1::513::8194",
            "name": "Etat Hors Gel",
            "type": "info",
            "subType": "string",
            "isHistorized": "1",
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "isVisible": "1"
        },
        {
            "logicalId": "attributes::1::in::513::8194::1",
            "name": "Hors Gel",
            "type": "action",
            "subType": "other",
            "isHistorized": "0",
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "isVisible": "1"
        },
        {
            "logicalId": "1::513::41",
            "name": "Etat Chauffage",
            "type": "info",
            "subType": "string",
            "isHistorized": "1",
            "configuration": {
                "minValue": "0",
                "maxValue": "1"
            },
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "isVisible": "1"
        },
        {
            "logicalId": "1::513::16",
            "name": "Etat Compensation",
            "type": "info",
            "subType": "numeric",
            "isHistorized": "0",
            "unite": "°C",
            "configuration": {
                "minValue": "-30",
                "maxValue": "30"
            },
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "isVisible": "1"
        },
        {
            "logicalId": "attributes::1::in::513::16::#slider#",
            "name": "Compensation",
            "type": "action",
            "subType": "slider",
            "isHistorized": "0",
            "configuration": {
                "minValue": "-30",
                "maxValue": "30",
                "lastCmdValue": "0.0"
            },
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "value": "Etat Compensation",
            "isVisible": "1"
        },
        {
            "logicalId": "1::2820::1291",
            "generic_type": "POWER",
            "name": "Puissance ",
            "type": "info",
            "subType": "numeric",
            "isHistorized": "1",
            "unite": "W",
            "configuration": {
                "historizeRound": 2,
                "calculValueOffset": "#value# \/ 10"
            },
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "isVisible": "1"
        },
        {
            "logicalId": "1::1794::0",
            "generic_type": "CONSUMPTION",
            "name": "Consommation",
            "type": "info",
            "subType": "numeric",
            "isHistorized": "1",
            "unite": "kWh",
            "configuration": {
                "historizeRound": 2,
                "calculValueOffset": "#value# \/ 1000"
            },
            "template": {
                "dashboard": "core::default",
                "mobile": "core::default"
            },
            "isVisible": "1"
        }
    ],
    "ref": "Sunricher.ZG9092"
}

 
```

owangen, 2022-04-20:
@Maxcrouz73 have you found out why its not working? I'm implementing support in z2m and experiencing same problem when trying to send custom attributtes to this thermostat. I dont get any error message, but the values im sending does not reflect on the thermostat.

owangen, 2022-04-23:
Got everything working with sending zigbee to the device, it didn't work because of a bug caused by me.

ipeacocks, 2022-08-02:
It's a bit off-top but where to buy it?

closed 2023-02-05 by github-actions[bot]