zigpy/zha-device-handlers #1343: [BUG] Tuya Air Quality detector exposes only VOC level sensor

sergey-suloev, 2022-02-12:
[image]

sergey-suloev, 2022-02-12:
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0x040d",
        "0x042b",
        "0x042e",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_yvx5lh6k",
  "model": "TS0601",
  "class": "zhaquirks.tuya.air.ts0601_air_quality.TuyaCO2SensorGPP"
}
```

sergey-suloev, 2022-02-12:
It seems like the problem was discussed here https://github.com/zigpy/zha-device-handlers/issues/1004#issuecomment-1002206182

javicalle, 2022-02-12:
Your device (already supported) is getting the correct quirks:
https://github.com/zigpy/zha-device-handlers/blob/23a166902e50d5c32019e0d4d32a968715041bb6/zhaquirks/tuya/air/ts0601_air_quality.py#L74-L86

Is your device reporting values?
Do you have any error in logs?

Can you enable and attach the debug logs for:
```yaml
  zigpy: debug
  zigpy.zcl: debug
  custom_zha_quirks: debug
  zhaquirks.tuya: debug
  homeassistant.components.zha: debug
```

Regards.

sergey-suloev, 2022-02-12:
@javicalle I know it is formally supported but I have this problem

sergey-suloev, 2022-02-12:
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/8053431/home-assistant.log)

sergey-suloev, 2022-02-12:
@javicalle  I noticed that if you put the json node below  under  TuyaCO2Sensor.ENDPOINTS I am able to see all the sensors.

            242: {
                  PROFILE_ID: 41440,
                  DEVICE_TYPE: 0x0061,
                  INPUT_CLUSTERS: [ ],
                  OUTPUT_CLUSTERS: [33],
            }

The guy here is saying the same thing https://github.com/zigpy/zha-device-handlers/issues/1004#issuecomment-1003404368

sergey-suloev, 2022-02-12:
Another workaround is to remove ("_TZE200_yvx5lh6k", "TS0601"),  from the TuyaCO2Sensor's signature MODELS_INFO node.

sergey-suloev, 2022-02-12:
@javicalle   I think it is a big mistake to support a device by 2 different quirks.

MattWestb, 2022-02-12:
Im sorry i was adding the GPP version and was not knowing if the same device ID is being used without GPP that was already in the quirk so i was leaving it then it cant being triggered if the device is having GPP endpoint and for not braking devices for user that  is having devices without it.

I have testing TS004F and if having more matching quirks the system is using the first class i the quirk (from start for the file).
But if putting the same device ID is 2 different quirk is being one other story that i thing its a bad one.

javicalle, 2022-02-12:
It is something here that I can't understand.

From the capture at the firts comment, the device is loading the `TuyaCO2SensorGPP` quirk.
But it seems that the problem is that device is using the `TuyaCO2Sensor` quirk. Then, the solution comes from adding the `GreenPowerProxy` cluster  or removing `_TZE200_yvx5lh6k` from `MODELS_INFO`, both in `TuyaCO2Sensor` quirk.

There's something weird here. I understand that the first quirk that matches will be loaded, but I don't understand why it indicates that it is loading the another one.

sergey-suloev, 2022-02-12:
@javicalle  I made a pull request for another fix previously. I don't know why new commits to my fork are added to this pull request.

sergey-suloev, 2022-02-12:
> It is something here that I can't understand.
>
> From the capture at the firts comment, the device is loading the `TuyaCO2SensorGPP` quirk. But it seems that the problem is that device is using the [...]

sergey-suloev, 2022-02-12:
> It is something here that I can't understand.
>
> From the capture at the firts comment, the device is loading the `TuyaCO2SensorGPP` quirk. But it seems that the problem is that device is using the [...]

It is very strange indeed. I think this problem comes for the fact that more than 1 class is handling the same device. This is absolutely wrong and causes issues.

javicalle, 2022-02-15:
Once PR #1346 is merged, this issue can be closed.
Fix (probably) will come in the next HA version.

closed 2022-02-15 by sergey-suloev

nazar-pc, 2023-12-18:
Has this regressed again? I also only see a single sensor on my `_TZE200_yvx5lh6k` and a few other people seem to have the same issue in https://github.com/zigpy/zha-device-handlers/pull/1410 and https://github.com/zigpy/zha-device-handlers/issues/2395

Reconfiguration sees all, but only one entity is being created.

nazar-pc, 2023-12-23:
Seems to work fine again on 2023.12.3, detects 5 entities

spusuf, 2024-03-12:
` _TZE200_c2fmom5z ` only detects VOC in 2024.3.0, havent tested with older versions as I just got the sensor today