zigpy/zha-device-handlers #1348: [Device Support Request]

ryanturnersmith, 2022-02-13:
**Is your feature request related to a problem? Please describe.**
New blind roller from Zemismart. When connecting to ZHA it connects fine but there are no controls. 

**Describe the solution you'd like**
Please add device to list

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_68nvbio9",
  "model": "TS0601",
   "class": "zigpy.device.Device"
}`

**Additional context**
I manually added __TZE200_68nvbio9_ it to ts0601_cover.py and __init__.py which resolve my problem

MattWestb, 2022-02-13:
If the device is working OK can you pleas making one PR for adding in the ZHA next release ?

ryanturnersmith, 2022-02-13:
I am very happy to do that, but this is new to me and not fully sure what I need to do. 
Can you assist me understand the process?

MattWestb, 2022-02-13:
Im not one coder only copy and past one.

Start going to one of the files you like to changing in the git and clicking on the edit icon and git like doing one copy in your git.
Then do the edit you like on your git (the branch you was forking and not  others!!).
Then all is made go it the root of the brash and clicking commit the changes and it shall sending the changes to the original as one PR.

Ask if getting problems !!

closed 2022-02-15 by dmulcahey

chrisborell, 2022-02-25:
I've successfully added this into HA. However i get this error when trying to move the blind up or down (see error log below)
Model: _TZE200_68nvbio9

"Failed to call service cover/open_cover. '_TZE200_68nvbio9"

[548004506400] '_TZE200_68nvbio9'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 190, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1630, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1667, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 668, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 930, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 705, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/cover.py", line 135, in async_open_cover
    res = await self._cover_channel.up_open()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 59, in wrapper
    result = await command(*args, **kwds)
  File "/config/deps/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py", line 1040, in command
    TUYA_COVER_COMMAND[manufacturer][command_id],
KeyError: '_TZE200_68nvbio9'