zigpy/zha-device-handlers #1352: [BUG] Lonsonho Tuya Zigbee Wireless Smart Scene Switch wrongly identified

SparkyDan555, 2022-02-13:
I have just got a [Lonsonho Tuya Zigbee Wireless Smart Scene Switch 4 Gang](https://www.aliexpress.com/item/4001095272467.html) which when added to ZHA shows up as 4 switches and a battery sensor. Nowhere is the button state exposed. I have added this using Zigbee2MQTT and it is exposed as expected and is useable in automations. My coordinator is a [Sonoff CC2531](https://itead.cc/product/cc2531-usb-dongle/) stick flashed with [Z-stack firmware](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20211115.zip). Please see the attached screenshots for details. 

What ZHA shows:
[image]

What Zigbee2MQTT shows:
[image]

Device signature:

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0xe000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_uaa99arv",
  "model": "TS0044",
  "class": "zigpy.device.Device"
}`

Logs:

Device 0x21f3 (a4:c1:38:fb:52:7a:17:20) left the network
[0x21F3:1:0x0006]: failed to get attributes '['on_off']' on 'on_off' cluster: Request failed after 5 attempts: <Status.MAC_TRANSACTION_EXPIRED: 240>
[0x21F3:1:0x0006]: async_initialize: retryable request #3 failed: Request failed after 5 attempts: <Status.MAC_TRANSACTION_EXPIRED: 240>. Retrying in 3.0s
[0x21F3:1:0x0001]: failed to get attributes '['battery_voltage', 'battery_percentage_remaining']' on 'power' cluster: Request failed after 5 attempts: <Status.NWK_INVALID_REQUEST: 194>
[0x21F3:1:0x0001]: async_initialize: retryable request #3 failed: Request failed after 5 attempts: <Status.NWK_INVALID_REQUEST: 194>. Retrying in 2.3s
[0x21F3:1:0x0006]: initializing channel: from_cache: False
[0x21f3] Extending timeout for 0x4f request
[0x21F3:1:0x0001]: initializing channel: from_cache: False
[... 175 lines cut ...]
[0x21F3:3:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>')]
[0x21f3] Extending timeout for 0x6f request
[0x21F3:3:0x0006]: 'async_initialize' stage succeeded
[0x21F3:4:0x0006]: failed to get attributes '['on_off']' on 'on_off' cluster: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>
[0x21F3:4:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>')]
[0x21F3:1:0x0000]: finished channel configuration
[0x21F3:1:0x0019]: finished channel configuration
[0x21F3:4:0x0006]: 'async_initialize' stage succeeded
[0x21f3] Extending timeout for 0x70 request
[0x21F3](TS0044): power source: Battery or Unknown
[0x21F3](TS0044): completed initialization
[0x21F3:1:0x0006]: Failed to bind 'on_off' cluster: 
[0x21f3] Extending timeout for 0x71 request
[0x21F3:1:0x0006]: failed to set reporting on 'on_off' cluster for: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>
[0x21F3:1:0x0006]: finished channel configuration
[0x21f3] Extending timeout for 0x72 request
[0x21F3:1:0x0001]: Failed to bind 'power' cluster: 
[0x21f3] Extending timeout for 0x73 request
[0x21F3:1:0x0001]: failed to set reporting on 'power' cluster for: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>
[0x21F3:1:0x0001]: finished channel configuration
[0x21f3] Extending timeout for 0x74 request
[0x21F3:1:0x0006]: 'async_configure' stage succeeded
[0x21F3:1:0x0001]: 'async_configure' stage succeeded
[0x21F3:1:0x0000]: 'async_configure' stage succeeded
[0x21F3:1:0x0019]: 'async_configure' stage succeeded

MattWestb, 2022-02-14:
Link for the (likely) same remote but working as one light switch https://de.aliexpress.com/item/4000298926256.html?gatewayAdapt=glo2deu&spm=a2g0o.detail.1000023.7.77d95bcdDpxTOx.

SparkyDan555, 2022-02-14:
I have just tested your quirk and it works perfectly! Thanks!

MattWestb, 2022-02-14:
Great !!
If you dont have any bad things i can making one PR so its in the next HA release and dont need local quirks.

MattWestb, 2022-02-14:
PR made https://github.com/zigpy/zha-device-handlers/pull/1354

PR #1354 (2022-02-15): Adding TS0044 _TZ3000_uaa99arv
New device class that is adding the secret 0xe000 tuya cluster.
Fixes #1352
closed 2022-02-15 by dmulcahey