zigpy/zha-device-handlers #1358: [BUG] Aqara Motion sensor always reports the temperature 29C

sergey-suloev, 2022-02-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here. Please include full debug logs as well as the device signature.

sergey-suloev, 2022-02-15:
[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/8069550/home-assistant.log)

dontinelli, 2022-02-15:
I have an issue with the Aqara Single Switch T1, which could be related to the above:
Since the device temperature is reported, it seems to be constantly at 0.2°C, while I would expect a temperature of approx. 15°C. Interestingly, when I get the device temperature over the "cluster" dialog, it reports a device temperature of (currently) 15°C.
Another switch (also located in the cellar, but in a different room) reports a temperature of 0.1°C or 0.2°C in the entity but 16°C in the "cluster" dialog (which could be correct). A third switch constantly reports 0.2°C in the entity but 25°C in the "cluster" dialog (which seems to be approx. 5° too high, but this could also be an inaccuracy of the measurement, I guess).

closed 2022-08-21 by github-actions[bot]