zigpy/zha-device-handlers #1365: [BUG?] Osram CLA60 TW

Tropaion, 2022-02-16:
I recently got this osram device: [Amazon](https://www.amazon.de/gp/product/B074PZKX4K).
This is the device signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.ExtendedSimpleDescriptorListAvailable|ExtendedActiveEndpointListAvailable: 3>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "3": {
      "profile_id": 49246,
      "device_type": "0x0220",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xfc0f"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "OSRAM",
  "model": "CLA60 TW OSRAM",
  "class": "cla60tw.CLA60TW"
}
```
It paired in ZHA fast and without problem, but got this error:
```
2022-02-16 23:58:05 WARNING (MainThread) [zigpy_znp.api] Failed to parse broken GeneralFrame(header=CommandHeader(id=0x9F, subsystem=Subsystem.ZDO, type=CommandType.AREQ), data=b'\xD4\x9B\x84\x01\x8B') as <class 'zigpy_znp.types.commands.ZDO.ParentAnnceRsp.Callback'>
2022-02-16 23:58:05 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MsgCbIncoming.Callback(Src=0x9BD4, IsBroadcast=<Bool.false: 0>, ClusterId=32799, SecurityUse=0, TSN=3, MacDst=0x0000, Data=b'\x84\x01\x8B')
2022-02-16 23:58:05 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 654, in on_zdo_message
args, data = list_deserialize(data, types)
File "/usr/local/lib/python3.9/site-packages/zigpy/types/__init__.py", line 9, in deserialize
value, data = type_.deserialize(data)
File "/usr/local/lib/python3.9/site-packages/zigpy/types/basic.py", line 622, in deserialize
item, data = cls._item_type.deserialize(data)
File "/usr/local/lib/python3.9/site-packages/zigpy/types/basic.py", line 648, in deserialize
item, data = cls._item_type.deserialize(data)
File "/usr/local/lib/python3.9/site-packages/zigpy/types/basic.py", line 140, in deserialize
raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
```
Now I thought this is a problem with the quirk since there isn't one for this device, so I created one: 
```
"""Osram CLA60 TW device."""
from zigpy.profiles import zll
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.osram import OSRAM, OsramLightCluster

class CLA60TW(CustomDevice):
    """Osram CLA60 TW device."""

    signature = {
        # <SimpleDescriptor endpoint=3 profile=49246 device_type=544
        # device_version=2 input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096, 64527]
        # output_clusters=[25]>
        MODELS_INFO: [(OSRAM, "CLA60 TW OSRAM")],
        ENDPOINTS: {
            3: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.COLOR_TEMPERATURE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                    OsramLightCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            3: {
                PROFILE_ID: zll.PROFILE_ID,
                DEVICE_TYPE: zll.DeviceType.COLOR_TEMPERATURE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                    OsramLightCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        }
    }
```
Now with this quirk applied I cant controll color temperature anymore and the errors are still there.

Tropaion, 2022-02-16:
Also, I noticed that always when sending a command to the device, there is an exception even though the device is accepting the command:
```
2022-02-17 00:06:51 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_off, service_data=entity_id=light.fabian_deckenlicht_on_off>
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x9BD4), DstEndpoint=3, DstPanId=0x0000, SrcEndpoint=1, ClusterId=6, TSN=57, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=30, Data=b'\x01\x39\x00')
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x9BD4, Relays=[])
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=57)
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=6, SrcAddr=0x9BD4, SrcEndpoint=3, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=65, SecurityUse=<Bool.false: 0>, TimeStamp=3000158, TSN=0, Data=b'\x18\x3D\x0A\x00\x00\x10\x00', MacSrcAddr=0x9BD4, MsgResultRadius=29)
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy.zcl] [0x9bd4:3:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=61 command_id=Command.Report_Attributes>
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy.zcl] [0x9bd4:3:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.false>)]]
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy.zcl] [0x9bd4:3:0x0006] Attribute report received: on_off=0
2022-02-17 00:06:51 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.fabian_deckenlicht_on_off, old_state=<state light.fabian_deckenlicht_on_off=on; min_mireds=153, max_mireds=370, supported_color_modes=['color_temp'], color_mode=color_temp, brightness=51, color_temp=208, hs_color=(26.842, 21.929), rgb_color=(255, 224, 199), xy_color=(0.379, 0.352), off_brightness=None, friendly_name=On/Off, supported_features=43 @ 2022-02-17T00:05:54.294599+01:00>, new_state=<state light.fabian_deckenlicht_on_off=off; min_mireds=153, max_mireds=370, supported_color_modes=['color_temp'], off_brightness=None, friendly_name=On/Off, supported_features=43 @ 2022-02-17T00:06:51.797585+01:00>>
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=8, SrcAddr=0x9BD4, SrcEndpoint=3, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=65, SecurityUse=<Bool.false: 0>, TimeStamp=3000802, TSN=0, Data=b'\x18\x3E\x0A\x00\x00\x20\x33', MacSrcAddr=0x9BD4, MsgResultRadius=29)
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy.zcl] [0x9bd4:3:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=62 command_id=Command.Report_Attributes>
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy.zcl] [0x9bd4:3:0x0008] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint8_t, value=51>)]]
2022-02-17 00:06:51 DEBUG (MainThread) [zigpy.zcl] [0x9bd4:3:0x0008] Attribute report received: current_level=51
2022-02-17 00:06:51 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9BD4:3:0x0008]: received attribute: 0 update with value: 51
2022-02-17 00:06:52 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=8, SrcAddr=0x9BD4, SrcEndpoint=3, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=72, SecurityUse=<Bool.false: 0>, TimeStamp=3024988, TSN=0, Data=b'\x18\x3F\x0A\x00\x00\x20\x33', MacSrcAddr=0x9BD4, MsgResultRadius=29)
2022-02-17 00:06:52 DEBUG (MainThread) [zigpy.zcl] [0x9bd4:3:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=63 command_id=Command.Report_Attributes>
2022-02-17 00:06:52 DEBUG (MainThread) [zigpy.zcl] [0x9bd4:3:0x0008] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint8_t, value=51>)]]
2022-02-17 00:06:52 DEBUG (MainThread) [zigpy.zcl] [0x9bd4:3:0x0008] Attribute report received: current_level=51
2022-02-17 00:06:52 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9BD4:3:0x0008]: received attribute: 0 update with value: 51
2022-02-17 00:06:56 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9BD4:3:0x0006]: command failed: 'off' args: '()' kwargs '{}' exception: ''
2022-02-17 00:06:56 DEBUG (MainThread) [homeassistant.components.zha.entity] light.fabian_deckenlicht_on_off: turned off: 
```

MattWestb, 2022-02-17:
Ok im Swedish and not "so young" (=one AS) but i can only recommending OSRAM light and plugs to person that is my enemies of some reason !!

Z2M have sniffing some OSRAM devices that is sending corrupted packages and i think you have the same problem.

**Pleas** do one sniff then sending the same commands to the device so we can taking o looking what is sending in the radio layer.

The next is trying finding updated firmware for the device if they have released one and if its still online (i think they was taking some server down in the new year) and updating it if you can.

As im one AS i can only recommending returning it if you can and getting somthing that is having better working firmware in there device but can being good having as "reference device" in the future.

Edit: its looks like one update is online for it from 2019 if using this link https://update.ledvance.com/firmware-overview?submit=all and putting in` CLA60`.

Configuring local OTA in HA and putting the file in the directory and restart HA and re power the light and see if its like the update.
The Image Type you is getting in the log then the device is responding to OTA request so you is knowing if its right firmware you is trying updating it with.

MattWestb, 2022-02-17:
I think you can changing the topic to "undocumented future" its sounds better then bug ;-)

closed 2022-02-22 by Tropaion