zigpy/zha-device-handlers #1366: [BUG] LIDL Livarno TS0502A brightness and color temp oddities

pojlFDlxCOvZ4Kg8y1l4, 2022-02-16:
**Describe the bug**
Changing the brightness color temperature makes HA do odd stuff.
"Flickering" of the light in HA when changing color temp, results in light set to off in lovelace but isnt actually off.
Switching on and off works well, as well as pairing and reconfiguring.
What does seem to be working but "single clicking" the brightness, lovelace will do odd stuff but if you click on 90% or 10% brightness it seems that the light would do that. Lovelace is not updating to the new setting. This does not work properly at all with temperature color, it will choose something random it seems.

**To Reproduce**
Steps to reproduce the behavior:
Change color or brightness with lovelace but also with stuff like "Adaptive Lighting". No difference if the latter is disabled or enabled.
Lovelace history will show something similar like this:
turned off by service light.turn_on
11:52:09 PM - 23 minutes ago
turned on by service light.turn_on
11:52:09 PM - 23 minutes ago
turned off by service light.turn_on
11:52:09 PM - 23 minutes ago
turned on by service light.turn_on
11:52:09 PM - 23 minutes ago

**Expected behavior**
Brightness and/or color temperature change as desired, no unexpected behavior, no control loss or inconsistency with the actual light setting.

**Screenshots**
screenshot is not really possible because it happens so fast

**Additional context**
Maybe this issue is relevant?  #1203 

When pairing, it showed two yellow warning signs. Under diagnostic, I also have two entities, both disabled. 

- sensor.tz3000_8uaoilu9_ts0502a_6b5713fe_basic_lqi

- sensor.tz3000_8uaoilu9_ts0502a_6b5713fe_basic_rssi

**Device info:**

TS0502A by _TZ3000_8uaoilu9 
Connected via Texas Instruments CC2531, Z-Stack Home 1.2 (build 20190608)
Zigbee info
IEEE: cc:86:ec:ff:fe:13:57:6b
Nwk: 0x7b5a
Device Type: Router
LQI: 115 
RSSI: Unknown 
Last Seen: 2022-02-17T00:20:24 
Power Source: Mains 
Quirk: zhaquirks.lidl.cct.CCTLight 

 Zigbee device signature:

> {
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_8uaoilu9",
  "model": "TS0502A",
  "class": "zhaquirks.lidl.cct.CCTLight"
}

Under system logs, I got these errore and both seem to be related

> Logger: homeassistant.components.zha.core.channels.base
> Source: components/zha/core/channels/base.py:428
> Integration: Zigbee Home Automation ([documentation](https://www.home-assistant.io/integrations/zha), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+zha%22))
> First occurred: 12:07:31 AM (1 occurrences)
> Last logged: 12:07:31 AM
> [0x7B5A:1:0x1000]: 'async_configure' stage failed: not enough values to unpack (expected 3, got 2)

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:96
Integration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
First occurred: February 16, 2022, 11:44:41 PM (1 occurrences)
Last logged: February 16, 2022, 11:44:41 PM
[140362881155024] Received invalid command: zha/devices/reconfigure
```

## System Health

version | core-2022.2.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.98
arch | x86_64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.22.0
Stage | running
Available Repositories | 976
Downloaded Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 7.2 GB
disk_used | 3.9 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), Terminal & SSH (9.3.0), Check Home Assistant configuration (3.9.0), Let's Encrypt (4.12.0), File editor (5.3.3), Google Assistant SDK (2.5.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 2
views | 9
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

MattWestb, 2022-02-17:
I think one dev have digging little and  trying making one workaround https://github.com/zigpy/zha-device-handlers/issues/1303.

In the ground is one bad behaving firmware that is not sending true status updated to ZHA and ZHA cant do so much for fixing it but perhaps can getting it working better with the patching hi have made.

The 2 disabled sensor is normal then its LQI and RSSI and if you like getting this updating enabling them and ZHA is updating them in the GUI.

pojlFDlxCOvZ4Kg8y1l4, 2022-02-17:
Yeah your issue is what I'm experiencing, the GIFs are accurate.

I can also crash my zigbee coordinator (CC2531) when I'm fiddling with the light. Don't know exactly how to reproduce it, but when I was trying to figure it out with repairing etc. it repeatedly made it crash.
Restarting HA was not enough, I had to physically disconnect the zigbee coordinator from the USB port.
I thought at first the zigbee stick was defective, so I've ordered a replacement with the CC2652RB chipset, but didn't cancel it. Maybe that will give some different results.

closed 2022-08-23 by github-actions[bot]

all43, 2025-02-21:
Regarding color temperature - I've ran into the same issue. I'm using zigbee2MQTT, but I think my results would be useful for ZHA as well. I experimented with different values while having two lights next to each other. One controlled by Lidl router and the other one by Z2M, so I could compare color temp.
Seems this device actually use 0-255 scale for color temperature instead of mireds. Where 0 means the warmest. If you set 256 it overflows to 0 making it warm again