zigpy/zha-device-handlers #1369: [Device Support Request] Aqara H1 EU With Neutral, Single Rocker Smart Wall Switch

Scoff123, 2022-02-17:
**Is your feature request related to a problem? Please describe.**
This device is supported by Zigbee2MQTT, DeConz and Zigbee for Domoticz. 

**Describe the solution you'd like**
I would like this device to be supported by ZHA, for the following features: 
device temperature
power monitoring
power outage memory
switch decouple
on/off

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "31": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n1aeu1",
  "class": "zigpy.device.Device"
}`

**Additional context**
URL for device: https://zigbee.blakadder.com/Aqara_WS-EUK03.html
Model WS-EUK03 manufactured by Aqara
Zigbee ID: "lumi.switch.n1aeu1"

ianfromnyc, 2022-03-03:
I have a number of these switches.  They work well in coupled relay mode, but decoupled mode won't work without a quirk.

Scoff123, 2022-03-03:
ZHA only currently gives me on/off state control and a device temperature reading of 0.3degrees which is obviously incorrect. Are you seeing a valid device temperature reading on your switches in ZHA?

ianfromnyc, 2022-03-03:
I don't get a temp reading, only on_off, electrical_measurement, and smartenergy_metering summation_delivered.  But since each switch only feeds about 10 or 20W of LED lighting, I'm not too concerned about temperature.  I just really want decoupled mode.

ianfromnyc, 2022-03-03:
I do see the faulty temp reading of 0.3C in addition to on_off, identify, electrical_measurement, and smartenergy_metering summation_delivered. But since each switch only feeds about 10 or 20W of LED lighting, I'm not too concerned about temperature. I just really want decoupled mode.

Scoff123, 2022-03-03:
Thanks for confirming. I'm also not interested in the switches temperature, I was just unsure if I had a faulty switch, or if this was a problem with ZHA. If you are also getting 0.3C then it looks to be ZHA. 

[image]

Can I ask how you get electrical_measurement, and smartenergy_metering summation_delivered? Does that information require a quirk?

ianfromnyc, 2022-03-03:
I get a Sensors card on my device. I'm just using default ZHA and the switches are paired direct to the coordinator. 

[image]

Scoff123, 2022-03-03:
That's interesting - I'm also using default ZHA and my single Aqara switch is paired to my coordinator. In fact, none of my ZHA devices show a Sensors card on the ZHA device info page - only 'Controls' and 'Diagnostics'. I wonder if this is an HA environment difference perhaps. I'm running HAOS on an Intel NUC, latest version available 2022.3.0 after just updating today, but this has always been the same, with none of my ZHA devices showing a Sensors card. Can I ask what environment you are running HA in? What Zigbee coordinator are you using also? I'm using a Tasmota flashed Sonoff ZBBridge

ianfromnyc, 2022-03-03:
I'm running HAOS on a RPi4 with an internal Raspbee stick.

Scoff123, 2022-03-03:
Thank you for the info, I've raised the question about the sensors card on the HA forum under the ZHA category, and I hope you don't mind me using your screenshot as an example ;) I will also eventually want to use the decoupled mode on my switch, so hopefully this can be picked up here as an issue.

Scoff123, 2022-03-04:
Just FYI - I have uninstalled ZHA and tried Zigbee2MQTT for the first time today, still using my Sonoff ZBBridge even though ezsp still isn't officially supported, and I added the Aqara light switch to Z2M and everything works perfectly. I am getting an internal switch temperature of 30C, measured power, and decoupled mode works perfectly too. Don't think I'll be going back to ZHA based no this alone.

[image]

dmulcahey, 2022-03-04:
Glad z2m works for you @Scoff123 but that doesn’t help us get things working here. Sorry we didn’t address this fast enough for you.

ianfromnyc, 2022-03-04:
I'd still like to have a device handler for these switches. I've started
working on one but I can't figure out why my Raspbee doesn't expose the
0xFCC0 cluster so the quirk fails to match.

janjurek, 2022-03-30:
Hi there guys I'm gonna chip in since I believe I'm having the same problem with probably slightly older revision of this device. I have RPi4 with original Raspbee and I've been using it with deCONZ up until now but assimilation of new devices is not great so I decided to switch to use the Home Assistant ZHA integration.

**My device is:** Xiaomi Aqara Wall Switch QBKG11LM https://zigbee.blakadder.com/Aqara_QBKG11LM.html 
It works and shows the exactly as @Scoff123 mentioned. I'm having on/off temperature and identify sensor. but power and power consumption sensors that @ianfromnyc have are not present for me.

I've already had some problem with this back when I was setting up the deCONZ and it looks like it's probably non standardly done by the manufacturer, but we we're able to resolve it in the end: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1249

And finally here is my zigpy node description I've go from the HA:
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0010"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x000c"
      ]
    },
    "5": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0010",
        "0x0012"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.ctrl_ln1.aq1",
  "class": "zigpy.device.Device"
}
```

It looks like the clusters are there "2" should be "ZHAPower" and "3" should  be "ZHAConsumption" so maybe _device_type_ is not correctly set for those? I'm really new to this so I'm just fishing here.

janjurek, 2022-04-01:
Well I've tried to dig into it a little bit, and tried to create custom quirk (fix the current one in handlers) for this to match the node decription (look like on 2 was "smartplug" instead of "power" then I've matched the replacements to mirror the signature and I believe I finally got some info in debug console
```
address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0xd263>, 2, 260, 12, b'\x18=\nU\x009q=\xecA', 0, 175, 255, 163, 0, 1, 2, -47]
2022-04-01 15:21:46 DEBUG (MainThread) [zigpy.zcl] [0xd263:2:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=61 command_id=Command.Report_Attributes>
2022-04-01 15:21:46 DEBUG (MainThread) [zigpy.zcl] [0xd263:2:0x000c] ZCL request 0x000a: [[Attribute(attrid=85, value=<TypeValue type=Single, value=29.530000686645508>)]]
2022-04-01 15:21:46 DEBUG (MainThread) [zigpy.zcl] [0xd263:2:0x000c] Attribute report received: present_value=29.530000686645508
2022-04-01 15:21:46 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK_AND_IEEE address=0xd263>, ep: 2, profile: 0x0104, cluster_id: 0x000c, data: b'183d0a550039713dec41'
```
If I read this well this means that I've received 29.53 watts when I'e turn on the switch, that is probably roughly what it should show given the bulbs on that line.

The problem that remains though is that there is no "power" (or "consumption" for that matter) sensor in Home assistant for that device, no matter how many times I re-pair the wall switch, so there has to be some part missing. I'm unable to figure out what though... here is my quirk for anybody interested in this:
```python
"""Xiaomi aqara single key wall switch devices."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    BinaryOutput,
    DeviceTemperature,
    Groups,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks import Bus
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.xiaomi import (
    LUMI,
    AnalogInputCluster,
    BasicCluster,
    OnOffCluster,
    XiaomiCustomDevice,
    XiaomiPowerConfiguration,
)

class CtrlLnTest(XiaomiCustomDevice):
    """lumi.ctrl_ln1.aq1.test"""

    def __init__(self, *args, **kwargs):
        """Init."""
        self.voltage_bus = Bus()
        self.power_bus = Bus()
        self.consumption_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        MODELS_INFO: [(LUMI, "lumi.ctrl_ln1.aq1")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    XiaomiPowerConfiguration.cluster_id,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    BinaryOutput.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [AnalogInput.cluster_id],
                OUTPUT_CLUSTERS: [Groups.cluster_id, AnalogInput.cluster_id],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                INPUT_CLUSTERS: [AnalogInput.cluster_id],
                OUTPUT_CLUSTERS: [AnalogInput.cluster_id],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [MultistateInput.cluster_id, BinaryOutput.cluster_id],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    BasicCluster,
                    XiaomiPowerConfiguration,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOffCluster,
                    Time.cluster_id,
                    BinaryOutput.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [AnalogInputCluster],
                OUTPUT_CLUSTERS: [Groups.cluster_id, AnalogInput.cluster_id],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                INPUT_CLUSTERS: [AnalogInput.cluster_id],
                OUTPUT_CLUSTERS: [AnalogInput.cluster_id],
            },
        5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [OnOffCluster, BinaryOutput.cluster_id],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

xhemp, 2022-04-15:
Have the same problem and been trying to find out how to get the power consumption to be shown on HA.

mtwilliams5, 2022-05-01:
I have this wall switch (`lumi.switch.n1aeu1`), and am hoping to be able to get it into decoupled mode (ideally without switching to zigbee2mqtt, since I don't want to migrate all my other devices over). Fairly new to ZHA stuff, but if there's anything I can do to help push this along, happy to help.

piomis, 2022-07-24:
Hi

Currently I have Aqara H1 WS-EUK03 (single rocker with neutral) and Aqara H1 WS-EUK04 (double rocker with neutral).
Looks that single rocker variant presents electrical measurements (power meter and energy meter), double rocker does not.

Reported temperature on both looks to be constant (and invalid) - 0.3 degree with EUK04 and 0.4 degree with EUK03.

Soon I plan to purchase also EUK01 variant (single rocker without neutral),

I confirm that on_off in coupled mode (both from HA and physical switch) works perfectly fine.

rosscullen, 2022-08-26:
I'm having similar issue to the comments above. No energy monitoring and 0.3 degrees sensor showing. On_off works fine.

sppre, 2022-09-14:
Hi

I also have Aqara H1 WS-EUK03 lumi.switch.n1aeu1 (single rocker with neutral) and Aqara H1 WS-EUK04 lumi.switch.n2aeu1 (double rocker with neutral).
Before upgrading to 8.2022 I had power meter and energy meter on both types.
After updating to 8.2022 I lost all sensors on the Aqara H1 WS-EUK04 (double rocker with neutral), but the single rocker still works.
Removing and repairing several times did not help.
[image]
[image]

rosscullen, 2022-09-16:
Anyone any thoughts on how this might be reviewed? Would it get better traction on the core HA issues respository? Thanks

mateuszlukasik-PL, 2022-09-28:
> Hi
>
> I also have Aqara H1 WS-EUK03 lumi.switch.n1aeu1 (single rocker with neutral) and Aqara H1 WS-EUK04 lumi.switch.n2aeu1 (double rocker with neutral). Before upgrading to 8.2022 I had power meter [...]

I have the same problem.
On Aqara H1 WS-EUK04 entity domain has changed to "switch" - was "light", power measurement is gone.
ZHA Quirk is now : zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2

On Aqara H1 WS-EUK03 entity domain is still "light", power measurement working fine.

sppre, 2023-01-11:
Dear All,
Is there any solution in sight?
Kind regards

ianfromnyc, 2023-01-11:
I tried to write a patch to add the single rocker switch, but for some
reason my switches wouldn't expose the 0xfcc0 cluster under ZHA and I
wasn't able to figure out why.

Eventually I decided I was spending more time trying to make ZHA work than
it would take to migrate to Z2M, so I migrated.

rosscullen, 2023-01-11:
> I tried to write a patch to add the single rocker switch, but for some
> reason my switches wouldn't expose the 0xfcc0 cluster under ZHA and I
> wasn't able to figure out why.
> 
> Eventually I decided I was spending more time trying to make ZHA work than
> it would take to migrate to Z2M, so I migrated.
> 
> 
> > Dear All,
> > Is there any solution in sight?
> > Kind regards

Might do similar with a separate Zigbee controller (dongle) and Zigbee2MQTT. Thanks

spitfire, 2023-01-23:
> I have the same problem.
>
> On Aqara H1 WS-EUK04 entity domain has changed to "switch" - was "light", power measurement is gone.
>
> ZHA Quirk is now : [...]

So does this mean the Aqara H1 EU double rocker with neutral works (with power metering), but the single rocker version doesn't?

kylehowells, 2023-01-30:
I have a `Aqara H1 EU With Neutral, Single Rocker Smart Wall Switch` - `WS-EUK03 lumi.switch.n1aeu1` like this items title lists, but mine shows a different signature (different in clusters):

```js
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n1aeu1",
  "class": "zigpy.device.Device"
}
```

Mine joins instantly, but then doesn't stay connected. The device never leaves pairing mode (even though HA can now control it) and after a while it times out, disconnects from HA and enters red LED "failed to connect" mode.

Lampyridae1984, 2023-02-11:
OMG, when did the progress go backwards? I switched to Z2M some time ago because there where still issues with the power metering. And now switching back because of HA Yellow I only have two switch entities. I can't even decouple the relay! 😕😑Why? What happened?

j0hann3s, 2023-02-27:
I have both switches ( lumi.switch.n1aeu1 [WS-EUK03] / lumi.switch.n2aeu1 [WS-EUK04] ) and everything worked on Z2M except for the single rocker version not being able to change to the decoupled mode. I just installed the new HA SkyConnect dongle and want to fully migrate over to ZHA to skip all the Z2M extra add-ons and integrations and also find it a bit weird how all this functionality has been working in Z2M for some while now :confused:

As of now in my ZHA setup, the single-rocker version can not go into decoupled mode but displays power information while the double-rocker version can go into decoupled mode but does not have any power reading information available.

(For those who, like me, have been searching for a way to put the double-rocker switch into decoupled mode. Here is how I did it: Go to the device page in ZHA into the option "Manage zigbee device" -> Clusters -> OppleSwitchCluster -> operation_mode -> Write "0" into the "Value" field and press "Write Attribute"... Done.)

I do not know how to program any quirks (tried and failed with understanding how to set up the equivalent to quirks in Z2M called custom_converters) so here is my best try to give relevant information for an experienced developer to take upon the quest of finally getting these popular switches to work in HA ZHA:

The single rocker switch:
[image]
[image]

Signature in ZHA:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n1aeu1",
  "class": "zigpy.device.Device"
}
```

The double rocker switch:
[image]
[image]

Signature in ZHA:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0012",
        "0x0702",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": []
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "31": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "42": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "51": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "61": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n2aeu1",
  "class": "zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2"
}
```

Feel free to ask me for any logs you might need so that all of us can finally use these switches like they were intended :smile:

szamfirov, 2023-05-08:
Just raised a PR for adding a quirk for the single rocker (with neutral) version of the switch: https://github.com/zigpy/zha-device-handlers/pull/2385

It's work in progress but would appreciate any support.

szamfirov, 2023-05-26:
The PR above was recently merged which should resolve any issues with the single rocker (with neutral version). Feel free to give the quirk a try. I've been using the wall switch in decoupled mode for quite some time and it works like a charm.

tarosk, 2023-07-09:
Thanks for the work @szamfirov 
Can this be replicated to double rocker version as well? Issue #2251

szamfirov, 2023-07-10:
The logic is already in there. I'm basing my implementation on the one for the double rocker switch.

Feel free to check [this comment](https://github.com/zigpy/zha-device-handlers/issues/2251#issuecomment-1466326352) for detailed instructions on how to enable decoupled mode.

danielshiloah, 2023-10-02:
I also have the double rocker (WS-EUK04), will there be a fix for that? looked at the device support request and it looks like nothing is happening... 
Did any one get it to work for them?

spitfire, 2023-10-02:
> I also have the double rocker (WS-EUK04), will there be a fix for that? looked at the device support request and it looks like nothing is happening... 
> 
> Did any one get it to work for them?

Yes, in Zigbee2MQTT;)

Edwhis, 2023-10-03:
> The PR above was recently merged which should resolve any issues with the single rocker (with neutral version). Feel free to give the quirk a try. I've been using the wall switch in decoupled mode for quite some time and it works like a charm.

Hi, I was checking your PR. You have mentioned, that you are using switch in decoupled mode. I've same switch, but I'm not able to find where I should enable this mode and I don't see anything related in PR. Can you help me?

tadaszelvys, 2023-10-18:
Almost two years and ZHA still not working with Aqara H1 EU :( Yesterday I connected it using Zigbee2MTTQ and it works like a charm. But I really like ZHA over Z2M :(

danielshiloah, 2023-12-19:
@TheJulianJES commented here that there was a push towards better support in ZHA for H1 double rocker, just checked it out and finally I get the Power consumption and overall power consumption. as well as voltage (stays on 0 for me although im 220v).
No temperature yet. 
Another issue is that the 2 buttons are identified as switch instead of lets say switch 1/2 so you cant do automations based on a specific button pressed / double pressed / hold.

But hay at least someone is working on it finally!

nlucansk, 2024-01-16:
`lumi.switch.n1aeu1`

Works like a charm

```
Core 2024.1.3
Supervisor 2023.12.0
Operating System 11.3
Frontend 20240104.0
```

Using Skyconnect USB `Firmware: 7.4.0.0 build 0`

Is there any way how to enable all those fancy double clicks? I've found guide for other type of switch but those values are not present on my end https://thesmarthomejourney.com/2024/01/15/aqara-h1-remote-switch-with-zha/

AlexKusnezov, 2024-01-31:
Hey guys, also having issue with `lumi.ctrl_ln1.aq1`

No power consumptionm, no tempreature.
I copied `ctrl_ln.py` into my quirks folder, but it seems not to load. Other quirk with other devices word perfectly.
Maybe you can point me in the right drirection on how to load this specific quirk
[image]
signature:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0010"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0009",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0053",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": [
        "0x000c"
      ]
    },
    "5": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0010",
        "0x0012"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.ctrl_ln1.aq1",
  "class": "zigpy.device.Device"
}
```
```
Diagnostic:
    "ieee": "**REDACTED**",
    "nwk": 56392,
    "manufacturer": "LUMI",
    "model": "lumi.ctrl_ln1.aq1",
    "name": "LUMI lumi.ctrl_ln1.aq1",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4447,
    "power_source": "Mains",
    "lqi": 180,
    "rssi": -55,
    "last_seen": "2024-01-31T15:19:00",
    "available": true,
    "device_type": "Router"
```
That's the quirk I'm trying to use:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/ctrl_ln.py

monsieurhannes, 2024-02-26:
Does anyone know if it's possible to disable "pairing mode" when holding one of the buttons for longer than 10 seconds? 
I would like to use it as a dimmer and I can see already, that someone is just pressing it forever (even though I put the dimmer steps so it would be bright after less than 10 seconds).
I just need it for one button, so the other one could still be used for pairing... :)

closed 2024-08-31 by github-actions[bot]