zigpy/zha-device-handlers #1372: [Device Support Request] TS004F Tuya remote in Scene mode

JoryHogeveen, 2022-02-18:
**Is your feature request related to a problem? Please describe.**
Currently te TS004F (which looks the same as the TS0044) is only supports as a dimmer switch.
However, this switch has two modes: Dimmer and Scene.

In Dimmer mode:
1 = on
2 = off
3 = brightness up (step, move and stop values)
3 = brightness down (step, move and stop values)

In Scene mode each button has a press, double press and long press value.

**Describe the solution you'd like**
The people at Z2MQTT are already working on it: https://github.com/Koenkk/zigbee2mqtt/discussions/7158
The people at Hubitat already integrated: https://raw.githubusercontent.com/kkossev/Hubitat/main/Drivers/Tuya%20TS004F/TS004F.groovy

Would be great if ZHA would support it as well.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0820",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_xabckq1v",
  "model": "TS004F",
  "class": "zhaquirks.tuya.ts004f.TuyaSmartRemote004F"
}
```

**Additional context**
I cannot pair the remote in Scene mode but you can switch modes when pressing the two right buttons on the remote at the same time for a longer period. It will send a zha event called `attribute_updated` in which it updates the `switch_mode` attribute. You can do the same action again to revert.

Related topic: https://github.com/zigpy/zha-device-handlers/issues/1322

MattWestb, 2022-03-05:
Some nice news https://github.com/zigpy/zha-device-handlers/issues/1322#issuecomment-1059771390 !!

JoryHogeveen, 2022-03-08:
Awesome!! Hope to see it in HA soon :)
So with that patch you can even just switch modes on the remote whilst still be connected to ZHA?

MattWestb, 2022-03-08:
With the dimmer switch / remote you need paring it one time with one tuya ZBGW and its working until battery is out or reading and writing some attributes to the device then its joining the network but its time critical and need being implanted in the future.
The Knop is working out of the box and very likely other tuya remotes type TS004F but can needing updating device class for getting all working OK.
Pressing the 2 let (upper and lower) button until the right lower LED flashing and mode is changes and also setting and reading the attribute on the Ob/Off cluster is working only need waking the remote so its receiving the command.

JoryHogeveen, 2022-03-08:
Ah too bad, I do not have a Tuya ZBGW unfortunately.
I thought I'd be able to change the mode merely by pressing the two left buttons and it would work :)
I'll try some cluster changes after the update is available. Might be able to create a workaround with ZHA Toolkit.

MattWestb, 2022-03-08:
Its possible sending one leave with rejoining flag sett with ZHA-ToolKit and sending the mode attribute but its very tricky (I have only getting it working one time but is possible).
If its not working you is only getting "scenes" commands from the first button (/endpoint).

Then I or some one can getting reading and writing attributes from quirk working its only some line codes and some testing so its working OK for all users but im not one code warier ;-/

The attribute you is finding on the OnOff cluster and looks like this:
https://github.com/zigpy/zha-device-handlers/blob/e3fa034284437595d1997326a8597086d623f413/zhaquirks/tuya/__init__.py#L837-L841

MattWestb, 2022-03-11:
@JoryHogeveen Can you trying this quirk and putting it in your local quirk directory (after unsiping it) and renaming / deleting the privies one so the system only loading the new one
[ts004f5.zip](https://github.com/zigpy/zha-device-handlers/files/8235738/ts004f5.zip).

If its working well it shall being possible witching mode with writing the switch mode attribute or log pressing the long pressing the right keys for 6 seconds.

It have working for my 2 times (deleting the device and taking the battery out and joining it new).

The problem is that restart of HA the system is getting problem then the device is sleeping so delete the latest quirk and putting the privies one in the local quirk folder for not getting problems. and shall working until the battery is out.

All 19 device automatons shall working OK if the tuya magic was working OK with  the files in the privies link also all events in in HA.

Pleas test and reporting back !!

MattWestb, 2022-03-19:
PR is made for adding the tuya magic and all device automatons https://github.com/zigpy/zha-device-handlers/pull/1437.

JoryHogeveen, 2022-03-19:
Hi @MattWestb 
Sorry, just saw you comments.
I'll test this next week when I'm back home!

MattWestb, 2022-03-19:
I have getting one from "Chinese Spain" (= taking 5 weeks) and its up and running nicely with both "tuya scenes" (Events) and commands (sending light commands) mode. Also binding it to one light group is working but cant  binding it only adding its one light group and you can controlling it also then your system is down.

MattWestb, 2022-03-22:
I have only trying in command mode and adding the remote as on light to one light group and its using the last added if doing more.
I have not testing if the event/scene commands is working OK then doing that but it shall working if the light group have more then one light and ZHA is adding the coordinator it it.

If its working you can switching mode in the remote and searing light and then like using HA automatons you only switching mod and can doing HA magic things.
Testing how its working and if its useful for you system.

JoryHogeveen, 2022-04-08:
Hi @MattWestb 

I just re-configured the device since the HA update and it's now being recognized as a TZ3000 switch `_TZ3000_xabckq1v`, is that correct?

JoryHogeveen, 2022-04-08:
Nevermind @MattWestb 

It already see it's correct.
The update is working perfect, thank you so much!
Closing topic :)

closed 2022-04-08 by JoryHogeveen

MattWestb, 2022-04-08:
ZHA is sometimes not loading the quirk and (normally then joining) and need one restart for doing it right.
I think its one fix on the way for the device automatons for the dimmer switch mode but i have not getting time looking if its released or not.

Glad your device is is working well !!

teklabania, 2023-02-26:
Hi, I have such device. I'm using it in scene mode and it's working almost fine. I'm catching zha events and assign desired actions to them but I have problem with button number 2. It works with a 1-3 seconds delay. I saw that it's been corrected in zigbee2mqtt with this PR:
https://github.com/Koenkk/zigbee-herdsman-converters/commit/9d011e20dbeab92ecaa6734bd831a2ec299d5313
The issue discussion is here:
https://github.com/Koenkk/zigbee2mqtt/issues/8149
Is it possible to implement similar solution in ZHA?