zigpy/zha-device-handlers #1373: [Device Support Request] LoraTap SS6400ZB-VS TS0044

frank8the9tank, 2022-02-18:
**Is your feature request related to a problem? Please describe.**
I have bought a zigbee 4 button remote from aliexpress, the device is succesfully connection to HA in the ZHA integration.
I see 4 entities for the 4 buttons, but no state change when i press a button.
I have asked for help on the Home assistant forum and for the ZHA integration on github.
Those guys point me here.
[HA forum](https://community.home-assistant.io/t/zigbee-4-button-remote-added-succesfull-but-not-working/394116/13)
[ZHA github](https://github.com/home-assistant/core/issues/66818)

**Describe the solution you'd like**
I like to have the 4 buttons to work in home assistant.
Trough an entity or an event, as long i can trigger an automation. 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
[0xa33b] Delivery error for seq # 0xbd, on endpoint id 0 cluster 0x0034: message send failure
Sending 'zdo_leave_req' failed: [0xa33b:0:0x0034]: Message send failure
[0x41d3:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=150 command_id=Command.Report_Attributes>
[0x41d3:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1291, value=<TypeValue type=int16s, value=0>)]]
[0x41d3:1:0x0b04] Attribute report received: active_power=0
[0x41d3:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=151 command_id=Command.Report_Attributes>
[0x41d3:1:0x0b04] ZCL request 0x000a: [[Attribute(attrid=1288, value=<TypeValue type=uint16_t, value=0>)]]
[0x41d3:1:0x0b04] Attribute report received: rms_current=0
[0x41D3:1:0x0b04]: async_update
[0x41d3:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=194 command_id=Command.Read_Attributes_rsp>
New device 0x64a2 (a4:c1:38:8f:b1:89:a8:e9) joined the network
[0x64a2] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0x64A2 ieee=a4:c1:38:8f:b1:89:a8:e9 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x0e\xa2d\xe9\xa8\x89\xb1\x8f8\xc1\xa4\x80'
[0x64a2:zdo] ZDO request ZDOCmd.Device_annce: [0x64A2, a4:c1:38:8f:b1:89:a8:e9, 128]
Tries remaining: 3
[0x64a2] Requesting 'Node Descriptor'
Tries remaining: 2
[0x64a2] Extending timeout for 0xc4 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x64A2 ieee=a4:c1:38:8f:b1:89:a8:e9 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\xc4\x00\xa2d\x02@\x80A\x11BB\x00\x00*B\x00\x00'
[0x64a2] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x64a2] Discovering endpoints
Tries remaining: 3
[0x64a2] Extending timeout for 0xc6 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x64A2 ieee=a4:c1:38:8f:b1:89:a8:e9 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\xc6\x00\xa2d\x04\x01\x02\x03\x04'
[0x64a2] Discovered endpoints: [1, 2, 3, 4]
[0x64a2] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=2 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=3 in=[] out=[] status=<Status.NEW: 0>>, <Endpoint id=4 in=[] out=[] status=<Status.NEW: 0>>]
[0x64a2:1] Discovering endpoint information
Tries remaining: 3
[0x64a2] Extending timeout for 0xc8 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x64A2 ieee=a4:c1:38:8f:b1:89:a8:e9 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\xc8\x00\xa2d\x14\x01\x04\x01\x00\x00\x01\x04\x06\x00\x01\x00\x06\x00\x00\x00\x02\x19\x00\n\x00'
[0x64a2:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0, device_version=1, input_clusters=[6, 1, 6, 0], output_clusters=[25, 10])
[0x64a2:2] Discovering endpoint information
Tries remaining: 3
[0x64a2] Extending timeout for 0xca request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x64A2 ieee=a4:c1:38:8f:b1:89:a8:e9 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\xca\x00\xa2d\x0e\x02\x04\x01\x00\x00\x01\x03\x01\x00\x06\x00\x00\xe0\x00'
[0x64a2:2] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=2, profile=260, device_type=0, device_version=1, input_clusters=[1, 6, 57344], output_clusters=[])
Unknown cluster 57344
[0x64a2:3] Discovering endpoint information
Tries remaining: 3
[0x64a2] Extending timeout for 0xcc request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x64A2 ieee=a4:c1:38:8f:b1:89:a8:e9 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\xcc\x00\xa2d\x0e\x03\x04\x01\x00\x00\x01\x03\x01\x00\x06\x00\x00\xe0\x00'
[0x64a2:3] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=3, profile=260, device_type=0, device_version=1, input_clusters=[1, 6, 57344], output_clusters=[])
Unknown cluster 57344
[0x64a2:4] Discovering endpoint information
Tries remaining: 3
[0x64a2] Extending timeout for 0xce request
Received frame on uninitialized device <Device model=None manuf=None nwk=0x64A2 ieee=a4:c1:38:8f:b1:89:a8:e9 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\xce\x00\xa2d\x0e\x04\x04\x01\x00\x00\x01\x03\x01\x00\x06\x00\x00\xe0\x00'
[0x64a2:4] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=4, profile=260, device_type=0, device_version=1, input_clusters=[1, 6, 57344], output_clusters=[])
Unknown cluster 57344
[0x64a2] Extending timeout for 0xd0 request
[0x64a2:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=208 command_id=Command.Read_Attributes_rsp>
[0x64a2] Read model 'TS0044' and manufacturer '_TZ3000_ee8nrt2l' from <Endpoint id=1 in=[on_off:0x0006, power:0x0001, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
[0x64a2] Discovered basic device information for <Device model='TS0044' manuf='_TZ3000_ee8nrt2l' nwk=0x64A2 ieee=a4:c1:38:8f:b1:89:a8:e9 is_initialized=True>
Device is initialized <Device model='TS0044' manuf='_TZ3000_ee8nrt2l' nwk=0x64A2 ieee=a4:c1:38:8f:b1:89:a8:e9 is_initialized=True>
Checking quirks for _TZ3000_ee8nrt2l TS0044 (a4:c1:38:8f:b1:89:a8:e9)
Considering <class 'zhaquirks.tuya.ts0044.TuyaSmartRemote0044TO'>
Fail because input cluster mismatch on at least one endpoint
[... 97 lines cut ...]
[0x64A2:1:0x0019]: 'async_initialize' stage succeeded
[0x64a2:4:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=241 command_id=Command.Read_Attributes_rsp>
[0x64A2:4:0x0006]: finished channel initialization
[0x64A2:4:0x0006]: 'async_initialize' stage succeeded
[0x64a2:3:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=239 command_id=Command.Read_Attributes_rsp>
[0x64A2:3:0x0006]: finished channel initialization
[0x64A2:3:0x0006]: 'async_initialize' stage succeeded
[0x64A2](TS0044): power source: Battery or Unknown
[0x64A2](TS0044): completed initialization
[0x64A2:1:0x0006]: attempting to update onoff state - from cache: True
[0x64A2:2:0x0006]: attempting to update onoff state - from cache: True
[0x64A2:3:0x0006]: attempting to update onoff state - from cache: True
[... cut ...]
[0x41d3:1:0x0702] ZCL request 0x000a: [[Attribute(attrid=512, value=<TypeValue type=bitmap8, value=bitmap8.0>)]]
[0x41d3:1:0x0702] Attribute report received: status=0
[0x41D3:1:0x0b04]: async_update
[0x41d3:1:0x0b04] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=243 command_id=Command.Read_Attributes_rsp>

**Additional context**
See both links for screenshot and more info.
Please let me know what i can do to help.

frank8the9tank, 2022-02-18:
I have tried to listen for events, but nothing is happening.

Signature:

{

"node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",

"endpoints": {

"1": {

"profile_id": 260,

"device_type": "0x0000",

"in_clusters": [

"0x0000",

"0x0001",

"0x0006"

],

"out_clusters": [

"0x000a",

"0x0019"

]

},

"2": {

"profile_id": 260,

"device_type": "0x0000",

"in_clusters": [

"0x0001",

"0x0006",

"0xe000"

],

"out_clusters": []

},

"3": {

"profile_id": 260,

"device_type": "0x0000",

"in_clusters": [

"0x0001",

"0x0006",

"0xe000"

],

"out_clusters": []

},

"4": {

"profile_id": 260,

"device_type": "0x0000",

"in_clusters": [

"0x0001",

"0x0006",

"0xe000"

],

"out_clusters": []

}

},

"manufacturer": "_TZ3000_ee8nrt2l",

"model": "TS0044",

"class": "zigpy.device.Device"

}

Is this what you need?

MattWestb, 2022-02-19:
I have adding your device as one new device class but i cant testing it.
Can you configure local quirk in HA https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274 and putting the the attached quirk in the new folder after renaming it and taking the  `.txt ` away so its looks like this `ts0044.py` and restarting HA.
[ts0044.py.txt](https://github.com/zigpy/zha-device-handlers/files/8101647/ts0044.py.txt)
If all is going OK the device is getting one quirk and device automation for all commands and also zh_events is being sen OK

Pleas reporting back so knowing if its working OK nor not.

frank8the9tank, 2022-02-19:
I have done the steps you requisted, but i dont see any change in HA or in the behaviour of the device.
I dont see the quirk getting applied:
[image]

Some other device on my zigbee network does have a quirk (and working fine) 
[image]

Have tried to removed and readded the remote, but that also does not make a change.
With mutiple restarts in between.
Am i doing something wrong or does the file need to be changed?

I have removed the quirk file now and removed and readded the device,
But now the battery charge does not get a value, 

The main thing i see that when i listen for zha_event in developer tools i see nothing, with or without the quirk.

MattWestb, 2022-02-19:
ZHA can "missing" loading quirks for devices if doing many paring and deleting of the device but its normally being OK after one restart of HA.

I taking one more look in the updated quirk and see if i have missing somthing.

Then the remote is sending no standard commands its need the quirk so ZHA can understanding the command its sending and shall being fixed then we is getting the quirk loaded for your device you is getting events from it.

MattWestb, 2022-02-19:
New version:
[ts0044.py,txt](https://github.com/zigpy/zha-device-handlers/files/8101964/ts0044.py.txt)
 Testing and reporting pleas !!

My Laptop is having the famous "corrupted clipboard syndrome" (= Copy and past error from the user) but  its looks being fixed after one more Coffey ;-))

frank8the9tank, 2022-02-19:
I see the quirk now beeing applied in the zha integration,
When i listen for events i see then now in the developments tools.
Look like a functions work, short, double and long press.

single press:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "a4:c1:38:8f:b1:89:a8:e9",
        "unique_id": "a4:c1:38:8f:b1:89:a8:e9:1:0x0006",
        "device_id": "daaf85b07ce143ac0d17009e7cbbdd69",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "remote_button_short_press",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-19T13:10:57.965517+00:00",
    "context": {
        "id": "5501537894c6982f14d999c1c1e090cf",
        "parent_id": null,
        "user_id": null
    }
}
```

So many thanks!!,
Will check if every thins keep working the comming days!

frank8the9tank, 2022-02-19:
One thing that i see that is that the battery charge level does not work, (this was before, without the quirk file).
If its possible that i would like that also be working again.

MattWestb, 2022-02-19:
I think its confusing ZHA then its having one power config cluster on all 4 endpoints and very likely its only the first that is working in real.
Perhaps better only having it on the first one that is having the basic cluster.

Shall i trying making one updated quirk ?

frank8the9tank, 2022-02-19:
If you think that get the battery level workings then yes please!,
Just to known: is this always the case with the Aliexpress range of devices? or did i just get a bad one?

MattWestb, 2022-02-19:
No you was getting one new not known device = normal with Chinese sellers.
Most TS004X devices have being very like only one new device found for one week ago and yours now but have working very OK around one year.

I doing on updated quirk so you can testing if its helping.

MattWestb, 2022-02-19:
I have trying deleting the not working switches in ZHA GUI but i dont knowing if its working or not. I hope you is not loosing the events from the remote but i have not testing it.  [ts0044.py.txt](https://github.com/zigpy/zha-device-handlers/files/8102326/ts0044.py.txt)
Very likely is the 4 switches being grayed out then they is deleted in the quirk and you can deleting them and it shall not coming back if deleting the device and re adding it.

Try is its working and reporting back.

frank8the9tank, 2022-02-19:
That worked,
Its now showing the battery level.
It did not work with only replacing the file (and deleting the compiled file) and a restart. (no change after this)
When i removed the remote from the network and restared HA and then added it again it worked.
[image]
So many thanks for helping me out!!

MattWestb, 2022-02-19:
looks very nice !!
Then changing local quirk files is always need restart HA or restart enough of ZHA for the quirk shall being re read from the system so i always doing reboot or HA.

I doing one PR for adding the new class in HA and asking the maintainer if i shall doing  the same change on the other device classes and also the 1 -3 gang switches.

You shall also have device automation's on the device card so you fast and easy doing nice automatons for the device.

Is you getting `zha_event` with the latest  quirk ??.

Good testing and hacking done !!

frank8the9tank, 2022-02-19:
What is see under events is the following:
(2 events per click, single, double and long)
This is only with the latest quirk, there was only one zha_event in the previous quirk.
So don't know if this is okay.

I've only tried automation with the single click,
They dont show up under automation in the device page for the integration.
I think the automation we make for the clicks don't get linked to the device as we listen for a zha_event, and don't use an entity or device in the automation.

`Event 7 fired 19:37:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "a4:c1:38:8f:b1:89:a8:e9",
        "unique_id": "a4:c1:38:8f:b1:89:a8:e9:1:0x0006",
        "device_id": "daaf85b07ce143ac0d17009e7cbbdd69",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "press_type",
        "args": [
            1
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-19T18:37:56.097618+00:00",
    "context": {
        "id": "cacd2b7b1b231be79317be8100431532",
        "parent_id": null,
        "user_id": null
    }
}

Event 6 fired 19:37:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "a4:c1:38:8f:b1:89:a8:e9",
        "unique_id": "a4:c1:38:8f:b1:89:a8:e9:1:0x0006",
        "device_id": "daaf85b07ce143ac0d17009e7cbbdd69",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "remote_button_double_press",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-19T18:37:56.097255+00:00",
    "context": {
        "id": "59cf1491646adf031bf484c255bd5cb1",
        "parent_id": null,
        "user_id": null
    }
}

Event 5 fired 19:37:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "a4:c1:38:8f:b1:89:a8:e9",
        "unique_id": "a4:c1:38:8f:b1:89:a8:e9:1:0x0006",
        "device_id": "daaf85b07ce143ac0d17009e7cbbdd69",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "press_type",
        "args": [
            2
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-19T18:37:49.925733+00:00",
    "context": {
        "id": "bc0b6bb27bed5b9649b761617127a9c5",
        "parent_id": null,
        "user_id": null
    }
}

Event 4 fired 19:37:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "a4:c1:38:8f:b1:89:a8:e9",
        "unique_id": "a4:c1:38:8f:b1:89:a8:e9:1:0x0006",
        "device_id": "daaf85b07ce143ac0d17009e7cbbdd69",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "remote_button_long_press",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-19T18:37:49.925151+00:00",
    "context": {
        "id": "325dabb15c39d306ccfdeb35aa6b8791",
        "parent_id": null,
        "user_id": null
    }
}

Event 3 fired 19:37:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "a4:c1:38:8f:b1:89:a8:e9",
        "unique_id": "a4:c1:38:8f:b1:89:a8:e9:1:0x0006",
        "device_id": "daaf85b07ce143ac0d17009e7cbbdd69",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "press_type",
        "args": [
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-19T18:37:35.392272+00:00",
    "context": {
        "id": "c4a24b47d790a902d36f2720b2788937",
        "parent_id": null,
        "user_id": null
    }
}

Event 2 fired 19:37:

{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "a4:c1:38:8f:b1:89:a8:e9",
        "unique_id": "a4:c1:38:8f:b1:89:a8:e9:1:0x0006",
        "device_id": "daaf85b07ce143ac0d17009e7cbbdd69",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "remote_button_short_press",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-19T18:37:35.391985+00:00",
    "context": {
        "id": "134f1613918c2e22310d33332c55294c",
        "parent_id": null,
        "user_id": null
    }
}

`

MattWestb, 2022-02-19:
I have one TS004F dimmer switch / that can being switched to one 4 scene switch like yours (not working OK in ZHA for the moment) and its one with the command number and (like 2) and one with the function (like long press) and i cant doing anything for the moment of it.
Then i getting the TS004F working in scene mode OK i can debugging it more.

On the left side of the device card you is having  `Automations` and one ` + ` then you clicking the ` + ` you is getting all device automatons for the device and if you is using them ZHA is also showing the automatons you have doing for the device on the device card.

frank8the9tank, 2022-02-19:
I had to change the automation from listening for events now to a device with a single click.
Now the automations also showup in the device page.
The remote now shows up as a device, this was not the case in the previous quirk.
[image]
So looks like this is done?

MattWestb, 2022-02-19:
Before it was one light switch (wall switch with complected lights) and can only riving commands from automatons.

My ts004f is looking like this if clicking on the plus for automatons:
[image]
but you shall having many more functions.

MattWestb, 2022-02-19:
PR is made and the maintainer must saying its OK for it getting merged and also i like doing all the  "extra fixes" (battery and deleting the switches so the automatons is working OK) for the other ts004x devices and classes but only after they is saying its OK.

frank8the9tank, 2022-02-19:
same here:
[image]

closed 2022-02-23 by dmulcahey

SkyaTura, 2022-12-24:
@MattWestb is it possible to create a similar quirk, but for `SS9600ZB-V2  TS0046`?

MattWestb, 2022-12-24:
Very likely yes.
Make one new device support request for the device then its different and need adding the 2 extra endpoints for button 5 and 6.

PR #1374 (2022-02-23): Adding one more TS0044 version
One more TS0044 cluster version. Have moving the OnOff cluster to out in replace and deleting the power-config on EP 2-3 then it was not reporting battery OK with all 4 active.
So no no working switches on the device card and working battery reporting.

Fixes: #1373