zigpy/zha-device-handlers #1375: [Device Support Request] TS0042 by _TZ3000_wjr2jjrm

bluechris, 2022-02-19:
Describe the solution you'd like
I have a Wireless Switch 2 GANG from Tuya that advertise with Tuya Hub that Have suport for Short Tap, Double Tap and Long Press. I bought them because i saw in this post here in github that this devices are supported 

https://github.com/zigpy/zha-device-handlers/issues/729

but unfortunately they can be added to ZHA but i dont get buttons but 4 switches that doesnt do anything. I suppose something is need to be added to ZHA for them to work correctly.
I bought them from [Aliexpress](https://www.aliexpress.com/item/1005002847761106.html?spm=a2g0o.9042311.0.0.1b8f4c4dCEWGQ2)

manufacturer": "_TZ3000_wjr2jjrm",
model": "TS0042",

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0xe000"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_wjr2jjrm",
  "model": "TS0042",
  "class": "zigpy.device.Device"
}

**ZHA Log when adding the device**

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xee46 (a4:c1:38:79:68:88:47:6c) left the network
Device 0xee46 (a4:c1:38:79:68:88:47:6c) joined the network
Device 0xee46 (a4:c1:38:79:68:88:47:6c) joined the network
[0xee46:zdo] ZDO request ZDOCmd.Device_annce: [0xEE46, a4:c1:38:79:68:88:47:6c, 128]
[... 149 lines cut ...]
[0xee46:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=29 command_id=Command.Read_Attributes_rsp>
[0xee46] Extending timeout for 0x23 request
[0xee46:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=31 command_id=Command.Read_Attributes_rsp>
[0xEE46:1:0x0006]: finished channel initialization
[0xEE46:3:0x0006]: initializing channel: from_cache: False
[0xee46] Extending timeout for 0x25 request
[0xee46:2:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=33 command_id=Command.Read_Attributes_rsp>
[0xEE46:2:0x0006]: finished channel initialization
[0xEE46:4:0x0006]: initializing channel: from_cache: False
[0xee46] Extending timeout for 0x27 request
[0xEE46:2:0x0006]: 'async_initialize' stage succeeded
[0xee46:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=35 command_id=Command.Read_Attributes_rsp>
[0xee46:3:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=37 command_id=Command.Read_Attributes_rsp>
[0xEE46:1:0x0001]: finished channel initialization
[0xEE46:1:0x0001]: 'async_initialize' stage succeeded
[0xEE46:1:0x0000]: 'async_initialize' stage succeeded
[0xEE46:1:0x0006]: 'async_initialize' stage succeeded
[0xEE46:1:0x0019]: 'async_initialize' stage succeeded
[0xEE46:3:0x0006]: finished channel initialization
[0xEE46:3:0x0006]: 'async_initialize' stage succeeded
[0xee46:4:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=39 command_id=Command.Read_Attributes_rsp>
[0xEE46:4:0x0006]: finished channel initialization
[0xEE46:4:0x0006]: 'async_initialize' stage succeeded
[0xEE46](TS0042): power source: Battery or Unknown
[0xEE46](TS0042): completed initialization

MattWestb, 2022-02-20:
I have making one new device class for you new device but i cant testing it then i dont have one. Can you testing if its working OK with the instruction in https://github.com/zigpy/zha-device-handlers/issues/1373#issuecomment-1045959371 and with the updated quirk [ts0042.py.txt](https://github.com/zigpy/zha-device-handlers/files/8103603/ts0042.py.txt) ?
If its loading OK you is getting the new quirk in the device class and you 4 switches is away and you is getting `zha_event` then pressing buttons and also device automatons shall working OK.

Pleas test and reporting back !!

bluechris, 2022-02-20:
Ouch, that was fast.... 
Yes they work totally fine, a million thx for this.

I need now to have this quirk forever or somehow it will be implemented in next ZHA version?

closed 2022-02-20 by bluechris

MattWestb, 2022-02-20:
GREAT !!

I can making one PR for adding it so its coming in the next HA release.

Then the update its in HA and you is updating HA you shall deleting the local quirk so you is not getting problem later then its coming updated.

MattWestb, 2022-02-20:
PR made https://github.com/zigpy/zha-device-handlers/pull/1376 waiting merging :-))

MattWestb, 2022-02-20:
By the way welcome to git hub and you first issue posted !!!

bluechris, 2022-02-20:
Thank you very much.