zigpy/zha-device-handlers #1377: [Device Support Request] Water leak sensor TZ3000_t6jriawg TS0207

njot0, 2022-02-20:
**Is your feature request related to a problem? Please describe.**
I have few water leak sensors from Aliexpress. There are no markings on the device, while box has markings Moes ZN323786-01.
This device does not work with Home Assistant using ZHA. I can pair it, but it does not change status when I put it into contact with water, and when it dries. After few hours, it goes unavailable.
Tried this with 2 separate devices, same issue.
[image]
**Describe the solution you'd like**
Proper integration into Home Assistant using ZHA, change state when it detects water, etc.

**Device signature**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0,user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>,manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66,descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False,*is_end_device=True,*is_full_function_device=False,*is_mains_powered=False,*is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_t6jriawg",
  "model": "TS0207",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Device creates these entities when paired:
```
binary_sensor.tz3000_t6jriawg_ts0207_99509336_ias_zone
button.tz3000_t6jriawg_ts0207_99509336_identify
sensor.tz3000_t6jriawg_ts0207_99509336_power
sensor.tz3000_t6jriawg_ts0207_99509336_basic_lqi (disabled by integration)
sensor.tz3000_t6jriawg_ts0207_99509336_basic_rssi (disabled by integration)
```
ias_zone is alway "dry", even when device is in contact with water.

These are the logs filtered to its Nwk, after pairing:
```
2022-02-20 13:21:18 DEBUG (MainThread) [bellows.zigbee.application] Received childJoinHandler frame with [0, <Bool.true: 1>, 0x6696, a4:c1:38:bc:36:93:50:99, <EmberNodeType.SLEEPY_END_DEVICE: 4>]
2022-02-20 13:21:18 DEBUG (MainThread) [bellows.zigbee.application] Received trustCenterJoinHandler frame with [0x6696, a4:c1:38:bc:36:93:50:99, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x0000]
2022-02-20 13:21:18 INFO (MainThread) [zigpy.application] New device 0x6696 (a4:c1:38:bc:36:93:50:99) joined the network
2022-02-20 13:21:18 DEBUG (MainThread) [zigpy.device] [0x6696] Scheduling initialization
2022-02-20 13:21:18 INFO (MainThread) [zigpy.device] [0x6696] Requesting 'Node Descriptor'
2022-02-20 13:21:18 DEBUG (MainThread) [zigpy.device] [0x6696] Extending timeout for 0xdd request
2022-02-20 13:21:18 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for a4:c1:38:bc:36:93:50:99/0x6696
2022-02-20 13:21:18 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x6696, EmberApsFrame(profileId=0, clusterId=2, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=221), 222, b'\xdd\x96f')
2022-02-20 13:21:19 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=0, clusterId=19, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=10), 172, -57, 0x6696, 255, 255, b'\n\x96f\x99P\x936\xbc8\xc1\xa4\x80']
2022-02-20 13:21:19 INFO (MainThread) [bellows.zigbee.application] ZDO Device announce: 0x6696, a4:c1:38:bc:36:93:50:99
2022-02-20 13:21:19 INFO (MainThread) [zigpy.application] Device 0x6696 (a4:c1:38:bc:36:93:50:99) joined the network
2022-02-20 13:21:19 DEBUG (MainThread) [zigpy.device] [0x6696] Scheduling initialization
2022-02-20 13:21:19 DEBUG (MainThread) [zigpy.device] [0x6696] Canceling old initialize call
2022-02-20 13:21:19 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x6696 ieee=a4:c1:38:bc:36:93:50:99 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\n\x96f\x99P\x936\xbc8\xc1\xa4\x80'
2022-02-20 13:21:19 DEBUG (MainThread) [zigpy.zdo] [0x6696:zdo] ZDO request ZDOCmd.Device_annce: [0x6696, a4:c1:38:bc:36:93:50:99, 128]
2022-02-20 13:21:19 INFO (MainThread) [zigpy.device] [0x6696] Requesting 'Node Descriptor'
2022-02-20 13:21:19 DEBUG (MainThread) [zigpy.device] [0x6696] Extending timeout for 0xdf request
2022-02-20 13:21:19 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for a4:c1:38:bc:36:93:50:99/0x6696
2022-02-20 13:21:19 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32770, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=11), 172, -57, 0x6696, 255, 255, b'\xdd\x00\x96f\x02@\x80A\x11BB\x00\x00*B\x00\x00']
2022-02-20 13:21:19 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x6696 ieee=a4:c1:38:bc:36:93:50:99 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\xdd\x00\x96f\x02@\x80A\x11BB\x00\x00*B\x00\x00'
2022-02-20 13:21:19 DEBUG (MainThread) [zigpy.zdo] [0x6696:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0x6696, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)]
2022-02-20 13:21:19 DEBUG (MainThread) [zigpy.zdo] [0x6696:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_rsp([<Status.SUCCESS: 0>, 0x6696, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)])
2022-02-20 13:21:19 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x6696, EmberApsFrame(profileId=0, clusterId=2, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=223), 224, b'\xdf\x96f')
[... 235 lines cut ...]
2022-02-20 13:23:52 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6696:1:0x0001]: initializing channel: from_cache: False
2022-02-20 13:23:52 DEBUG (MainThread) [zigpy.device] [0x6696] Extending timeout for 0x0d request
2022-02-20 13:23:52 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for a4:c1:38:bc:36:93:50:99/0x6696
2022-02-20 13:23:52 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x6696, EmberApsFrame(profileId=260, clusterId=1, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=13), 14, b'\x00\r\x00 \x00!\x00')
[... cut ...]
2022-02-20 13:24:14 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x6696](TS0207): power source: Battery or Unknown
2022-02-20 13:24:14 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x6696](TS0207): completed initialization
```
Few more details - there is nothing in the logs (filtered to either its Nwk or to t6jriawg or to TS0207) when I press the button on the device or when I put it into contact with water, although LED on the device is flashing at this time. When it dries, there is no LED flashing nor anything in the logs.

Any help is greatly appreciated.

maurocordioli, 2022-04-17:
same behavior (always dry) for another similar water sensor
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4619, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_ezhgerjm",
  "model": "TY0207",
  "class": "zigpy.device.Device"
}

closed 2022-10-21 by github-actions[bot]

ncareau, 2023-06-19:
I'm having the same issue on this device. The device reports data once and then becomes unavailable.

Bohemski, 2024-02-20:
i'm havin exactly the same problem now.

Bohemski, 2024-02-20:
I just found the solution. It's a battery problem, you need to put new ones in.

manuscho, 2024-08-25:
Same problem here - I have the issue with 3 seperate NEO Water Leak Sensors ([LINK](https://zigbee.blakadder.com/Neo_NAS-WS05B.html)), which use a Tuya TS0207 chip and should theoretically be compatable to ZHA on Home Assistant. 

However, even with new batteries - No updates from the devices after the initial pairing is completed.

**Did anybody ever find a solution to this? :(**