zigpy/zha-device-handlers #1382: [BUG] LLKZMK11LM: no power metering entity

mikoff, 2022-02-20:
**Describe the bug**
On Home Assistant restart, the power monitoring entity of the device is lost. If we go to the cluster management screen we see that the device can read proper wattage, but the entities report “Unavailable” status.
The option to "Reconfigure" does not help.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair LLKZMK11LM
2. Check that power monitoring works and reports
3. Reboot HA
4. Power monitoring utilities are missing

**Screenshots**
[image]
[image]

**Additional context**
The problem was discussed by other users on HA forum, for example:
https://community.home-assistant.io/t/aqara-lumi-2-channel-relay-with-zha/356984

mazerte, 2022-02-22:
It may be related to this one to https://github.com/zigpy/zha-device-handlers/issues/1337.

On my side I get the same problem with the Aqara Smart Plug. Clusters are reporting good metrics but entities are unavailable.

dmulcahey, 2022-02-27:
Do you have the same errors in the logs that the linked issue above does?

leonardpitzu, 2022-02-28:
Logs are empty. Same the entities.

jims-code, 2022-04-03:
The strange behaviour mentioned in this issue is shown by devices that were connected some days ago. Devices that were connected last year work fine.

Maybe there are some hints in these issues that help finding the reason for this strange behaviour:
https://github.com/home-assistant/core/issues/67644
https://github.com/zigpy/zha-device-handlers/issues/1337

The file plug_mmeu01.py has been updated 3 months ago. It might be that this caused an unwanted side effect.
https://github.com/zigpy/zha-device-handlers/pull/1252

kvvoff, 2022-06-11:
Is there any progress in solving the problem? I have the same with LLKZMK11LM

jims-code, 2022-06-11:
For the Aqara Smart Plug the problem has been solved in zigpy/zha-device-handlers Release 0.0.73. The installation was done automatically by a Home Assistant update.

More details concerning the bug can be found here: https://github.com/home-assistant/core/issues/67644

kvvoff, 2022-06-11:
for LLKZMK11LM still doesn't work. Latest version of HA

closed 2022-12-15 by github-actions[bot]