zigpy/zha-device-handlers #1392: [Device Support Request]Missing feature on Develco Frient SMSZB-120 Intelligent Smoke Alarm

nomad-cyanide, 2022-02-23:
**Is your feature request related to a problem? Please describe.**
- Current support of the SMSZB-120 will only produce a 5 loud sounds from the device, when used as a siren with ZHA. 

**Describe the solution you'd like**
I would like to use the SMSZB-120 a regular siren with ZHA, like it does when detecting smoke.

**Device signature **
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.NONE: 0>, manufacturer_code=4117, maximum_buffer_size=80, maximum_incoming_transfer_size=80, server_mask=10752, maximum_outgoing_transfer_size=80, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=False, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 49353,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0003",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "35": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0x0500",
        "0x0502"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "38": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0402"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "frient A/S",
  "model": "SMSZB-120",
  "class": "zhaquirks.develco.smoke_alarm.SMSZB120"
}
```
**Additional context**
Tried making a button to activate the siren like so:
```
type: button
tap_action:
  action: toggle
entity: siren.frient_1sal_ias_wd
```
This just produces 3-5 loud sound, not an actual siren.
The firmware seems to be 4.02 on all 3 devices I have of them. 

Further info on the device support: https://github.com/zigpy/zha-device-handlers/issues/854

nomad-cyanide, 2022-02-23:
Found this in a thread for zigbee2MQTT: https://github.com/Koenkk/zigbee2mqtt/issues/3012?fbclid=IwAR1UQ2EuifzOGT_TOExc3XZypGmxtDgT6zOC30oShBpprJmDFvrpaN0nut4#issuecomment-596243748

Perhaps we can use this somehow?
```
To get the siren to start
{"warning": {"mode": "burglar", "level": "high", "strobe": false, "duration": 300}}

To stop it
{"warning": {"mode": "stop", "level": "low", "strobe": false, "duration": 300}}

```

byrongamatos, 2022-03-08:
I am also very interested in this as I have 3 of those devices

closed 2022-09-11 by github-actions[bot]

nikobiker12, 2023-12-02:
> Found this in a thread for zigbee2MQTT: [Koenkk/zigbee2mqtt#3012](https://github.com/Koenkk/zigbee2mqtt/issues/3012)
> 
> Perhaps we can use this somehow?
> 
> ```
> To get the siren to start
> {"warning": {"mode": "burglar", "level": "high", "strobe": false, "duration": 300}}
> 
> To stop it
> {"warning": {"mode": "stop", "level": "low", "strobe": false, "duration": 300}}
> ```

Hello, same problem for me.
i dont undertand :( How can I use you tip ?
Thanks

DamGranger, 2024-02-12:
Hi everyone, would be nice to have this one reopen, as I believe there is no other device capable of what you can do with this one.

That being said, if anyone stumble upon this thread, I did manage to make mine scream for 4 minutes using a script
```
alias: SE Déclenchement sirènes
sequence:
  - parallel:
      - service: zha.issue_zigbee_cluster_command
        data:
          ieee: 00:15:bc:00:31:01:23:7b
          endpoint_id: "35"
          cluster_id: 1282
          cluster_type: in
          command: 0
          command_type: server
          args:
            - 255
            - 240
            - 255
            - 4
        alias: Déclencher 240s de sonnerie sur détecteurs de fumée Bureau
      - alias: Déclencher 240s de sonnerie sur détecteurs de fumée Salle de jeux
        service: zha.issue_zigbee_cluster_command
        data:
          ieee: 00:15:bc:00:31:01:21:8f
          endpoint_id: "35"
          cluster_id: 1282
          cluster_type: in
          command: 0
          command_type: server
          args:
            - 255
            - 240
            - 255
            - 4
mode: single
icon: mdi:bullhorn
```

I do have another script sending args as 0, 0, 0, 0 and stopping everything. These two scripts are called as actions in Alarmo when the alarm is triggered or when it's disarmed. As Alarmo is configured to have the smoke sensor as "always active" smoke detection in one place triggers all the alarms.

So far pretty satisfied with the solution but would be way better if this could be an entity or a command instead of calling a ZHA cluster command.