zigpy/zha-device-handlers #1396: [Device Support Request] TS011F Led status indicator

PonyOny, 2022-02-25:
**Describe the solution you'd like**
I would like to be able to select the power button led color when device is turned on, I have some switches that are red color when ON and some are blue color when on, I can't find where to change this behaviour in the Clusters.

Red Led:
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_ew3ldmgx",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```
Blue Led:
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_typdpbpg",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```

MattWestb, 2022-02-25:
I think (= dont knowing) you need modifying the hardware and changing the LED in the plug to one that have the color you like.

Or if its one multi color LED (that i think its not) we need finding the right commands for changing the color and implanting in in the system as one normal attribute.

MattWestb, 2022-02-25:
Then it must being done in the firmware.
You is having one OnOff cluster named `TuyaZBOnOffAttributeCluster` and its having one attribute `SwitchBackLight` with 3 different settings  `Mode_0`,  `Mode_1` and  `Mode_2`.
Is the color changing then writing the attribute to the device ?

PonyOny, 2022-02-25:
Changing values to 0,1,2 does not seem to affect led color.

MattWestb, 2022-02-25:
Is it possible changing it from tuya APP with there ZBGW ?
If then we need one sniffing then changing the color from the TBGW..

MattWestb, 2022-02-25:
By the way the normal is normal, inverted function or off for my tuya (LIDL) plugs.

PonyOny, 2022-02-25:
> Is it possible changing it from tuya APP with there ZBGW ?
> If then we need one sniffing then changing the color from the TBGW..

I don’t have the tuya ZBGW I use a Zigbee usb device.

YozenPL, 2022-05-12:
> Then it must being done in the firmware. You is having one OnOff cluster named `TuyaZBOnOffAttributeCluster` and its having one attribute `SwitchBackLight` with 3 different settings `Mode_0`, `Mode_1` and `Mode_2`. Is the color changing then writing the attribute to the device ?

It seems like those values are not send/saved. Changing from Mode_1 to Mode_2 and setting the attribute via button does nothing. When I click "show zigbee attribute" it's still set to Mode_1

If I connect the switch for a moment to Tuya app and set backlight light to ON (when lights are turned off) I have:
[image]
And for switch which was not configured via Tuya app I have:
[image]

MattWestb, 2022-05-12:
Then tuya is doing some magic as mor or less norla tuya is doing and we is not knowing wot they is doing.
Then you is having the device and one tuya ZBGW its great but do you jave one device for sniffing the zigbee traffic  ?
If yes one sniff from pairing and also one then changing the mode so we can see how tuya is doing it is more then welcome.

PonyOny, 2022-10-17:
> @YozenPL To set these values at the moment, you need to put `0`, `1`, or `2` in the upper field (where it says `SwitchBackLight.Mode_1` <- remove that and replace with either `0`, `1`, or `2`) (and then write the attribute). Also see #1442

Just tried that,
value o = light OFF
value 1 = light ON 
value 2 = light OFF

MattWestb, 2022-10-17:
One is permanent off and the other is flow the sate of the switch so being on or off.

TheJulianJES, 2022-10-17:
I was thinking whether or not these values should be exposed to HA as a config entity, but I'm not sure it's a good idea/really needed for LED indicators.

MattWestb, 2022-10-17:
I think its the same class as the power on behavior that is used one or 2 times of the user but is nice to having in the GUI.
Also tuya Zigbee power on is different from the ZCL one and need added with tuya manufacture code for not using the ZCL ones.

TheJulianJES, 2022-10-17:
Yeah, I've tested the Tuya power-on-state: https://github.com/home-assistant/core/pull/80486

The LED indicator modes on the BW-SHP13 I tested with seems to have the following modes:
- LED always off
- LED on when plug is on + LED off when plug is off
- LED off when plug is on + LED on when plug is off (inversed as above)

They could be exposed to the UI in a similar fashion to the power-on-state. Not sure if all Tuya plugs have the same three modes though and how they should even be named in the UI.

(Child lock could also be exposed for supporting plugs. BW-SHP13 doesn't, BW-SHP15 does)

MattWestb, 2022-10-17:
It the same with the first gen LIDL plug (WO power measuring).
Some "tuya extras" found from dev docks and other pleases but its not 100% used the same in all devices then tuya is very "flexible".
https://github.com/zigpy/zigpy/discussions/823#discussioncomment-1539469

TheJulianJES, 2023-01-04:
Config entities for most Tuya plugs were implemented in HA a couple of versions ago:
[image]

- https://github.com/home-assistant/core/pull/80486
- https://github.com/home-assistant/core/pull/81218
- https://github.com/home-assistant/core/pull/83233

A differently colored LED is hardware related for these plugs and the color can't be changed by software.

closed 2023-01-04 by TheJulianJES