zigpy/zha-device-handlers #1397: [Device Support Request] Insta GmbH Nexentro Blinds Actuator Mini

cremor, 2022-02-26:
**Is your feature request related to a problem? Please describe.**
I bought a few "Nexentro Blinds Actuator Mini" devices made by "Insta GmbH". Those are mounted behind existing blind switches and enable the blinds to be controlled both by the old switches and by Zigbee.

What works:
- cover.close_cover
- cover.open_cover
- cover.stop_cover
- cover.set_cover_position

What doesn't work:
- Cover position is always reported as 100.
- Cover state only changes with HA commands, never changes with events from device.
  But I do see zha_events being triggered when the device starts/stops the blinds. See examples below.
- Nothing related to tilt works. Tilt position is not shown and tilt related cover services don't do anything.

**Describe the solution you'd like**
All cover services should work. State and all attributes should be reported correctly.

**Device signature**
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4474, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "6": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0102"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    },
    "7": {
      "profile_id": 260,
      "device_type": "0x0203",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0019",
        "0x0102"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Insta GmbH",
  "model": "Generic UP Device",
  "class": "zigpy.device.Device"
}
```

**ZHA add device log**
<details>
  <summary>Click to show</summary>
[... 187 lines cut ...]
        "endpoint_id": 7,
        "cluster_id": 258,
        "command": "up_open",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2022-02-26T14:24:13.059192+00:00",
    "context": {
        "id": "b18490e2add2d7c56f91ba2b5e184873",
        "parent_id": null,
        "user_id": null
    }
}
```

</details>

**Additional context**
Manufacturer device page (contains links to data sheet, Zigbee certification and device manual): https://www.insta.de/product-solutions/nexentro/zigbee_blinds_actuator_mini
Entry in Zigbee Device Compatibility Repository: https://zigbee.blakadder.com/Insta_57008000.html
Zigbee2MQTT device page: https://www.zigbee2mqtt.io/devices/57008000.html
deCONZ thread: https://forum.phoscon.de/t/insta-nexentro-blinds-actuator-mini/687

Picture of the device:
[image]

cremor, 2022-03-01:
Update: One of my four devices now reports a correct state and cover_position attribute. I don't know why only that one does this. I've configured them all the same and they also all use the same firmware version.

Hedda, 2022-03-17:
As started by @mdeweerd ([le_top](https://community.home-assistant.io/u/le_top)) in https://community.home-assistant.io/t/does-zha-not-support-tilt-for-covers-blinds/402038

cover.py for zha component inside Home Assistant core looks to be missing tilt for covers  as cover does include tilting blinds:

https://github.com/home-assistant/core/tree/dev/homeassistant/components/cover

For reference it means developer need to extend ZHA cover support in cover.py for zha component inside Home Assistant core:

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/cover.py

https://github.com/home-assistant/core/tree/dev/homeassistant/components/zha

PS: Workaround could be to experiment with framework for generic exposure of device configuration attributes by @dumpfheimer

See -> https://github.com/zigpy/zigpy/discussions/934

Hedda, 2022-03-17:
> cover.py for zha component inside Home Assistant core looks to be missing tilt for covers as cover does include tilting blinds from:
> 
> https://github.com/home-assistant/core/tree/dev/homeassistant/components/cover
> 
> For reference it means developer need to extend ZHA cover support in cover.py for zha component inside Home Assistant core:
> 
> https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/cover.py

For comparison can look at some of the other components in Home Assistant core that have already implemented tilt, like mqtt:

https://github.com/home-assistant/core/blob/fe38e6ba875cb267da2ed066109b946a124166cc/homeassistant/components/mqtt/cover.py

https://github.com/home-assistant/core/blob/fe38e6ba875cb267da2ed066109b946a124166cc/tests/components/mqtt/test_cover.py

https://github.com/home-assistant/core/blob/9b5939a7a5424a8545c78e9b6b5f20e1a7ab7ac7/homeassistant/components/demo/cover.py

https://github.com/home-assistant/core/blob/3f7b7187ab410fd8dbf04fcfc834fda731dc8087/tests/components/group/test_cover.py

https://github.com/home-assistant/core/blob/e9a9b119c40c9337b1c61b7780e97b689cbd9f0f/homeassistant/components/template/cover.py

cremor, 2022-03-17:
@dmulcahey and @adminiuga: The ZHA manifest mentiones you two as code owners. Would you be interested in implementing tilt support for covers in ZHA?

And are the other problems I mentioned also features that need to be implemented in ZHA directly, or would those be things for a quirk?

cremor, 2022-03-18:
Thanks to the documentation of [zha-toolkit](https://github.com/mdeweerd/zha-toolkit) I now figured out how to set the tilt position. (Btw, I would have figured this out way earlier if home-assistant/frontend#10633 would have been fixed already.)

The following script works:

```yaml
alias: Set cover tilt position
mode: single
icon: mdi:boom-gate-up
fields:
  entity:
    name: Entity
    required: true
    selector:
      entity:
        domain: cover
  value:
    name: Value
    required: true
    selector:
      number:
        min: 0
        max: 100
sequence:
  - service: zha.issue_zigbee_cluster_command
    data:
      ieee: '{{ (device_attr(entity, "identifiers") | list).0.1 }}'
      endpoint_id: 6
      cluster_id: 258
      command: 8
      command_type: server
      args:
        - '{{ 100 - value }}'

```

(Getting the IEEE would be easier/nicer with zha-toolkit, but that's fine for me now.)

Endpoint, cluster and command ids are the ones from the Zigbee standard, so nothing device or vendor specific is needed. Zigpy also already supports those as a `go_to_tilt_percentage` command: https://github.com/zigpy/zigpy/blob/0c4b9a8a25c5eef5b008d66c2e56aae3f8d5e769/zigpy/zcl/clusters/closures.py#L568
It's just ZHA which is not using the functionality provided by Zigpy.

cremor, 2022-09-17:
Not resolved. The initial comment is still up to date.

Hedda, 2022-09-23:
> Temporary workaround until there is native tilt support for cover in ZHA could be to experiment with automations using ZHA toolkit and/or dumpfheimer framework for generic exposure of device configuration attributes. See -> [zigpy/zigpy#934](https://github.com/zigpy/zigpy/discussions/934) and/or https://github.com/mdeweerd/zha-toolkit + https://community.home-assistant.io/t/zha-toolkit-toolkit-providing-low-and-high-level-zigbee-commands-through-zha-zigpy/373346 but that is probably more work than extending zha component cover support.

Again, be sure to check out a the related zigpy development discussion here -> https://github.com/zigpy/zigpy/discussions/934

Developers there discuss ideas to expose more generic device configuration attributes in an easy way for ZHA UI/GUI.

cremor, 2023-03-22:
Please leave this open for now. Someone is currently working on tilt support for covers in ZHA, see https://community.home-assistant.io/t/support-tilt-for-covers-in-zha/403146/5
Once tilt for covers is available in ZHA I will update this issue.

Hedda, 2023-09-18:
> * **Update:** ZHA actually doesn't support tilt for covers at all. So this is not something that would be fixed in a quirk. See comments below for more information.

FYI, check out this work-in-progress draft for ZHA cover tilt support by @josef109 -> https://github.com/home-assistant/core/pull/93301

cremor, 2023-09-18:
I know that PR and I'd really like to see it merged. But sadly there has been no real change since May and the author doesn't respond to any messages (neither on GitHub nor in the HA forum).

cremor, 2023-11-03:
I've updated the first comment since HA 2023.11 now finally supports tilt for covers in ZHA.

Hedda, 2023-12-20:
Sounds like this might still need one or both of these fixes from @tomasbedrich and @GuillaumeB-GitHub to work as it should?

https://github.com/home-assistant/core/pull/105467

https://github.com/home-assistant/core/pull/99646

closed 2024-06-24 by github-actions[bot]