zigpy/zha-device-handlers #1399: [Device Support Request] Zemismart Curtain ZM79E-DT - direction inverted

rjbungay, 2022-02-27:
**Is your feature request related to a problem? Please describe.**
I recently purchased 2 Zemismart Curtain Tracks ZM79E-DT having checked that they were compatible with ZHA on the zigbee.blakadder website. I am running Home Assistant 2022.2.9 supervised on a Pi 4 4GB with a SonOff ZBBridge running ZHA.

If I just add the Zemismart to ZHA the ZM79E is recognised but fails due to "input cluster mismatch on at least one endpoint" and shows with no control entities, just LQI and RSSI.

I have managed to identify that this is down to the ts0601_cover.py showing a different cluster configuration for the _TZE200_3i3exuay with "time" as an input cluster rather than an out put cluster. I have managed to edit a copy of ts0601_cover.py show "time" as an output cluster and load this as a custom quirk.

From 0601_cover.py with "time" shown as an input cluster:-
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Time.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],

The ZM79E is now recognised and loaded as a cover device with control entities. Howver, the controls don't work as the position report appears to work inverted 0=open, 100=closed. I am able to invert this with the remote control but now the Up/Down controls on the remote and HA are inverted. This means that the curtains can't be controlled in HA as the cover thinks it is already at the end stop and "greys" out the control needed to move it in the correct direction.

I am not a programmer or familiar with ZHA/quirks so am now out of my depth. I have done a lot of googleing and think that this should be correctable by editing the __init__.py file to add "_TZE200_3i3exuay" to the inverted by default list and/or re-mapping the 0x0000, 0001, 0002 Up, Down, Stop commands as needed. I have tried editting these parameters in a copy of the __init__.py file and saving this in the custom quirks directory but this doen't seem to be doing anything to affect the position reporting of altering the direction of the Up/Down controls. As this is a supwervised install I don't believe I can access the original files in the \usr\ directory to edit them. Am I doing this correctly or is ther something I'm missing?

I am trying to get hold of a Tuya Zigbee hub so that I can see if it is possible to update the firmware on the ZM79E-DT _TZE200_3i3exuay as mine appear to be "different" to the version that the quirk is written for (different in/out cluster configuration and direction), not sure if there is an update available that will change things.

Going forward, would it not make more sense to be able to invert the position reporting and movement directio controls in HA or ZHA somehow. The direction of movement/position can simply be changed by assembling the carridges in the opposite sides of the track so a simple method of chaing direction would seem logical. This would not sort out the issue I have though as the position and direction are currently in opposition so won't work as the motor already thinks it is in an "open" position so greys out the open control which would actually close it etc.

**Describe the solution you'd like**
Going forward, would it not make more sense to be able to invert the position reporting and movement directio controls in HA or ZHA somehow. The direction of movement/position can simply be changed by assembling the carridges in the opposite sides of the track so a simple method of chaing direction would seem logical. This would not sort out the issue I have though as the position and direction are currently in opposition so won't work as the motor already thinks it is in an "open" position so greys out the open control which would actually close it etc.

As a short term solution, how can I invert the position and/or Up/Down controls so that the motor will work, is there a way to get the edited __init__.py to load the corrected directions?

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

As reported with standard ZHA Quirk:-
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_3i3exuay",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

As reported with my edited custom Quirk:-
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
[... 68 lines cut ...]
polling current state
read pos=None
Error handling '_unsupported_attribute_added' event with (cc:86:ec:ff:fe:9a:b2:8b, 1, 258, 8) params: FOREIGN KEY constraint failed
[0x9cff:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0x9cff:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_req([0x0000])
[0x9cff:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=27 command_id=Command.Read_Attributes>
[0x9cff:1:0x000a] ZCL request 0x0000: [[7]]
[0x9cff:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=27 command_id=Command.Read_Attributes>
[0x9cff:1:0x000a] ZCL request 0x0000: [[7]]
Duplicate 27 TSN
[0x9cff:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=28 command_id=36>
[0x9cff:1:0xef00] ZCL request 0x0024: [[0, 95]]
[0x9cff:1:0xef00] No handler for cluster command 36
[0x9cff:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=29 command_id=17>
[0x9cff:1:0xef00] ZCL request 0x0011: [MCUVersionRsp(tsn=24576, version=64)]
[0x9cff:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=29 command_id=17>
[0x9cff:1:0xef00] ZCL request 0x0011: [MCUVersionRsp(tsn=24576, version=64)]
[0x9cff:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=29 command_id=17>
[... cut ...]
[0x9cff:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=29 command_id=17>
[0x9cff:1:0xef00] ZCL request 0x0011: [MCUVersionRsp(tsn=24576, version=64)]
[0x9cff:1:0xef00] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=29 command_id=17>
[0x9cff:1:0xef00] ZCL request 0x0011: [MCUVersionRsp(tsn=24576, version=64)]

Any help sorting this out would be much appreciated.

rjbungay, 2022-02-27:
I have just managed to get this installed and working correctly with a crucial bit of info from this post

https://github.com/zigpy/zha-device-handlers/issues/1294

If I remember correctly the structure for the custom quirk directory should look like this then:
/config/custom_quirks/tuya:
-- __init.py__ (modified)
-- ts0601_cover.py (modified)

Note:
Also edit the ts0601_cover.py file and change the following line (13):
from zhaquirks.tuya import (
to
from . import (

rjbungay, 2022-02-27:
Can these settings be added to the official quirk or are there different versions of _TZE200_3i3exuay that require different settings?
In ts0601_cover.py
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
and the associated "replacement"

In init.py
    "_TZE200_3i3exuay": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},

TUYA_COVER_INVERTED_BY_DEFAULT = [
    "_TZE200_3i3exuay",
]

rjbungay, 2022-02-27:
The last issue I have but I'm not sure if this is a ZHA problem or in HA is I can't set the type to "show as curtain" to display the correct icons and and "open" and "close" symbols, type "door" is the closest I can get.

[image]

TheJulianJES, 2022-04-17:
@rjbungay So the only change required for you is to add ``"_TZE200_3i3exuay"`` to ``TUYA_COVER_INVERTED_BY_DEFAULT``?

The person who originally added the model to Zigbee2MQTT [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3454) doesn't seem to have tested the change before. 

The command was added to ZHA [here](https://github.com/zigpy/zha-device-handlers/pull/912).
@holdestmade Can you perhaps confirm/deny that your position is currently inverted?

rjbungay, 2022-04-17:
@TheJulianJES, I confirm that this motor needs to be added to the "inverted by default" but also the clusters need to be changed as I posted above. Without the cluster change the quirk is rejected with the error "Fail because input cluster mismatch on at least one endpoint". 

In addition to adding to the change adding to inverted by default in init.py this is my current custom quirk with the cluster changes:-

"""Tuya based cover and blinds."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from . import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)

class TuyaZemismartSmartCover0601_RJB(TuyaWindowCover):
    """Tuya Zemismart curtain cover motor."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # input_clusters=[0x0000, 0x000a, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=81 input_clusters=[0, 10, 4, 5, 61184] output_clusters=[25]>
        MODELS_INFO: [
            ("_TZE200_3i3exuay", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

TheJulianJES, 2022-04-17:
Ah, I see. I really wonder if there are two separate Tuya devices with the exact same model/manufacturer with one being inverted and the other one not.
The device signature can be added easily (as another class, so the other signature/quirk also works).
However, the inverted thing might get a bit tricky (if the current implementation is actually correct for the "other" cover with the same name).

Otherwise, the ``cover_inverted`` attribute on the ``TuyaWindowCoverControl`` could possibly be written to invert the cover.

I hope the original quirk author can give some feedback if his device also needs to be inverted (or not).

blauret, 2022-05-07:
I can give some feedback, as I wrote the original code for the _TZE200_fzo2pocs. The current implementation has reversed the controls and this is a breaking change for me. The breaking change was introduced in #1319.

closed 2022-11-10 by github-actions[bot]