zigpy/zha-device-handlers #1401: [Device Support Request] LIDL _TZ3000_5bsf8vaj, _TZ3000_utagpnzs and _TZ3000_odygigth of "model" TS0505A

mdeweerd, 2022-02-28:
These lights are not referenced in quirks yet:
All have "Manufacturer" TS0505A  bough @LIDL last year.

_TZ3000_5bsf8vaj and _TZ3000_utagpnzs : extérieur lights.
_TZ3000_odygigth: intérior light bulb.

All have RGB colors.

This is a scan of endpoints with their attributes and attribute values:
[light_scans.zip](https://github.com/zigpy/zha-device-handlers/files/8157471/light_scans.zip)

I was trying to find a way to set their default "on" state after power up.  there are several non-standard attributes but they do not seem to match those already defined for Tuya.

MattWestb, 2022-03-01:
I have trying documenting attribute and commands that is little no standard and most tuya https://github.com/zigpy/zigpy/discussions/823#discussioncomment-1539469.

tuya is normally using one no standard for power on status and its implanted in INIT but i dont knowing if its working on all devices but looks like its working on all plugs i is having.

mdeweerd, 2022-03-01:
Because I helped a user program the power_on_state of his device and because I have it in some devices, I thought I could set it in the lights as well, but they do not have the 0x8002 attribute (in the onoff cluster).

They do have a bunch of attributes that are not yet present in the quirks or that can not have the same meaning:

# Basic cluster 0x0000
- "attribute_id": "0xffde",
- "attribute_id": "0xffe0",
- "attribute_id": "0xffe1",
- "attribute_id": "0xffe2",
- "attribute_id": "0xffe3",
# Color Control cluster 0x0300
- "attribute_id": "0xf000",
- "attribute_id": "0xf001",
- "attribute_id": "0xf002",
- "attribute_id": "0xf003",
- "attribute_id": "0xf004",

# Found in code, but in (Tuya) Window Covering cluster:
- tuya/ts130f.py:    attributes.update({0xF000: ("tuya_moving_state", t.enum8)})
- tuya/ts130f.py:    attributes.update({0xF001: ("calibration", t.enum8)})
- tuya/ts130f.py:    attributes.update({0xF002: ("motor_reversal", t.enum8)})
- tuya/ts130f.py:    attributes.update({0xF003: ("calibration_time", t.uint16_t)})

closed 2022-10-19 by github-actions[bot]