zigpy/zha-device-handlers #1402: GitHub bug? zha-device-handlers Git 0.0.67 tagged but no 0.0.67 release yet on GitHub

Hedda, 2022-03-01:
@dmulcahey zha-device-handlers Git 0.0.67 tagged but there is no 0.0.67 release yet on GitHub yet as it still lists 0.0.66 as the last release:

https://github.com/zigpy/zha-device-handlers/releases

0.0.67 should be listed with changes which it does not now:

https://github.com/zigpy/zha-device-handlers/releases/tag/0.0.67

https://github.com/zigpy/zha-device-handlers/tags

PS: zha-quirks package on PyPI uses tags and not GitHub releases so at least it does list 0.0.67 there:

https://pypi.org/project/zha-quirks/#history

LittleFox94, 2022-03-09:
GitHub releases are based on tags and creating a release will create a tag (if you don't create it from an existing tag)
Git tags will not create a GitHub release automagically - someone has to create a release from this tag

mdeweerd, 2022-03-09:
May be a pre-release - it adds a tag for sure but it's possibly not visible for all users?
Or the release has been engaged and then abandoned?

LittleFox94, 2022-03-09:
@dmulcahey can probably answer this - I guess creating the release on GitHub was just forgotten

dmulcahey, 2022-03-09:
It was accidentally left in draft. The release was on PYPI and is already in HA

closed 2022-03-09 by dmulcahey

cremor, 2022-03-09:
@dmulcahey But the release notes are not correct now, are they? Because they reference PRs that were only merged recently, although 0.0.67 was tagged 14 days ago.

Hedda, 2022-03-09:
Also no release notes under the tag 

https://github.com/zigpy/zha-device-handlers/releases/tag/0.0.67

vs.

https://github.com/zigpy/zha-device-handlers/releases/tag/0.0.66

and it is that link that is refered to in https://github.com/home-assistant/core/pull/67109 for the changelog:

dmulcahey, 2022-03-09:
I’ll clean up the release notes when I have a minute

dmulcahey, 2022-03-09:
Should be fixed