zigpy/zha-device-handlers #1403: [Device Support Request]

kiwizznz, 2022-03-01:
**Is your feature request related to a problem? Please describe.**
Just added to my Mercator (Mercator White Smart Zigbee Isaac Double Outlet Power Point) setup a whole bunch of their smart power points. All seems well (not officially supported but pairs fine, reports in HA and can turn on/off switches) other than the ‘metering_summation_delivered’ which shows an insane amount of power used in kWh (this example is after a couple weeks): 18,281,013,456.0 kWh which is quite annoying as I got them to measure power usage on devices etc.

A new switch out the box shows: 2,097,168.0 kWh where I assume it should be 0?

These entities show up for the device (plug named Laundry Smart Plug)

sensor.laundry_smart_plug_electrical_measurement
sensor.laundry_smart_plug_electrical_measurement_rms_current
sensor.laundry_smart_plug_electrical_measurement_rms_voltage
sensor.laundry_smart_plug_smartenergy_metering_summation_delivered

**Describe the solution you'd like**
Would like the readings in HA to match correct kWh usage.

[image]

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

Zigbee Device Signature:
```
{
"node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
"endpoints": {
"1": {
"profile_id": 260,
"device_type": "0x0051",
"in_clusters": [
"0x0000",
"0x0003",
"0x0004",
"0x0005",
"0x0006",
"0x0702",
"0x0b04",
"0xef00"
],
"out_clusters": [
"0x000a",
"0x0019"
]
},
"2": {
"profile_id": 260,
"device_type": "0x0051",
"in_clusters": [
"0x0003",
"0x0004",
"0x0005",
"0x0006",
"0xef00"
],
"out_clusters": []
},
"242": {
"profile_id": 41440,
"device_type": "0x0061",
"in_clusters": [],
"out_clusters": [
"0x0021"
]
}
},
"manufacturer": "_TZ3210_pfbzs1an",
"model": "TS011F",
"class": "zigpy.device.Device"
}
```

**Additional context**
System Health

version: core-2022.2.9
installation_type: Home Assistant OS
dev: false
hassio: true
docker: true
user: root
virtualenv: false
python_version: 3.9.7
os_name: Linux
os_version: 5.10.98
arch: x86_64
timezone: Australia/Melbourne

host_os: Home Assistant OS 7.4
update_channel: stable
supervisor_version: supervisor-2022.01.1
docker_version: 20.10.9
disk_total: 228.5 GB
disk_used: 13.7 GB
healthy: true
supported: true
board: generic-x86-64
supervisor_api: ok
version_api: ok

Cheers!

MattWestb, 2022-03-05:
Your device is need one quirk for getting the attribute being fixed.
It shall being added as one new device class in the https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py.
Use one of the existing ones and adding your device signature and also correcting the replacement and all shall working OK.

kiwizznz, 2022-03-05:
@MattWestb,

I've tried to do that, but for some reason the quirk isn't working. I'm guessing its something to do with "Unknown cluster 61184". 

[image]

Here is the signature which I've come up with, I believe the only thing missing is where I've put '???'. In HA, that cluster ID is labelled 'Cluster' and if I use 'Cluster' in the signature, it doesn't load/work.

[image]

```
class Plug(CustomDevice):
    """Mercator TS0011F plug."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # device_version=1
            # input_clusters=[0, 4, 5, 6, 1794, 2820, 61184, 3]
            # output_clusters=[25, 10]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    ???.cluster_id,
                    Identify.cluster_id,                    
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=81
            # device_version=1
            # input_clusters=[4, 5, 6, 61184, 3]
            # output_clusters=[25, 10]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    ???.cluster_id,
                    Identify.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    Cluster.cluster_id,
                    Identify.cluster_id,                    
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],

            },
        },
    }
```

I have another question, do the INPUT_CLUSTER entries need to be in the same order as the ids listed in the SimpleDescriptor? If so, great as that's how I have listed them. If not, could that be why the signature isn't applying to the device (other than the cluster ID 61184 thing)?

Cheers!

MattWestb, 2022-03-06:
Your device is little tricky then its one hybrid device = is have one "normal" Zigbee module that is doing most things plus one tuya MCU that is doing little more things.
I think getting the normal function i can helping with but the tuya MCU over my head.
First use this cluster for the unknown one in the signature:
https://github.com/zigpy/zha-device-handlers/blob/e4e62504738ced0c54c983ee5094a3a6af0a741d/zhaquirks/tuya/ts0601_motion.py#L364
and i think you can putting the same in replacement.
Then you also need importing the function so the system is knowing what to do with it like this:
https://github.com/zigpy/zha-device-handlers/blob/e4e62504738ced0c54c983ee5094a3a6af0a741d/zhaquirks/tuya/ts0601_motion.py#L42

If all is being OK the system shall loading the quirk for your device and getting more functions.

The `TuyaNewManufCluster` is sending commands to and from the MCU but for the moment we dont knowing witch commands is using but it can being fixed later if we is "finding" them.

kiwizznz, 2022-03-12:
Thanks @MattWestb ,

Tried with the updated reference to TuyaNewManufCluster with no luck. It didn't produce an error though. Is there any way to identify if the quirk is being loaded/processed somewhere?

Here is what I have so far:

```
"""TS011F plug."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
    Cluster,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBOnOffAttributeCluster,
    TuyaNewManufCluster,
)

class Plug(CustomDevice):
    """Mercator TS0011F plug."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # device_version=1
            # input_clusters=[0, 4, 5, 6, 1794, 2820, 61184, 3]
            # output_clusters=[25, 10]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                    Identify.cluster_id,                    
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=81
            # device_version=1
            # input_clusters=[4, 5, 6, 61184, 3]
            # output_clusters=[25, 10]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                    Identify.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaNewManufCluster.cluster_id,
                    Identify.cluster_id,                    
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],

            },
        },
    }
```

I did some more testing to work out the correct power usage, for one switch HA reports 12583008 kw/h, using a multiplier of 2097168. The multiplier shows up in the cluster information, but the correct kw/h is 0.06 (12583008 divided by the multiplier 2097168), which is 6 and likely reported in w/h rather than kw.

I'd also like to fix the device type, HA shows it as a switch rather than a plug (and such, integrations with Google mean it appears as a light)

The other thing I've seen on the native Mercator app is 'relay state' and 'child lock'. I don't really care about the child lock, but relay state is important as the switches default to off (rather than last state) on a power outage. Once I can work out how to get the quirk working, I'd like to try solve this challenge too.

One more question - is there a way to dump all the cluster information at once, or do you have to pick the cluster info then query for each one?

Cheers!

MattWestb, 2022-03-12:
I think i can see one error in the signatures and its the `"device_type": "0x0051"` that is:
https://github.com/zigpy/zigpy/blob/0c4b9a8a25c5eef5b008d66c2e56aae3f8d5e769/zigpy/profiles/zha.py#L23. 
Fast look on the cluster looks OK but not 110% checked.
Fix the device profile and restarting HA.
If its **not**  matching then the system is not loading the quirk and you is getting in the device signature `"class": "zigpy.device.Device"`.
If its loading its getting the device class from the quirk.

Also using the `TuyaZBOnOffAttributeCluster,`  as replacment  for the on/off cluster and you is getting some more attributes for power on state and little more but we have not found the attribute for the children lock.
But with your device it can being that all this things is being done with the `TuyaNewManufCluster` and then the extra attributes on the normal cluster is not  working.

First you must getting the device signature OK in the quirk so the system is loading then you can start testing the device.

kiwizznz, 2022-03-13:
@MattWestb good spotting! I've changed the device type now, and of course run into another issue, I think this is the related error log (ZHA/zigbee won't load unless I disable the custom quirks now):

```
2022-03-13 22:42:35 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry ConBee II, s/n: DE2435826 - dresden elektronik ingenieurtechnik GmbH for zha
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
result = await component.async_setup_entry(hass, self)
File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 102, in async_setup_entry
await zha_gateway.async_initialize()
File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 152, in async_initialize
self.application_controller = await app_controller_cls.new(
File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 61, in new
await app._load_db()
File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 53, in _load_db
await self._dblistener.load()
File "/usr/local/lib/python3.9/site-packages/zigpy/appdb.py", line 470, in load
await self._load_unsupported_attributes()
File "/usr/local/lib/python3.9/site-packages/zigpy/appdb.py", line 521, in _load_unsupported_attributes
ep = dev.endpoints[endpoint_id]
KeyError: 2
```

My guess is that its the last bit, something about unsupported_attributes or dev.endpoints[endpoint_id].

MattWestb, 2022-03-13:
I dont knowing how the code is working but i think your device have saving some attributes on endpoint 2 before you have adding the new quirk and with the quirk is not needed or its wrong so the system is not knowing how to using it.

Start the system without the quirk and deleting the device and then copy the quirk back and restarting HA.
Then its shall not being any strange things in the DB.
Trying adding the device and see if the system is doing it OK and if yes restart HA and see if its working.

kiwizznz, 2022-03-14:
Hmmm. I'm not keen on removing the device as I have ~10 others (same model) setup and would mean I have to redo all the configuration. Any ideas on how to work out what the actual attributes are?

Brazen00, 2022-03-30:
I have the same issue with my new 1- and 2-gang Mercator sockets. I'll try this new quirk when I get a spare moment, but I'm just curious as to why you don't keep endpoints 2 and 242 in your replacement. Especially endpoint 2 as it controls the right socket on/off. That might be why it's throwing an error?

kiwizznz, 2022-03-31:
@Brazen00 great to have someone else in the same boat. Have yours also randomly turned off? I've moved some back to the Mercator Hub and they have stopped misbehaving.

Re the quick / replacement, I'm not/wasn't aware the replacement had to be for everything (i.e if you only want to replace one bit, you don't have to repeat the entire thing). If this is wrong, it could very likely be the reason the quirk isn't being applied!

Cheers!

Brazen00, 2022-03-31:
Mine aren't randomly turning off I don't think. I'll set up some automations to tell me if they turn off without my consent :)

I've hacked your quirk to the below and it works. I did have to remove the devices, restart HA and re-pair them to get the Energy Metering to show properly. I'm going to add one for the single-gang plugs later but thought you might like the below to get moving:

`

class Plug_TZ3210_2AC(CustomDevice):
    """Mercator TS0011F plug."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 1794, 2820, 61184]
            # output_clusters=[25, 10]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,                    
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=81
            # device_version=1
            # input_clusters=[3, 4, 5, 6, 61184]
            # output_clusters=[25, 10]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,                    
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],

            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
`

kiwizznz, 2022-03-31:
Awesome @Brazen00 I'll take a look, thanks for helping getting it sorted! With them randomly turning off (I have ~10 and it seems to only happen every few days, not always the same switch) they seem to be turning themselves off as there is no related event in HA, it doesn't see any action that triggered it turning off, but it can see when it is turned on (when I push the on switch).

Brazen00, 2022-04-01:
Weird. Just so I understand - when they switch off randomly does HA show them as off or still on?

Brazen00, 2022-04-01:
I couldn't wait, so I hacked up the 1-gang socket just now. Working like a charm :)
```

class Plug_TZ3210_1AC(CustomDevice):
    """Mercator TS0011F 1 outlet plug."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 1794, 2820, 61184]
            # output_clusters=[25, 10]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,                    
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,                    
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

kiwizznz, 2022-04-01:
@Brazen00 - HA show them as off when they turn themselves off, not sure if its instant or takes a while to update. I've removed the plugs that I would have been able to point out in the logging, but will try and capture the event the next time if happens. 

Looking forward to testing out your quirks later this evening!

MattWestb, 2022-04-01:
If some one is interested is it very likely the device have over current alarm but its not implanted but can being.
The parameters shall being basic cluster and attribute:
- 0x8003 Over Current Alarm (t.Bool) 0/1 (Default 0x00)
  - 0x00 = Over Current OK
  - 0x01 = Over Current Alarm

If having the auto off problem it can being that  the protection is being triggered false and can being detected by implanting the attribute.

Brazen00, 2022-04-01:
Hmm, that's odd. The full quirk I'm using is:

```
"""TS011F plug."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBOnOffAttributeCluster,
    TuyaNewManufCluster,    
)

class Plug(CustomDevice):
    """Tuya plug with restore power state support."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=266 device_type=81
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 9, 1794, 2820, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=1
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
[... 506 lines cut ...]
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,                    
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

Brazen00, 2022-04-01:
It's weird. I've been testing another quirk so I had a few different files floating around. When I tried restoring some to the prod versions I got the same error as you. I fixed it by re-downloading the latest HA build, then just replacing the ts001f_plug.py file. It's working fine now.

MattWestb, 2022-04-01:
@Brazen00 You was very likely patching the `/zhaquirks/__init__.py` by mistake and ZHA is not liking it. I have doing that some time copy one INIT for one manufacture to the `/zhaquirks/__init__.py` and is getting no loading ZHA.

kiwizznz, 2022-04-02:
@Brazen00 tried deleting, recreating and using your quirk - loaded fine! Did notice some strange behavior though, even after pairing the white pairing light was still flashing. I had 2 usb devices plugged in (my plugs are the USB version) and when I unplugged one of the USB devices the light stopped flashing AND one of the switches turned on. Plugged USB back in and the switch turned off! Tried a third time and it did nothing (expected behavior). Power metering looks to be working properly now though:

[image]

@MattWestb I'm not sure how to test the overcurrent alarm. I've not had any devices that should trip it (everything <1000w) but if its buggy would be good to be able to test to confirm.

Brazen00, 2022-04-02:
Looks like it's all good then. I don't think the behaviour you're describing is due to the quirk but obviously couldn't be sure.

Do you want to raise a PR to get this implemented?

Brazen00, 2022-04-03:
@kiwizznz So I've started getting random switch-offs too. Always the same socket though. I'll plug something else in there to see if it keeps happening.

MattWestb, 2022-04-03:
If interested of the auto power off its well documented in Z2M https://github.com/Koenkk/zigbee2mqtt/issues/11648 and looks being one hard/software future (bug / fault).

Back to the original question is the updated quirk working OK (beside auto power off for some individual devices) ?
If OK then i suggesting making one PS so its can being in the next HA release (likely 2022.05 then 2022.04 is in beta).

Brazen00, 2022-04-03:
Yes, as far as I'm concerned it's working well. @kiwizznz do you want to raise the PR for this?

kiwizznz, 2022-04-03:
@Brazen00 that would be great. Do you know if there is a way to fix the device and vendor name using quirks?

MattWestb, 2022-04-03:
The device name and manufacture is being red from the device in ZHA and cant being changes only if making one new firmware for the device.
Then all this devices is using tuya hard and software and only labeling with different brands its in the end "tuya" devices.

Give the device the name you like and also putting the the "manufacture" and "model" if you like then paring the device or later.

Also in the quirk deleting the "Mercator" and making it neutral then its more then likely shoeing up some more device with the same firmware and different labeled on the outside that is using the same device class in the quirk..

Brazen00, 2022-04-03:
OK so I've raised the PR, but I'm getting a check error when it looks for cluster 61184. I've run out of ideas as to where to fix this. Any thoughts @MattWestb ?

MattWestb, 2022-04-04:
@Brazen00 I think (= not knowing) that the tuya MCU / DP cluster is making the problem.
Its normally used in TS0601 devices that is having one tuya MCU and sending DP commands and only using the Zigbee module as one tunnel for the comport.
We have seen some "hybrid devices" with normal Zigbee function and the tuya MCU cluster but the MCU part looks not working or we is not knowing how to using it in this devices.

Then we dont knowing how we shall using it and its never being used in the replacement (only "new" functions added from the MCU cluster on new cluster is being added) i think the right and the best is deleting it from the replacement and looking if the PR is going thru the test.

Also after getting it working you can re power the device and see is its sending some MCU / DP commands on it but i think its dont doing that (normaly tuya MCU is normally sending all there DP commands then booting up).

closed 2022-04-07 by dmulcahey

SgtBatten, 2022-09-18:
> Have yours also randomly turned off? I've moved some back to the Mercator Hub and they have stopped misbehaving.

Hey kiwi, did you ever resolve this. I've been begging mercator to update the firmware to change the overvoltage protection but they are basically ignoring me now. I don't have a mercator hub but have heard from others the issue stops if you are using one. 

Out of curiosity if i did have a hub, can you still use the GPOs in home assistant? i've not seen an ikuu integration anywhere

kiwizznz, 2022-09-27:
@SgtBatten Nope, still randomly turning off. You can use their official hub and the generic tuya app which has ha integration but then you aren't using local control. I wonder if their next version is going to support matter.....

kiwizznz, 2022-10-11:
Nope, still randomly turning off. You can use their official hub and the
generic tuya app which has ha integration but then you aren't using local
control.