zigpy/zha-device-handlers #1406: [Device Support Request] Smart Air House keeper TS0601 _TZE200_dwcarsat v2

JoryHogeveen, 2022-03-03:
See https://github.com/zigpy/zha-device-handlers/issues/1313

The current implementation isn't 100%.
It assumes it's the same as the `_TZE200_ryfmq5rl` and `_TZE200_yvx5lh6k` devices but it isn't.

The `_TZE200_dwcarsat` actually has six sensors.

- Temperature
- Humidity
- CO2
- Formaldehyde / CHOH (CH2O)
- PM2.5
- VOC

Most of the above are visible, however, incorrect:

- CO2 > It looks like this shows the CHOH value (formaldehyde). eg. 0 - 10 ppm (I'm not even sure if this is actually correct...)
- Formaldehyde / CHOH (CH2O) > It looks like this shows the CO2 value. eg, 100+ ppm
- PM2.5 > Still isn't shown

So it looks like the CO2 and Formaldehyde sensors are mixed up though I'm not sure, the readings I get are quite deadly though haha.

JoryHogeveen, 2022-03-03:
More info on the MQTT implementation here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/3745/files

budcalloway, 2022-03-04:
Hi Jory,
I have the same concerns with this exact same device :)

ajpalm, 2022-03-08:
Can confirm under ZHA I'm seeing similar issues with the TZE200_dwcarsat TS0601 device not showing all the sensors and the data being mismatched.

JoryHogeveen, 2022-03-09:
@LittleFox94 Thanks for this, if even the 5 sensor model is showing inaccurate then I assume the whole integration for all smart air models is incorrect.

@MattWestb Do you have more information on how the sensor params are linked to the actual HA params? Besides the values being reported, was it tested that the values we're linked correctly?

LittleFox94, 2022-03-09:
I just noticed the values don't even change a lot after having the window open for half an hour

The original values in the comment above were before letting fresh air in after the night (bedroom+home office), only difference is formaldehyde now being at 368ppm - but Hass history suggests the 378ppm in the comment above were a typo (it shows constant 368ppm)

This could be the sensor of course, but might also point at a problem in the integration

JoryHogeveen, 2022-03-09:
Hi @LittleFox94 
A Formaldehype  value above 100 would be deadly ;). The exposure limit is 8 hours for 0.75 ppm and 15 minutes for 2 ppm (on the workfloor).

Your Formaldehype value probably stands for CO2 which is a normal "daily" value. Hence opening a window won't change it very much.

LittleFox94, 2022-03-09:
Yep, I'm aware that formaldehyde value being true would mean I'm dead ^^'

But shouldn't CO2 change when opening the window, by letting CO2 out? I thought it concentrates in rooms from .. breathing it out?

JoryHogeveen, 2022-03-09:
Hi @LittleFox94 
If your usual value is around 350 then it means your house is "breathing naturally" so that's a good sign :)
Of course this would depend on the isolation of your house and the room size but just from breathing this difference wouldn't be very large and very local in your room.

MattWestb, 2022-03-09:
@JoryHogeveen I think all magic is being made in the INIT https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/air/__init__.py and i also think the DPs from some  one that is having tuya integration and have looking in tuya dev for getting the information what its function liked to the device but its not 110% sure is correct.

JoryHogeveen, 2022-03-09:
@MattWestb Yep looks like it's handled there.
Also looks like this is an issue that is also happening at Zigbee2MQTT: https://github.com/Koenkk/zigbee2mqtt/issues/11033

MattWestb, 2022-03-09:
We have founding that many tuya device need some "tuya magic" being casting on them then joining (first was TS004F) but we have not getting it implanted in ZHA but HB and Z2M have getting it working and we need reading and writing attributes very early (or kicking the device so it rejoining) then the device is paring for getting it working. 
Then we can getting it working in quirks we can fixing nearly 10 bad devices and perhaps i can helping in this case 2 but we cant testing it for the moment.

Bad things is missing endpoints (2 devices), Not reporting attributes and having connecting problems / stooping committing or falling in sleep or cant  getting time synced.

LittleFox94, 2022-03-10:
just for completeness, my sensor doesn't seem to actually send any updated data - it's not shown as connected in ZigBee visualization and all data is last updated "Yesterday" and right now there is a lot of Isopropanol in the air it should probably pick up as VOC ^^'

stephenjamieson, 2022-03-15:
Can confirm that I'm having the same issues with this device. watching

JoryHogeveen, 2022-03-17:
Maybe related error logs:

`Logger: zigpy.zcl
Source: /usr/local/lib/python3.9/site-packages/zhaquirks/tuya/__init__.py:1295
First occurred: 6:09:32 PM (2 occurrences)
Last logged: 6:09:36 PM

[0x4093:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=1, dp=20, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))]
[0x4093:1:0xef00] No 'handle_set_data_response' tuya handler found for [TuyaCommand(status=0, tsn=3, dp=20, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))]`

buddy31130, 2022-03-24:
I have to add that this device is sending a huge amount of messages everyday. About 15,000-20,000 entries for some sensors and linkquality each day. Is it normal ?

maxperron, 2022-05-12:
I just bought this device and have the same issues as described above. Is there any unofficial quirk we can download?

sj3fk3, 2022-05-15:
Seeing the same issues.. Deadly Formaldehyde Concentration (about 300 ppm) and strange CO2 levels (either around 8 or 40.000)? Temp and Hum seem ok, no PM2.5 and VOC also looks kinda okish.

Sesshoumaru-sama, 2022-05-17:
I also died of Formaldehyde several times over, if the value would be correct and PM2.5 is missing.
Fix would be greatly appreciated.

AssetBurned, 2022-06-11:
Isn't this a pretty similar unit compared to the _TZE200_yvx5lh6k in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4990 ?

Also to link the topics https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5733

arohl, 2022-07-19:
> Can confirm under ZHA I'm seeing similar issues with the TZE200_dwcarsat TS0601 device not showing all the sensors and the data being mismatched.

I have the same model and only see one sensor - Voc_level - any idea why I can't even see temperature and humidity?

gudvinr, 2022-07-27:
So, I have this sensor too and also didn't have anything other than VOC right after it is paired.

Other sensors appeared only after HA restart and CO₂/HCHO sensors are swapped. PM2.5 is missing as well.

chaples55, 2022-07-30:
I have several of these TZE200_dwcarsat TS0601 and can confirm the behavior that several others have reported. Would love to see these supported! For the time being, I have them integrated via Tuya, though even that isn't perfect since the temperature units use a weird ° character that isn't recognized. Here's how that shows up:
[image]

chaples55, 2022-08-03:
Just updated HA Core from 6.6 to 8.0 and the temperature reading shows up correctly in °F now via the Tuya integration.

arohl, 2023-01-08:
> Just updated HA Core from 6.6 to 8.0 and the temperature reading shows up correctly in °F now via the Tuya integration.

@chaples55 Doing it via the Tuya integration means you need a Tuya hub? Looks like it works in zigbee2mqtt but not ZHA.

chaples55, 2023-01-08:
Yeah, I'm still using 5 of these things via a Tuya hub. I have a 6th one connected via ZHA to keep tabs on whether it is fully supported yet, but unfortunately that still is not the case. 

I have a second ZigBee dongle coming soon so I can use both z2m and ZHA and ditch the Tuya hub entirely.

arohl, 2023-01-08:
@chaples55 My SkyConnect arrived on Friday so I guess I can use my old dongle to run z2m - why didn't I think of that :-)

marinswk, 2023-01-19:
Having the same exact issues with the same model and ZHA with Skyconnect on homeassistant OS.
Mismatched sensors values and missing pm2.5 sensor report.
Any update here? any way to fix it? :)

n00b001, 2023-02-10:
I'm also having this exact same issue

jdopson0624, 2023-02-16:
Same issue here...

fhaust, 2023-02-17:
It's a bit strange ... the sensors come up after a home assistant restart.
But CO2 and CH2O (Formaldehyde) seems to be switched? (But even then the
CH20 level is higher than safe standard).

gudvinr, 2023-02-17:
Don't bother to make it make sense. HCHO and CO2 values are artificial and just calculated from VOC sensor data inside box.

fhaust, 2023-02-17:
@gudvinr Is that so? So the "4 in 1" and "6 in 1" sensors are basically the same, just with some more code?

Walmtech, 2023-02-17:
Well nothing suprise me with these cheap sensors. I took a look inside and can confirm in my 6in1 there are only 3 sensors.

[image]

Walmtech, 2023-02-17:
That MP501sensor for VOC only has 4 levels of output not very useful really

[image]

chaples55, 2023-02-17:
Can confirm that the HCHO, CO2, and VOC values are complete garbage. Bought some proper sensors from Airthings and the values are completely different. The CO2 level on the Airthings reads 2-3x higher and actually changes in response to room occupancy and ventilation, unlike these cheap multi-sensors.

tobixen, 2023-03-05:
Bought a heap of cheap tuya airsensors a while ago - _TZE200_yvx5lh6k - using ZHA and they are only reporting VOC (varying between 20 and 50 micrgram / cubic metre.

The layout looks different from what's posted above, but I can recognize the MP501 gas sensor, the AHT20 temperature/humidity sensor and I also have that ZT3L-chip.

Even just the information from the humidity/temperature sensor would be useful for me.

closed 2023-09-08 by github-actions[bot]

schwaboy, 2024-12-27:
Using ZHA and I only see the VOCs. I bought this primarily for the CO2. How can we get this to show up using ZHA?